stevebuk1/pv_opt #228: Battery doesn't get charged in cheap periods

SzosszeNET, 2024-05-24:
In the past couple days noticed that the battery doesn't get charged as much as I would expect and not sure if PV operates with the most efficient cost.
My battery is almost empty have some cheap period followed by more expensive period but there is no charge scheduled

[image]
[image]
[error.log](https://github.com/fboundy/pv_opt/files/15434711/error.log)
[main.log](https://github.com/fboundy/pv_opt/files/15434714/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/15434715/pv_opt.log)

SzosszeNET, 2024-05-24:
main.log
pv_opt.log
error.log

fboundy, 2024-05-25:
Since the weather got better I’ve noticed a problem with the optimiser which I think I’ve just got to the bottom of and should get fixed this weekend. Not sure if your issue is related but will try to have a look at the same time.

fboundy, 2024-05-25:
> [error.log](https://github.com/fboundy/pv_opt/files/15434711/error.log) [main.log](https://github.com/fboundy/pv_opt/files/15434714/main.log) [pv_opt.log](https://github.com/fboundy/pv_opt/files/15434715/pv_opt.log)

I can't see these logs for some reason. Could you please re-upload? Thanks

SzosszeNET, 2024-05-26:
[main.log](https://github.com/fboundy/pv_opt/files/15446799/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/15446800/pv_opt.log)
[error.log](https://github.com/fboundy/pv_opt/files/15446801/error.log)

Odd, just downloaded and re-uploaded them.

Sine updated to 3.14.9 and indeed that seems have addressed it to some extent

solarjones, 2024-05-26:
Seeing similar too.

solarjones, 2024-05-26:
[error.log](https://github.com/fboundy/pv_opt/files/15448186/error.log)
[main.log](https://github.com/fboundy/pv_opt/files/15448187/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/15448188/pv_opt.log)

My battery is predicted to be flat at 23:00 and OE tariff showing cheaper slots before 05:00 when the price then ramps up. Notice it isn't showing the optimised blue trace line.

[image]

fboundy, 2024-05-26:
Try changing the “pass threshold” to zero

solarjones, 2024-05-26:
> Try changing the “pass threshold” to zero
> 
> 
> > error.log
> 
> > main.log
> 
> > pv_opt.log
> 
> > My battery is predicted to be flat at 23:00 and OE tariff showing cheaper slots before 05:00 when the price then ramps up. Notice it isn't showing the optimised blue trace line.
> 
> > image.png (view on web)

Unsure which parameter that is as I don't have that?

SzosszeNET, 2024-05-26:
@solarjones it's the "charge threshold" in the bottom of the dashboard. By default it's 4

fboundy, 2024-05-26:
Its called "Charge Threshold" on the dashboard

solarjones, 2024-05-26:
> @solarjones it's the "charge threshold" in the bottom of the dashboard. By default it's 4

Thanks! Obvious now you said it 😀

SzosszeNET, 2024-05-27:
Still something off. Not sure why the 15:30 slot doesn't trigger a charge and why only discharge is scheduled :/

[image]

[image]

SzosszeNET, 2024-05-27:
[error.log](https://github.com/fboundy/pv_opt/files/15457105/error.log)
[main.log](https://github.com/fboundy/pv_opt/files/15457108/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/15457111/pv_opt.log)
[pv_opt.log.patch](https://github.com/fboundy/pv_opt/files/15457120/pv_opt.log.patch)

fboundy, 2024-05-28:
> Yep I'm having the same problem with my setup as well.
> Recently there seems to be very few charging slots allocated, despite there being a large difference in unit price between now (for example) and my peak period of 16:00 - 19:00.
> I adjusted the "Charge Threshold" to 0 as noted above, and it seemed to have started working again. However, it's had another recalculation and again has totally ignored my cheapest slot.
> I've ended up putting it in read only and manually setting my charging slots so I can at least get through the peak times later, but there seems to be something fundamentally wring with the calculations as it's missing key (relatively) cheap slots to charge.
> My screenshot FYI:
> image.png (view on web)

It’s ignoring the cheap slots because they aren’t cheap enough to save you any money. The SOC shows your solar getting you to after the peak slots.

punkymuzzle, 2024-05-28:
But at a basic level, it should be charging the battery at 14.51p, 15.54p etc, ready for discharging at the expensive times, which are approx twice the price? It didn't do this, which was why I had to put it in read only and manually charge, ready for peak slots.

> It’s ignoring the cheap slots because they aren’t cheap enough to save you any money. The SOC shows your solar getting you to after the peak slots.

But after the peak slots it's still more expensive than before, so why wouldn't it charge up as cheap as possible to last as far as possible overnight?

So roughly what does the difference in price between charging and discharging need to be before it considers it enough to set a charge?  From the above I have a 14 / 15p rate before peak, with approx. 30p peak - I would have thought that would have been plenty of difference?

fboundy, 2024-05-28:
Sorry but I don’t see what you’re seeing. Battery isn’t forecast to be empty till midnight and you don’t have actual prices that far out. You’ve also got to factor in the round trip loss of charging and discharging the battery which is about 85% so a direct price of 16.5p is still cheaper than a battery charge price of 14p.

The whole point of the app is that it only charges the battery when it saves you money to do so. The detailed calculations are all in the log file and to some extent you can tune these with the tuning parameters. If you just want to charge it when the price is below a particular threshold there are much simpler approaches.

closed 2024-08-04 by github-actions[bot]