stevebuk1/pv_opt #230: Consumption Shape

solarjones, 2024-05-26:
I run the optimiser with shaped consumption rather than historic as we use the washing machine/tumble drier based on the days PV output (where possible) to maximise good generation.  Historic usage wouldn't work well with this, hence why I run with shaped as a fairly accurate baseload dataset.

Could the CONSUMPTION_SHAPE config currently in PV_OPT.py be moved to CONFIG.yaml to be preserved as user defined parameters?

Also, how many data points are allowed? I'd think 12 could be useful to spread across the 24 hours.

Thanks.

fboundy, 2024-05-26:
Yes, this could be done. At the moment it uses a very simple curve but if it was custom I could set it to read an arbitrary number.

fboundy, 2024-06-02:
This will be in 3.15.0. To use it add the following to `config.yaml':

    use_consumption_history: false
    # consumption_history_days: 6
    #
    daily_consumption_kwh: 17
    shape_consumption_profile: true
    consumption_shape:
      - hour: 0
        consumption: 300
      - hour: 0.5
        consumption: 200
      - hour: 6
        consumption: 150
      - hour: 8
        consumption: 500
      - hour: 15.5
        consumption: 500
      - hour: 17
        consumption: 750
      - hour: 22
        consumption: 750
      - hour: 24
        consumption: 300

Consumption is nominally in watts but it is scaled to match the daily total so is actually dimensionless

closed 2024-06-02 by fboundy