stevebuk1/pv_opt #235: Set up help

Gorwanvfr, 2024-06-01:
Looking for some help please.
I have gone so far with the install but have reached the end of my knowledge. I know I'm doing something wrong but have no idea what. I was having issues getting the mqtt broker to work but I finally figure out my mistake and I think it is working now. When I start AppDaemon it runs and creates the different log files. The issue is I can not see any of the entities within home assistant  
[image].
Thanks in advance [
[Pv_opt_log.txt](https://github.com/user-attachments/files/15522409/Pv_opt_log.txt)
[Main_Log.txt](https://github.com/user-attachments/files/15522410/Main_Log.txt)
[Error_log.txt](https://github.com/user-attachments/files/15522411/Error_log.txt)
](url)

fboundy, 2024-06-02:
Can you please also share what you have in your `config.yaml` file?

Gorwanvfr, 2024-06-03:
HI fboundy,
            Please find attached the config file

[config.yaml.txt](https://github.com/user-attachments/files/15531172/config.yaml.txt)

stevebuk1, 2024-07-31:
Hi @Gorwanvfr , did you manage to get up and running ok?