stevebuk1/pv_opt #239: Multiple Inverters

neilrfuller, 2024-06-06:
Hi, I have a system with twin 3.6kW inverters each with 4kW of panels. Could you advise me if pv_opt can handle multiple inverters, or would I need 2 instances. Many thanks.

fboundy, 2024-06-07:
It's not currently set up to do that but it might be possible. What brand and model are the inverters?

neilrfuller, 2024-06-07:
Thanks for getting back to me. The inverters are 2 x Solis S5-EH1P3.6K-L. Tbh wish I had a single larger inverter for simplicity, but was told at the time a twin install offered benefits. I suspect it was just that larger inverters weren't in stock back then. I have the inverters communicating with HA via wills106/homeassistant-solax-modbus. Managed to install Solcast via BJReplay. Think I will get going using simple automations as suggested by GordonMarkus. My knowledge of HA very much at entry level!

lazy-pete, 2024-06-07:
I’d be interested to see if this works also - I have two 5kWh batteries and
a single solis inverter, however can only push 3.6kw to the house at a
time, so high power items (showers etc) would utilise more PV power if I
had two on the go.

Cheers,

P

neilrfuller, 2024-06-07:
If free to choose again, I don't think I would go down the twin inverter route. I have a separate CT clamp for each inverter. I suspect that really an EPM (Export Power Monitor) should have been fitted to balance the inverters.

closed 2024-08-14 by github-actions[bot]