stevebuk1/pv_opt #243: ERROR: id_consumption_today : Neither the entities listed in the YAML sensor.solmod_house_load_today nor the system default of sensor.solmod_house_load_today exist in HA.

swests, 2024-06-17:
**Describe the bug**
Trying to start the integration after upgrades and get the following, stopping the app from running:
ERROR: id_consumption_today : Neither the entities listed in the YAML sensor.solmod_house_load_today nor the system default of sensor.solmod_house_load_today exist in HA.

**Expected behavior**
App to start

** Versions**
HA: 2024.06.3
PV_OPT:  v3.15.1
Solcast: 4.0.31 (new version)
Solis SolAX Modbus integration

Error in AppDaemon Log: 
 WARNING pv_opt: Traceback (most recent call last): File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 162, in initialize_app await utils.run_in_executor(self, init) File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor response = future.result() ^^^^^^^^^^^^^^^ File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run result = self.fn(*self.args, **self.kwargs) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock return f(*args, **kw) ^^^^^^^^^^^^^^ File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 521, in initialize self._load_args() File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1205, in _load_args raise ValueError(e) ValueError: id_consumption_today : Neither the entities listed in the YAML sensor.solmod_house_load_today nor the system default of sensor.solmod_house_load_today exist in HA.

I've checked again in HA:
[image]

swests, 2024-06-17:
Rolled back to 2024.05.5 but still the same. App won’t start.

Bart39, 2024-06-17:
Same here also

swests, 2024-06-18:
Also see this when I restart AppDaemon
WARNING: pv_opt: Entity sensor.pvopt_status not found in namespace default

fboundy, 2024-06-18:
Could someone please upload error.log and Pvopt.log?

Are you all using the HA Core ModBus integration? Does that appear to be working? For example are you getting the inverter dc power (solar) in HA or are all the entities offline?

Bart39, 2024-06-18:
See attached, yes using solax modbus

swests, 2024-06-18:
Using the Solax solis modus integration and its reporting all entities. Will upload logs shortly

swests, 2024-06-18:
I stopped AppDaemon, cleared the logs and restarted AppDaemon.  Logs below.

[error.log](https://github.com/user-attachments/files/15882294/error.log)
[main.log](https://github.com/user-attachments/files/15882295/main.log)
[pv_opt.log](https://github.com/user-attachments/files/15882296/pv_opt.log)

swests, 2024-06-18:
New set of logs.  I uninstalled PV_OPT, restarted and then installed the latest, with the default config.  Made minor edits to config.yaml to pick up my entities (device_name: solmod).  Still get the same issue.  Attached config, and logs

[error.log](https://github.com/user-attachments/files/15882698/error.log)
[pv_opt.log](https://github.com/user-attachments/files/15882699/pv_opt.log)
[main.log](https://github.com/user-attachments/files/15882700/main.log)
[config.yaml.txt](https://github.com/user-attachments/files/15882738/config.yaml.txt)

fboundy, 2024-06-18:
Thanks

Will do the update later and see if I can diagnose what’s going on

swests, 2024-06-18:
Detailed version info:
HA: 2024.5.5 
AppDaemon: 0.16.6 
PV_OPT: 3.15.1
OE: 11.0.2
SolaX Inverter Modbus: 2024.05.7
Solcast: 4.0.31 (new repo by BJReplay)
MQTT: 6.4.1

fboundy, 2024-06-18:
This is an incompatibility between the version of `deepdiff` installed with HA and the `2.0.0` release of `numpy`. 

As a fix please add `numpy==1.26.4` to the AppDaemon config page. In due couse I expect versions will be bumped to fix this.

[image]

dolce08, 2024-06-18:
The fix worked. As always thanks for your work and the great integration.

fboundy, 2024-06-18:
This is the root cause issue with deepdiff: https://github.com/seperman/deepdiff/issues/464

fboundy, 2024-06-18:
Also raised as an issue on the AppDaemon addon: https://github.com/hassio-addons/addon-appdaemon/issues/345

swests, 2024-06-18:
Perfect!  Thank-you for jumping right in.  Working again...

Did you want to keep it open to track the downstream issue?

fboundy, 2024-06-18:
Yes - leave it open for now

fboundy, 2024-08-15:
The root cause of this is in Appdaemon 4.4.2 requirements which need bumping to deepdiff >6.6.0

closed 2024-10-23 by github-actions[bot]