stevebuk1/pv_opt #244: PV-OPT stopped working after update to 3.15.2

punkymuzzle, 2024-06-19:
**Describe the bug**
upgraded pv_opt to version 3.15.2 and all seemed to work well initially.
However, the Dashboard no longer works and it seems like there are entities missing.  
Looking at the logs it appears to be trying to load the solis config, but my inverter is SOLAX.  So need to understand how to tell pv_opt to load the solax config

Logs attached also
Screenshot attached below:

**To Reproduce**
upgrade to 3.15.2

**Expected behaviour**
Dashboard shows correctly in HA

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: RaspPi running HA

**Smartphone (please complete the following information):**
N/A

**Additional context**
Add any other context about the problem here.
[error.log](https://github.com/user-attachments/files/15906129/error.log)
[pv_opt.log](https://github.com/user-attachments/files/15906130/pv_opt.log)

fboundy, 2024-06-19:
Do you read the release notes about the changes to the AppDaemon config?

punkymuzzle, 2024-06-19:
I saw there was an update and there was some mention of one of the repositories changing version?
Is there an easy way to load and re-read the release notes?  Assuming I need to do something in AppDaemon to reconfigure it?
Thanks

punkymuzzle, 2024-06-19:
Yep that worked, thanks very much for the pointer.

fboundy, 2024-06-19:
https://github.com/fboundy/pv_opt/releases/tag/3.15.2

punkymuzzle, 2024-06-19:
All working now, thank you very much for the help

closed 2024-06-19 by punkymuzzle