stevebuk1/pv_opt #245: Decreasing SOC despite battery charge

stevebuk1, 2024-06-23:
**Describe the bug**
A battery charge shows a decreasing SOC rather than an increasing one. 

**To Reproduce**
Putting Pv_opt into Optimed charging only (no forced discharge) and selecting sufficient manual consumption to generate a charging schedule should reproduce the issue. 

**Expected behavior**
Battery charging should result in increasing SOC. 

**Screenshots**
[image]

**Additional context**

Logging of self.opt shows "Forced" is equal to "Grid", whereas "Grid" should be "Forced + Consumption". 

`
                              import  export  Solcast  Solcast_p10  Solcast_p90  weighted  consumption      chg  chg_end      battery    grid  forced        soc    soc_end
2024-06-23 20:30:00+00:00  26.520375       0     13.8          9.5         14.5      13.8  1403.118040  10887.0  10170.9  1389.234000     0.0       0  95.500000  89.218421
2024-06-23 21:00:00+00:00  26.520375       0      0.0          0.0          0.0       0.0  1403.118040  10170.9   9447.6  1403.202000    -0.0       0  89.218421  82.873684
2024-06-23 21:30:00+00:00  26.520375       0      0.0          0.0          0.0       0.0  1403.118040   9447.6   8724.3  1403.202000    -0.0       0  82.873684  76.528947
2024-06-23 22:00:00+00:00  26.520375       0      0.0          0.0          0.0       0.0  1403.118040   8724.3   8001.0  1403.202000    -0.0       0  76.528947  70.184211
2024-06-23 22:30:00+00:00   7.499900       0      0.0          0.0          0.0       0.0  1192.650334   8001.0   7400.7  1164.582000    28.0      28  70.184211  64.918421
2024-06-23 23:00:00+00:00   7.499900       0      0.0          0.0          0.0       0.0   982.182628   7400.7   6908.9   954.092000    28.0      28  64.918421  60.604386
2024-06-23 23:30:00+00:00   7.499900       0      0.0          0.0          0.0       0.0   771.714922   6908.9   6525.5   743.796000    28.0      28  60.604386  57.241228
2024-06-24 00:00:00+00:00   7.499900       0      0.0          0.0          0.0       0.0   561.247216   6525.5   6250.6   533.306000    28.0      28  57.241228  54.829825
2024-06-24 00:30:00+00:00   7.499900       0      0.0          0.0          0.0       0.0   374.164811   6250.6   6072.2   346.096000    28.0      28  54.829825  53.264912
2024-06-24 01:00:00+00:00   7.499900       0      0.0          0.0          0.0       0.0   365.661065   6072.2   5898.1   337.754000    28.0      28  53.264912  51.737719
2024-06-24 01:30:00+00:00   7.499900       0      0.0          0.0          0.0       0.0   357.157319   5898.1   5728.4   329.218000    28.0      28  51.737719  50.249123
2024-06-24 02:00:00+00:00   7.499900       0      0.0          0.0          0.0       0.0   348.653574   5728.4   5563.1   320.682000    28.0      28  50.249123  48.799123
2024-06-24 02:30:00+00:00   7.499900       0      0.0          0.0          0.0       0.0   340.149828   5563.1   5402.2   312.146000    28.0      28  48.799123  47.387719
2024-06-24 03:00:00+00:00   7.499900       0      0.0          0.0          0.0       0.0   331.646082   5402.2   5245.7   303.610000    28.0      28  47.387719  46.014912
2024-06-24 03:30:00+00:00   7.499900       0     11.7          9.8         12.3      11.7   323.142337   5245.7   5099.6   283.434000    28.0      28  46.014912  44.733333
2024-06-24 04:00:00+00:00   7.499900       0     44.1         41.9         46.3      44.1   314.638591   5099.6   4974.6   242.500000    28.0      28  44.733333  43.636842
2024-06-24 04:30:00+00:00  26.520375       0     69.8         66.3         73.3      69.8   306.134845   4974.6   4852.8   236.292000     0.0       0  43.636842  42.568421
2024-06-24 05:00:00+00:00  26.520375       0     81.4         77.3         85.5      81.4   297.631099   4852.8   4741.3   216.310000    -0.0       0  42.568421  41.590351
2024-06-24 05:30:00+00:00  26.520375       0     88.7         84.3         93.1      88.7   289.127354   4741.3   4638.0   200.402000     0.0       0  41.590351  40.684211
2024-06-24 06:00:00+00:00  26.520375       0     96.1         91.3        100.9      96.1   280.623608   4638.0   4542.9   184.494000     0.0       0  40.684211  39.850000
`

The issue appears to be caused by this change made at version 3.15.1:

[image]

When the Flows function is called by the High cost swaps algorithm, timed_slot_flows is the battery charging power. 
Battery_flows therefore must be set to be equal the battery charging power, as it was prior to the change being made.

Reverting the change resolves the issue:

[image]

fboundy, 2024-06-24:
Hmm - yes I think you are right. This was changed when I was fixing a bug related to charging from battery and solar together so need to look at the impact of reverting on that scenario.

fboundy, 2024-06-24:
Fixed by #246

commit 13140ca6 (2024-06-24): Fix #245
closed 2024-06-24 by fboundy