stevebuk1/pv_opt #248: Integration with solis-sensor and solis-control

CraigCallender, 2024-07-01:
Not really a feature request as I've already done it.  But I'm curious, @fboundy, if you'd be interested in a pull request with the necessary config and documentation to get pv_opt working with the [solis-sensor](https://github.com/hultenvp/solis-sensor) and [solis-control](https://github.com/stevegal/solis_control) integrations.  My Solis inverter's datastick isn't compatible with Solarman - so I've had to go the route of using soliscloud via those integrations/scripts.  There weren't any code changes needed, but quite a lot of fiddling to get the configuration reading the correct sensors from the solis-sensor integration and then some automation fiddling to get pv_opt's data into the solis-control script.  

Like I said, I'm happy to do the PR - I just didn't want to go through all the work of writing it up if you're not open to it going in alongside this repo.  In which case I can fork this repo and host it there.

Thanks!  And thanks for all this hard work!

fboundy, 2024-07-01:
Definitely! Please go ahead and submit the PR

CraigCallender, 2024-07-01:
I'll work on it today and hopefully have it done tomorrow for review.  :)

fboundy, 2024-08-15:
Hi - I haven't had time to look at this for a while. How far did you get?

fboundy, 2024-08-15:
No worries - I've been tied up with other stuff. More focussed on this in the winter but it's horrible out today

CraigCallender, 2024-09-18:
@fboundy - Work gave us all a week to work on whatever we wanted to (hackathon).  So I wanted to make some improvements.  I finally got around to documenting this!  PR linked above. :)

stevebuk1, 2024-10-07:
Hi Craig, I've been making a few mods to the documentation in a "Patch2" branch in my own fork to better describe the Solarman integration, it won't be long before I raise a PR and commit to fboundys main. After that I think the time might be right to pull in your changes currently within the "Patch" branch to main also. 

I however note that in the Patch branch, config.yaml looks to be unchanged - in other words you are running with the defaults for the Solax integration - is that right? I assume you haven't got the Solax integration installed? Would you mind posting your config.yaml? 

Reason I ask is that extensive changes to config.yaml are needed for a new Solarman repo so now feels like the time to ensure it supports the solis-sensor/solis-control options.

closed 2024-12-09 by CraigCallender