stevebuk1/pv_opt #249: Discharge Power Greater Than `battery_current_limit_amps`

CraigCallender, 2024-07-01:
**Describe the bug**
It looks like the discharge windows calculation ignores the `battery_current_limit_amps` setting.  Possibly discharges have a negative value and the code is just finding the minimum of either "the perfect" discharge or the setting.  E.g.: if "the perfect" discharge is -100 amps and the `battery_current_limit_amps` is set to 37 => -100 is set/selected.

**To Reproduce**
Steps to reproduce the behavior:
1. Set you the `battery_current_limit_amps` to something fairly low (in my case, it's set to 37 amps).
2. Run a plan that includes discharging.  It seems if there's a plan that would include somewhat quick cycling to achieve optimum performance that will do it.
3. Look at the plan and see that while your charge power is limited to 37 amps, your discharge can exceed -37.

**Expected behavior**
I would expect that the `battery_current_limit_amps` is respected for both charge and discharge windows.  If it's not symmetrical for all batteries/inverters - perhaps it would make sense to have two settings then?

**Screenshots**
(On the last line you can see the value of `-48 A` even though I've set `battery_current_limit_amps` to 37)
[image]

**Desktop (please complete the following information):**
 - OS: HomeAssistant
 - Browser Chrome
 - Version

**Smartphone (please complete the following information):**
 - Device: iPhone 13 Pro Max
 - OS: iOS 17.5.1
 - Browser: Chrome
 - Version [e.g. 22]

**Additional context**
One thing that might cause confusion... I've modified my dashboard output (screenshot above) to show charge/discharge in amps rather than W as that's what my inverter requires.  I've used the simple formula of `watts / 48 = amps`.  So perhaps the discharge window is trying to use the `inverter_power_watts` as it's maximum discharge - but I think if `battery_current_limit_amps` is set it should probably be that value as that's the fastest you can charge/discharge a battery.

CraigCallender, 2024-07-01:
I actually just fixed this locally.  I'll submit a PR for it in the morning.

fboundy, 2024-07-02:
Thanks for this. I won't get a chance to look at it till later this week.

PR #250 (2024-07-03): Fix: #249 - Limit Discharging to battery_current_limit_amps
Fixes #249.  The issue was that in the charge window it was correctly using `inverter.charger_power` to calculate max charge, but in the discharge window it was using `inverter.inverter_power`.  The quickest fix would be to just use `inverter.charger_power` in the discharge window calculation (and I've confirmed that this works as expected).  

I thought it might be cleaner though to have the BatteryModel understand the properties of the battery.  And since we're talking about charging/discharging the battery - to switch over to use t [... cut]
commit cbbeb65b (2024-07-03): Merge pull request #250 from CraigCallender/main

Fix: #249 - Limit Discharging to battery_current_limit_amps
closed 2024-07-03 by fboundy