stevebuk1/pv_opt #255: Pvopt Pass Threshold P

pughster25, 2024-07-20:
somehow i have changed the default value from the dashboard slider and its reset it to max 1 not 4

i have it set in the pv_opt.py file to be 4  (it was 1) and overwrite_ha_on_restart: false was set

    "pass_threshold_p": {
        "default": 4.0,
        "attributes": {
            "min": 0.0,
            "max": 4.0,
            "step": 0.5,
            "mode": "slider",
        },
        "domain": "number",

i have since updated it to 4 as above and also set the overwrite_ha_on_restart: true in config.yaml but whenever i restart HA or appdeamon the value is still 1 even though the pv_opt.py file after restarting eithe ha or appdeamon shows the above config default 4

any ideas?

SzosszeNET, 2024-07-20:
Logs would help - can you have a look there?

pughster25, 2024-07-20:
cant get the logging to work, i have added in the below to the appdaemon.yaml

  logs:
  main_log:
    filename: /share/logs/main.log
    date_format: '%H:%M:%S'
  error_log:
    filename: /share/logs/error.log
    date_format: '%H:%M:%S'    
  pv_opt_log:
    name: PV_Opt
    filename: /share/logs/pv_opt.log
    date_format: '%H:%M:%S'      
    format: '{asctime} {levelname:>8s}: {message}'

installed samba in HA and added HA as an allowed host

is there anything else i should add to get logs?

pughster25, 2024-08-01:
@SzosszeNET is the above log any help?

fboundy, 2024-08-15:
> cant get the logging to work, i have added in the below to the appdaemon.yaml
> 
> logs: main_log: filename: /share/logs/main.log date_format: '%H:%M:%S' error_log: filename: /share/logs/error.log date_format: '%H:%M:%S' pv_opt_log: name: PV_Opt filename: /share/logs/pv_opt.log date_format: '%H:%M:%S' format: '{asctime} {levelname:>8s}: {message}'
> 
> installed samba in HA and added HA as an allowed host, and created the logs folder
> 
> is there anything else i should add to get logs?

Did you make any progress.  I think you need to restart the addon to update the logging

closed 2024-09-21 by fboundy

philicibine, 2024-09-28:
I see this issue aswell.  in pv_opt.py the default is set at 4p but the max of the slider is set at 1p

```"pass_threshold_p": {
        "default": 4.0,
        "attributes": {
            "min": 0.0,
            "max": 5.0,
            "step": 0.5,
            "mode": "slider",
        },
        "domain": "number",
```

Changing the max to 4p has no effect.

fboundy, 2024-09-28:
I've corrected this in the code now but once the entity exists in HA you will need to change the `min` and `max` attributes under `STATES` in `Developer Tools`

pughster25, 2024-10-01:
i get Value 3 for number.pvopt_pass_threshold_p is outside valid range 0.0 - 1.0. popup when setting the value in dev tools and then trying to change the slider to 4 on the dashboard, the stat then resets back to max 1