stevebuk1/pv_opt #256: error: unhashable type: list - Solax X1-Hybrid-7.5 Inverter

luismsousa, 2024-07-23:
_Obvious thanks for all the hard work that has gone into developing this solution, it looks truly amazing!_ 🎉 

**Describe the bug**
pv_opt fails to start due to unhashable type: list error

**To Reproduce**
Steps to reproduce the behavior:
1. use the attached config
2. Attempt to start pv_opt and tail logs in appdaemon

**Attempted troubleshooting**
1. Fixed errors with yaml indentation
2. renamed entities in dashboard template to fit my own
3. separated 
**Screenshots**
Current state of the dashboard:
[image]

**Desktop (please complete the following information):**
 - OS: Mac OS
 - Browser Brave
 - Version latest

**Additional context**
log files
[config.yaml.txt](https://github.com/user-attachments/files/16354108/config.yaml.txt)
[pv_opt.log.txt](https://github.com/user-attachments/files/16354109/pv_opt.log.txt)
[error.log.txt](https://github.com/user-attachments/files/16354110/error.log.txt)
[appdaemon.yaml.txt](https://github.com/user-attachments/files/16354191/appdaemon.yaml.txt)

Inverter: X1-Hybrid-7.5

I'm using the https://github.com/wills106/homeassistant-solax-modbus integration with the wifi dongle. I've attached a file containing all the enabled entities: 

[entities.txt](https://github.com/user-attachments/files/16354142/entities.txt)

I've been trying for a few hours to follow the troubleshooting steps in https://github.com/fboundy/pv_opt/issues/149 but I've hit a bit of a wall regarding a potential root cause. I'm at a bit of a loss as to what might be incorrect.

Cheers

stevebuk1, 2024-07-27:
The logfile is showing that you are calling up the solis.py file, whereas it should be Solax.py.

Try adding the following lines to config.yaml as per this post:
https://github.com/fboundy/pv_opt/issues/149#issuecomment-2002547227

stevebuk1, 2024-07-27:
Forgot to add that in the bit in config.yaml that says this:

```
dependencies:
- pvpy
- solis
```

you might need to make it say this:

```
dependencies:
- pvpy
- solis
- solax
```

stevebuk1, 2024-07-29:
When "Status" - top of 2nd column in your Dashboard reads "Idle" you know you are up and running. 
In logfile terms you'll progress beyond where you are and get a line that looks like:
"Optimiser will run every 10 minutes from 23:00  or on PV_OPT Event"

Your current issue is that Pv_opt can't find an HA entity to get house consumption data from. 

The default Config.yaml has this set:

```
  id_consumption_today: sensor.{device_name}_consumption_today
  id_consumption:
    - sensor.{device_name}_house_load
    - sensor.{device_name}_bypass_load
```

These entities are used for loading historic consumption into Pv_opt, to use in forward prediction. 

As your device name is solax its looking for "sensor.solax_house _load for power (in W) and "sensor.solax_consumption_today" for consumption (in kWh).  I think Pv opt can use power or energy but one or the other is needed. Alternatively, disable "Use Consumption History" on your Dashboard and set a fixed daily consumption. I suggest doing this as a first step to get past this issue for now, try that and post logs plus your config.yaml.

stevebuk1, 2024-07-30:
> p.s.: I'm happy to help contribute on the documentation side if you need help.

Unfortunately I don't have write access to the code -  but have contributed to development for those on IOG and have EVs so have a good working knowledge of it. If you'd like to contribute then its a case of taking a fork, making changes and opening a "Pull Request" to have your changes incorporated by the Pv Opt owner. (You'll see I have an open PR on the documentation above for the EV stuff I've been working on). 

By the looks of it the development done in #149 included a specific config.yaml for the Solax X1 but thats not yet in the main release, and the changes I suggested were based on that thread. I think once you are up and running there would be scope for a PR on both the docuentation and the contents of config.yaml, but I haven't checked the other thread thats running on the Solax X1 to see whether its being dealt with there.

fboundy, 2024-08-15:
Solax X1 should be working - because everyone's config.yaml is different it doesn't update through HASS.

@punkymuzzle has a working X1 system. I'll try to get his config and update the docs.

punkymuzzle, 2024-08-15:
> Solax X1 should be working - because everyone's config.yaml is different it doesn't update through HASS.
> 
> @punkymuzzle has a working X1 system. I'll try to get his config and update the docs.

Sure, more than happy to help.
My Inverter is an X1-AC, so quite old (5 years+ ), so entities may be different to later versions?

My config.yaml as follows:
[config.yaml.txt](https://github.com/user-attachments/files/16626944/config.yaml.txt)

Let me know if you need anything further.

closed 2024-10-23 by github-actions[bot]