stevebuk1/pv_opt #257: pv_opt crashing after hours of use

punkymuzzle, 2024-08-01:
**Describe the bug**
Everything works fine, then (usually in the morning) pv_opt seems to stop working.  AppDaemon reports errors.
Inverter settings (usually self use mode with a time of 00:00) seem to stick to a previous time.

Seems to work again after AppDaemon restart

**To Reproduce**
I just leave pv_opt running overnight.

**Screenshots**
AppDaemon Log:

 * 09:55:42 timestamp was from yesterday morning.
[image]

**Desktop (please complete the following information):**
N/A

**Smartphone (please complete the following information):**
N/A

**Additional context**
Logs attached:

[pv_opt.log](https://github.com/user-attachments/files/16452661/pv_opt.log)
[main.log](https://github.com/user-attachments/files/16452662/main.log)
[error.log](https://github.com/user-attachments/files/16452663/error.log)

fboundy, 2024-08-15:
Looks like it is because the battery voltage is reading zero. I've added a default of 100 in the latest patch.

punkymuzzle, 2024-08-15:
Awesome, thank you

commit b3424d60 (2024-08-15): Fix #257 and  partial fix for #261
closed 2024-08-15 by fboundy