stevebuk1/pv_opt #260: Ignoring export pricing because Use Export is turned off?

SzosszeNET, 2024-08-12:
**Describe the bug**
Been wondering about this a while now, as oddly see almost 0 export even to decrease battery charge when negative pricing is coming.
Also noticed "Ignoring export pricing because Use Export is turned off" but not sure what this does? Is that relevant or what does this switch do or how do you flick it to true?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

SzosszeNET, 2024-08-23:
Looking at the code that would happen if I don't have an export tarrif, but I do?

18:40:46     INFO: Checking tariff prices vs Octopus Energy Integration:
18:40:46     INFO: -----------------------------------------------------
18:40:46     INFO: 
18:40:46  WARNING: pv_opt: Entity sensor.pvopt_tariff_import_OK not found in namespace default
18:40:46     INFO:   Import: Average Prices - PV_OPT:  8.38 p/kWh  OE Integration:  8.38 p/kWh  Mean difference:  0.00 p/kWh
18:40:46  WARNING: pv_opt: Entity sensor.pvopt_tariff_export_OK not found in namespace default
18:40:46     INFO:   Export: Average Prices - PV_OPT: 15.00 p/kWh  OE Integration: 15.00 p/kWh  Mean difference:  0.00 p/kWh
18:40:46     INFO: 
18:40:46     INFO: Checking tariff start and end times:
18:40:46     INFO: ------------------------------------
18:40:46     INFO:   Import: E-1R-AGILE-23-12-06-N                    Start: 2024-08-14 12:00:00+0000 End: 2024-08-24 22:00:00+0000 
18:40:46     INFO:   Export: E-1R-OUTGOING-FIX-12M-19-05-13-N         Start: 2019-05-15 23:00:00+0000 End: N/A 
18:40:46     INFO:   AGILE tariff detected. Rates will update at 16:00 daily
18:40:46     INFO:   Tariffs OK
18:40:46     INFO: 
18:40:46     INFO: Solcast forecast loaded OK
18:40:46     INFO: 
18:40:46     INFO: Getting expected consumption data for 2024-08-23 00:00:00+0000 to 2024-08-25 00:00:00+0000:
18:40:53     INFO:   - Got 14 days history from sensor.solis_bypass_load from 09-Aug 17:30 UTC to 23-Aug 17:00 UTC
18:40:53     INFO:   - 14 days was expected. OK
18:40:53     INFO:   - Estimated consumption from ['sensor.solis_house_load', 'sensor.solis_bypass_load'] loaded OK 
18:40:53     INFO:   - Total consumption: 28.0 kWh
18:40:53     INFO: 
18:40:53     INFO: Initial SOC: 97.0
18:40:53     INFO: Calculating Base flows:
18:40:54     INFO: 
18:40:54     INFO: Optimising for Solcast 30% confidence level forecast from 23-Aug 17:30 UTC to 24-Aug 23:30 UTC
18:40:54     INFO: Ignoring export pricing because Use Export is turned off
18:40:54     INFO: []
18:40:57     INFO: []
18:41:01     INFO: Optimiser prices loaded for period 23/08 17:30 UTC - 24/08 23:30 UTC
18:41:01     INFO: Base cost:  14.7
18:41:01     INFO: 
18:41:01     INFO: High Cost Usage Swaps
18:41:01     INFO: ---------------------
18:41:01     INFO: 
18:41:01     INFO: []
18:41:03     INFO: Charge net cost delta:  0.0p: < Pass Threshold (4.0p) => Slots Excluded
18:41:03     INFO: 
18:41:03     INFO: Low Cost Charging
18:41:03     INFO: ------------------
18:41:03     INFO: 
18:41:03     INFO: Max export price when there is no forced charge: 15.00p/kWh.
18:41:03     INFO: 33 slots have an import price less than the max export price
18:41:03     INFO: 32 Min import price -0.54p/kWh at 24/08 23:30 UTC    0W   SOC:  25.3%-> 21.1% Net:  13.6 New SOC:  25.3%-> 40.0% Max export: 1448W 
18:41:04     INFO: 21 Min import price 10.14p/kWh at 24/08 06:00 UTC    0W   SOC:  51.7%-> 51.0% Net:   5.7 New SOC:  51.7%-> 65.8% Max export: 1448W 
18:41:04     INFO: 19 Min import price 12.45p/kWh at 24/08 03:00 UTC    0W   SOC:  58.7%-> 57.1% Net:   0.9 New SOC:  58.7%-> 73.4% Max export: 1448W 
18:41:04     INFO: 18 Min import price 12.57p/kWh at 23/08 23:30 UTC    0W   SOC:  69.8%-> 68.9% Net:  -3.4 New SOC:  69.8%-> 84.5% Max export: 1448W 
18:41:05     INFO: 
18:41:05     INFO: Charge net cost delta: 18.1p: > Pass Threshold 4.0p => Slots Included
18:41:05     INFO: 
18:41:05     INFO: Forced Discharging
18:41:05     INFO: ------------------
18:41:05     INFO: 
18:41:05     INFO: 57 slots have an export price greater than the min import price
18:41:08     INFO: 
18:41:08     INFO: Discharge net cost delta: -0.0p: > Discharge Threshold (0.0p) => Slots included
18:41:08     INFO: Iteration  1: Slots added:   4
18:41:08     INFO: 
18:41:08     INFO: Low Cost Charging
18:41:08     INFO: ------------------
18:41:08     INFO: 
18:41:08     INFO: Max export price when there is no forced charge: 15.00p/kWh.
18:41:08     INFO: 33 slots have an import price less than the max export price
18:41:09     INFO: 
18:41:09     INFO: Charge net cost delta: -0.0p: < Pass Threshold 4.0p => Slots Excluded
18:41:09     INFO: 
18:41:09     INFO: Forced Discharging
[... 41 lines cut ...]
18:41:13     INFO: Current inverter status:
18:41:13     INFO: ------------------------
18:41:13     INFO:   mode              : Self-Use
18:41:13     INFO:   code              : 35
18:41:13     INFO:   switches          :
18:41:13     INFO:     SelfUse         : True
18:41:13     INFO:     Timed           : True
18:41:13     INFO:     OffGrid         : False
18:41:13     INFO:     BatteryWake     : False
18:41:13     INFO:     Backup          : False
18:41:13     INFO:     GridCharge      : True
18:41:13     INFO:     FeedInPriority  : False
18:41:13     INFO:   charge            :
18:41:13     INFO:     start           : 23-Aug 18:00 BST
18:41:13     INFO:     end             : 23-Aug 18:00 BST
18:41:13     INFO:     current         : 60.0
18:41:13     INFO:     active          : False
18:41:13     INFO:   discharge         :
18:41:13     INFO:     start           : 23-Aug 18:25 BST
18:41:13     INFO:     end             : 23-Aug 18:25 BST
18:41:13     INFO:     current         : 60.0
18:41:13     INFO:     active          : False
18:41:13     INFO:   hold_soc          :
18:41:13     INFO:     active          : False
18:41:13     INFO:     soc             : 100.0

philicibine, 2024-09-02:
I am seeing this also... even tho the Include Export switch is turned on.

closed 2024-11-10 by github-actions[bot]