stevebuk1/pv_opt #263: Charge/Hold during cheap/free periods

solarjones, 2024-08-23:
v3.15.5

Have I set something up incorrectly, as I can't understand why the charging plan would not force max charge current during free/cheap periods? Instead it's showing a charge window into expensive time slots.  I've turned off Use Solar option to test this out.

Would you expect it for force charge and hold charge 15:00-16:00?

[image]

[image]

[image]

[main.log](https://github.com/user-attachments/files/16729110/main.log)
[pv_opt.log](https://github.com/user-attachments/files/16729111/pv_opt.log)

Thanks!

stevebuk1, 2024-08-25:
Its because your battery is at 92% at the start of the 15:00 slot, and 2300W is enough to get it to 100% in that 1/2 hour. This is shown by this part of the log:

14:10:00     INFO: >>> Slot: 23/08 14:00 UTC Factor: 1.000 Forced:      0W  End SOC: 88.0%  SPR:   4146W  SCPA:   6000W  SAC:   2301W  Min Power:   2301W
14:10:00     INFO: 24/08 18:00 BST:  2.07 kWh at  51.95p. <==> 23/08 15:00 BST: -1.32p/kWh -2.74p  SOC:  91.5%-> 88.0% New SOC:  91.5%->100.0% Net:  532.4

Its then held at 100% until 17:00 because its been worked out that letting it discharge at say 16:00 or 16:30 would then mean there would be insufficient battery to cover a later slot that costs more than 10p and then 17p. Holding at 100%  for 16:00 to 16:30 covers off the 19:30 slot that would then would otherwise draw from grid, as shown by this part of the log:

14:10:02     INFO: >>> Slot: 23/08 15:00 UTC Factor: 1.000 Forced:      0W  End SOC: 96.5%  SPR:   1699W  SCPA:   6000W  SAC:    671W  Min Power:    671W
14:10:02     INFO: 23/08 19:30 BST:  0.85 kWh at  16.02p. <==> 23/08 16:00 BST: 10.67p/kWh  9.06p  SOC: 100.0%-> 96.5% New SOC: 100.0%->100.0% Net:  382.6

The slightly misleading part of the screengrab is that the inverter isnt actually drawing 2300W from grid for 15:00 to 17:00, its just doing it for the 1st half hour and then holding the battery at 100%.

stevebuk1, 2024-09-29:
Closing, as what is reported is expected behaviour.

closed 2024-09-29 by stevebuk1