stevebuk1/pv_opt #265: Referenced entities switch.pvopt_charge_active are missing or not currently available

philicibine, 2024-09-02:
Referenced entities switch.pvopt_charge_active are missing or not currently available 
Referenced entities switch.pvopt_discharge_active are missing or not currently available
Referenced entities switch.pvopt_hold_soc_active are missing or not currently available

Keeps being logged

Im on HassOS on rpi4.  Using Solax Modbus (latest) with an ESP32 modbus bridge on a solis hybrid inverter.

stevebuk1, 2024-09-05:
As far as I can see, these were three entities added for future development. They are created, but neither read nor set by Pv_opt. On my system they haven't appeared. Which log are the messages appearing in - Pv_opt.log or Appdeamon.log? If you could attach the log in question I'll take a look.

stevebuk1, 2024-09-29:
Hi @philicibine, did my response above answer your question?

philicibine, 2024-09-30:
I guess it did.  Thanks.

closed 2024-09-30 by philicibine

stevebuk1, 2024-10-04:
I closed this too hastily. I have now found the cause of this and put in a Beta release for testing. Will let you know when there is an update to download.