stevebuk1/pv_opt #266: Unable to locate pv_opt_log

Pyinthesky99, 2024-09-14:
I have, hopefully, installed all prerequisites and installed pv_opt via HACS but am unable to locate pv_opt_log to check that it is running. I assume it is not as there are new entities in /Settings/Devices and Services/Entities in Home Assistant.

I am running the Solarman Integration on HAOS running in a Proxmox vm accessed via Edge on Win 11.

Pyinthesky99, 2024-09-14:
[AppDaemonlog.txt](https://github.com/user-attachments/files/17002054/AppDaemonlog.txt)
no entry
"added log pv_opt-log"  in Appdaemon log

stevebuk1, 2024-09-14:
It doesnt look like appdeamon is finding your Pv_opt install. 

Section 11 of the install guide talks about editting appdeamon.yaml, here:

https://github.com/fboundy/pv_opt?tab=readme-ov-file#11-configure-appdaemon

The important line is this:
app_dir: /homeassistant/appdaemon/apps

as HACS installs Pv_opt to this location (/homeassistant/appdaemon/apps)

Check:

1) that the Pv_opt files exist at /homeassistant/appdaemon/apps/pv_opt (they are pv_opt.py, pypy.py and solis.py, plus others)
2) that your appdeamon.yaml contains a line app_dir: /homeassistant/appdaemon/apps

Note that appdeamon.yaml (and helloworld.py, which from your logs is loading correctly) is in a completely different area. Mine is in /addon_configs/a0d7b954_appdaemon/

Pyinthesky99, 2024-09-15:
Hi Steve. Thanks for the response. As below, assuming the ‘top’ CONFIG is “/homeassistant’, immediately below. in the tree is /addon_configs/a0d7b954_appdaemon/ which contains appdaemon.yaml with the necessary pv_opt changes
On the same ‘level’ is /appdaemon/apps/pv_opt/ which contains pv_opt files but critically not pv_opt_log.
Doing a linux find . -name “pv_opt_log” from root returns no result.
From root, I am assuming again, /homeassistant resides in /mnt/data/supervisor. Interestingly there is, along with /homeassistant , another(?) /addon_configs directory which contains (another?) /a0d7b954_appdaemon directory which contains an appdaemon.yaml and /compiled /dashboards /namespaces and /www directories and an /apps directory that contains apps.yaml and hello.py. This appdaemon.yaml is un-modded, it contains none of the pv_opt script, unlike the one below.

 
[image]

All very confusing, am I looking at the correct appdaemon.yaml?

stevebuk1, 2024-09-15:
Ok, really not sure about whats at `/mnt/data/supervisor`. I'm on a Pi4 and `/mnt `is empty.

Ignoring that path for the moment, your directory structure looks the same as mine, i.e.:
`/addon_configs/a0d7b954_appdaemon`
and 
`/homeassistant/appdaemon/apps/pv_opt`

The log file is called pv_opt.log (not pv_opt_log) and confusingly isnt in the pv_opt area or even the apps area, it should be here:
`/homeassistant/appdaemon/pv_opt.log`

All errors are logged to this file:
`/homeassistant/appdaemon/error.log`

Do you have these? 

If so, please post these and also this file:
`/homeassistant/appdaemon/appspv_opt/config/config.yaml`

Pyinthesky99, 2024-09-16:
Apologies, pv_opt_log was a typo, a find on pv_opt.log returned nothing.

Using the path in the previous comment to /homeassistant the only thing in /appdaemon is another folder /apps which contains folder /pv_opt, no log files so no error.log

[image]

This is a copy of pv_opt/config/config.yaml  (I haven’t the technology for a direct copy so copy/pasted via notepad)
[config.yaml.txt](https://github.com/user-attachments/files/17011054/config.yaml.txt)

stevebuk1, 2024-09-16:
Thanks for the screengrab. This is all sounding remarkably similar to this issue : #264 and specifically this post: https://github.com/fboundy/pv_opt/issues/264#issuecomment-2318202733.

I think the correct appdaemon.yaml to edit is the one at mnt/data/supervisor/addon_configs/a0d7b954_appdaemon/appdaemon.yaml, and not the one inside the homeassistant directory. 

@pbcjh10, would you mind sharing the additional init.py file you had to create and also the location of your appdaemon.yaml file and its contents?

Pyinthesky99, 2024-09-16:
Thanks for the info, I was looking at #264  earlier and thought it looked rather familiar.
I have edited the other appdaemon.yaml and will await the info from @pbcjh10 
Regards

Pyinthesky99, 2024-09-25:
@stevebuk1 For want of nothing better to do, I have added  as per @pbcjh10 issue (which now seems to have disappeared)  the pv_opt three-liner into apps.yaml , restarted HA and appdaemon and this has now appeared in the appdaemon log
[image]
It looks like progress of sorts, none of theses lines have been converted into entities and I am still unable to see any pv_opt.log

Pyinthesky99, 2024-09-25:
@stevebuk1 just reading #264 which, obviously has been closed.
Regards
Clueless newbie

stevebuk1, 2024-09-25:
Sorry you're still having difficulties. At least appdeamon has at least now found Pv_opt, but with no pv_opt.log theres probably no error.log either. 

We talked about thinking the real install of appdeamon is at `mnt/data/supervisor/addon_configs/a0d7b954_appdaemon`. 
1) Please attach appdaemon.yaml from that path
2) Please attach a screengrab from the "Configuration" tab, which is the one next to the "Log" tab that you screengrabbed above. 
3) Please attach your apps.yaml file 

Finally, if appdeamon.yaml contains the entries required by https://github.com/fboundy/pv_opt?tab=readme-ov-file#11-configure-appdaemon then there should be the following 3 log files:
/share/logs/main.log
/share/logs/error.log
/share/logs/pv_opt.log

although as you can't find pv_opt.log we may not have got as far as creating them. As far as I'm aware the content of the "Log" tab that you snapshotted before does not get stored anywhere, but I can confirm a normal startup of appdeamon does not contain any of the entries in the snapshot, so I suspect the problem here is that appdaemon.yaml is either incorrect or its not being found.

Pyinthesky99, 2024-09-25:
Hi Steve, this is a copy of the appdaemon.yaml at that location.
[appdaemon-yaml.txt](https://github.com/user-attachments/files/17137917/appdaemon-yaml.txt)
 This is a grab of the Configuration Tab
[image]
No sign of the log files, if their location is /share/logs/... what is the location of the /share folder on your system? A grab of the location on your Pi (HAOS?) might enable me to find it (when it appears). I assume appdaemon creates the /share/logs directories?
I have Windows Explorer access to the Home Assistant file system now.
Nothing concrete from Peter then?
Alan

Pyinthesky99, 2024-09-25:
This is tail-end of the previous grab
[image]
which may be of interest
I have stopped/started AppDaemon and this is a text of the log output.
[Stop-Start AppDaemon.txt](https://github.com/user-attachments/files/17138472/Stop-Start.AppDaemon.txt)
Do the 'entity' names listed appear on your system in Settings/Devices and Services/Entities(tab)?
Alan

stevebuk1, 2024-09-25:
Appdeamon.yaml looks fine. 

/share is effectively off root, i.e. alongside config and homeassistant (these latter two being the same directory):

[image]
My appdaemon.yaml file redirects my logs to homeassistant/appdaemon. This is only because I'd installed predbat and that was the default location for them. Might be worth a try? Here is my appdaemon.yaml:

```
# write log records to a file, retaining 9 versions, rather than the standard appdaemon log
logs:
  main_log:
    filename: /homeassistant/appdaemon/appdaemon.log
    date_format: '%H:%M:%S'
    log_generations: 9
    log_size: 10000000
  error_log:
    filename: /homeassistant/appdaemon/error.log
    date_format: '%H:%M:%S'    
  pv_opt_log:
     name: PV_Opt
     filename: /homeassistant/appdaemon/pv_opt.log
     log_generations: 9
     log_size: 10000000
     date_format: '%H:%M:%S'      
     format: '{asctime} {levelname:>8s}: {message}'
```

Anyway, your screengrab shows you are definitely up and running, although very annoying that that output isnt being redirected to a log file. I can see enough to see the problem however. 

Pv_opt is looking for entities that provide consumption data. Its looking for:

sensor.solis_consumption_today  (which is in your config.yaml)
sensor.solis_house_load_today     (which is the default)

but can't find either of them.  

You'll need to find the sensor in HA that reports house consumption. I'm using the Solax Modbus integration, and mine is called sensor.solis_house_load_today:

[image]

Are you using the Solax modbus integration for inverter control? ((https://github.com/wills106/homeassistant-solax-modbus)

Pyinthesky99, 2024-09-26:
Thanks for all that info.
No I am currently running Solarman but was considering swapping to Solax Modbus
config.yaml suggests id_consumption_today: sensor.{device_name}_daily_house_backup_consumption
The entity concerned generated by Solarman is sensor.solis_daily_house_backup_load_consumption so have put that into config.yaml then stop/started appdaemon. It now appears to get a reading of some sort but I get a bunch more errors in the log. See below
[image]

stevebuk1, 2024-09-26:
The default is the Solax integration. You'll need to uncomment this line in
config.yaml  for Solarman:

# inverter_type: SOLIS_SOLARMAN

Pyinthesky99, 2024-09-26:
Thanks Steve
Have done, progress, but....
[image]

Pyinthesky99, 2024-09-26:
Interesting it says pv_opt_log not found, not pv_opt.log  That was where I came in!

stevebuk1, 2024-09-26:
Indeed you did! Pv_opt_log is the variable that references the log,
ov_opt.lig is the actual filename.  I guess changing the create path to
homeassistant/appdeamon didn't fix the problem?

stevebuk1, 2024-09-26:
> Thanks Steve
> Have done, progress, but....
> image.png (view on web)

Progress! I'm at work so will take a look this eve. Please post your latest
config.yaml and also a complete list of HA entities for the Solarman
integration.

Note, in section 5 it does say inverter writes with Solarman are currently
disabled. This may well be the time to move to Solax Modbus or HA core
Modbus which are fully working with multiple users. I use the former.

Pyinthesky99, 2024-09-26:
It probably doesn't help that I have two appdaemon.yaml files, one under /addon_configs/a0176954_appdaemon/...  and one under /Config (Home Assistant)/addon_configs/a0176954_appdaemon/...
I have changed both to direct the log files to the location you use but still with the same result.
Thanks very much for your continued work on this. I have also installed pySolarman alongside Solarman, with the help of Gordon Markus and his youtube vid, with a reasonable degree of success as far as writing to the inverter is concerned, but point taken. I wonder what peter (#264) was using.

Pyinthesky99, 2024-09-26:
[config.yaml.txt](https://github.com/user-attachments/files/17149867/config.yaml.txt)
I have 57 entities in Solarman, 
[Solarman Entities List.docx](https://github.com/user-attachments/files/17150022/Solarman.Entities.List.docx)

stevebuk1, 2024-09-26:
Ok suggest we carry on with Solarman for now, you won't need write access for your inverter in readonly mode. (Note the issue here is with Pv_opt controlling Solarman, not Solarman itself). 

Re config.yaml, you've now uncommented inverter_type: SOLIS_SOLARMAN but you still have uncommented assignments within the Solax_modbus setup area. These may now conflict with the Solarman setup area, so best to comment out the following lines:

  id_consumption_today: sensor.{device_name}_consumption_today
  id_consumption:
    - sensor.{device_name}_house_load
    - sensor.{device_name}_bypass_load

Re appdaemon.yaml - I see the problem, sorry I missed it earlier. The "logs:" entry must not be indented, i.e. no spaces. 
Remove the 2 spaces from "logs:" and the same number of spaces from all the next lines, so it looks like mine.

[image]

stevebuk1, 2024-09-26:
> Thanks Steve
> Have done, progress, but....

Your image shows a bug in the Pv_opt code for the Solarman integration. 

Are you ok in replacing files directly? If so, overwrite solis.py with this one,:

https://github.com/stevebuk1/pv_opt/blob/patch2/apps/pv_opt/solis.py

Pyinthesky99, 2024-09-26:
Thanks very much for this info, much agree about pursuing with Solarman, particularly considering the amount of hours getting that and pysolarman into a reasonable state. Also didn't want to disappear down another rabbit hole.  Will apply your suggestions and let you know how it goes later tomorrow.
Indents and yaml, who'd have thunk it :-)
Take the rest of the day off!
Regards
Alan
Alan

Pyinthesky99, 2024-09-26:
@stevebuk1 see above

stevebuk1, 2024-09-26:
> Indents and yaml, who'd have thunk it :-)

Yaml and python are based on indents, It saves all the curly bracket stuff that C needs, but its easy to get wrong on copy and paste. I'm sure that will solve the logging issue. 

Happy to continue on Solarman route for now until at least you have something up and running, but I'm unsure myself of how mature the Solarman development is and you might be facing Beta development of Pv_opt for it.  

Don't worry about tagging me; I get a notification on this thread regardless.

fboundy, 2024-09-27:
Hi both - I haven’t been following all the details but  a couple of points to bear in mind:

• If you are using Solarman then presumably you are using an older DLS-W dongle. These do not work with the Solax integration or anything that directly uses Modbus over TCP so you have to stick with Solarman
• PV_opt should work ok with Solarman but if I recall correctly only one other user has configured it. You might be able to see who by looking back through GitHub issues.

stevebuk1, 2024-09-27:
@fboundy thanks for the background. 

I found the previous issue, its #48 . Looks like it progressed to a first test of v3.0.0, but the trail went quiet after that so no real idea whether it works or not. Also the issue confirms what is stated in the documentation, namely that writes to the inverter remaim disabled. Note there is definitely some tweaks required as "online" is missing from  INVERTER_DEFS for Solarman in solis.py. I've created a Patch2 branch in my fork with this fix in. If it becomes involved I'll create a seperate feature branch. 

@Pyinthesky99 , getting an entities list will be necessary so Pv_opt can be updated. You'd attached a word doc, but that only included 27 of the 57 entities you mentioned and there don't appear to be any that are related to setting charge and discharge times. Can you have another go at this please? Not urgent as we've got a bit of work to go before we are ready for Pv_opt to do inverter writes anyway.

Pyinthesky99, 2024-09-27:
We have logs!!
The indents did the trick. Apologies about the entities, don't know what happened there, Here should be all 59 entities
[Solarman Entities.docx](https://github.com/user-attachments/files/17166882/Solarman.Entities.docx)
Appdaemon started clean after sorting out the logs.
Then copied solis.py and have noticed no visible difference. Are most of the dashboard entities relating to those of Solax Integrations?
Thanks for the pointers @fboundy

stevebuk1, 2024-09-27:
> We have logs!!

Great! 

> Are most of the dashboard entities relating to those of Solax Integrations?

I don't think any are. All of the dashboard entities are generated/updated by Pv_opt, but as its almost the very last step nothing will appear until everything before it is working. This is why the logs are essential. 

Can you attach pv_opt.log and error.log so I can take a look at how far we've got.

stevebuk1, 2024-09-27:
> I don't think any are. All of the dashboard entities are generated/updated by Pv_opt

Apologies, slight correction, most are. You'll know if they are as they will include "pvopt" in the name, e.g. sensor.pvopt_charge_current. If your dashboard is not blank, you'll find the bar graphs in the Status window may say "entity not found" and these can be modified now to suit the entity names that Solis SolarMan supplies.

Pyinthesky99, 2024-09-28:
Thanks for your efforts
[error.log](https://github.com/user-attachments/files/17173968/error.log)
[pv_opt.log](https://github.com/user-attachments/files/17173969/pv_opt.log)

Here are the logs.

stevebuk1, 2024-09-28:
We've got further. Its now looking for id_consumption as sensor.solis_house_load and can't find it. 

Have you definitely commented out all 3 of these lines in config.yaml in the SOLIS_SOLAX_MODBUS area?

```
  id_consumption:
    - sensor.{device_name}_house_load
    - sensor.{device_name}_bypass_load
```

```
Make it look like this:

#  id_consumption:
#   - sensor.{device_name}_house_load
#  - sensor.{device_name}_bypass_load
```
Not Pv_opt doesnt actually need "id_consumption" to load as long as it finds "id_consumption_today" (sensor.solis_daily_house_backup_load_consumption) which I can tell from your logs that it has. But if you give it an HA entity to look for and its wrong, it won't start.

Pyinthesky99, 2024-09-28:
These are in config.yaml at lines 172-174 in the fboundy section where they are commented out. But they also appear in the wills106 Solax_Modbus section at lines 134-136 where they are not commented out. I have now commented these out. There is also id_consumption _today at line 133, which I have also commented out as I have the same id at line 207 in the Stephan Joubert section which I had previously un-commented out!
[error.log](https://github.com/user-attachments/files/17174670/error.log)
[pv_opt.log](https://github.com/user-attachments/files/17174671/pv_opt.log)
[config.yaml (2).txt](https://github.com/user-attachments/files/17174676/config.yaml.2.txt)

Here are updated logs and updated config.yaml (text)

Pyinthesky99, 2024-09-28:
seems to be looking for pv_total_power which is lurking in the sunsynk section. better comment that out too?!

Pyinthesky99, 2024-09-28:
have and we now have a few more sensors!!

stevebuk1, 2024-09-28:
> have and we now have a few more sensors!!

Yep, one by one. 

The next failure is because of this line right at the bottom of config.yaml:

id_solar_power: sensor.{device_name}_pv_total_power

You'll need to comment this out (as your solar power sensor is 'sensor.total_solis_power' which should be picked up by default)

stevebuk1, 2024-09-28:
If you've already done the commenting, attach a new pv_opt.log so I can see where the next problem is.

Pyinthesky99, 2024-09-28:
[pv_opt.log](https://github.com/user-attachments/files/17175447/pv_opt.log)
Here is a recent pv_opt.log
There are no entities remotely resembling timed charge in my list in settings. If I understand this correctly, which I probably don't, the entity needs to be first created by Solarman so that Pv_opt can locate and id it?
On the same theme, I have been looking at the dashboard and it obviously wont load. .Some days ago,  I chopped it down to the first single card which loaded with ‘unable to find entity’  errors. I replaced sensor.solis_pv_total_power with sensor.solis_inverter_dc_power. I was spoilt for choice as I had that and sensor.solis_inverter_ac_power, sensor.solis_inverter_active_power, sensor.solis_inverter_apparent_power and sensor.solis_inverter_reactive_power to chose from! But I did get some semblance of a dashboard with some creative licence. There are quite a few items that appear on Solis Cloud that are not that easy to find in HA entities.
[image]
Are these timed 'register items' that can be read? Perhaps Stephan and the Solarman repo had no use for them as it wasn't ever intending to write them.

Pyinthesky99, 2024-09-28:
Minor issue, also cannot find sensor.pvopt_charge_current, or something suitable to replace sensor.solis_battery_current or sensor.solis_battery_current which appear in the dashboard config.

stevebuk1, 2024-09-28:
> [pv_opt.log](https://github.com/user-attachments/files/17175447/pv_opt.log)
> Here is a recent pv_opt.log

Thanks. You are through all of the entity reading, consumptions are being read and the Optimser is running every 10 minutes but stopping when calculating costs.  Can you post error.log? 

> There are no entities remotely resembling timed charge in my list in settings. If I understand this correctly, which I probably don't, the entity needs to be first created by Solarman so that Pv_opt can locate and id it?

Yes thats basically it.  Pc_opt has no method of controlling the inverter, thats Solarmans job. It modifies Solarmans entities to program the inverter. I assume you've used Solarman on its own to programme your inverter to charge at a particular time? If so, just describe how you'd go about it. If not, you will have to get Solarman to be able to do it. 

> On the same theme, I have been looking at the dashboard and it obviously wont load. .Some days ago, I chopped it down to the first single card which loaded with ‘unable to find entity’ errors. I [...]

Yes, the Dashboard is setup on Solax modbus names. You'll need edit the dashboard to use your names. I'm afraid niether of them bear much resemblence to either SolisCloud or the inverter itself! - its a bit trial and error. 

> Are these timed 'register items' that can be read? Perhaps Stephan and the Solarman repo had no use for them as it wasn't ever intending to write them.

Here is the excerpt from Solarmans "solis_hybrid.yaml" file I mentioned earlier. The register values are there and the naming suggest that the file should generates entities. If you can't programme the inverter to charge at a particular time slot from HA as it stands, Pv_opt can't help you. 

```
 - group: TimedCharge
   items: 
    - name: "Timed Charge Current"
      class: ""
      state_class: "measurement"
      uom: "A"
      scale: 0.1
      rule: 1
      registers: [43141]
      icon: 'mdi:wrench-clock'

    - name: "Timed Discharge Current"
      class: ""
      state_class: "measurement"
      uom: "A"
      scale: 0.1
      rule: 1
      registers: [43142]
      icon: 'mdi:wrench-clock'

    - name: "Timed Charge Start Hour"
      class: ""
      state_class: "measurement"
      uom: "H"
      scale: 1
      rule: 1
      registers: [43143]
      icon: 'mdi:wrench-clock'

    - name: "Timed Charge Start Minute"
      class: ""
      state_class: "measurement"
      uom: "M"
      scale: 1
      rule: 1
      registers: [43144]
      icon: 'mdi:wrench-clock'

    - name: "Timed Charge End Hour"
      class: ""
      state_class: "measurement"
      uom: "H"
      scale: 1
      rule: 1
      registers: [43145]
      icon: 'mdi:wrench-clock'

    - name: "Timed Charge End Minute"
      class: ""
      state_class: "measurement"
      uom: "M"
      scale: 1
      rule: 1
      registers: [43146]
      icon: 'mdi:wrench-clock'
```

stevebuk1, 2024-09-28:
> Minor issue, also cannot find sensor.pvopt_charge_current, or something suitable to replace sensor.solis_battery_current or sensor.solis_battery_current which appear in the dashboard config.

Don't change any entity in the dashboard with "pvopt" in the name. These will come from Pv_opt, but your installation hasn't got far enough to create them in HA yet. Its really the very last step after the optimiser run has completed. 

Note: sensor.pvopt_charge_current is what Pv_opt will program the next charge slot with. At the right time, Pv_opt will write that value to a Solarman entity which in turn will program your inverter - but we haven't found the Solarman entity that does this.

Pyinthesky99, 2024-09-28:
Thanks immensely for all this and your efforts, will look at it tomorrow.
here is latest error.log
[error.log](https://github.com/user-attachments/files/17176640/error.log)
Do you ever think, blow all round this, set to charge 2:30-3:00 and 3:30-5:00 every winter evening, perhaps 14:30-15:00 to ensure passing the dreaded 4 to 7 rates, and adjourn to the pub!
Kind regards
Alan

stevebuk1, 2024-09-28:
> here is latest error.log
> [error.log](https://github.com/user-attachments/files/17176640/error.log)

Thanks - I can see the problem. Will have a look tomorrow.

> Do you ever think, blow all round this, set to charge 2:30-3:00 and 3:30-5:00 every winter evening, perhaps 14:30-15:00 to ensure passing the dreaded 4 to 7 rates, and adjourn to the pub!

Yep! Welcome to the life of HomeAssistant - it eats time! My main reason for using Pv_opt is really EV charging - I need something to prevent house battery discharge during EV charge. I'm on IOG so it needs to deal with extra slots outside the core 6 hours. But to be honest, if it wasnt for that I'd have probably said just charge the battery fully overnight. Agile is however a different case - perhaps worth it in the end? Home Automation has to be just that - automation. The goal is no manual intervention and when its all up and running I do find its genuinely just that - I now plugin the car and everything is automatic.

Pyinthesky99, 2024-09-28:
Fair enough, I suppose the answer to my point would be "where's the fun in that"!
You've probably seen this, copied from a Home Assistant group, posted a few times on Solis Facebook groups............

So you decided to install Home Assistant, huh? Congratulations on your new full-time job as a "smart home systems integrator," where your main responsibility is praying your Wi-Fi doesn't decide to take a nap. 
Welcome to the world where your light switches will now have more complicated relationships than your friends on Facebook. It’s like your house got a PhD in neediness, and now every single device has its own personality disorder. 
You wanted a smart home, and instead, you got a *high-maintenance* home. Your lights flicker like they’re at a rave because the latest update decided your firmware wasn’t moody enough. Speaking of updates, every time there's a new one, it's like rolling the dice—will your home be smarter or will you spend the next 48 hours trying to figure out why your fridge thinks it's a speaker?
Oh, and that dashboard you spent hours customizing? Hope you enjoy re-doing it from scratch every time Home Assistant decides to "improve" its interface. But hey, at least you'll always know what your neighbor’s Wi-Fi signal is, because your automations will probably connect to that by mistake, just for fun.
But sure, it's totally worth it, right? I mean, who wouldn't want to control their lights from their phone in three extra steps instead of just flipping a switch like a normal person? Welcome to the future, where your house is smarter than you—and knows it.
So you’ve committed to Home Assistant, the one platform that turns your home into a glorified science project. You know, because life wasn’t already complicated enough. Now you get to explain to guests why your "smart" house has a nervous breakdown every time someone tries to turn on a light manually. 
Home Assistant is like that friend who knows everything but never shuts up about it—constantly needing your attention for updates, bug fixes, or just because it decided to forget how to connect to your TV. Want to dim the lights? Better hope you remembered the exact YAML syntax, or you'll be sitting in the dark, googling "why won't my smart lights work?"
And speaking of YAML, who doesn’t love a configuration language where a single misplaced space can ruin your entire weekend? It's like you're coding for NASA, but the mission is just getting your thermostat to not go haywire at 3 AM. You’ve basically become a full-stack developer just to turn off your porch light.
Let’s not forget about the *joy* of adding new devices. Every time you introduce a new gadget, it’s like bringing home a new pet, except this one speaks a different language and might randomly decide to stop working out of sheer spite. But sure, it’s super convenient when it’s all working perfectly—which is approximately never.
So go ahead, pour another cup of coffee, because you're going to need it to stay awake through another night of debugging automations that worked perfectly fine until you dared to breathe in the same room as your router. Your home may be smart, but you? You’re just tired."

 Many a true word, etc..... :-)

Thanks for the Solarman info, as per Gordon Markus' investigations Solarman needs another device running pysolarman5 to write to the inverter, the traffic being passed from Solarman and pysolarman5 by MQTT,  Don't know if the py under AppDaemon is quite capable of that, I am using a vm running RaspPiOS. Possibly above my pay grade and limited abilities. Will have a look in the clear(er) light of day.
Thanks again for all your help.
Alan

Pyinthesky99, 2024-09-29:
Progress, well regress actually. Those entities you have listed from solarman do not exist in the solis_hybrid.yaml in my installation, downloaded maybe a year or so ago. So, I downloaded the latest version from github, renamed my solis_hybrid.yaml to xsolis_hybrid.yaml and copied the new version to my HA. After restart the new entities are listed but all entities now show their values as unknown. Copying back restores values. Another rabbit hole. I have started a side-by-side-compare and nothing significantly different  so I can copy the relevant lines across, or ask the question on Facebook, or raise an issue on Solarman github, unless you can offer a simple answer as I am doing something clueless.
Regards
Alan

stevebuk1, 2024-09-29:
We can change the device name in PvOpt so it finds xsolis. Will take a look
at that and the other error this eve.

Pyinthesky99, 2024-09-29:
thanks again Steve

pbcjh10, 2024-09-29:
@stevebuk1 do you still want me to post my logs or are we past it now? Sorry for the late reply, we were out of the house having builders in!

Hoping I can piggyback here (let me know if you want me to make a new thread) because I think I'm at a similar point now, in that I think I'm running, I have a log, but not generating a charge/discharge plan;

[image]

I attach all the logs!
[pv_opt.log](https://github.com/user-attachments/files/17179123/pv_opt.log)
[main.log](https://github.com/user-attachments/files/17179124/main.log)
[error.log](https://github.com/user-attachments/files/17179125/error.log)

Pyinthesky99, 2024-09-29:
hi @pbcjh10  !!
Welcome to the party. I assume you are running Solarman Integration. This is a bit of a read but I seem to have pv_opt running but entities are missing when I try and build the dashboard. It looks like I am running an older version of solis_hybrid.yaml compared to the one now on Stephan Joubert's repo on github but if I copy it across all entities go unavailable. Sounds familiar? :-)

pbcjh10, 2024-09-29:
When I copied Stephen's repo the dashboard didn't work until I edited in my Octopus details, once I did that it sprang to life! I'm just stuck on the last step and unsure why pvopt isn't generating the opt cost and base cost sensors

stevebuk1, 2024-09-29:
@pbcjh10 , welcome, thanks for the logs and yes feel free to piggyback here.

By the sounds of it you are both at the same stage (or were until all of Alan's entities got renamed form Solis to xsolis). Basically the Solarman defaults for the entities in PvOpt are a guess, so we need to edit config.yaml to map to your actual sensors until everything is working. Once that's done I'll edit the main code with the new defaults.

pbcjh10, 2024-09-29:
I am Solax modbus not Solarman! 
So I have 196 entities loaded, I thought that my logs/screenshot suggested that it was a pv opt issue not a sensor naming problem?

fboundy, 2024-09-29:
Steve

You should now have write access to the main repo so happy for you to update as you see fit. This stuff should probably go in a separate branch to your ongoing EV dev branch.

If you’re not able to access for some reason let me know.

Cheers

Francis

Pyinthesky99, 2024-09-29:
Hi Steve. Just to be clear, I renamed solis_hybrid.yaml  to xsolis_hybrid.yaml as a backup before copying solis_hybrid.yaml from github to my HA. All Solarman entities then went unavailable until I 'removed' the new solis_hybrid.yaml and renamed xsolis_hybrid.yaml back to solis_hybrid.yaml in my HA,  then all entity reading came back, so SNAFU!
I may have to bite the bullet and reload Solarman in it's entirety.

Pyinthesky99, 2024-09-29:
Have removed Solarman from Settings/Integrations and HACs and downloaded about three times but still only get 59 entities. I downloaded the zip from github Solarman and manually copied inverter_definitions folder to my HA, I then got 70 odd entities but none of them loaded with a value. Another remove and reload and we are back to 59 working.

stevebuk1, 2024-09-29:
> Steve
> 
> You should now have write access to the main repo so happy for you to update as you see fit. This stuff should probably go in a separate branch to your ongoing EV dev branch.
> 
> If you’re not able to access for some reason let me know.
> 
> Cheers
> 
> Francis
> > I am Solax modbus not Solarman!
> > So I have 196 entities loaded, I thought that my logs/screenshot suggested that it was a pv opt issue not a sensor naming problem?

Thanks for the collaborator access. I've started a Patch2 branch at my fork for the Solarman changes (I avoided using the name patch as I see you have one in use that has commits and needs sync already.  For now I'll do everything in my fork and then open PRs when ready. I assume as long as it's tested then it's ok to release to main without doing a beta?

stevebuk1, 2024-09-29:
> Hi Steve. Just to be clear, I renamed solis_hybrid.yaml  to xsolis_hybrid.yaml as a backup before copying solis_hybrid.yaml from github to my HA. All Solarman entities then went unavailable until I [...]

Ok understood now. Did the additional entities include anything to do with charge and discharge?

stevebuk1, 2024-09-29:
> Have removed Solarman from Settings/Integrations and HACs and downloaded about three times but still only get 59 entities. I downloaded the zip from github Solarman and manually copied [...]

Annoying! I'd open an issue in Solarman for this - I can't test Solarman as I don't think my data stick is compatible.

Stick with the 59 working entities until we get the optimiser running correctly and all your dashboard entities populated.

stevebuk1, 2024-09-29:
> I am Solax modbus not Solarman! 
> So I have 196 entities loaded, I thought that my logs/screenshot suggested that it was a pv opt issue not a sensor naming problem?

If you are running Solax then the issues you have are probably unrelated to this thread. It would be easier to track if you opened another issue.with where you are at now.

[92 more replies]

closed 2024-10-11 by stevebuk1