stevebuk1/pv_opt #272: AppDaemon main.log WARNING HASS: Code: 400, error: {"message":"Invalid JSON specified."}"

philicibine, 2024-10-05:
Seeing ```06:59:43 WARNING HASS: Code: 400, error: {"message":"Invalid JSON specified."}"``` over and over in my main AppDaemon log.  Any ideas what is causing it?  The JSON is below

(Ommitted code tags so its easier to see)

06:59:43 WARNING HASS: Error setting Home Assistant state default.sensor.pvopt_opt_cost_Flux, {'state': -0.0, 'attributes': {'state_class': 'measurement', 'device_class': 'monetary', 'unit_of_measurement': 'GBP', 'friendly_name': 'PV Opt Comparison Flux', 'net_base': 0.1, 'soc': [{'period_start': '2024-10-04T01:00:00+01:00', 'soc': 20.416666666666668}, {'period_start': '2024-10-04T01:30:00+01:00', 'soc': 20.0}, {'period_start': '2024-10-04T02:00:00+01:00', 'soc': 20.0}, {'period_start': '2024-10-04T02:30:00+01:00', 'soc': 22.43131313131313}, {'period_start': '2024-10-04T03:00:00+01:00', 'soc': 24.816161616161615}, {'period_start': '2024-10-04T03:30:00+01:00', 'soc': 27.160606060606064}, {'period_start': '2024-10-04T04:00:00+01:00', 'soc': 29.463636363636365}, {'period_start': '2024-10-04T04:30:00+01:00', 'soc': 31.729292929292928}, {'period_start': '2024-10-04T05:00:00+01:00', 'soc': 33.95353535353535}, {'period_start': '2024-10-04T05:30:00+01:00', 'soc': 32.23535353535354}, {'period_start': '2024-10-04T06:00:00+01:00', 'soc': 29.679797979797982}, {'period_start': '2024-10-04T06:30:00+01:00', 'soc': 27.5989898989899}, {'period_start': '2024-10-04T07:00:00+01:00', 'soc': 23.299999999999997}, {'period_start': '2024-10-04T07:30:00+01:00', 'soc': 20.147474747474746}, {'period_start': '2024-10-04T08:00:00+01:00', 'soc': 20.0010101010101}, {'period_start': '2024-10-04T08:30:00+01:00', 'soc': 21.872727272727275}, {'period_start': '2024-10-04T09:00:00+01:00', 'soc': 22.55757575757576}, {'period_start': '2024-10-04T09:30:00+01:00', 'soc': 29.924242424242426}, {'period_start': '2024-10-04T10:00:00+01:00', 'soc': 43.61919191919192}, {'period_start': '2024-10-04T10:30:00+01:00', 'soc': 56.53030303030303}, {'period_start': '2024-10-04T11:00:00+01:00', 'soc': 65.07171717171718}, {'period_start': '2024-10-04T11:30:00+01:00', 'soc': 79.48787878787878}, {'period_start': '2024-10-04T12:00:00+01:00', 'soc': 87.48989898989899}, {'period_start': '2024-10-04T12:30:00+01:00', 'soc': 100.0}, {'period_start': '2024-10-04T13:00:00+01:00', 'soc': 100.0}, {'period_start': '2024-10-04T13:30:00+01:00', 'soc': 100.0}, {'period_start': '2024-10-04T14:00:00+01:00', 'soc': 100.0}, {'period_start': '2024-10-04T14:30:00+01:00', 'soc': 100.0}, {'period_start': '2024-10-04T15:00:00+01:00', 'soc': 100.0}, {'period_start': '2024-10-04T15:30:00+01:00', 'soc': 100.0}, {'period_start': '2024-10-04T16:00:00+01:00', 'soc': 100.0}, {'period_start': '2024-10-04T16:30:00+01:00', 'soc': 100.0}, {'period_start': '2024-10-04T17:00:00+01:00', 'soc': 100.0}, {'period_start': '2024-10-04T17:30:00+01:00', 'soc': 98.17171717171718}, {'period_start': '2024-10-04T18:00:00+01:00', 'soc': 95.37373737373738}, {'period_start': '2024-10-04T18:30:00+01:00', 'soc': 92.97272727272727}, {'period_start': '2024-10-04T19:00:00+01:00', 'soc': 90.99292929292928}, {'period_start': '2024-10-04T19:30:00+01:00', 'soc': 88.83939393939394}, {'period_start': '2024-10-04T20:00:00+01:00', 'soc': 80.12626262626263}, {'period_start': '2024-10-04T20:30:00+01:00', 'soc': 70.72828282828283}, {'period_start': '2024-10-04T21:00:00+01:00', 'soc': 64.93535353535353}, {'period_start': '2024-10-04T21:30:00+01:00', 'soc': 57.48585858585859}, {'period_start': '2024-10-04T22:00:00+01:00', 'soc': 57.48585858585859}, {'period_start': '2024-10-04T22:30:00+01:00', 'soc': 57.48585858585859}, {'period_start': '2024-10-04T23:00:00+01:00', 'soc': 57.48585858585859}, {'period_start': '2024-10-04T23:30:00+01:00', 'soc': 57.48585858585859}, {'period_start': '2024-10-05T00:00:00+01:00', 'soc': 57.48585858585859}, {'period_start': '2024-10-05T00:30:00+01:00', 'soc': 57.48585858585859}], 'forced': [{'period_start': '2024-10-04T01:00:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T01:30:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T02:00:00+01:00', 'forced': 529}, {'period_start': '2024-10-04T02:30:00+01:00', 'forced': 519}, {'period_start': '2024-10-04T03:00:00+01:00', 'forced': 510}, {'period_start': '2024-10-04T03:30:00+01:00', 'forced': 501}, {'period_start': '2024-10-04T04:00:00+01:00', 'forced': 493}, {'period_start': '2024-10-04T04:30:00+01:00', 'forced': 484}, {'period_start': '2024-10-04T05:00:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T05:30:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T06:00:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T06:30:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T07:00:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T07:30:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T08:00:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T08:30:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T09:00:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T09:30:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T10:00:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T10:30:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T11:00:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T11:30:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T12:00:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T12:30:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T13:00:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T13:30:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T14:00:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T14:30:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T15:00:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T15:30:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T16:00:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T16:30:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T17:00:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T17:30:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T18:00:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T18:30:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T19:00:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T19:30:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T20:00:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T20:30:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T21:00:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T21:30:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T22:00:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T22:30:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T23:00:00+01:00', 'forced': 0}, {'period_start': '2024-10-04T23:30:00+01:00', 'forced': 0}, {'period_start': '2024-10-05T00:00:00+01:00', 'forced': 0}, {'period_start': '2024-10-05T00:30:00+01:00', 'forced': 0}], 'import': [{'period_start': '2024-10-04T01:00:00+01:00', 'import': 25.21008}, {'period_start': '2024-10-04T01:30:00+01:00', 'import': 25.21008}, {'period_start': '2024-10-04T02:00:00+01:00', 'import': 15.12609}, {'period_start': '2024-10-04T02:30:00+01:00', 'import': 15.12609}, {'period_start': '2024-10-04T03:00:00+01:00', 'import': 15.12609}, {'period_start': '2024-10-04T03:30:00+01:00', 'import': 15.12609}, {'period_start': '2024-10-04T04:00:00+01:00', 'import': 15.12609}, {'period_start': '2024-10-04T04:30:00+01:00', 'import': 15.12609}, {'period_start': '2024-10-04T05:00:00+01:00', 'import': 25.21008}, {'period_start': '2024-10-04T05:30:00+01:00', 'import': 25.21008}, {'period_start': '2024-10-04T06:00:00+01:00', 'import': 25.21008}, {'period_start': '2024-10-04T06:30:00+01:00', 'import': 25.21008}, {'period_start': '2024-10-04T07:00:00+01:00', 'import': 25.21008}, {'period_start': '2024-10-04T07:30:00+01:00', 'import': 25.21008}, {'period_start': '2024-10-04T08:00:00+01:00', 'import': 25.21008}, {'period_start': '2024-10-04T08:30:00+01:00', 'import': 25.21008}, {'period_start': '2024-10-04T09:00:00+01:00', 'import': 25.21008}, {'period_start': '2024-10-04T09:30:00+01:00', 'import': 25.21008}, {'period_start': '2024-10-04T10:00:00+01:00', 'import': 25.21008}, {'period_start': '2024-10-04T10:30:00+01:00', 'import': 25.21008}, {'period_start': '2024-10-04T11:00:00+01:00', 'import': 25.21008}, {'period_start': '2024-10-04T11:30:00+01:00', 'import': 25.21008}, {'period_start': '2024-10-04T12:00:00+01:00', 'import': 25.21008}, {'period_start': '2024-10-04T12:30:00+01:00', 'import': 25.21008}, {'period_start': '2024-10-04T13:00:00+01:00', 'import': 25.21008}, {'period_start': '2024-10-04T13:30: [... cut ...] ar': 0.0}, {'period_start': '2024-10-04T21:00:00+01:00', 'solar': 0.0}, {'period_start': '2024-10-04T21:30:00+01:00', 'solar': 0.0}, {'period_start': '2024-10-04T22:00:00+01:00', 'solar': 0.0}, {'period_start': '2024-10-04T22:30:00+01:00', 'solar': 0.0}, {'period_start': '2024-10-04T23:00:00+01:00', 'solar': 0.0}, {'period_start': '2024-10-04T23:30:00+01:00', 'solar': 0.0}, {'period_start': '2024-10-05T00:00:00+01:00', 'solar': 0.0}, {'period_start': '2024-10-05T00:30:00+01:00', 'solar': 0.0}]}}

philicibine, 2024-10-05:
Seeing the same for "default.sensor.pvopt_opt_cost_Eco7_Fix," and also "default.sensor.pvopt_opt_cost_Agile_Fix"

fboundy, 2024-10-28:
Is this problem still ive?

philicibine, 2024-10-30:
Hi, apologies... have been away for a cpl of days.  From what I recall it seemed to happen during the night / first thing in the morning.  So will check tonight and tomorrow and get back to you

philicibine, 2024-10-31:
Hi.  I just double checked and can see the following from this morning.

05:58:37 WARNING HASS: Error setting Home Assistant state default.sensor.pvopt_opt_cost_Agile_Fix, {'state': 0.42, 'attributes': {'state_class': 'measurement', 'device_class': 'monetary', 'unit_of_measurement': 'GBP', 'friendly_name': 'PV Opt Comparison Agile_Fix', 'net_base': 0.42, 'soc': [{'period_start': '2024-10-30T00:00:00+00:00', 'soc': 74.0}, {'period_start': '2024-10-30T00:30:00+00:00', 'soc': 72.98080808080807}, {'period_start': '2024-10-30T01:00:00+00:00', 'soc': 71.16464646464647}, {'period_start': '2024-10-30T01:30:00+00:00', 'soc': 70.08181818181819}, {'period_start': '2024-10-30T02:00:00+00:00', 'soc': 69.08585858585859}, {'period_start': '2024-10-30T02:30:00+00:00', 'soc': 68.35757575757575}, {'period_start': '2024-10-30T03:00:00+00:00', 'soc': 67.27474747474747}, {'period_start': '2024-10-30T03:30:00+00:00', 'soc': 66.04949494949494}, {'period_start': '2024-10-30T04:00:00+00:00', 'soc': 65.0070707070707}, {'period_start': '2024-10-30T04:30:00+00:00', 'soc': 64.13232323232324}, {'period_start': '2024-10-30T05:00:00+00:00', 'soc': 63.02626262626263}, {'period_start': '2024-10-30T05:30:00+00:00', 'soc': 62.06969696969696}, {'period_start': '2024-10-30T06:00:00+00:00', 'soc': 60.76868686868687}, {'period_start': '2024-10-30T06:30:00+00:00', 'soc': 59.80707070707071}, {'period_start': '2024-10-30T07:00:00+00:00', 'soc': 58.84040404040404}, {'period_start': '2024-10-30T07:30:00+00:00', 'soc': 57.791919191919185}, {'period_start': '2024-10-30T08:00:00+00:00', 'soc': 57.6030303030303}, {'period_start': '2024-10-30T08:30:00+00:00', 'soc': 57.78383838383839}, {'period_start': '2024-10-30T09:00:00+00:00', 'soc': 58.44242424242424}, {'period_start': '2024-10-30T09:30:00+00:00', 'soc': 60.26262626262626}, {'period_start': '2024-10-30T10:00:00+00:00', 'soc': 62.02828282828283}, {'period_start': '2024-10-30T10:30:00+00:00', 'soc': 65.54242424242423}, {'period_start': '2024-10-30T11:00:00+00:00', 'soc': 68.05454545454545}, {'period_start': '2024-10-30T11:30:00+00:00', 'soc': 69.86060606060606}, {'period_start': '2024-10-30T12:00:00+00:00', 'soc': 74.16060606060606}, {'period_start': '2024-10-30T12:30:00+00:00', 'soc': 78.15151515151516}, {'period_start': '2024-10-30T13:00:00+00:00', 'soc': 80.57272727272728}, {'period_start': '2024-10-30T13:30:00+00:00', 'soc': 83.38181818181818}, {'period_start': '2024-10-30T14:00:00+00:00', 'soc': 89.23232323232322}, {'period_start': '2024-10-30T14:30:00+00:00', 'soc': 89.51313131313131}, {'period_start': '2024-10-30T15:00:00+00:00', 'soc': 89.90505050505051}, {'period_start': '2024-10-30T15:30:00+00:00', 'soc': 90.2050505050505}, {'period_start': '2024-10-30T16:00:00+00:00', 'soc': 89.07777777777778}, {'period_start': '2024-10-30T16:30:00+00:00', 'soc': 87.38282828282829}, {'period_start': '2024-10-30T17:00:00+00:00', 'soc': 86.53232323232324}, {'period_start': '2024-10-30T17:30:00+00:00', 'soc': 85.37373737373737}, {'period_start': '2024-10-30T18:00:00+00:00', 'soc': 83.83131313131312}, {'period_start': '2024-10-30T18:30:00+00:00', 'soc': 82.47070707070708}, {'period_start': '2024-10-30T19:00:00+00:00', 'soc': 66.6060606060606}, {'period_start': '2024-10-30T19:30:00+00:00', 'soc': 49.686868686868685}, {'period_start': '2024-10-30T20:00:00+00:00', 'soc': 46.24747474747475}, {'period_start': '2024-10-30T20:30:00+00:00', 'soc': 43.76262626262626}, {'period_start': '2024-10-30T21:00:00+00:00', 'soc': 41.42929292929293}, {'period_start': '2024-10-30T21:30:00+00:00', 'soc': 38.54848484848485}, {'period_start': '2024-10-30T22:00:00+00:00', 'soc': 38.54848484848485}, {'period_start': '2024-10-30T22:30:00+00:00', 'soc': 38.54848484848485}, {'period_start': '2024-10-30T23:00:00+00:00', 'soc': 38.54848484848485}, {'period_start': '2024-10-30T23:30:00+00:00', 'soc': 38.54848484848485}], 'forced': [{'period_start': '2024-10-30T00:00:00+00:00', 'forced': 0}, {'period_start': '2024-10-30T00:30:00+00:00', 'forced': 0}, {'period_start': '2024-10-30T01:00:00+00:00', 'forced': 0}, {'period_start': '2024-10-30T01:30:00+00:00', 'forced': 0}, {'period_start': '2024-10-30T02:00:00+00:00', 'forced': 0}, {'period_start': '2024-10-30T02:30:00+00:00', 'forced': 0}, {'period_start': '2024-10-30T03:00:00+00:00', 'forced': 0}, {'period_start': '2024-10-30T03:30:00+00:00', 'forced': 0}, {'period_start': '2024-10-30T04:00:00+00:00', 'forced': 0}, {'period_start': '2024-10-30T04:30:00+00:00', 'forced': 0}, {'period_start': '2024-10-30T05:00:00+00:00', 'forced': 0}, {'period_start': '2024-10-30T05:30:00+00:00', 'forced': 0}, {'period_start': '2024-10-30T06:00:00+00:00', 'forced': 0}, {'period_start': '2024-10-30T06:30:00+00:00', 'forced': 0}, {'period_start': '2024-10-30T07:00:00+00:00', 'forced': 0}, {'period_start': '2024-10-30T07:30:00+00:00', 'forced': 0}, {'period_start': '2024-10-30T08:00:00+00:00', 'forced': 0}, {'period_start': '2024-10-30T08:30:00+00:00', 'forced': 0}, {'period_start': '2024-10-30T09:00:00+00:00', 'forced': 0}, {'period_start': '2024-10-30T09:30:00+00:00', 'forced': 0}, {'period_start': '2024-10-30T10:00:00+00:00', 'forced': 0}, {'period_start': '2024-10-30T10:30:00+00:00', 'forced': 0}, {'period_start': '2024-10-30T11:00:00+00:00', 'forced': 0}, {'period_start': '2024-10-30T11:30:00+00:00', 'forced': 0}, {'period_start': '2024-10-30T12:00:00+00:00', 'forced': 0}, {'period_start': '2024-10-30T12:30:00+00:00', 'forced': 0}, {'period_start': '2024-10-30T13:00:00+00:00', 'forced': 0}, {'period_start [... cut ...] ar': 0.0}, {'period_start': '2024-10-30T20:00:00+00:00', 'solar': 0.0}, {'period_start': '2024-10-30T20:30:00+00:00', 'solar': 0.0}, {'period_start': '2024-10-30T21:00:00+00:00', 'solar': 0.0}, {'period_start': '2024-10-30T21:30:00+00:00', 'solar': 0.0}, {'period_start': '2024-10-30T22:00:00+00:00', 'solar': 0.0}, {'period_start': '2024-10-30T22:30:00+00:00', 'solar': 0.0}, {'period_start': '2024-10-30T23:00:00+00:00', 'solar': 0.0}, {'period_start': '2024-10-30T23:30:00+00:00', 'solar': 0.0}]}}

closed 2025-01-07 by github-actions[bot]