stevebuk1/pv_opt #274: Unexpected error in worker for App pv_opt

philicibine, 2024-10-19:
Seeing this today.  PV Opt seems to freeze at "Calculating Base Flows:"  I did notice the octopus addon was updated yesterday.  I wonder if this is the cause?

```
07:40:23 WARNING pv_opt: ------------------------------------------------------------
07:40:23 WARNING pv_opt: Unexpected error in worker for App pv_opt:
07:40:23 WARNING pv_opt: Worker Ags: {'id': 'cbb16cf49a4449cb83a38862608c6c19', 'name': 'pv_opt', 'objectid': '66b112c1d7da413c843b8d5e1ffb7481', 'type': 'scheduler', 'function': <bound method PVOpt.optimise_time of <pv_opt.PVOpt object at 0x7f8e6b9410>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 600, '__thread_id': 'thread-0'}}
07:40:23 WARNING pv_opt: ------------------------------------------------------------
07:40:23 WARNING pv_opt: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1572, in optimise_time
    self.optimise()
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1644, in optimise
    consumption = self.load_consumption(
                  ^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 2418, in load_consumption
    power = self.riemann_avg(power)
            ^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 2931, in riemann_avg
    dt = x.index.diff().total_seconds().fillna(0)
         ^^^^^^^
AttributeError: 'NoneType' object has no attribute 'index'

07:40:23 WARNING pv_opt: ------------------------------------------------------------
```

solarjones, 2024-10-25:
Getting similar errors too.

23:00:05 WARNING pv_opt: ------------------------------------------------------------
23:00:05 WARNING pv_opt: Unexpected error in worker for App pv_opt:
23:00:05 WARNING pv_opt: Worker Ags: {'id': '9b0e678e811946218635a1a3643c2570', 'name': 'pv_opt', 'objectid': 'db2102f5f7f2476d96a40f3a63f35495', 'type': 'scheduler', 'function': <bound method PVOpt.optimise_time of <pv_opt.PVOpt object at 0x7fb3f85cc810>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 600, '__thread_id': 'thread-0'}}
23:00:05 WARNING pv_opt: ------------------------------------------------------------
23:00:05 WARNING pv_opt: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1572, in optimise_time
    self.optimise()
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1716, in optimise
    self.optimised_cost = {"Base": self.contract.net_cost(self.flows["Base"])}
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pvpy.py", line 531, in net_cost
    imp_df = self.tariffs["import"].to_df(start, end, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pvpy.py", line 248, in to_df
    self.agile_predict = self._get_agile_predict()
                         ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pvpy.py", line 309, in _get_agile_predict
    df.index = pd.to_datetime(df.index).tz_convert("UTC")
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'Index' object has no attribute 'tz_convert'

23:00:05 WARNING pv_opt: ------------------------------------------------------------

[error.log](https://github.com/user-attachments/files/17527883/error.log)
[main.log](https://github.com/user-attachments/files/17527884/main.log)
[pv_opt.log](https://github.com/user-attachments/files/17527885/pv_opt.log)

As the PV_Opt hasn't changed in a while, I assume it's being caused by a dependant integration perhaps?  

Thanks
Simon

ThomasPrior, 2024-10-26:
Seeing the same issue here since 16:00 BST 25/10. Downgrading the Octopus custom component to the last v12 release has not restored PV Opt's functionality.

swests, 2024-10-26:
Seeing the same as @solarjones .  PV_OPT is effectively off-line

ThomasPrior, 2024-10-26:
@solarjones and (possibly @swests):

Can you try changing your copy of `pvpy.py` as recorded here: https://github.com/ThomasPrior/pv_opt/blob/main/apps/pv_opt/pvpy.py

Line 309, replace:
`df.index = pd.to_datetime(df.index).tz_convert("UTC")`

with

`df.index = pd.to_datetime(df.index, utc=True).tz_convert("UTC")`

Things are working again for me with this change and I'd be curious to see how you get on too.

swests, 2024-10-26:
Outstanding @ThomasPrior - that’s got it working for me.

solarjones, 2024-10-26:
@ThomasPrior  - excellent! This resolved things for me too, much appreciated. 

Any idea why this suddenly caused an issue?

ThomasPrior, 2024-10-26:
I don't really have a concrete answer for that I'm afraid. 

Given that multiple older versions of pandas had the same issue _and_ downgrading the Octopus integration had no effect, I believe this would be an issue with the data Octopus provides.

FarmingMartin, 2024-10-26:
seems to have solved the problem for me to. 
Not created a charging plan yet but that might be due to the high energy prices on Agile today, but the everything else looks to be working. I'll monitor it and keep you updated here. Thanks for the quick answer.

ThomasPrior, 2024-10-26:
No worries, thank you for testing. I had to reboot appdaemon to get things chugging along properly afterwards.

lazy-pete, 2024-10-26:
That worked for me too. Thanks.

Pete

punkymuzzle, 2024-10-26:
Thanks for this.

Tried the fix but unfortunately mine is still not working.

Mine is still stuck on "**Loading Tariffs**"  and I get this error in my AppDaemon Log:

`/homeassistant/appdaemon/apps/pv_opt/pv_opt.py:2758: FutureWarning: In a future version of pandas, parsing datetimes with mixed time zones will raise an error unless `utc=True`. Please specify `utc=True` to opt in to the new behaviour and silence this warning. To create a `Series` with mixed offsets and `object` dtype, please use `apply` and `datetime.datetime.strptime`
  df.index = pd.to_datetime(df.index) `

ThomasPrior, 2024-10-27:
> Thanks for this.
>
> Tried the fix but unfortunately mine is still not working.
>
> Mine is still stuck on "**Loading Tariffs**" and I get this error in my AppDaemon Log: [...]

The file I changed was pvpy.py. This would appear to be a separate but related issue with the same fix. Apply the same change as in my PR and test. I think it should work.

swests, 2024-10-27:
So this has come back in a slightly different form.  Now seeing 

pytz.exceptions.AmbiguousTimeError: 2024-10-27 01:22:22.274169

I *suspect* this is related to the clock going back…

swests, 2024-10-27:
I tried reverting the change from yesterday to no avail.

punkymuzzle, 2024-10-27:
Thank you for the advice.

Just to be clear, I replace the line referenced in the error file with the full amendment above?
I have tried this and now have the following in my pv_opt.py file:

`                df.index = pd.to_datetime(df.index, utc=True).tz_convert("UTC")`

 `                 #df.index = `pd.to_datetime(df.index)`
                
The bottom line is the original, the top is the amended.

Is this correct?
Thanks

ThomasPrior, 2024-10-27:
Yeah I'm seeing this myself now too. Can of worms well and truly opened...

swests, 2024-10-27:
Full Error:

pytz.exceptions.AmbiguousTimeError: 2024-10-27 01:22:22.274169
raise AmbiguousTimeError(dt)
File "/usr/lib/python3.11/site-packages/pytz/tzinfo.py", line 366, in localize
^^^^^^^^^^^^^^^^^^^^^^^^^
dt = self.localize(dt, is_dst)
File "/usr/lib/python3.11/site-packages/pytz/tzinfo.py", line 425, in utcoffset
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
return self.AD.tz.utcoffset(self.datetime()).total_seconds() / 60
File "/usr/lib/python3.11/site-packages/appdaemon/adapi.py", line 2003, in get_tz_offset
^^^^^^^^^^^^^^^^^^^^
self.log(f"Time Zone Offset: {self.get_tz_offset()} minutes")
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 473, in initialize
^^^^^^^^^^^^^^
return f(*args, **kw)
File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
result = self.fn(*self.args, **self.kwargs)
File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
^^^^^^^^^^^^^^^
response = future.result()
File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
await utils.run_in_executor(self, init)
File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 162, in initialize_app
01:22:22 WARNING pv_opt: Traceback (most recent call last):
01:22:22 WARNING pv_opt: ------------------------------------------------------------
01:22:22 WARNING pv_opt: Unexpected error running initialize() for pv_opt
01:22:22 WARNING pv_opt: ------------------------------------------------------------

punkymuzzle, 2024-10-27:
yeah I'm getting more and more errors too....

FarmingMartin, 2024-10-27:
Looks like mine has got another issue.
I'm not getting any charge times generated and PV-OPT results keeps flashing on and off.
Did a HA reboot and status has not changed ftom idle.

https://github.com/user-attachments/assets/3bd22f9c-4242-4ef0-b4e3-6c43ed01d0eb

ThomasPrior, 2024-10-27:
That's where we all are at the moment

swests, 2024-10-27:
It’s failing at

self.log(f"Time Zone Offset: {self.get_tz_offset()} minutes")

In pv_opt.py, around line 475

ThomasPrior, 2024-10-27:
I'm out of my depth. I've closed the PR and will wait for more informed advice on how to deal with this problem. Sorry all.

solarjones, 2024-10-27:
Getting the following error this morning:

06:20:00 WARNING pv_opt: ------------------------------------------------------------
06:30:00 WARNING pv_opt: ------------------------------------------------------------
06:30:00 WARNING pv_opt: Unexpected error in worker for App pv_opt:
06:30:00 WARNING pv_opt: Worker Ags: {'id': '6f62c83fa0d94de4bec7f0311f284c55', 'name': 'pv_opt', 'objectid': 'd2f63b137c874cb091a6de1f541af16d', 'type': 'scheduler', 'function': <bound method PVOpt.optimise_time of <pv_opt.PVOpt object at 0x7fd4676a9910>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 600, '__thread_id': 'thread-0'}}
06:30:00 WARNING pv_opt: ------------------------------------------------------------
06:30:00 WARNING pv_opt: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1572, in optimise_time
    self.optimise()
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1614, in optimise
    if self._check_tariffs():
       ^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 981, in _check_tariffs
    self._check_tariffs_vs_bottlecap()
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 2759, in _check_tariffs_vs_bottlecap
    df.index = df.index.tz_convert("UTC")
               ^^^^^^^^^^^^^^^^^^^
AttributeError: 'Index' object has no attribute 'tz_convert'

lazy-pete, 2024-10-27:
I think I am having the same issue.

Wrote a plan yesterday, but then failed to deliver it and now seems stuck.
Below from error file, with last bit being -

AttributeError: 'Index' object has no attribute 'tz_convert'

00:10:00 WARNING pv_opt:
------------------------------------------------------------ 00:10:00
WARNING pv_opt: Traceback (most recent call last): File
"/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in
worker funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in
f_app_lock return f(*args, **kw) ^^^^^^^^^^^^^^ File
"/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1572, in
optimise_time self.optimise() File
"/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in
f_app_lock return f(*args, **kw) ^^^^^^^^^^^^^^ File
"/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1614, in optimise if
self._check_tariffs(): ^^^^^^^^^^^^^^^^^^^^^ File
"/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 981, in
_check_tariffs self._check_tariffs_vs_bottlecap() File
"/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 2759, in
_check_tariffs_vs_bottlecap df.index = df.index.tz_convert("UTC")
^^^^^^^^^^^^^^^^^^^ AttributeError: 'Index' object has no attribute
'tz_convert'

fboundy, 2024-10-27:
Mine has hung up too. I should have time to look at it this evening. It looks like it’s time-zone related.

lazy-pete, 2024-10-27:
IIRC there were issues last time the clocks changed as well.

fboundy, 2024-10-27:
Probably. It’s a pain because you only properly get to test the fix 6 months later

fboundy, 2024-10-27:
Octopus sticks to UTC which should be simple but the inverter uses local time.

punkymuzzle, 2024-10-27:
Just checked my SolaX app and my inverter has the correct time, so assuming
this is using UTC (as I never changed it for DST).
If that'd the case then the Octopus info should work?
Mines still not working.

fboundy, 2024-10-27:
Should be fixed in 3.16.1 - just releasing now

FarmingMartin, 2024-10-27:
How do I update?

fboundy, 2024-10-27:
Via HACS

stjohncanning, 2024-10-27:
Works for me - thankyou!

punkymuzzle, 2024-10-27:
Still doesn't seem to be working for me (X1-AC Inverter)

Don't seem to be getting any errors in AppDaemon which is great, but my Dash looks like this, it's stuck on "Optimising Charge Plan, and the "PV Opt Results" table keeps flashing on and off (with no content).

[image]

punkymuzzle, 2024-10-27:
Working for me as well.  Thank you very much

FarmingMartin, 2024-10-27:
All sorted now thanks

swests, 2024-10-27:
Thanks! Sorted!