stevebuk1/pv_opt #275: "Calculating Base Flow" Error

stjohncanning, 2024-10-26:
Been using PVOPT (current versions) for months without error.   On 24/10 it did not restart after routine reboot of HA, or after reboot of Pi4.
Status shows 'Loading Tariffs' and does not move on.
However error log (attached) shows tariffs loaded ok, and last message in error log is 'calculating base flows'.

Reloaded backup from mid October, stopped at same place (loading tariffs/calculating base flows)  BUT warning messages below from the current error log were not present, not sure if relevant?

09:12:39     INFO: 
09:12:39  WARNING: pv_opt: Entity sensor.pvopt_tariff_import_OK not found in namespace default
09:12:40     INFO:   Import: Average Prices - PV_OPT: 22.58 p/kWh  OE Integration: 22.58 p/kWh  Mean difference:  0.00 p/kWh
09:12:40  WARNING: pv_opt: Entity sensor.pvopt_tariff_export_OK not found in namespace default
09:12:40     INFO:   Export: Average Prices - PV_OPT: 15.00 p/kWh  OE Integration: 15.00 p/kWh  Mean difference:  0.00 p/kWh
09:12:40     INFO: 

Any help appreciated, don't think I changed anything, but not sure where to go on this one.

error log and config files attached:
[config.txt](https://github.com/user-attachments/files/17529604/config.txt)
[pv_opt.log](https://github.com/user-attachments/files/17529605/pv_opt.log)

FarmingMartin, 2024-10-26:
I'm having the same issue. Was working great but around 4.00pm on 25/10/24 getting the sane error message " comparing tarriffs" I've also done the same ,rebooted HA reloaded backup from 7 days ago removed and reinstalled octopus intergration still not solved. Have noticed some octopus intergration registers are now not showing any data
[image]
[image]
[image]

ThomasPrior, 2024-10-26:
@stjohncanning and @FarmingMartin, please see this potential fix. If you can please let me know how you get on with it, it would help validate whether the pull request I've submitted fully resolves this.

https://github.com/fboundy/pv_opt/issues/274#issuecomment-2439180295

stjohncanning, 2024-10-26:
> #274 (comment)

Works for me, back to normal.
Thankyou

stjohncanning, 2024-10-26:
Looks like this is same as #274.   Closing to remove duplication

closed 2024-10-26 by stjohncanning