stevebuk1/pv_opt #280: Charge threshold per pass parametet setting

FarmingMartin, 2024-10-27:
I have a question which I can't find the answer to.
Although my PV-OPT seems to be working correctly, I have a setting showing on my dashboard, which is not listed in the setup instructions.
"Charge Threshold per pass (p)"
The settings for this are 0, 0.5, and 1
Can anyone tell me if this is correct. And what the default setting should be.
I am missing  "pass threshold"
[image]
[image]
 has this been replaced but not updated in the documents.
I have also not got the ""optimising charging" switch showing also again has this been removed or am I missing somthing in the dashboard YAML code.

fboundy, 2024-10-28:
Charge Threshold and Threshold per Pass are the same thing. The default is 4p. Not sure why your setup is showing such a narrow range. It also doesn't make much sense that the Slot Threshold (ie the minimum benefit you have to get from any 30 minute slot) is higher than the Pass Threshold (the benefit of all the slots). In reality though neither will make much difference.

I'd suggest you take a new copy of the dashboard from the Github repo and then restart HA and Appdaemon,

FarmingMartin, 2024-11-03:
I have redone the dadhboatd twice, and it's still giving me the same parameter.
Is there anywhere else I should look in case It has been altered there by mistake?
How ever with the setting at 0.5 PV-OPT does still seem to be working.

stevebuk1, 2024-11-08:
I've always had the same problem:

[image]

Pv_opt doesnt overwrite the MQTT generated entities if they already exist, and if not overwritten it doesn't appear to grab changes such as max and min settings. Deleting the entity would resolve, then Pv_opt will regenerate it, but that can't be done from the GUI for MQTT entities as far as I can see. Googling leads me to believe one needs to install something called "MQTT Explorer" which is a tool that will allow you to delete the entities. I'll get round to it eventually!

FarmingMartin, 2024-11-08:
Thanks for that info. I'll also have a look at getting that tool and see if I can delete the entities and see if that corrects the issue, obviously I'll do a backup first just incase I mess it up.
I'll report back of any updates.

FarmingMartin, 2024-11-10:
I've had a look at what you have suggested as a possible cure but to be honest it's way above me, so I have left mine as it is. It seems to work OK, but if you manage to get it sorted I would greatly appreciate step by step instructions on how you did it.
But for now I'll leave mine as it is.

fboundy, 2024-11-10:
MQTT explorer won’t help I’m afraid.

I played around with this a lot when I was writing this bit of the code originally - should have documented it but you know how it is…

My recollection is that if you disable PV Opt (or appdaemon) and restart HA you will then be able to delete the entities. When PV Opt starts it will the use its defaults to create new ones via MQTT discovery.

FarmingMartin, 2024-11-10:
Thanks for the update, but where/how do I delete the entities as I can't see that one in any of the YAML code.
Thanks.

fboundy, 2024-11-10:
OK - here's an easier way to change the range on the slider. 

It is in the attributes to the entity so:

- Click on Developer Tools on the menu panel
- Click on States
- Type "threshold" on the "Filter entities box"
- Scroll down to the one you want to edit and click on the name. This will bring all the details up in the "State attributes" box.
- Edit max and min to what you wanyt
- Click on `SET STATE`

That should do it

FarmingMartin, 2024-11-10:
I have done as advised above. And although it does chane the range, if I move the slider it says " outside range of 0 to 1"
I change the initial setting to 4 and again it did show that value on the slider. But again if I move it either higher or lower I get the same error message as above.

fboundy, 2024-11-10:
OK try this then:

- Go to Add-Ons and Stop both the Mosquito MQTT broker and Appdaemon
- Go to Services and delete the MQTT integration
- Restart HA
- Start the Mosquito Broker
- Start Appdaemon
- Add the MQTT Integration under Services with the default settings
- Re-start Appdaemon

This should get the defaults from `pv_opt.py` re-applied via MQTT discovery

FarmingMartin, 2024-11-10:
Great followed your instructions above and it's now all fixed.

philicibine, 2024-11-11:
I have had this issue for a while.  Tried following the above and still not working for me.  Still defaults to 1 and the slider goes from 0 to 1.

fboundy, 2024-11-11:
I think you are using Steve’s dev branch aren’t you?

It looks like the defaults are set wrong in pv_opt.py (lines 223-231). Can you try changing max to 10 and repeat?

fboundy, 2024-11-11:
Sorry - those might not be the correct lines but you are looking for the “max” entry for pass_threshold_p

philicibine, 2024-11-11:
> I think you are using Steve’s dev branch aren’t you? It looks like the defaults are set wrong in pv_opt.py (lines 223-231). Can you try changing max to 10 and repeat?

No.  I don't think i am.  I even re-downloaded the whole thing today to ensure i have all the most recent changes.  Have tried changing the lines in pv_opt.py before and it seemed to have no effect.

philicibine, 2024-11-11:
Definitely on your branch 3.17

fboundy, 2024-11-11:
OK so you have the following at line 182 in `pv_opt.py`?

    "pass_threshold_p": {
        "default": 4.0,
        "attributes": {
            "min": 0.0,
            "max": 20.0,
            "step": 0.5,
            "mode": "slider",
        },
        "domain": "number",

In which case I'm a bit baffled. Once you've deleted the entities using the above it has to create them from these defaults.

philicibine, 2024-11-11:
Ok, I think I figured it out!  I was using a different MQTT broker than the standard one in HA.  Running on a seperate pi.  This was setup with persistence.  So I did as the above.. but then cleared the mosquitto.db file on the other pi.  It now appears to be sorted.  Thanks for your help.

fboundy, 2024-11-11:
Yes, that’ll be it.

MQTT is a bit of a pain but it allows the entities to be created in a controlled way

closed 2024-11-11 by fboundy