stevebuk1/pv_opt #283: Setting ups for Solis Cloud

exlabrat, 2024-11-07:
Hi

I'm excited to see the new code merged for Solis Cloud as this should fit with my aims for my system. I'm trying to set up PV_opt for Solis Cloud based on the documentation but so far both PV_opt and Use PV_opt are giving me errors. I wonder if you can point me in the right direction (I'm a relative novice with scripting in HA). 

This is the log for PV_opt: 

15:50:59 ERROR: id_solar_power : Neither the entities listed in the YAML sensor.solis_pv_total_power nor the system default of None exist in HA.
--
15:50:59 INFO: >>> id_solar_power sensor.solis_pv_total_power False
15:50:59 INFO: >>> Test: True
15:50:59 INFO: battery_current_limit_amps = 50 50: value in YAML
15:50:59 INFO: inverter_power_watts = 3000 3000: value in YAML
15:50:59 INFO: charger_power_watts = 3000 3000: value in YAML
15:50:59 INFO: maximum_dod_percent = 20 20: value in YAML
15:50:59 INFO: battery_capacity_wh = 96000 96000: value in YAML
15:50:59 INFO: allow_cyclic = False False: value in YAML
15:50:59 INFO: forced_discharge = False False: value in YAML
15:50:59 INFO: read_only = True True: value in YAML
15:50:59 INFO: debug_categories = W W: value in YAML
15:50:59 INFO: -----------------------------------
15:50:59 INFO: Reading arguments from YAML:
15:50:59 INFO:
15:50:59 INFO: sensor.solis_meter_item_c_volt (voltage): 0.0 V
15:50:59 INFO: sensor.solis_meter_item_c_current (current): 0.0 A
15:50:59 INFO: sensor.solis_meter_item_b_volt (voltage): 0.0 V
15:50:59 INFO: sensor.solis_meter_item_b_current (current): 0.0 A
15:50:59 INFO: sensor.solis_meter_item_a_volt (voltage): 239.5 V
15:50:59 INFO: sensor.solis_meter_item_a_current (current): 0.0 A
15:50:59 INFO: sensor.solis_backup_load_power (power): 0.0 W
15:50:59 INFO: sensor.solis_force_discharge_soc (battery): 20.0 %
15:50:59 INFO: sensor.solis_force_charge_soc (battery): 18.0 %
15:50:59 INFO: sensor.solis_battery_state_of_health (battery): 98.0 %
15:50:59 INFO: sensor.solis_plant_total_consumption_power (power): 443.0 W
15:50:59 INFO: sensor.solis_grid_phase3_reactive_power (None): 0.0 var
15:50:59 INFO: sensor.solis_grid_phase2_reactive_power (None): 0.0 var
15:50:59 INFO: sensor.solis_grid_phase1_reactive_power (None): 0.0 var
15:50:59 INFO: sensor.solis_grid_phase3_apparent_power (None): 0.0 VA
15:50:59 INFO: sensor.solis_grid_phase2_apparent_power (None): 0.0 VA
15:50:59 INFO: sensor.solis_grid_phase1_apparent_power (None): 0.0 VA
15:50:59 INFO: sensor.solis_grid_phase3_power (power): 0.0 W
15:50:59 INFO: sensor.solis_grid_phase2_power (power): 0.0 W
15:50:59 INFO: sensor.solis_grid_phase1_power (power): 0.0 W
15:50:59 INFO: sensor.solis_total_energy_used (energy): 3776.0 kWh
15:50:59 INFO: sensor.solis_total_energy_purchased (energy): 11513.0 kWh
15:50:59 INFO: sensor.solis_total_consumption_power (power): 415.0 W
15:50:59 INFO: sensor.solis_power_grid_total_power (power): -459.0 W
15:50:59 INFO: sensor.solis_total_on_grid_energy (energy): 670.8 kWh
15:50:59 INFO: sensor.solis_yearly_on_grid_energy (energy): 75.8 kWh
15:50:59 INFO: sensor.solis_yearly_grid_energy_purchased (energy): 3753.0 kWh
15:50:59 INFO: sensor.solis_monthly_on_grid_energy (energy): 0.0 kWh
15:50:59 INFO: sensor.solis_monthly_grid_energy_purchased (energy): 137.64 kWh
15:50:59 INFO: sensor.solis_daily_grid_energy_used (energy): 13.2 kWh
15:50:59 INFO: sensor.solis_daily_grid_energy_purchased (energy): 13.2 kWh
15:50:59 INFO: sensor.solis_daily_on_grid_energy (energy): 0.0 kWh
15:50:59 INFO: sensor.solis_daily_energy_discharged (energy): 7.0 kWh
15:50:59 INFO: sensor.solis_daily_energy_charged (energy): 7.0 kWh
15:50:59 INFO: sensor.solis_total_energy_discharged (energy): 3459.0 kWh
15:50:59 INFO: sensor.solis_total_energy_charged (energy): 4692.0 kWh
15:50:59 INFO: sensor.solis_remaining_battery_capacity (battery): 20.0 %
15:50:59 INFO: sensor.solis_battery_current (current): 0.0 A
15:50:59 INFO: sensor.solis_battery_voltage (voltage): 48.6 V
15:50:59 INFO: sensor.solis_battery_power (power): 0.0 W
15:50:59 INFO: sensor.solis_energy_total (energy): 1.0 kWh
15:50:59 INFO: sensor.solis_energy_this_year (energy): 1.0 kWh
15:50:59 INFO: sensor.solis_energy_this_month (energy): 0.0 kWh
15:50:59 INFO: sensor.solis_energy_today (energy): unknown kWh
15:50:59 INFO: sensor.solis_ac_frequency (None): 49.85 Hz
15:50:59 INFO: sensor.solis_ac_output_total_power (power): 0.0 W
15:50:59 INFO: sensor.solis_ac_current_r (current): 1.0 A
15:50:59 INFO: sensor.solis_ac_voltage_r (voltage): 237.9 V
15:50:59 INFO: sensor.solis_temperature (temperature): 21.8 ��C
15:50:59 INFO: sensor.solis_timestamp_measurements_received (None): 1730994518.999
15:50:59 INFO: sensor.solis_state (None): 1
15:50:59 INFO: sensor.solis_power_state (None): 3
15:50:59 INFO:
15:50:59 INFO: ------------------------------------
15:50:59 INFO: Available entities for device solis:
15:50:59 INFO:
15:50:59 INFO: Inverter appears to be online
15:50:59 INFO: Serial number:
15:50:59 INFO: Device name: solis
15:50:59 INFO: Inverter type: SOLIS_SOLAX_MODBUS: inverter module: solis.py
15:50:59 INFO: Time Zone Offset: 0.0 minutes
15:50:59 INFO: Local timezone set to GB
15:50:59 INFO:
15:50:59 INFO: ******************* PV Opt v3.16.1 *******************

For Use PV_opt the trace gives an error at

Executed: 7 November 2024 at 17:00:05
Error: In 'template' condition: AttributeError: 'NoneType' object has no attribute 'tzinfo'

But I'm guessing this might be due toPV_opt failing.

Many thanks
Ian

stevebuk1, 2024-11-08:
Hi, thanks for the log excerpt. Could you attach the full log file please? (pv_opt.log). On startup Pv_opt will log all entities with "solis" in the name so it then will be possible to work out which entity in the Solis Cloud integration reports solar output. Its a sensor in W that reports Solar (noting mine was particularly poor today!)

[image]

If you can find the name of the entity then you can add it to config.yaml as a map entry which will solve the issue.

exlabrat, 2024-11-09:
Many thanks. Could that be sensor.solis_ac_output_total_power (power)? 

I've attached the log, although I can't see anything else there. 

[pv_opt.log](https://github.com/user-attachments/files/17686891/pv_opt.log)

exlabrat, 2024-11-09:
Actually using that mapping seems to solve that problem and I can regenerate the PV_opt dashboard. 

I'm now getting other errors on the consumption data - new pv_opt.log attached
[pv_opt-2.log](https://github.com/user-attachments/files/17686934/pv_opt-2.log)

stevebuk1, 2024-11-09:
> I'm thinking that the consumption sensor is solis_daily_grid_energy_purchased

I think you are right. In the SOLIS_SOLAX_MODBUS area of config.yaml uncomment id_consumption_today and set it to sensor.{device_name}__daily_grid_energy_purchased

On the line below, check that any entries for `id_consumption` are commented out. 

By way of explanation, Pv_opt can either use an incrementing kWh sensor or a live W sensor to work out consumption, but increasingly reading the live W sensor with 7 days history is a very large file and can cause issues for some systems.

fboundy, 2024-11-10:
Worth also noting that using the instantaneous power rather than the integrated energy is potentially less accurate. My preference is always to use energy where available. Even better would be to down-sample it to 30 minutes using a jinja template in Home Assistant first

exlabrat, 2024-11-10:
Thanks both. I have a choice between sensor.solis_daily_grid_energy_purchased and sensor.solis_daily_grid_energy_used

[image]

[image]

There's also one called sensor.solis_plant_total_consumption_power which looks to give 5 min power numbers. 

I'm still working through some other sensors which seem to need mapping - currently I can generate the control dashboard but not the output dashboard(s).

stevebuk1, 2024-11-10:
> I'm still working through some other sensors which seem to need mapping - currently I can generate the control dashboard but not the output dashboard(s).

@CraigCallender would you mind uploading your config.yaml to this thread so we can see what sensors you've mapped to what on your system?

stevebuk1, 2024-11-10:
> Worth also noting that using the instantaneous power rather than the integrated energy is potentially less accurate. My preference is always to use energy where available.

I use energy on my system but it was more accident that deliberate. From memory it was because the default config.yaml was setup that way, but certainly for the Solax integration users the current config.yaml on main does not suggest using energy in preference to power, as if both exist and are correct then power will be used in preference to energy. I've corrected this in Dev. 

```
  # ===============================================================================================================
  # Brand / Integration Specific Config: SOLIS_SOLAX_MODBUS: https://github.com/wills106/homeassistant-solax-modbus
  # ===============================================================================================================
  #
  # These are the default entities used with the Solis Solax Modbus integration. You can change them here and over-ride the defaults

  # inverter_type: SOLIS_SOLAX_MODBUS
  # device_name: solis
  
  # battery_voltage: sensor.{device_name}_battery_voltage
  # update_cycle_seconds: 15
  # maximum_dod_percent: number.{device_name}_battery_minimum_soc 
  
  id_consumption_today: sensor.{device_name}_consumption_today
  id_consumption:
    - sensor.{device_name}_house_load
    - sensor.{device_name}_bypass_load

  # id_grid_import_today: sensor.{device_name}_grid_import_today
  # id_grid_export_today: sensor.{device_name}_grid_export_today

```

fboundy, 2024-11-12:
Hoping to temporarily swap my system over to using this rather than the Solax integration from this evening so should be able to better support it.

@stevebuk1 @CraigCallender - I'll create a new branch `solis-cloud`

CraigCallender, 2024-11-12:
Hey @stevebuk1  and @fboundy - Here's my config.yaml for all the mapping:

```yaml
# Internal configuration --- DO NOT EDIT ---
pvpy:
  module: pvpy
  global: true

solis:
  module: solis
  global: true

inverters:
  module: inverters
  global: true

pv_opt:
  module: pv_opt
  class: PVOpt
  dependencies:
    - pvpy
    - solis

  log: pv_opt_log
  prefix: pvopt
  debug: false

# User configuration ---  EDIT AWAY! ---
  # ========================================
  # System parameters
  # ========================================

  # If true the current config in HA will be over-written with that in the config.yaml. 
  overwrite_ha_on_restart: true
  list_entities: true

  # If true the personal data will be redacted from the log files. 
  # redact_personal_data_from_log: false
  
  #=======================================
  #Logging Category Control
  #=======================================
  #Defines Logging subjects to add to logfile
  #If commented out, everything is logged
  #Ignored if "debug" is set to False above
  #
  # S = Startup/Initialisation Logging
  # T = Tariff loading Logging
  # P = Power consumption history Logging
  # C = Charge algorithm Logging
  # D = Discharge algorithm Logging
  # W = Charge/Discharge Windows Logging
  # F = Power Flows Logging
  # V = Power Flows debugging (extra verbose)
  # I = inverter control/commands Logging
  # E = EV debugging
  # Letters can be added to "debug_categories" in any order
  debug_categories: W

  # ========================================
  # Basic parameters
  # ========================================
  read_only: true # If true the inverter will not be controlled
  forced_discharge: false # Enable forced discharging
  allow_cyclic: false # Enable alternate charge/discharge windows
  optimise_frequency_minutes: 5
  pass_threshold_p: 2.0
  discharge_threshold_p: 0.4

  # ========================================
[... 215 lines cut ...]
  # id_grid_import_today: sensor.{device_name}_{inverter_sn}_day_grid_import
  # id_grid_export_today: sensor.{device_name}_{inverter_sn}_day_grid_export
  # supports_hold_soc: false
  # update_cycle_seconds: 300

# Tariff comparison
  # id_daily_solar: sensor.{device_name}_power_generation_today
  id_solar_power: 
  - sensor.{device_name}_dc_power_pv1
  - sensor.{device_name}_dc_power_pv2
  # id_solar_power: sensor.{device_name}_dc_power_combined
    
  alt_tariffs:
    - name: Agile_Fix
      octopus_import_tariff_code: E-1R-AGILE-23-12-06-G
      octopus_export_tariff_code: E-1R-OUTGOING-FIX-12M-19-05-13-G

  #   - name: Eco7_Fix
  #     octopus_import_tariff_code: E-2R-VAR-22-11-01-G
  #     octopus_export_tariff_code: E-1R-OUTGOING-FIX-12M-19-05-13-G
      
  #   - name: Flux
  #     octopus_import_tariff_code: E-1R-FLUX-IMPORT-23-02-14-G
  #     octopus_export_tariff_code: E-1R-FLUX-EXPORT-23-02-14-G
```

stevebuk1, 2024-11-12:
Ok. Was about to do the same. I think this needs a separate inverter
default as the entity names don't match the other integration options.

fboundy, 2024-11-12:
Leave this one with me for now. May not get to it this evening but I should have time over the next couple of days.

stevebuk1, 2024-11-12:
Ok will do!

exlabrat, 2024-11-12:
Thanks @CraigCallender, the config.yaml is really helpful. I've got this running now, and briefly had the dashboard working with a couple of cards absent, but the addition of the octopus rates card seems to kill the visualisation. Hopefully I can sort that soon

fboundy, 2024-11-13:
This is very interesting!

I've dug a bit into SolisControl and I am able to move its functionality into PV_Opt. I'm also updating the `solis.py` module so it will read the inverter status (control mode etc) via the API. I hope to get a Beta version of this out to test over the next couple of days. This will mean that there is no need for a separate automation to run to control the inverter.

What I'd really like to do is put this functionality into the SolisCloud integration but I'm not sure my coding is up to it. I might have a go on a fork of it when I'm done with this.

CraigCallender, 2024-11-15:
@fboundy I just tried 3.18 and one change I had to make to the `config.yaml` (newly downloaded from the repo) was on line 322 the `device_name` was set to `soliscloud`.  I _think_ this was so you could test the `solis-sensor` integration alongside your normal integration?  But the default sensor names when installing `solis-sensor` is just `solis`, so all the mappings were broken.

exlabrat, 2024-11-15:
I've got this up and running now with the dashboard. I just have two things to resolve. 

1. Where does sensor.pvopt_opt_cost_actual and sensor.pvopt_opt_cost_current come from? I don't seem to be running those right now

2. A minor point on dashboards - The vertical stack card occupies a lot of space for me. Is there a setting to reduce the pace between cards? I can't find this readily.

[image]

exlabrat, 2024-11-15:
This is the widget that requires pvopt_opt_cost_actual

[image]

fboundy, 2024-11-15:
Yes that’s right - good catch

CraigCallender, 2024-11-19:
I don’t have the new firmware on my inverter, but if someone could set their values via the Control Inverter section of the web interface with Chrome’s developer console open, you can capture the JSON sent. If you give me that, I can update my fork of solis_control.

fboundy, 2024-11-19:
I’ve got a Python script that runs independently to HA so easy enough to test. I’ve raised an issue on soliscontrol too so will update that with what I find.

Now need to find the Modbus register details too…

closed 2024-12-23 by fboundy