stevebuk1/pv_opt #284: Feature to put the inverter in backup mode for a number of hours to protect battery SoC whilst EV charging.

punkymuzzle, 2024-11-07:
**Is your feature request related to a problem? Please describe.**
I want to be able to set the inverter to set to Backup mode for a number of hours (either a single or multiple slots) when I put my EV on charge, so the battery does bot drain to charge the car.
When charging my Electric car overnight, some of this load will be provided by my batteries.  I don't want this to happen as then I need to charge the batteries again before any price increase in the half hourly tariff rates.
So a way to let the inverter know what times the car is charging (so it can not provide battery power to charge the car) would be ideal.

**Describe the solution you'd like**

1. What Would be great is a dashboard option to set PV-Opt into "EV Charge" mode and assign time slots & durations for this to happen. 
2. PV-Opt would then set the inverter to not discharge the battery during these times
3. if the time slots selected coincided with the PV-Opt charging schedule, the battery would also charge as per the schedule (but not discharge)
4. Once the last EV charging slot had completed, PV-Opt would revert back to its calculated schedule.

I know that other automations (cough - Predbat - cough) have the ability to integrate car charging into the schedule, but that is dependent on a car charger that can be controlled.  This is more of a process to protect the battery from discharging whilst EV charging is occurring when controlled via a separate app - i.e my ProjectEV app for my charger.

**Describe alternatives you've considered**
If the hourly rates slots overnight are relatively cheap then I place PV-Opt in read only and manually set the inverter to Backup mode to prevent the battery from discharging to charge the EV.  Then reset everything in the morning.
However the process is not particularly easy:
1. Sometimes PV-Opt takes numerous cycles for the read only selection to "stick" so the inverter keeps resetting back to its schedule.
6. Sometimes PV-Opt resets the read only toggle back to "off", effectively ignoring my request for read only and so ends up powering the EV charge.
7. In the event of the above I sometimes have to stop AppDaemon to force PV-Opt to stop running.  This is sometimes a problem as if I forget to reset things in the morning, PV-Opt does not resume running and I am usually left with an empty battery and consuming expensive electricity.
8. I can't do any of the above if the overnight price fluctuates and I want to charge my EV at certain time blocks, unless I stay up overnight to keep configuring PV-Opt.

**Additional context**
More than willing to test this via beta versions if needed.
My Inverter is a Solax X1AC.

Thanks very much in advance

stevebuk1, 2024-11-08:
Hi, not sure whether you saw it already but over the last six months the basics of an EV integration has been added on to Pv_opt, see issue #121 for full details. 

For users on IOG, it picks up the charge slots for the car and holds the inverter if not already charging, and also subtracts the car charging consumption from the total consumption to leave the house consumption only.  A few of us have been using this for a good few months now without issue. Note for this the inverter is put into charge mode at zero current. 

For Agile users, code is in place to generate a car schedule based on a "car ready by" time and a "charge to add", and will select the cheapest slots between car plugin and car ready by time to meet the charge target. This sets an entity that can then be used to start and stop the car charger automatically. (This is very much as Predbat does it) I've done some limited testing with this by forcing an Agile tariff and getting it to control my car charger  - all appears fine but it does need some more testing. Can your ProjectEV charger be controlled by HA at all? 

If so, then all of the above has been written for Zappi chargers but if you are ok with keeping "Use Consumption History" as Off for the time being (I imagine it is already) then I "think" it will work for any charger as the Agile car charging part doesn't communicate directly with the charger - maybe a few tweaks needed. 

The ability for Pv_opt to read an external car charging schedule is for the future. Its queued up behind the ability to set manual tariff times (for Eon NextDrive and Tomato Energy users) and the ability to integrate with two Zappis.

stevebuk1, 2024-11-08:
> * Sometimes PV-Opt takes numerous cycles for the read only selection to "stick" so the inverter keeps resetting back to its schedule.
> * Sometimes PV-Opt resets the read only toggle back to "off", effectively ignoring my request for read only and so ends up powering the EV charge.

Neither of these should happen (and don't on my system). Separately to the EV query/request, if you can post Pv_opt.log and error.log from an overnight run where it does I'll take a look. In the interim it may be worth setting Read only = True in your config.yaml, so if Pv_opt resets for any reason that will be its default state:

```
  # ========================================
  # Basic parameters
  # ========================================
  read_only: true # If true the inverter will not be controlled
```

punkymuzzle, 2024-11-08:
> > * Sometimes PV-Opt takes numerous cycles for the read only selection to "stick" so the inverter keeps resetting back to its schedule.
> > * Sometimes PV-Opt resets the read only toggle back to "off", effectively ignoring my request for read only and so ends up powering the EV charge.
> 
> Neither of these should happen (and don't on my system, where I've regularly set read only for overnight if I'm working on new code which allows me to disable Pv_opt and just do the inverter setting manually).
> 
> Separately to the EV query/request, if you can post Pv_opt.log and error.log from an overnight run where it does I'll take a look. In the interim it may be worth setting Read only = True in your config.yaml, so if Pv_opt resets for any reason that will be its default state:
> 
> ```
>   # ========================================
>   # Basic parameters
>   # ========================================
>   read_only: true # If true the inverter will not be controlled
> ```

Will do, thanks you very much

punkymuzzle, 2024-11-08:
> Hi, not sure whether you saw it already but over the last six months the basics of an EV integration has been added on to Pv_opt, see issue #121 for full details.
> 
> For users on IOG, it picks up the charge slots for the car and holds the inverter if not already charging, and also subtracts the car charging consumption from the total consumption to leave the house consumption only. A few of us have been using this for a good few months now without issue. Note for this the inverter is put into charge mode at zero current - backup mode isnt ideal as depending on Solis inverter model this can oscillate between full rate charging and/or self use discharge.
> 
> For Agile users, code is in place to generate a car schedule based on a "car ready by" time and a "charge to add", and will select the cheapest slots between car plugin and car ready by time to meet the charge target. This sets an entity that can then be used to start and stop the car charger automatically. (This is very much as Predbat does it) I've done some limited testing with this by forcing an Agile tariff and getting it to control my car charger - all appears fine but it does need some more testing. Can your ProjectEV charger be controlled by HA at all?
> 
> If so, then all of the above has been written for Zappi chargers but if you are ok with keeping "Use Consumption History" as Off for the time being (I imagine it is already) then I "think" it will work for any charger as the Agile car charging part doesn't communicate directly with the charger - maybe a few tweaks needed.
> 
> The ability for Pv_opt to read an external car charging schedule is for the future. Its queued up behind the ability to set manual tariff times (for Eon NextDrive and Tomato Energy users) and the ability to integrate with two Zappis.

Thanks very much for this information, much appreciated.
Yes I did take a look at the thread you mentioned but I don't think it will work in my case (plus I have other use cases for pausing charging / discharging, will describe this later).

Unfortunately my Project EV is not controllable via HA.  I did try to find an integration but there is nothing official to access it. Someone has tried to put something together by hacking into it with varying degrees of success, but there is nothing official that will work.  So at the moment I'm stuck with using the app to charge my car (which works surprisingly well to be honest).
The next issue is that I don't want to put my car n charge regularly.  I do quite low mileage so tend not to charge the car every day, few days etc.  My charging is more sporadic than that.
So this is why I requested this functionality to be added, so I can effectively set the inverter / battery to pause when I put the car on charge, thereby charging from the mains and not the battery, which I then have to re-charge, etc.

Other use cases for this functionality are:
1. When my kids use the shower, again I would like the ability to isolate the inverter / battery so it doesn't discharge, thereby saving its charge to supply power during the expensive times (as it's programmed to do).  Switching on a 10kW Shower will draw 2kW from my battery for however long the shower is in use (and as I have two kids, that can be a while).  Again, draining the battery which is not desirable.
2. When Electricity is cheap / negative I would like the ability to isolate the battery for the duration, so the house load draws directly from the grid.  Unless I intervene, PV-Opt will supply house load even when the price is low / negative.  Yes I know I can set PV-Opt to read only and then set the Inverter to backup mode (which is what I do now) but that means effectively switching PV-Opt off whilst I do this and then remembering to reset it all back again when I want it to resume.

When @fboundy was kindly adding functionality for my Solax X1AC inverter, he setup an additional dashboard for me to send commands directly to the inverter to see if charging commands etc worked:
[image]

In my mind, something similar to set the inverter not to use the battery to supply load at these times would be the ideal thing.

Hope all that makes sense, and if you need any further information I'm happy to provide it.
This isn't essential (it is an enhancement after all, not a necessity).  But if something like this can be added I'm sure others would use it as well (there are probably more use cases than the three I've provided).

Thanks

stevebuk1, 2024-11-09:
Without a way to control the charger then agreed, the EV functionality I've added won't help you. 

Thanks for explaining your use case - that makes good sense and I can see how it would be a useful addition in its own right I'll have a think and see what I can quickly add. Gut reaction at the moment is to provide a switch called "prevent battery discharge". When set, PV_opt would still charge the battery at the slots it needs to but would prevent discharge when not charging. You could then just toggle it manually, and/or use an external automation with start and end times for automated use.

stevebuk1, 2024-11-09:
> When @fboundy was kindly adding functionality for my Solax X1AC inverter, he setup an additional dashboard for me to send commands directly to the inverter to see if charging commands etc worked:

Are you using this unmodified from what was originally provided? If not, can you:
1) put the dashboard into edit mode
2) click the 3 dots
3) select "raw configuration editor"
4) copy and paste the contents into a .txt file and attach to this issue.

punkymuzzle, 2024-11-09:
That's fantastic, thanks very much for this.  This would be really helpful.
yaml code for the "test" dashboard (that @fboundy kindly created for me) as follows:

```
type: entities
entities:
  - entity: select.pvopt_test_function
    name: Function
  - entity: select.pvopt_test_enable
    name: Enable / Disable
  - entity: text.pvopt_test_start
    name: Start Time (Local Time Zone)
  - entity: text.pvopt_test_end
    name: End Time (Local Time Zone)
  - entity: number.pvopt_test_power
    name: Power
  - entity: number.pvopt_test_target_soc
    name: Target SOC
  - entity: button.pvopt_test_button
    name: Send to Inverter
title: PV Opt Test Card
```

Thanks once again.

stevebuk1, 2024-11-13:
Ok theres a Beta release to try on my fork at https://github.com/stevebuk1/pv_opt/tree/main/apps/pv_opt.

Best probably not use HACS to install this, just overwrite the files you need, which is:
https://github.com/stevebuk1/pv_opt/blob/main/apps/pv_opt/pv_opt.py
https://github.com/stevebuk1/pv_opt/blob/main/apps/pv_opt/pvpy.py

Essentially I've added a new entity switch.pvopt_prevent_discharge which does exactly that for as long as it is set. 

I've created a test dashboard update, which is here, 
https://github.com/stevebuk1/pv_opt/blob/main/dashboards/pvopt_test_card%20V2.yaml
and looks like this:
[image]
as you can see there is a new entity at the bottom called "Prevent Discharge". 

When set, it will set the battery charge current to zero and maintain that until cleared.

You'll be able to see that in the charge plan via a charge power of 1W and a "<=Car" string, as the new entity is effectively being coupled into the car charging logic that prevents house battery discharge during car charging. 
[image]
Toggling "Prevent Discharge" will trigger an optimser run so latency will be a minute or two at most. 

I've tested it with my system and all seems fine, but theres been a lot of updates into Beta-15 for other reasons and control of Solax inverters probably hasn't been tested. Post with logs here for any issues.

stevebuk1, 2024-11-13:
Forgot to add that setting this switch this will not affect any house battery charge plan. So you can go to bed at 10pm with it set and the battery will charge to the charge plan as normal.

punkymuzzle, 2024-11-13:
That's fantastic, thank you very much for this.
I'm away from home at the moment so won't be able to install and test this
until Friday, but will install and test as soon as I can.
Thanks once again, really appreciate it.

punkymuzzle, 2024-11-14:
OK, copied the files over and rebooted.
Can see the following error in the AppDaemon log.

> /homeassistant/appdaemon/apps/pv_opt/pv_opt.py:2385: SettingWithCopyWarning: 
A value is trying to be set on a copy of a slice from a DataFrame.
Try using .loc[row_indexer,col_indexer] = value instead
See the caveats in the documentation: https://pandas.pydata.org/pandas-docs/stable/user_guide/indexing.html#returning-a-view-versus-a-copy
  y["start"] = y.index.tz_convert(self.tz)
/homeassistant/appdaemon/apps/pv_opt/pv_opt.py:2386: SettingWithCopyWarning: 
A value is trying to be set on a copy of a slice from a DataFrame.
Try using .loc[row_indexer,col_indexer] = value instead
See the caveats in the documentation: https://pandas.pydata.org/pandas-docs/stable/user_guide/indexing.html#returning-a-view-versus-a-copy
  y["end"] = y.index.tz_convert(self.tz) + pd.Timedelta(30, "minutes")

> 

Logs:
[pv_opt.log](https://github.com/user-attachments/files/17759207/pv_opt.log)
[error.log](https://github.com/user-attachments/files/17759208/error.log)

I can't test the dashboard yet as my battery SoC is 10% so it won't discharge any lower (so I won't be able to test if it's working).  I'll test tomorrow when my battery has some charge in it.

Thanks

punkymuzzle, 2024-11-15:
Tried the "Prevent discharge" option without success:
[image]
Battery still providing power to the house.

[image]

stevebuk1, 2024-11-15:
Hmmm. Thanks for the appdaemon log, I'll get that fixed but I don't think its preventing operation.

Can you post the pv_opt.log from when you toggled "Prevent Discharge" please.

Also, in your config.yaml, can you change this:

 ```
 log: pv_opt_log
  prefix: pvopt
  debug: false

  # User configuration ---  EDIT AWAY! ---
  # ========================================
  # System parameters
  # ========================================
```
to read this:

```
  log: pv_opt_log
  prefix: pvopt
  debug: true
  debug_categories: O

  # User configuration ---  EDIT AWAY! ---
  # ========================================
  # System parameters
  # ========================================

```
ie change debug from false to true and add " debug_categories: O" (the letter O, not the number 0) just underneath.

punkymuzzle, 2024-11-15:
Set the discharging to this:
[image]
Switched on my immersion heater to draw 3Kw, but it's still coming from the battery and not the grid:
[image]

Here are my three logs:
[main.log](https://github.com/user-attachments/files/17772557/main.log)
[pv_opt.log](https://github.com/user-attachments/files/17772558/pv_opt.log)
[error.log](https://github.com/user-attachments/files/17772559/error.log)

Also, copies of the oldest logs I have on HA:
[pv_opt.log.3.txt](https://github.com/user-attachments/files/17773183/pv_opt.log.3.txt)
[error.log.1.txt](https://github.com/user-attachments/files/17773188/error.log.1.txt)
[main.log.3.txt](https://github.com/user-attachments/files/17773214/main.log.3.txt)

Hope this helps.  Many thanks

punkymuzzle, 2024-11-15:
Update.
Whilst I was replying, it started working:
[image]

so that's good progress.
I'm going to retest this to coincide with when the battery should also charge to see if the settings get overwritten

punkymuzzle, 2024-11-15:
SCratch that, this was set via PV-Opt:
[image]

stevebuk1, 2024-11-15:
> SCratch that, this was set via PV-Opt:

Good news. I expected the tag "<=Car" to appear as its gated on being on IOG and/or having a car charger. As part of this EndSOC is set to the same as StartSOC to keep things neat. You therefore may notice end SOC is not the same as start SOC because the discharge prevention is an override after the SOC calculations are done. 

Let us know when you've checked out that charging is unaffected.

punkymuzzle, 2024-11-15:
Please tell me if I'm misunderstanding.
So I think what's happening is that when I set the "Prevent Discharge" option in the test dashboard and send to the inverter it's inserting a 30 minute slot in the PV_opt schedule, but setting the W to 1W?
Then, when I set the option back to deselect "prevent Discharge" then it's removing the charging slot from the schedule?
Is that correct?
If it is then I think it's working.

If the above is correct then I have a couple of follow on questions:
1. When I set it, the 1W charging slot was for 30 minutes (and then my next "real" charging slot was scheduled).  I'm assuming that if my next "real" charging slot was a few hours away, the 1W charging slot would extend until my "real" charging slot was active?
2. If so, then what happens after the charging slot has completed?  Will PV-Opt put another 1W slot in place until I cancel it from the test dashboard?
3. If that is the case then can is it just a case of toggling the "Prevent Discharge" on and off or can this be scheduled?
I.e. if I had a charging schedule in place, I want to prevent discharge from, say, 0200 - 0400 to charge my car, then resume my actual PV_opt charging schedule.
Will the current setup cater for this?  or are we just testing that the "Prevent Discharge" option works?

Thanks very much

punkymuzzle, 2024-11-15:
Also, not sure it's 100% working.
The normal schedule has restarted, but when I look at my inverter settings the charge maximum current is still set to 0A, meaning the inverter won't charge is the sun actually comes out.
Here's the settings for the inverter (which by default sets to self use mode"

[image]

But here's the charging settings:
[image]

The 0A should be set to 20 so it can charge / discharge as it sees fit until PV-Opt tells it otherwise

stevebuk1, 2024-11-15:
> Please tell me if I'm misunderstanding. So I think what's happening is that when I set the "Prevent Discharge" option in the test dashboard and send to the inverter it's inserting a 30 minute slot [...]

Yes thats correct. 
> If the above is correct then I have a couple of follow on questions:
>
> 1. When I set it, the 1W charging slot was for 30 minutes (and then my next "real" charging slot was scheduled).  I'm assuming [...]

Yes thats correct also. It will only set a current of zero if Pv_opt isnt requesting a charge. 

> 2. If so, then what happens after the charging slot has completed?  Will PV-Opt put another 1W slot in place until I cancel it from the test dashboard?

Also correct. 

> 3. If that is the case then can is it just a case of toggling the "Prevent Discharge" on and off or can this be scheduled?
>    I.e. if I had a charging schedule in place, I want to prevent discharge from, say, 0200 - 0400 to charge my car, then resume my actual PV_opt charging schedule.

You can use an HA automation with a couple of datetime helpers to set Prevent_Discharge at 2am and clear it at 4am. If Pv_opt wants to force charge the battery in this time then it can and will. 

>    Will the current setup cater for this?  or are we just testing that the "Prevent Discharge" option works?
> 
> Thanks very much

punkymuzzle, 2024-11-15:
OK
Tried setting the "Prevent Discharge" time for a longer time to span PV-Opt charging schedul slots:
[image]

However it still looks like it defaults to a 30 minute slot:
[image]

And then when once PV_opt sets the inverter for Self use mode / Forced Time mode, it's leaving the charging rate at 0A, meaning the battery will not charge:
[image]

stevebuk1, 2024-11-15:
> The normal schedule has restarted, but when I look at my inverter settings the charge maximum current is still set to 0A, meaning the inverter won't charge is the sun actually comes out.

Yes, this is the way that the function has been designed. For Solis inverters, there are two ways of preventing discharge:

1) Set backup mode
2) Set the inverter to charge, but at a current of 0.

For Solis inverters, 1) is that backup mode will charge the battery at full rate until it reaches backup_soc, then slowly discharge until it falls beneath it, then charge at full rate again. Using 2) is basically a "don't use battery" command which avoids the above issues. 

The disadvantage of 2) is that as you've spotted, if its daytime and there is good solar generation, the battery cannot charge and so the excess goes to grid. For EV charging (which is what the function was designed for) this doesnt really come into play as the 7kW draw usually exceeds solar and overnight charging on IOG means theres no solar anyway. For Agile users though, its less than ideal because cheap daytime slots are a thing. 

> But here's the charging settings: [...]

Whilst Prevent_Discharge in set, the inverter should be in "Force Time Use" so it uses the time registers, sets Charge mode and use the value on the charge current slider, which Pv_opt will set to zero. Basically, charging at a very low rate prevents any discharge. 

Whilst Prevent_Discharge is cleared, the inverter should Revert back to Self use. In Solis inverters, the charge and discharge current slides have no effect in Self use. Are you saying that if that charge current remains at zero its a problem in self use? 

For Solis inverters, these two sliders are to set the actual charge current when in a charge slot and to set the discharge current when in a discharge slot (there are separate start/end time registers for charge and discharge). These currents are used irrespective of solar or house consumption, and remain fixed for the entire period between start and end (obviously charge windows and discharge windows cannot overlap).

If I've understood you correctly,

punkymuzzle, 2024-11-15:
> 1. When I set it, the 1W charging slot was for 30 minutes (and then my next "real" charging slot was scheduled).  I'm assuming that if my next "real" charging slot was a few hours away, the 1W [...]
Thanks for this.
However it looks like it didn't insert the next 1W slot in (as per my latest post)?

stevebuk1, 2024-11-15:
> Thanks for this.
> However it looks like it didn't insert the next 1W slot in (as per my latest post)?

Ah ok. It should do, I'll try that on my system this evening (I don't think I tested it out across a slot transition so there may be a problem there).

punkymuzzle, 2024-11-15:
OK great thank you.

Ref the earlier reply:

> Whilst Prevent_Discharge in set, the inverter should be in "Force Time Use" so it uses the time registers, sets Charge mode and use the value on the charge current slider, which Pv_opt will set to [...]

Yes, when this is set to zero, my inverter will not charge (as that's the setting which controls how much power is being used to charge the battery) 

Can this be set to a default of whatever the maximum power of the inverter (in my case it will set to 20A)?

stevebuk1, 2024-11-15:
> Yes, when this is set to zero, my inverter will not charge (as that's the setting which controls how much power is being used to charge the battery)
> 
> Can this be set to a default of whatever the maximum power of the inverter (in my case it will set to 20A)?

Yes that should be an easy fix. I can't imagine Pv_opt doesnt already have code for doing that after a charging slot ends, but perhaps the override logic we've added is operating slightly differently.

punkymuzzle, 2024-11-15:
Awesome, thank you

punkymuzzle, 2024-11-15:
> You can use an HA automation with a couple of datetime helpers to set Prevent_Discharge at 2am and clear it at 4am. If Pv_opt wants to force charge the battery in this time then it can and will.

One more thing :)

If the plan is to use an HA automation in order to allow us to select a slot that "Prevent Discharge" is active for, can't we use the time selection option in the dashboard instead?
[image]

The previous dashboard created by @fboundy allowed these times to be used to set charging times.  So can't we use them to select "Prevent Discharge" instead?

stevebuk1, 2024-11-15:
> The previous dashboard created by @fboundy allowed these times to be used to set charging times. So can't we use them to select "Prevent Discharge" instead?

From what I made of the Solax inverter support threads, this test dashboard was created to test out writes to the inverter. So you'd set up these values, press "send to inverter" and Pv_opt would read these and send to the inverter. The key here is that the actual times were just sent to the inverter along with Power and Target SOC, Pv_opt didn't do much more with them than just program both time register sets to be the same thing.  Adding a "Prevent Discharge" switch to this dashboard was done because it was a convienient place to put it, not because there is any connection between the Discharge prevent functionality and those other sliders. Reusing these to form some sort of new input to Pv_opt for Pv_opt to then create a series of hold points in advance is far more involved. This is what I meant by "The ability for Pv_opt to read an external car charging schedule is for the future." in my early posts. It might be a bit earlier than I originally envisaged because I note fboundy has already done the manual tariff overrides, but current focus is getting the existing EV functionality stable and ready for production release, then adding multiple Zappi support.

punkymuzzle, 2024-11-15:
OK thanks, understood. I`ll look to see if i can use automatons for this.

Just so I fully understand, the prevent discharge button is in isolation, not attached to anything else?
Reason I ask is that selected prevent discharge a while ago (and the 30 min slot created) and it worked, but then hasn't worked afterwards, despite it being left on.
[image]

[image]

stevebuk1, 2024-11-15:
> Just so I fully understand, the prevent discharge button is in isolation, not attached to anything else?

Yes thats right. 

> Reason I ask is that selected prevent discharge a while ago (and the 30 min slot created) and it worked, but then hasn't worked afterwards, despite it being left on.

If left set it should just roll into the next slot on the charge plan when the time comes. If its not doing this its a bug - I'll look at this this evening, alongside the setting of the <=Car tag and the endSOC value.

punkymuzzle, 2024-11-15:
Thank you very much, really appreciate your help

punkymuzzle, 2024-11-15:
For what it's worth I've created some automations, helpers and a dashboard to start / stop the "prevent discharge" at certain times (once the code is fully working):

### **_Helpers:_**
**_created two date/time helpers:_**

_input_datetime.prevent_battery_discharge_time_start
input_datetime.prevent_battery_discharge_time_start_

### **_Automations:_**
**_Turn on Prevent Battery Discharge:_**

```
alias: Prevent Battery Discharge - Turn On
description: Prevent Battery Discharge - Turn On
triggers:
  - value_template: >
      {{ now().strftime('%H:%M') ==
      states('input_datetime.prevent_battery_discharge_time_start')[0:5] }}
    trigger: template
conditions: []
actions:
  - target:
      entity_id: switch.pvopt_prevent_discharge
    action: switch.turn_on
    data: {}
mode: single
```
**_Turn off Prevent Battery Discharge:_**

```
alias: Prevent Battery Discharge - Turn Off
description: Prevent Battery Discharge - Turn Off
triggers:
  - value_template: >
      {{ now().strftime('%H:%M') ==
      states('input_datetime.prevent_battery_discharge_time_stop')[0:5] }}
    trigger: template
conditions: []
actions:
  - target:
      entity_id: switch.pvopt_prevent_discharge
    action: switch.turn_off
    data: {}
mode: single
```

### **_Dashboard Card:_**

```
type: entities
title: Prevent Battery Discharge Settings
entities:
  - entity: input_datetime.prevent_battery_discharge_time_start
  - entity: input_datetime.prevent_battery_discharge_time_stop
  - entity: switch.pvopt_prevent_discharge
    name: Prevent Discharge
state_color: true
```

This is the result:
[image]

Please feel free to ignore / reuse / include in any instructions once this gets released, etc.

stevebuk1, 2024-11-18:
Hi, ok I have a Beta version (3.19.0-Beta-16) to try that 

- continues to prevent battery discharge on contiguous slots (basically it needed extending from 1/2 hour to an hour)
- sorts out the <=Car tagging and SOC values
- resets the inverter current back to 20A once finished. (this is hard coded for now)
- 
https://github.com/fboundy/pv_opt/blob/dev/apps/pv_opt/pv_opt.py
https://github.com/fboundy/pv_opt/blob/dev/apps/pv_opt/pvpy.py
https://github.com/fboundy/pv_opt/blob/dev/apps/pv_opt/solis.py
https://github.com/fboundy/pv_opt/blob/dev/apps/pv_opt/solax.py

I've tested this out on my system without issue so hopefully this will work better.

punkymuzzle, 2024-11-18:
I've installed so will look at how this performs.
Thank you

punkymuzzle, 2024-11-18:
It's been running a short while now and the AppDaemon log has no errors, which is great.
it also seems to be calculating charging slots etc.

I won't have time to test further for a couple of days due to work, but I'll report back as soon as I have finished testing.
Thank you

punkymuzzle, 2024-11-19:
I`ve selected the "prevent discharge" botton.
[image]

It looks like it is set for one hour at the moment so I`ll keep an eye on it to see what happens after the hour is up.
[image]

Looking good so far

punkymuzzle, 2024-11-19:
Spoke too soon. its now disappeared from tge schedule

[image]

stevebuk1, 2024-11-19:
Ok, please grab the log at that point and attach.

punkymuzzle, 2024-11-19:
Here you go (not sure which log you need, so here's all three of them):
[pv_opt.log](https://github.com/user-attachments/files/17814104/pv_opt.log)
[main.log](https://github.com/user-attachments/files/17814106/main.log)
[error.log](https://github.com/user-attachments/files/17814107/error.log)

punkymuzzle, 2024-11-19:
Looks like it has recalculated the slot (11:12 when I noticed)

[image]

punkymuzzle, 2024-11-19:
It's just updated and rescheduled the slot again (11:35):
[image]

So it looks like it's working OK.

I'll uncheck the "prevent discharge" button and wait for it to update the charging schedule, then I'll look to see if the automations / helpers I mentioned in an earlier post work

stevebuk1, 2024-11-19:
The appearing/disappearing I suspect this is coupled with the optimiser taking 13 minutes when you only have 10 (issue #299) , which according to main.log is equally affecting this release (which is expected). 

I think the cause of this is that once the inverter is written to, it waits 300 seconds (5 minutes) for the inverter to update itself before Pv_opt checks the update has gone through, and it needs to do this twice when trying to write to the inverter. Add on the rest of the computation time gets us roughly to 13 minutes. 

I looked through the Solax development issue and found this post:

[image]

You can see "update_cycle_seconds" is set to 300. 
This however is an override for the default value in solax.py, which is set to 15. 

I can't tell from the Solax threads (both over 200 posts long) whether the inverter really needs 300 seconds (I doubt it) or can be reduced to 15. I suggest reducing it to 60 in config.yaml and see what happens. The other alternative is to back off the optimiser frequency from 10 minutes to 30 minutes, to give it time.

punkymuzzle, 2024-11-19:
So we may have three bugs:
1.
Unchecked the "prevent discharge" button at 11:35 but it hasn't removed the 1W slot until the time went past 12:00, then it removed it. So I think the charging calculation isn't removing this until 30 mins have passed.
[image]

2. Although the "prevent discharge" has been deselected, it's reappeared in the charging schedule:
[image]

3.
One other thing I have noticed is that the charging schedule does not take into account the SoC when it's being prevented from discharging (ie the current SoC is 21% but PV-Opt is still calculating at 10%) :
[image]

punkymuzzle, 2024-11-19:
> The appearing/disappearing I suspect this is coupled with the optimiser taking 13 minutes when you only have 10 (issue #299) , which according to main.log is equally affecting this release (which is [...]

I'll take a look at changing the schedule (mine seems to count down from 360 seconds, pretty sure I haven't changed anything with this).
Will look later

Thanks

punkymuzzle, 2024-11-19:
Changed the setting in config.yaml from 300 to 60 and rebooted.
Now the status keeps flipping (timewise) between different readings as to how long is left (example below).  it also flickers between "Waiting for Inverter Read" and "Writing to HA", "Idle", etc.

[image]
[image]
[image]

So it's not happy

stevebuk1, 2024-11-19:
I'll take a look at changing the schedule (mine seems to count down from 360 seconds, pretty sure I haven't changed anything with this).
Will look later

> I'll take a look at changing the schedule (mine seems to count down from 360 seconds, pretty sure I haven't changed anything with this).
> Will look later

Pv_opt takes the value of updated_cycle_seconds (set to 300) and multiplies it by 1.2, which gets you the 360. 

I appreciate nothings changed with this, but we have to wait 5 minutes for an inverter command to work then new functionality to write to the inverter is going to cause a massive backlog of optimiser iterations. Its probably been fine up to now because the inverter is only written to a few times, and the other optimiser runs then have a chance to catchup. Toggling Prevent_discharge is going to take 15 minutes to actually operate, and thats assuming the optimiser queue is empty. If its queued up already, its potentially hours before any external changes are going to take effect. We have to solve this issue first.

stevebuk1, 2024-11-19:
> Changed the setting in config.yaml from 300 to 60 and rebooted.
> Now the status keeps flipping (timewise) between different readings as to how long is left (example below). it also flickers between "Waiting for Inverter Read" and "Writing to HA", "Idle", etc.

Given the screenshots show figures above 60 the change hasnt worked or Pv_opt is still processing events that occured prior to the change. It should go:
Writing to HA
Waiting for Inverter Read (countdown from 60)
Waiting for Inverter Read (countdown from 60)
Idle

I'll relook at the code to see if its picking up 300 (360) from anywhere else, but I deduced this from the log on issue #299. Please post main.log and pv_opt.log with every post - theres nothing I can really do or investigate without them. All the information from screenshots is in the logs anyway.

punkymuzzle, 2024-11-19:
Changed the value to 120 so there is now a pause between it optimising, which seems to have calmed it down.
However, still problems
1. The 1W charge disappeared from the schedule again (assuming this is as per your earlier messages).
2. As the SoC dropped to 10% my inverter hibernated, meaning that HA cannot update it any further.  This issue was overcome by @fboundy adding some code to make PV-Opt write the charging schedule to my inverter when it got to 13%.
So I have to use my (SolaX) app to communicate with the inverter and manually reset chargiing times to effectively bring it to life

I've had to put PV-Opt in read only mode now as my battery is flat so I need to charge it before my expensive peak time (16:00 - 19:00) so I have a full battery.

Do we need a rethink on how this works?  or do youthink we are still on the right track?

Thanks

stevebuk1, 2024-11-19:
> Do we need a rethink on how this works? or do you think we are still on the right track?

From the log you posted today everything actually appears to be working, its just long delays between each optimiser run and lags of whats done where that makes it look like it isn't. We just need to solve the 300/360 seconds delay, which is an issue from the original Solax integration development that hadn't affected things too much up to now but certainly does now. For reference my delay is 15 seconds, so a full optimiser run from idle through to idle takes a minute or so. We need to achieve the same, otheriwse, any method of preventing discharge is going to result in more inverter writes and certainly needing 10 minutes for each one will going to cause issues for anything more than the most basic charge window. 

> Also, I noticed there is an update to PV-Opt (to 3.18.2). Assuming that if I update to this version it will overwrite the beta files. Is it worth me updating first and then re-adding the files from your beta?

No need - the files you have already include the changes made to 3.18.2. I merge production changes into the development code as soon as they are released. 

> Changed the value to 120 so there is now a pause between it optimising, which seems to have calmed it down.
> However, still problems

I really do need a Pv_opt.log from you with every post that describes a behaviour. I really can't do anything without them......

stevebuk1, 2024-11-19:
For now its probably best that you revert to the production release of Pv_opt and make sure you have stable operation with an inverter delay of 60seconds and see if that resolves the issue you raised . If you think its working a pv_opt.log and main.log would be very useful to see if this solves the issues within #299. I'll also put a similar response in #299. We'll leave this open and come back to it once #299 is resolved.

punkymuzzle, 2024-11-19:
Fantastic, thank you very much.
To confirm, now I have reverted back to 3.18.2 and set the optimiser time to 120 seconds (as 60 seconds was causing issues with the dashboard showing various states of optimisation).
I will drop this to 60 seconds as requested and leave overnight to see if this is still OK.  If it is I will then re-download the bera files and start to test the "prevent discharge" option again.

Many thanks

stevebuk1, 2024-11-19:
The recommendation for 60secs was entirely arbitrary. If 120s is reliable
we can stick with that. That means everything completes in 5 minutes which
is less than the optimiser scheduled time of 10 minutes.

Can I just confirm that the value of 60 (or 120) is regarding the value of
id_cycle_seconds in config yaml? You referred to it as optimiser time and I
can't see this needing to be set any lower than 10 minutes.

punkymuzzle, 2024-11-19:
This is the setting I am referring to (in the config.yaml)

>   update_cycle_seconds: 60

> 

This is set towards teh end of the .yaml file under the section:

>  ===============================================================================================================
Brand / Integration Specific Config: SOLAX_X1: 
===============================================================================================================
 These are the default entities used with the Solax X1integration. You can change them here and over-ride the `defaults`

Set it to 60 seconds and no adverse behaviour yet.

Is this the wrong setting?

stevebuk1, 2024-11-19:
> This is set towards the end of the .yaml file under the section:

Yep that's the one!

punkymuzzle, 2024-11-20:
OK so I just went for it last night and tried it out end to end:

1. Installed the latest beta files from the earlier post - **Worked OK**
2. My charging schedule calculated OK with no issues.  It had several charging slots allocated overnight.  I wanted to charge my car overnight for approx 4 hours, which overlapped the charging slots. - **Worked OK**
3. Used the Automations and helpers I created (mentioned in[ earlier post ](https://github.com/fboundy/pv_opt/issues/284#issuecomment-2479430438)) to set the "prevent discharge" to switch on just before my car started charging, and to switch off just after my car stopped charging. **Worked OK**

I watched the 1W charging slot be inserted into the schedule just before the car charger started - **Worked OK**
I watched as the original charging schedule kicked in at the right time (and the battery charged successfully) - **Worked OK**
I watched as the next 1W charging slot was inserted into the schedule - **Worked OK**

In the morning I came down to a car that was fully charged as well as a battery that was charged as per the PV-Opt schedule, which is fantastic.
Looking at the charging graph for my battery it showed that there was no discharge, as planned.
So I reckon that it`s all working correctly.

Attached logs if you need them:
[main.log](https://github.com/user-attachments/files/17828626/main.log)
[pv_opt.log](https://github.com/user-attachments/files/17828627/pv_opt.log)
[error.log](https://github.com/user-attachments/files/17828628/error.log)

my only two comments are:
1. When the charging slot completed and the next 1W slot was configured, there is a slight delay in applying the config to the inverter. This is obviously due to the update cycle, and is expected.  So the way to minimise this delay is to make the update cycle as small as possible, which results in HA constantly recalculating. So there is a balance to be decided upon.
2. Regarding the automations and helpers I created, I will look to add some more, so we have more charging slots.  I also want to look at how the times that are entered can be reset once used, to avoid them recurring at the same time if left as they are.
Once I get this all up and running I will upload the yaml for the automations, helpers and dashboard so they can be included in the PV-Opt dashboard if you want.

Once again, huge thanks for working on this.  Hopefully this can be added into the production version of PV-Opt, but please let me know if you need anything from me.

Thanks

stevebuk1, 2024-11-21:
Pleased it all now seems to be working. 

Regarding your observations:

> * When the charging slot completed and the next 1W slot was configured, there is a slight delay in applying the config to the inverter. This is obviously due to the update cycle, and is expected. [...]

Yes that's it. 
In general, the optimiser runs every 10 minutes. 
If it doesn't need to do anything with the inverter, it takes a minute or two complete. 
If if does need to do anything with the inverter, there is an additional 2 minutes (2 lots of "update cycle_seconds")

However, when you toggle Prevent_discharge, the optimiser will be triggered to run immediately, so its about 4 minutes for the toggle to take effect. This is a minimum though, this will be longer if done midway through a scheduled optimise, as it has to wait for that one to finish first. Not so convienient for your use case of preventing discharge if someone wants a shower. 

The delays are unfortunately avoidable as Pv_opt is reading the status of Prevent_discharge live. The ideal (long term) is that the particular 1/2 hour slots are all setup in advance, then there would be no delays because Pv_opt does the inverter loading 10 minutes before the slot starts. This functionality is mostly incorporated already as PV_opt can read the Intelligent Octopus Go car charging schedule and load all the prevent slots for the night, but thats a very bespoke format that the Octopus Energy Integration generates and it would be quite hard to generate that from a dashboard that has a number of start and end time pairs. I think the way forward on this is to simply have 48 slots in a some sort of drop down list (24 hours of 1/2 hour slots) and one simply sets each slot individually for whether discharge is presented? Or do you think say 3 pairs of start/end times would be sufficient? 

> * Regarding the automations and helpers I created, I will look to add some more, so we have more charging slots.  I also want to look at how the times that are entered can be reset once used, to [...]

Yes please, I'll add them into the repo and documentation. 

> Hopefully this can be added into the production version of PV-Opt, but please let me know if you need anything from me.

Theres a few things to iron out in the dev version (unrelated to this) but yes, the intention is to incorporate all of this into the production version.

punkymuzzle, 2024-11-25:
Hi
I've been using this over the past few days and this has generally been working well.  I have set the "prevent discharge" on several occasions so I can charge my car and it works well.

However I have noticed a couple of anomalies which I'm not sure if we can fix.  These are observations at the moment so I'll keep an eye on it and try and capture the logs when I see this happening.
1. Sometimes the 1W slot is added to the schedule fine and gets written to the inverter.  However if the SoC drops to 10% and the inverter hibernates, when it wakes up (for the 1W slot) PV-Opt cannot update the schedule again as the SoC is still at 10% and still in hibernate mode.
2. Sometimes after the 1W slot has finished, if PV-Opt writes the next slot which happens to be a "self use" mode, then it doesn't amend the Amperage of the inverter, leaving at 0A.  This means that the battery will discharge to provide power to the house, but won't ever charge.  When I've noticed this I have intervened to up the Amperage to allow the battery to charge, but if this happens overnight then the batter will still remain at 10% SoC until it's manually changed.

Apart from the two above issues it seems to be working great.

I still have slight issues with the scheduling slots from my automations / helpers (as I need to think of a way to disable the automation once it's been applied at a certain time, to stop it repeating the next day) but that's a minor thing I need to fix.

Thanks

stevebuk1, 2024-11-25:
> 1. Sometimes the 1W slot is added to the schedule fine and gets written to the inverter.  However if the SoC drops to 10% and the inverter hibernates, when it wakes up (for the 1W slot) PV-Opt cannot update the schedule again as the SoC is still at 10% and still in hibernate mode.

Thanks for pointing this out. Because the Car slots are seen as charge slots (at 1W), then the code Francis wrote to load the next charge slot once the battery SOC approached a "Sleep_SOC" will load a charge of 1W, which isnt much use in lifting the inverter out of its sleep state. I'm not entirely sure I can fix this without designating the car slots as something other than a charge slot, which is a major piece of work. 

> 2\. Sometimes after the 1W slot has finished, if PV-Opt writes the next slot which happens to be a "self use" mode, then it doesn't amend the Amperage of the inverter, leaving at 0A.  This means [...]

I'd added code to write a current of 20A once any form of timed charge period exited, which should equally encompass true charge slots as well as car slots. This is probably an easy fix but I'll need a log to see where its going wrong.

punkymuzzle, 2024-11-25:
> > > 1. Sometimes the 1W slot is added to the schedule fine and gets written to the inverter.  However if the SoC drops to 10% and the inverter hibernates, when it wakes up (for the 1W slot) PV-Opt cannot update the schedule again as the SoC is still at 10% and still in hibernate mode.
> > 
> Thanks for pointing this out. Because the Car slots are seen as charge slots (at 1W), then the code Francis wrote to load the next charge slot once the battery SOC approached a "Sleep_SOC" will load a charge of 1W, which isnt much use in lifting the inverter out of its sleep state when the time comes. I'm not entirely sure I can fix this without designating the car slots as something other than a charge slot, which is a major piece of work.
> 

Not sure if this will work, but, I have two charging slots available on my inverter.  PV-Opt always sets these two sets of sots to be identical, but could we use both slots (one with the 1W slot and the other with the next "real" slot)?  Not sure if this would work, but maybe worth thinking about?

> I'd added code to write a current of 20A once any form of timed charge period exited, which should equally encompass true charge slots as well as car slots. This is probably an easy fix but I'll [...]
Sure, log attached:

[pv_opt.log](https://github.com/user-attachments/files/17909762/pv_opt.log)
[error.log](https://github.com/user-attachments/files/17909764/error.log)
[main.log](https://github.com/user-attachments/files/17909766/main.log)

You might see other errors in these files because PV-Opt has crashed on me in the last few hours and despite me restarting everything multiple times it's hanging on "Optimising Charging Plan" - but that will be for another issue that I will raise.

Thanks

[74 more replies]