stevebuk1/pv_opt #296: Unexpected error in worker for App pv_opt:

philicibine, 2024-11-17:
Hi.  My PV opt doesnt appear to be charging according to the schedule it sets.  Recently for a few mornings I have woken up to the battery being at 20% which is unusual.  Having looked in the app daemon error logs i am seeing the following.

```
07:50:05 WARNING pv_opt: ------------------------------------------------------------
07:50:05 WARNING pv_opt: Unexpected error in worker for App pv_opt:
07:50:05 WARNING pv_opt: Worker Ags: {'id': '28e141cf029f4b3cbc8770702a717b6d', 'name': 'pv_opt', 'objectid': 'b922d7baae0044dc8030cd7835b8036c', 'type': 'scheduler', 'function': <bound method PVOpt.optimise_time of <pv_opt.PVOpt object at 0x7f7c1579d0>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 600, '__thread_id': 'thread-0'}}
07:50:05 WARNING pv_opt: ------------------------------------------------------------
07:50:05 WARNING pv_opt: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1622, in optimise_time
    self.optimise()
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1694, in optimise
    consumption = self.load_consumption(
                  ^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 2468, in load_consumption
    power = self.riemann_avg(power)
            ^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 2980, in riemann_avg
    dt = x.index.diff().total_seconds().fillna(0)
         ^^^^^^^
AttributeError: 'NoneType' object has no attribute 'index'

07:50:05 WARNING pv_opt: ------------------------------------------------------------
```

fboundy, 2024-11-17:
It’s not loading the consumption data correctly. Does the source you are using look correct?

philicibine, 2024-11-17:
The source is correct.  It seems this is an intermittent issue as it is getting the consumption history fine now. That log was from 7.50am this morning.. it then doesnt happen again until 2.55pm and then happens for several runs but stops erroring out at about 3.40pm.  Very strange.  Don't know what happened.

fboundy, 2024-11-17:
It looks like the error is because the consumption array has zero length so it may be that the day isn’t being properly transferred to appDaemon from HA. What is your overall setup? ie what hardware are you running HA on? Are you using the default database (I’ve seen issues with MariaDB)?

philicibine, 2024-11-17:
Just checked and i seem to be using sqlite

fboundy, 2024-11-17:
Yes - that’s the default.

fboundy, 2024-11-17:
How frequently do you record inverter data?

philicibine, 2024-11-18:
I think it polls every 30s.  Not really sure how to check.

philicibine, 2024-11-18:
> It’ll be in the setup of whatever integration you use to communicate with the inverter
> […](#)

Sorry .. I'm using Solax Modbus to communicate with the inverter.  It polls every 60s.  Which again is the default.  The Modbus device I'm using is an esp32 with a modbus TCP gateway flashed onto it

philicibine, 2024-11-20:
Hi, the issue appears to have resolved itself for now.  Will check the firmware version next time I go up into my loft.  I did a full reset of the modbus adapter and the pi.  It seemed to charge last night.

closed 2024-12-23 by fboundy