stevebuk1/pv_opt #305: Modbus Registers - Support for Firmware 4A004C

alienatedsec, 2024-11-19:
@fboundy Following your Facebook post, I did follow up and found the following registers:

- Slot 2
```
43151 - Slot 2 - Charge Current (not tested)
43152 - Slot 2 - Discharge Current (not tested)
43153 - Slot 2 - Charge Start Hour
43154 - Slot 2 - Charge Start Minute
43155 - Slot 2 - Charge Stop Hour
43156 - Slot 2 - Charge Stop Min
43157 - Slot 2 - Discharge Start Hour
43158 - Slot 2 - Discharge Start Minute
43159 - Slot 2 - Discharge Stop Hour
43160 - Slot 2 - Discharge Stop Min
```
A continuous pattern goes up to Slot 5 with the following exception.

- Slot 6 starts outside of the continuous pattern range
```
43491 - Slot 6 - Charge Current (not tested)
43492 - Slot 6 - Discharge Current (not tested)
43493 - Slot 6 - Charge Start Hour
43494 - Slot 6 - Charge Start Minute
43495 - Slot 6 - Charge Stop Hour
43496 - Slot 6 - Charge Stop Min
43497 - Slot 6 - Discharge Start Hour
43498 - Slot 6 - Discharge Start Minute
43499 - Slot 6 - Discharge Stop Hour
43500 - Slot 6 - Discharge Stop Min
```

fboundy, 2024-11-19:
Great stuff!

I will get my various repos updated

fboundy, 2024-11-19:
Is there any sign of on/off switches, target Soc or voltage?

fboundy, 2024-11-19:
Have you upgraded your firmware?

If you’re raising an issue or a PR on the Solax integration it will need to use the HMI f/w as a switch to be backwards compatible

Happy to test once my system is back up

alienatedsec, 2024-11-19:
> Is there any sign of on/off switches, target Soc or voltage?

Nothing that I am aware of (yet). As we can see per `Slot 6` registers, Solis started using registers far outside of the normal range.

> Have you upgraded your firmware?

Yes, this was last week. The additional slots were spotted after your post which kicked my investigation.

> If you’re raising an issue or a PR on the Solax integration it will need to use the HMI f/w as a switch to be backwards compatible

I need to find the register for this. However, the old registers are still working, at least for Slot 1 :)

jmccrohan, 2024-11-20:
@alienatedsec It seems Solis have removed the `3X_0{3,4}.json` register lists from the APKs as of v4.4.0 onwards. Are the registers above from v4.3.8 or have you found the obfuscated version in later APKs?

alienatedsec, 2024-11-20:
@jmccrohan I noticed the same. Additionally, some APK download services could have a banking Trojan embedded inside the SolisCloud apk, so be careful before installing. I also looked for a JSON file, and I found a list of banking institutions, which is odd.

Nevertheless, I had to configure times (specific for each slot) on the inverter's screen first and I queried several ranges with QModbus thereafter.

fboundy, 2024-11-20:
With rev FB00 it looks the the registers are from 43700 onwards and the old registers don't work.  

I have found the registers for everything apart from the "switch"

fboundy, 2024-11-20:
It also appears that Bit 1 of the Energy Control Switch is no longer used

jmccrohan, 2024-11-20:
Bit 1 or Bit 0?

Bit 0 was already deprecated, but Bit 1 corresponded to Time Charging Mode.

fboundy, 2024-11-20:
> Bit 1 or Bit 0?
> 
> Bit 0 was already deprecated, but Bit 1 corresponded to Time Charging Mode.

Bit 1 - it seems to have been replaced by individual switches on each slot

alienatedsec, 2024-11-20:
So I dont see many differences when using `4A004C` - I hope to get the latest HMI soon and I will test thereafter.

fboundy, 2024-11-20:
The full table (excluding `switch` which I have yet to find is:

| Direction |Parameter|Slot|API Code| API Value|Register|Register Value|
|:--|:--|--|:--:|:--:|:--:|:--:|
|charge|start_hours|1|5946|1|43711|1|
|charge|start_minutes|1|5946|30|43712|30|
|charge|end_hours|1|5946|2|43713|2|
|charge|end_minutes|1|5946|30|43714|30|
|charge|current|1|5948|60|43709|600|
|charge|volt|1|5947|49.6|43710|496|
|charge|soc|1|5928|60|43708|60|
|discharge|start_hours|1|5964|1|43753|1|
|discharge|start_minutes|1|5964|30|43754|30|
|discharge|end_hours|1|5964|2|43755|2|
|discharge|end_minutes|1|5964|30|43756|30|
|discharge|current|1|5967|60|43751|600|
|discharge|volt|1|5966|49.6|43752|496|
|discharge|soc|1|5965|60|43750|60|
|charge|start_hours|2|5949|1|43718|1|
|charge|start_minutes|2|5949|30|43719|30|
|charge|end_hours|2|5949|2|43720|2|
|charge|end_minutes|2|5949|30|43721|30|
|charge|current|2|5951|60|43716|600|
|charge|volt|2|5950|49.6|43717|496|
|discharge|start_hours|2|5968|1|43760|1|
|discharge|start_minutes|2|5968|30|43761|30|
|discharge|end_hours|2|5968|2|43762|2|
|discharge|end_minutes|2|5968|30|43763|30|
|discharge|current|2|5971|60|43758|600|
|discharge|volt|2|5970|49.6|43759|496|
|discharge|soc|2|5969|60|43757|60|
|charge|start_hours|3|5952|1|43725|1|
|charge|start_minutes|3|5952|30|43726|30|
|charge|end_hours|3|5952|2|43727|2|
|charge|end_minutes|3|5952|30|43728|30|
|charge|current|3|5954|60|43723|600|
|charge|volt|3|5953|49.6|43724|496|
|charge|soc|3|5930|60|43722|60|
|discharge|start_hours|3|5972|1|43767|1|
|discharge|start_minutes|3|5972|30|43768|30|
|discharge|end_hours|3|5972|2|43769|2|
|discharge|end_minutes|3|5972|30|43770|30|
|discharge|current|3|5975|60|43765|600|
|discharge|volt|3|5974|49.6|43766|496|
|discharge|soc|3|5973|60|43764|60|
|charge|start_hours|4|5955|1|43732|1|
|charge|start_minutes|4|5955|30|43733|30|
|charge|end_hours|4|5955|2|43734|2|
|charge|end_minutes|4|5955|30|43735|30|
|charge|current|4|5957|60|43730|600|
|charge|volt|4|5956|49.6|43731|496|
|charge|soc|4|5931|60|43729|60|
|discharge|start_hours|4|5976|1|43774|1|
|discharge|start_minutes|4|5976|30|43775|30|
|discharge|end_hours|4|5976|2|43776|2|
|discharge|end_minutes|4|5976|30|43777|30|
|discharge|current|4|5979|60|43772|600|
|discharge|volt|4|5978|49.6|43773|496|
|discharge|soc|4|5977|60|43771|60|
|charge|start_hours|5|5958|1|43739|1|
|charge|start_minutes|5|5958|30|43740|30|
|charge|end_hours|5|5958|2|43741|2|
|charge|end_minutes|5|5958|30|43742|30|
|charge|current|5|5960|60|43737|600|
|charge|volt|5|5959|49.6|43738|496|
|charge|soc|5|5932|60|43736|60|
|discharge|start_hours|5|5980|1|43781|1|
|discharge|start_minutes|5|5980|30|43782|30|
|discharge|end_hours|5|5980|2|43783|2|
|discharge|end_minutes|5|5980|30|43784|30|
|discharge|current|5|5983|60|43779|600|
|discharge|volt|5|5982|49.6|43780|496|
|discharge|soc|5|5981|60|43778|60|
|charge|start_hours|6|5961|1|43746|1|
|charge|start_minutes|6|5961|30|43747|30|
|charge|end_hours|6|5961|2|43748|2|
|charge|end_minutes|6|5961|30|43749|30|
|charge|current|6|5963|60|43744|600|
|charge|volt|6|5962|49.6|43745|496|
|charge|soc|6|5933|60|43743|60|
|discharge|start_hours|6|5987|1|43788|1|
|discharge|start_minutes|6|5987|30|43789|30|
|discharge|end_hours|6|5987|2|43790|2|
|discharge|end_minutes|6|5987|30|43791|30|
|discharge|current|6|5986|60|43786|600|
|discharge|volt|6|5985|49.6|43787|496|
|discharge|soc|6|5984|60|43785|60|

fboundy, 2024-11-20:
And here in register order:

|   register | direction   | parameter     |   api cid |   slot |   API Value |   Modbus Value |
|-----------:|:------------|:--------------|----------:|-------:|------------:|---------------:|
|      43708 | charge      | soc           |      5928 |      1 |        60   |             60 |
|      43709 | charge      | current       |      5948 |      1 |        60   |            600 |
|      43710 | charge      | volt          |      5947 |      1 |        49.6 |            496 |
|      43711 | charge      | start_hours   |      5946 |      1 |         1   |              1 |
|      43712 | charge      | start_minutes |      5946 |      1 |        30   |             30 |
|      43713 | charge      | end_hours     |      5946 |      1 |         2   |              2 |
|      43714 | charge      | end_minutes   |      5946 |      1 |        30   |             30 |
|      43715 | charge      | soc           |      5929 |      2 |        60   |             60 |
|      43716 | charge      | current       |      5951 |      2 |        60   |            600 |
|      43717 | charge      | volt          |      5950 |      2 |        49.6 |            496 |
|      43718 | charge      | start_hours   |      5949 |      2 |         1   |              1 |
|      43719 | charge      | start_minutes |      5949 |      2 |        30   |             30 |
|      43720 | charge      | end_hours     |      5949 |      2 |         2   |              2 |
|      43721 | charge      | end_minutes   |      5949 |      2 |        30   |             30 |
|      43722 | charge      | soc           |      5930 |      3 |        60   |             60 |
|      43723 | charge      | current       |      5954 |      3 |        60   |            600 |
|      43724 | charge      | volt          |      5953 |      3 |        49.6 |            496 |
|      43725 | charge      | start_hours   |      5952 |      3 |         1   |              1 |
|      43726 | charge      | start_minutes |      5952 |      3 |        30   |             30 |
|      43727 | charge      | end_hours     |      5952 |      3 |         2   |              2 |
|      43728 | charge      | end_minutes   |      5952 |      3 |        30   |             30 |
|      43729 | charge      | soc           |      5931 |      4 |        60   |             60 |
|      43730 | charge      | current       |      5957 |      4 |        60   |            600 |
|      43731 | charge      | volt          |      5956 |      4 |        49.6 |            496 |
|      43732 | charge      | start_hours   |      5955 |      4 |         1   |              1 |
|      43733 | charge      | start_minutes |      5955 |      4 |        30   |             30 |
|      43734 | charge      | end_hours     |      5955 |      4 |         2   |              2 |
|      43735 | charge      | end_minutes   |      5955 |      4 |        30   |             30 |
|      43736 | charge      | soc           |      5932 |      5 |        60   |             60 |
|      43737 | charge      | current       |      5960 |      5 |        60   |            600 |
|      43738 | charge      | volt          |      5959 |      5 |        49.6 |            496 |
|      43739 | charge      | start_hours   |      5958 |      5 |         1   |              1 |
|      43740 | charge      | start_minutes |      5958 |      5 |        30   |             30 |
|      43741 | charge      | end_hours     |      5958 |      5 |         2   |              2 |
|      43742 | charge      | end_minutes   |      5958 |      5 |        30   |             30 |
|      43743 | charge      | soc           |      5933 |      6 |        60   |             60 |
|      43744 | charge      | current       |      5963 |      6 |        60   |            600 |
|      43745 | charge      | volt          |      5962 |      6 |        49.6 |            496 |
|      43746 | charge      | start_hours   |      5961 |      6 |         1   |              1 |
|      43747 | charge      | start_minutes |      5961 |      6 |        30   |             30 |
|      43748 | charge      | end_hours     |      5961 |      6 |         2   |              2 |
|      43749 | charge      | end_minutes   |      5961 |      6 |        30   |             30 |
|      43750 | discharge   | soc           |      5965 |      1 |        60   |             60 |
|      43751 | discharge   | current       |      5967 |      1 |        60   |            600 |
|      43752 | discharge   | volt          |      5966 |      1 |        49.6 |            496 |
|      43753 | discharge   | start_hours   |      5964 |      1 |         1   |              1 |
|      43754 | discharge   | start_minutes |      5964 |      1 |        30   |             30 |
|      43755 | discharge   | end_hours     |      5964 |      1 |         2   |              2 |
|      43756 | discharge   | end_minutes   |      5964 |      1 |        30   |             30 |
|      43757 | discharge   | soc           |      5969 |      2 |        60   |             60 |
|      43758 | discharge   | current       |      5971 |      2 |        60   |            600 |
|      43759 | discharge   | volt          |      5970 |      2 |        49.6 |            496 |
|      43760 | discharge   | start_hours   |      5968 |      2 |         1   |              1 |
|      43761 | discharge   | start_minutes |      5968 |      2 |        30   |             30 |
[... cut ...]
|      43786 | discharge   | current       |      5986 |      6 |        60   |            600 |
|      43787 | discharge   | volt          |      5985 |      6 |        49.6 |            496 |
|      43788 | discharge   | start_hours   |      5987 |      6 |         1   |              1 |
|      43789 | discharge   | start_minutes |      5987 |      6 |        30   |             30 |
|      43790 | discharge   | end_hours     |      5987 |      6 |         2   |              2 |
|      43791 | discharge   | end_minutes   |      5987 |      6 |        30   |             30 |

fboundy, 2024-11-20:
> So I dont see many differences when using `4A004C` - I hope to get the latest HMI soon and I will test thereafter.

So are you still able to change the time slot settings using the 431xx registers you've listed above? 

I suspect 4A00 follows the old protocol but with an additional 3 sets of register. On the API there is a control code 5936 which creates a longer version of code 103 which is used to control the original 3.

It looks like they dropped this in favour of the completely new set of registers and API codes in 4B00. This would make 4A backwards compatible whereas 4B is not.

Updating the Solax integration will take a bit of work.

alienatedsec, 2024-11-21:
> So are you still able to change the time slot settings using the 431xx registers you've listed above?

Correct.

All 'new' registers are still `0` on the `4A004C` firmware

[image]

fboundy, 2024-12-18:
Now implemented