stevebuk1/pv_opt #307: Tuning Help

solarjones, 2024-11-24:
******************* PV Opt v3.17.1 *******************
When Agile turns negative, I struggle to understand why the charing plan doesn't seem to select the cheapest slots. I suspect it's a combination of the tuning parameters I've selected and my lack of understanding. I've checked the readme/help but would really appreciate an idiots (me!) guide to simply charge up my batteries to 100% and keep them there when the rates pay the most.

Example, why is it set to charge at 01:30 and not 02:30?

[image]
[image]
[image]

[main.log](https://github.com/user-attachments/files/17894188/main.log)
[pv_opt.log](https://github.com/user-attachments/files/17894189/pv_opt.log)

Thanks in advance!

Simon

stevebuk1, 2024-11-24:
Hi Simon,

Its not easy to see that the cheapest slot is being used because the charge plan is displayed on the dashboard as a series of charging windows rather than 1/2 hour slots. 

However, we can see from your log that the cheapest slot is having charges added into it to avoid expensive slots later on. The log excerpt shows the first eight of these "swaps", I'll describe each one, 1 to 8. The important part here is that these swaps are run in price order (cheapest first):

```
17:20:19     INFO: >>> Slot: 25/11 02:30 UTC Factor: 1.000 Forced:      0W  End SOC: 15.0%  SPR:   3671W  SCPA:   6000W  SAC:  16320W  Min Power:   3671W

17:20:19     INFO: >>> Slot: 25/11 02:30 UTC Factor: 1.000 Forced:   3671W  End SOC: 32.4%  SPR:   3671W  SCPA:   2329W  SAC:  12979W  Min Power:   2329W

17:20:19     INFO: >>> Slot: 25/11 02:00 UTC Factor: 1.000 Forced:      0W  End SOC: 15.0%  SPR:   3671W  SCPA:   6000W  SAC:  16320W  Min Power:   3671W

17:20:19     INFO: >>> Slot: 25/11 02:00 UTC Factor: 1.000 Forced:   3671W  End SOC: 32.4%  SPR:   3671W  SCPA:   2329W  SAC:  12979W  Min Power:   2329W

17:20:20     INFO: >>> Slot: 25/11 01:30 UTC Factor: 1.000 Forced:      0W  End SOC: 15.0%  SPR:   3671W  SCPA:   6000W  SAC:  16320W  Min Power:   3671W

17:20:20     INFO: >>> Slot: 25/11 01:30 UTC Factor: 1.000 Forced:   3671W  End SOC: 32.4%  SPR:   3671W  SCPA:   2329W  SAC:  12979W  Min Power:   2329W

17:20:20     INFO: >>> Slot: 25/11 03:00 UTC Factor: 1.000 Forced:      0W  End SOC: 93.8%  SPR:   3671W  SCPA:   6000W  SAC:   1186W  Min Power:   1186W

17:20:20     INFO: >>> Slot: 25/11 03:30 UTC Factor: 1.000 Forced:      0W  End SOC: 92.9%  SPR:   3671W  SCPA:   6000W  SAC:   1371W  Min Power:   1371W

```

The first entry needs a charge rate of 3671W (SPR) @ 02:30 (the cheapest slot), as its the first one the full 6000W (SCPA) is available. 
The 2nd entry wants a charge rate of 3671W (SPR) @ 2.30 also, but as there is 2329W available it doesnt get all of it, it only gets the remainder (also 2329W, indicated by Min Power being 2329W).

At this point, the 2:30 to 3:00 slot is full, in that its now charging at full rate. 

Pv_opt then moves to the next cheapest slot, which is the 02:00 to 02:30 slot, as shown by the 3rd entry. 
The same behaviour as described in entry 1 and 2 is largely repeated in entries 3 and 4, this then fills the 02:00 to 02:30 slot. 

Entries 5 and 6 fill the next cheapest slot, which is the 01:30 to 02:00 slot. 
Finally entries 7 and 8 fill the next cheapest slot, which is the 03:00 to 03:30 slot. 

Other entries go on to fill other slots, but I stopped there because these slots are well away from the 4 slots above. 

Once all the various swaps are done, a windowing function is then run. It notices that there are four slots all charging at 6000W, these being:
01:30 to 02:00 (entries 5 and 6)
02:00 to 02:30 (entries 3 and 4)
02:30 to 03:00 (entries 1 and 2)
03:30 to 03:30 (entries 7 and 8)

As this is the inverter running at 6000W from 01:30 to 3:30, that is what is displayed in the charge plan. 
Joining slots together to form windows is really just a display thing to save a massive long list of 1/2 hour slots.

stevebuk1, 2024-11-24:
Here is a full algorithmic description as written by Francis a while ago. It was written for a specific issue but I think its worthy of being added to the documentation, as it explains a lot of the fundamental operation of PV_opt. 

Note:  plunge pricing is currently disabled. 

1 - Run the "Base Case": this just predicts what your SOC and Grid in/out will be based on the solar and load (which are treated as givens throughout. The net cost of this is calculated.

2 - If there is "Plunge Pricing" with the price below the plunge_threshold (default of 2p/kWh) for any 30 minute "slots" then we assume that you will charge at full power for these because it's a no-brainer. This sets a new base net cost. In the absence of Plunge Pricing the base stays the same.

3 - We then look for the slots with the highest predicted spend (kWh x price) and see if we can buy the same amount of energy for a cheaper price beforehand (allowing for conversion efficiencies). This can only be done before the high cost slot and if the SOC is <100% and we aren't already force charging at full power in that slot. We look for the cheapest slot that meets these criteria, add forced charging at the necessary power to give the energy we need and re-run the prediction. This should reduce the net cost but sometimes it doesn't because the battery empties before we get the benefit but that's OK because we keep re-running this loop until there are no more "high cost slots" which can be swapped out. The reduction in net cost at the end needs to be greater that the pass_threshold (default 4p).
This is the bit that isn't being logged in your version but mine currently looks like this:
11:50:02     INFO: High Cost Usage Swaps
11:50:02     INFO: ---------------------
11:50:02     INFO: 
11:50:02     INFO: 03/04 17:30:  0.79 kWh at  20.86p. <==> 02/04 21:30: 11.69p/kWh  9.19p  SOC:  15.0%-> 15.0% New SOC:  15.0%-> 16.4% Net:  510.7
11:50:02     INFO: 03/04 17:30:  0.79 kWh at  20.86p. <==> 02/04 21:30: 11.69p/kWh  9.19p  SOC:  15.0%-> 16.4% New SOC:  15.0%-> 17.9% Net:  511.1
11:50:02     INFO: 03/04 17:30:  0.79 kWh at  20.86p. <==> 02/04 21:30: 11.69p/kWh  9.19p  SOC:  15.0%-> 18.9% New SOC:  15.0%-> 19.3% Net:  510.8
11:50:02     INFO: 03/04 17:30:  0.79 kWh at  20.86p. <==> 02/04 21:30: 11.69p/kWh  9.19p  SOC:  15.0%-> 23.2% New SOC:  15.0%-> 20.7% Net:  510.5
11:50:02     INFO: 03/04 17:30:  0.79 kWh at  20.86p. <==> 02/04 21:30: 11.69p/kWh  9.19p  SOC:  15.0%-> 28.6% New SOC:  15.0%-> 22.1% Net:  508.4
11:50:03     INFO: 03/04 17:30:  0.79 kWh at  20.86p. <==> 02/04 21:30: 11.69p/kWh  9.19p  SOC:  15.0%-> 34.3% New SOC:  15.0%-> 23.6% Net:  505.5
11:50:03     INFO: 03/04 17:30:  0.79 kWh at  20.86p. <==> 02/04 21:30: 11.69p/kWh  9.19p  SOC:  15.0%-> 40.1% New SOC:  15.0%-> 25.0% Net:  503.4
11:50:03     INFO: 03/04 17:30:  0.79 kWh at  20.86p. <==> 02/04 21:30: 11.69p/kWh  9.19p  SOC:  15.0%-> 45.8% New SOC:  15.0%-> 26.4% Net:  502.0
11:50:03     INFO: 03/04 17:30:  0.79 kWh at  20.86p. <==> 02/04 21:30: 11.69p/kWh  9.19p  SOC:  15.0%-> 51.5% New SOC:  15.0%-> 27.9% Net:  500.9
11:50:03     INFO: 03/04 17:30:  0.79 kWh at  20.86p. <==> 02/04 21:30: 11.69p/kWh  9.19p  SOC:  15.0%-> 57.2% New SOC:  15.0%-> 28.6% Net:  500.6
11:50:03     INFO: 03/04 17:30:  0.79 kWh at  20.86p. <==> 03/04 04:00: 11.77p/kWh  9.25p  SOC:  60.4%-> 57.8% New SOC:  60.4%-> 67.5% Net:  491.8
11:50:03     INFO: 03/04 17:30:  0.79 kWh at  20.86p. <==> 03/04 04:00: 11.77p/kWh  9.25p  SOC:  60.4%-> 67.5% New SOC:  60.4%-> 74.0% Net:  481.6
11:50:03     INFO: 03/04 17:30:  0.79 kWh at  20.86p. <==> 03/04 02:00: 11.84p/kWh  9.31p  SOC:  37.7%-> 35.5% New SOC:  37.7%-> 44.8% Net:  465.2
11:50:03     INFO: 02/04 17:30:  0.77 kWh at  20.38p. <==> 02/04 13:30: 14.15p/kWh 10.87p  SOC:  15.1%-> 15.0% New SOC:  15.1%-> 22.1% Net:  460.0
11:50:03     INFO: 02/04 17:30:  0.77 kWh at  20.38p. <==> 02/04 13:30: 14.15p/kWh 10.87p  SOC:  15.1%-> 22.1% New SOC:  15.1%-> 28.7% Net:  451.4
11:50:03     INFO: 02/04 17:30:  0.77 kWh at  20.38p. <==> 02/04 12:30: 14.47p/kWh 11.11p  SOC:  15.0%-> 15.0% New SOC:  15.0%-> 22.0% Net:  441.3
11:50:03     INFO: 03/04 18:30:  1.11 kWh at  17.76p. <==> 03/04 02:00: 11.84p/kWh 13.18p  SOC:  37.7%-> 44.8% New SOC:  37.7%-> 51.3% Net:  435.2
11:50:03     INFO: 03/04 18:30:  1.11 kWh at  17.76p. <==> 02/04 23:00: 11.91p/kWh 13.25p  SOC:  19.5%-> 26.2% New SOC:  19.5%-> 22.9% Net:  429.3
11:50:03     INFO: 02/04 18:30:  1.11 kWh at  17.74p. <==> 02/04 12:30: 14.47p/kWh 16.08p  SOC:  15.0%-> 22.0% New SOC:  15.0%-> 28.6% Net:  423.3
11:50:03     INFO: 02/04 18:30:  1.11 kWh at  17.74p. <==> 02/04 13:00: 14.55p/kWh 16.17p  SOC:  28.6%-> 28.7% New SOC:  28.6%-> 38.8% Net:  421.3
11:50:03     INFO: 02/04 19:00:  1.05 kWh at  15.45p. <==> 02/04 13:00: 14.55p/kWh 15.28p  SOC:  28.6%-> 38.8% New SOC:  28.6%-> 42.3% Net:  420.7
11:50:03     INFO: 03/04 19:00:  1.05 kWh at  15.45p. <==> 03/04 05:30: 13.93p/kWh 14.62p  SOC:  94.7%-> 91.8% New SOC:  94.7%->100.0% Net:  418.9
11:50:03     INFO: 03/04 19:00:  0.63 kWh at   9.31p. <==> 03/04 13:30: 14.15p/kWh  8.96p  SOC:  66.0%-> 64.2% New SOC:  66.0%-> 71.7% Net:  418.3
11:50:03     INFO: 03/04 22:00:  0.52 kWh at   6.58p. <==> 03/04 21:30: 11.69p/kWh  6.08p  SOC:  15.0%-> 15.0% New SOC:  15.0%-> 19.7% Net:  417.8
11:50:04     INFO: 02/04 18:30:  0.25 kWh at   3.98p. <==> 02/04 12:00: 15.39p/kWh  3.84p  SOC:  15.0%-> 15.0% New SOC:  15.0%-> 17.3% Net:  417.7
[... cut ...]
5 - If you had the ability to force discharge we would then do the same looking for possible discharge slots. In mine I have the following. There is one possible slot but it only nets me 2p so it is excluded.
11:50:04     INFO: Forced Discharging
11:50:04     INFO: ------------------
11:50:04     INFO: 
11:50:04     INFO: 58 slots have an export price greater than the min import price
11:50:05     INFO: 39 Max export price 15.00p/kWh at 02/04 22:00   SOC:  28.6%-> 23.9% Net: 415.7 New SOC:  28.6%-> 15.0% Max export: 1730W 
11:50:06     INFO: 
11:50:06     INFO: Discharge net cost delta:  2.0p: < Discharge threshold (15.0p) => Slots excluded
11:50:06     INFO: Iteration  1: Slots added:   0

solarjones, 2024-11-24:
Thanks both. Starting to make more sense to me now. 

However could do with a little more information on the plunge pricing tuning parameters for scenario 2 from above. You say plunge defaults to 2p.  Does the algorithm look at slots that are cheaper, say -4p and select those, or just finds any slot that exceeds the plunge threshold?  Eg. slot 1 priced at -2.5p and slot 2 priced at -4p. Which would it select (aside from demand, bat SOC etc)?

Thanks again.

stevebuk1, 2024-11-25:
Hi, you must have missed my sentence above  "Note: plunge pricing is currently disabled." - i.e. 2) does not run. Francis disabled it around July time, I guess there were issues with it.

solarjones, 2024-11-25:
Ah I did miss that part sorry and didn't realise it had been disabled.  

Thanks again.

closed 2024-11-26 by stevebuk1