stevebuk1/pv_opt #308: Optimizer appears to run multiple times per Cycle

Pyinthesky99, 2024-11-24:
**Describe the bug**
Optimizer appears to run multiple times per Cycle

**To Reproduce**
Runs every cycle

**Expected behavior**
a single run of Optimizer per Cycle

**Screenshots**
Word Search of pv_opt.log
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser Edge

**Smartphone (please complete the following information):**
N/A

**Additional context**

[pv_opt.log](https://github.com/user-attachments/files/17895171/pv_opt.log)

stevebuk1, 2024-11-25:
Hmm I see what you mean. Every 10 minutes it runs 4 times in quick succession, then waits for the next 10 minute trigger.
```
12:40:37     INFO: Optimiser triggered by Scheduler
12:41:15     INFO: Optimiser triggered by Scheduler
12:41:53     INFO: Optimiser triggered by Scheduler

12:50:00     INFO: Optimiser triggered by Scheduler
12:50:38     INFO: Optimiser triggered by Scheduler
12:51:16     INFO: Optimiser triggered by Scheduler
12:51:53     INFO: Optimiser triggered by Scheduler

13:00:00     INFO: Optimiser triggered by Scheduler
13:00:40     INFO: Optimiser triggered by Scheduler
13:01:22     INFO: Optimiser triggered by Scheduler
13:02:03     INFO: Optimiser triggered by Scheduler 

13:10:00     INFO: Optimiser triggered by Scheduler
13:10:40     INFO: Optimiser triggered by Scheduler
13:11:21     INFO: Optimiser triggered by Scheduler
13:12:02     INFO: Optimiser triggered by Scheduler 

13:20:00     INFO: Optimiser triggered by Scheduler
13:20:40     INFO: Optimiser triggered by Scheduler
13:21:21     INFO: Optimiser triggered by Scheduler
13:22:02     INFO: Optimiser triggered by Scheduler 

13:30:00     INFO: Optimiser triggered by Scheduler
13:30:37     INFO: Optimiser triggered by Scheduler 
13:31:14     INFO: Optimiser triggered by Scheduler
13:31:51     INFO: Optimiser triggered by Scheduler

13:40:00     INFO: Optimiser triggered by Scheduler
13:40:36     INFO: Optimiser triggered by Scheduler
13:41:13     INFO: Optimiser triggered by Scheduler
13:41:50     INFO: Optimiser triggered by Scheduler
13:42:27     INFO: State change detected for switch.pvopt_read_only [config item: read_only] from off to on:

13:50:00     INFO: Optimiser triggered by Scheduler
13:50:36     INFO: Optimiser triggered by Scheduler
13:51:13     INFO: Optimiser triggered by Scheduler
13:51:49     INFO: Optimiser triggered by Scheduler
13:53:18     INFO: State change detected for switch.pvopt_read_only [config item: read_only] from on to off:

14:00:00     INFO: Optimiser triggered by Scheduler 
```

Dashboard changes don't appear to be the issue - theres only 3 of these in the log, and they all trigger their own optimiser run, as they should, in any case the repeated triggers say they are the Scheduler. 

In the most recent log you are showing 4 runs per 10 minutes, but other logs are showing two. However also have seen long periods where you get the desired one run per 10 minutes. 

I think the first thing to try is a restart of AppDaemon ( not HA). 
Give it 1/2 hour to settle down
If after that you are still seeing two (or more) iterations every 10 minutes add the letter "S" to the debug_categories in config.yaml, and post a log of that 1/2 hour later. 

I suspect this might be going back to where we started on the integration where we were having problems getting Appdaemon up and running as it appeared to be in two different places? i.e. we may have two instances of Pv_opt being initialised that are resulting in duplicate Scheduler triggers.

Pyinthesky99, 2024-11-25:
Hello Steve.
Have restarted Appdaemon and still duplicating
Just 
  debug: true
  debug_categories: O S

??

stevebuk1, 2024-11-25:
Not sure if it matters, but no space between O and S......

```
debug: true
debug_categories: OS
```

stevebuk1, 2024-11-25:
Also if you can post appdaemon.log (it might be called main.log) alongside Pv_opt.log once you have 1/2 hour of logging after making the debug_categories change above.

Pyinthesky99, 2024-11-25:
Made changes to config.yaml at 14:35 and it appears to have done one cycle OK at 14:40 and behaving itself every 10 minutes since.. typical!

[appdaemon.log](https://github.com/user-attachments/files/17905049/appdaemon.log)
[error.log](https://github.com/user-attachments/files/17905051/error.log)
[pv_opt.log](https://github.com/user-attachments/files/17905052/pv_opt.log)

I will flick a switch or slider and see if it changes things.

stevebuk1, 2024-11-25:
I think I know what it is. Flick forced discharge on and then off. (Or the
other way around)

Pyinthesky99, 2024-11-25:
Funnily enough, I flicked it to off just after 15:30, as the default is on and my current disposition is for it to be off, the Inverter at the same time had dropped off the net and it has doubled up ever since..
[appdaemon.log](https://github.com/user-attachments/files/17905755/appdaemon.log)
[error.log](https://github.com/user-attachments/files/17905758/error.log)
[pv_opt.log](https://github.com/user-attachments/files/17905759/pv_opt.log)

stevebuk1, 2024-11-25:
Managed to reproduced it on my system. Its an old bug that schedules an optimiser run every 10 minutes if the Optimise Discharge button is toggled, or indeed any entity with "forced" in the internal variable name.  I think its a hangover from a time where charging and discharging routines ran separately on their own schedule and if discharging wasn't needed it wouldn't be scheduled. 

Fixed in 3.19.0-Beta-18:

https://github.com/fboundy/pv_opt/blob/dev/apps/pv_opt/pv_opt.py

Pyinthesky99, 2024-11-25:
Sorry, this one stopped on loading tariffs at 21:22:01 then on the 21:30 cycle it stopped on Optimising Charge Plan

[pv_opt.log](https://github.com/user-attachments/files/17909166/pv_opt.log)

stevebuk1, 2024-11-25:
Hmmm, strange, can you post error.log please?

Pyinthesky99, 2024-11-25:
Even worse now, copied back 'old' pv_opt.py and it stuck again , restarted HA still same, stopped Optimising Charge Plan, so now something more fundamental?
[error.log](https://github.com/user-attachments/files/17909266/error.log)
[pv_opt.log](https://github.com/user-attachments/files/17909267/pv_opt.log)

Pyinthesky99, 2024-11-25:
Just noticed Battery Capacity 10000, Inverter and Charger Power 3600 all changed. Also EV Charger turned on so reset but still currently no difference, stuck

Pyinthesky99, 2024-11-26:
Been out, was prepared to roll it back to 3xpy files you sent a few days ago, but it had gone to Idle!
From last message above at 11:29:30 it did nothing until 14:30:08 when it burst into life and been doing 10 minute cycles ever since.
[pv_opt.log](https://github.com/user-attachments/files/17921971/pv_opt.log)
[appdaemon.log](https://github.com/user-attachments/files/17921968/appdaemon.log)
[error.log](https://github.com/user-attachments/files/17921970/error.log)

Pyinthesky99, 2024-11-26:
Thanks Steve,
I will give it a day or three to make sure it is stable, then apply Beta 18 again.
Apologies for being a pedant, but py script for Beta 18 says at line 17
VERSION = "3.19.0-Beta-17"     :-)

stevebuk1, 2024-11-26:
I like the fact you disabled all the logging - you've worked out that's where I make most of my errors ;-)

Good spot on the version number - corrected here, also added some more logging in case it happens again:

https://github.com/fboundy/pv_opt/blob/dev/apps/pv_opt/pv_opt.py
https://github.com/fboundy/pv_opt/blob/dev/apps/pv_opt/pvpy.py

stevebuk1, 2024-11-26:
> Did a reboot of VM and when it came back EV Charger was set to Zappi. Set it back to none and my Inverter defaults came back to Normal as per my config.yaml.

I'm intrigued about the Ev charger being set to zappi. Can you post your config.yaml please?

stevebuk1, 2024-11-26:
Agreed, nothing in there.

Pyinthesky99, 2024-11-28:
Hi Steve.
I loaded the two beta18 files at around 11:30 yesterday and restored OS logging in my config.yaml.
I have toggled Include Export and Optimise Discharging several times and it is doing an optimise cycle just the once every 10 minutes.
I am happy for this issue to be closed at this time and will await events.

The only thing left on Solarman V2 that is slightly out are the timing switches, as referenced in #273. They work but always report errors
[image]
because it's expecting hours and minutes but as soon as you enter the first number it kicks off a write?

Some sort of local storing of the times and a write button to commit them would be helpful but that would be a while down the road. There was something similar on a Dashboard posted on here but I think that was on a Solax Integration which appears a completely different beast. If you can point me in the right direction I can do some digging.

Thanks again for your help.
Alan

stevebuk1, 2024-11-28:
> There's interesting, to tidy things up I uncommented the Comparing Tariffs
> at the bottom of config.yaml and this morning it kicked off good old 11:58!
> I think I will comment them out and remove the graph from the Dashboard.
> pv_opt.log <https://github.com/user-attachments/files/17947732/pv_opt.log>
>
> Edit.. It carried on at 12 after apparently stopping at 11:58 so did no
> harm. May leave it in and remove later if it causes a problem, obviously
> it's a feature!

I've yet to run compare tariffs, but whilst investigating this I note it
has its own scheduler separate from the 10 minute runs.

stevebuk1, 2024-11-28:
> I am happy for this issue to be closed at this time and will await events.

Ok, will close. 

> The only thing left on Solarman V2 that is slightly out are the timing switches, as referenced in #273. They work but always report errors [...]

The code was changed to combine hours and minutes together so that a single write could be performed, so its not that.
 Please raise in a fresh issue and attach the complete log from where that screenshot came from and I'll investigate.

closed 2024-11-28 by stevebuk1