stevebuk1/pv_opt #310: Error displayed when changing Time Entity (Solarman V2)

Pyinthesky99, 2024-11-28:
**Describe the bug**
If you change the value of any of the hour or minute entities, an error message is displayed.
The Entity value does change as requested. 

**To Reproduce**
Steps to reproduce the behaviour:
Change any hour or Minute timer on Dashboard

**Expected behaviour**
Timer should change without error displayed.

**Screenshots**
The value of the entity time.solis_timed_charge_start_2 was changed via a Dashboard from 11 to 10 hours and on tabbing to minutes an error is displayed.

[image]
[appdaemon.log](https://github.com/user-attachments/files/17949591/appdaemon.log)
[error.log](https://github.com/user-attachments/files/17949595/error.log)
[pv_opt.log](https://github.com/user-attachments/files/17949597/pv_opt.log)
[home-assistant_2024-11-28T14-51-35.385Z.log.txt](https://github.com/user-attachments/files/17949612/home-assistant_2024-11-28T14-51-35.385Z.log.txt)

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser Edge

**Smartphone (please complete the following information):**
 - Device: N/A

**Additional context**
Add any other context about the problem here.

stevebuk1, 2024-11-29:
The entry in home-assistant.log is:

```
Exception: [4065705146] Unexpected response: Invalid length! (Length: 1, Expected: 2)
2024-11-27 01:50:20.564 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py", line 462, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_app.py", line 537, in _handle
    resp = await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_middlewares.py", line 114, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 85, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/api/__init__.py", line 430, in post
    response = await shield(
               ^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2761, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2804, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 996, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1068, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/time/__init__.py", line 35, in _async_set_value
    return await entity.async_set_value(service_call.data[ATTR_TIME])
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/time.py", line 58, in async_set_value
    await self.write(get_t_as_list_int(value, self._multiple_registers), value.strftime(TIME_FORMAT))
  File "/config/custom_components/solarman/entity.py", line 140, in write
    if await self.coordinator.inverter.call(self.code, self.register, value, ACTION_ATTEMPTS_MAX) > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 263, in call
    response = await self.safe_read_write(code, start, arg)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 167, in safe_read_write
    raise Exception(f"[{self.serial}] Unexpected response: Invalid length! (Length: {length}, Expected: {expected})")

```

However, this is an error that doesnt involve Pv_opt (none of the lines of code executed involve pv_opt.py, pypy.py or solis.py). The issue needs to be raised with David Rapan on the Solarman repo, copy and paste the code above to show the error.

Pyinthesky99, 2024-11-30:
Hi Steve
Will do.
Thanks for your response. Hope you are on a commission :-)
Just fyi, it seems to work if I use entity: number.solis_timed_<charge_start/end>etc where the dashboard displays the times in hours and minutes in individual sliders but not if I use entity: time.solis_timed_charge_<start/end>etc although the sliders aren't the easiest things to use, they usually result in (with me, probably) four or five write before it settles on the number I want.
Is there a recommended way to use sliders?
Regards
Alan

Pyinthesky99, 2024-12-01:
Hi Steve
Don't know if you saw my issue raised on David's repo (issue 272)
Have loaded the latest version and no error after using the time 'entities' on my Dashboard.
The number entities are not in his latest version so no problem with the sliders, they are not there.
Rather a pity as I think pv_opt uses the number entity to do the writing!!?

stevebuk1, 2024-12-01:
Pv_opt uses entities of type time entities to do the write (as per the commented defaults in config.yaml) so there should be no issue. I didnt realise you had both number.solis_xxx and time.solis.xxx setup, that doesn't sound like a good idea which from the face of it has now been resolved. I assume Pv_opt is still controlling the inverter ok?

Pyinthesky99, 2024-12-01:
Hi Steve,
That's good! PV_Opt is currently writing OK but I have reverted to the non-updated Solarman. I had a problem when I loaded the latest Solarman and assumed that was why.
I will give it another go tomorrow and let you know how I get on.
Thanks for the info.
Regards
Alan

stevebuk1, 2024-12-01:
Ok no worries, let me know when you've tried it again with the latest Solarman version (try to use HACS or the updates facility in HA for updates - downloading new code direct as I asked you to do in Pv_opt is very unique and not to be repeated!!)  

Entity names PV_opt expects are:

time.solis_timed_charge_start
time.solis_timed_charge_end
number.solis_timed_charge_current
time.solis_timed_discharge_start
time.solis_timed_discharge_end
number.solis_timed_discharge_current

(I don't think you overrode any of these in config.yaml)

Regards,

Steve

Pyinthesky99, 2024-12-02:
I had taken a clone of my Home Assistant VM and installed the updated Solarman_HA on that yesterday. I did initially make a hash of the install but eventually installed it (hopefully) correctly.
I have run up that clone again today and it took a while to settle down but looks ok atm, I was probably a trifle impatient yesterday and thrown off course by the number sliders not loading.
I will run it for 24hrs to make sure.
In the unlikely event someone else will find it useful, here is my pv_opt Dashboard which has been tweaked to accommodate the different Solarman entities. It does include my Meter and MPAN numbers, I don't know if there is any risk publishing them, yours or Francis' are already on the github repo!

[Solarman V2 Dashboard .txt](https://github.com/user-attachments/files/17977605/Solarman.V2.Dashboard.txt)

> 
> (I don't think you overrode any of these in config.yaml)
> 
Wouldn't dream of it!
Alan

Pyinthesky99, 2024-12-03:
All good now.
Thanks for your help
Regards
Alan

closed 2024-12-03 by Pyinthesky99