stevebuk1/pv_opt #311: This is not working for me any more.  Doesn't charge on a daily basis.

philicibine, 2024-12-02:
This is happening on a daily basis now.  We have had some cloudy days and it has not been charging the battery.  Could really do with some help.  The pv_opt log below seems to run right after a reboot for a few times and then just stops.  See below.

```
22:39:37     INFO: Starting Opimisation with discharge enabled
22:39:37     INFO: -------------------------------------------
22:39:37     INFO:
22:39:37     INFO: Checking tariffs:
22:39:37     INFO: -----------------
22:39:37     INFO:
22:39:37     INFO:   Contract last loaded at 01-Dec 19:50 UTC
22:39:37     INFO:
22:39:37     INFO: Checking tariff prices vs Octopus Energy Integration:
22:39:37     INFO: -----------------------------------------------------
22:39:37     INFO:
22:39:37     INFO:   Import: Average Prices - PV_OPT: 21.44 p/kWh  OE Integration: 21.44 p/kWh  Mean difference:  0.00 p/kWh
22:39:37     INFO:   Export: Average Prices - PV_OPT: 15.00 p/kWh  OE Integration: 15.00 p/kWh  Mean difference:  0.00 p/kWh
22:39:37     INFO:
22:39:37     INFO: Checking tariff start and end times:
22:39:37     INFO: ------------------------------------
22:39:37     INFO:   Import: E-1R-AGILE-23-12-06-A                    Start: 2024-11-22 13:00:00+0000 End: 2024-12-02 23:00:00+0000
22:39:37     INFO:   Export: E-1R-OUTGOING-FIX-12M-19-05-13-A         Start: 2019-05-15 23:00:00+0000 End: N/A
22:39:37     INFO:   AGILE tariff detected. Rates will update at 16:00 daily
22:39:37     INFO:   Tariffs OK
22:39:37     INFO:
22:39:37     INFO: Solcast forecast loaded OK
22:39:37     INFO:
22:39:37     INFO: Getting expected consumption data for 2024-12-01 00:00:00+0000 to 2024-12-03 00:00:00+0000:
22:39:37     INFO: Getting consumpion
22:39:38     INFO:   - Got 10 days history from sensor.solax_house_load_today from 21-Nov 23:00 UTC to 01-Dec 22:30 UTC
22:39:38     INFO:   - 10 days was expected. OK
22:39:39     INFO:   - Estimated consumption from sensor.solax_house_load_today loaded OK
22:39:39     INFO:   - Total consumption: 47.2 kWh
22:39:39     INFO:
22:39:39     INFO: Initial SOC: 20.0
22:39:39     INFO: Calculating Base flows:
22:39:39     INFO:
22:39:39     INFO: Optimising for Solcast 50% confidence level forecast from 01-Dec 22:30 UTC to 02-Dec 23:30 UTC
22:40:05     INFO: Optimiser prices loaded for period 01/12 22:30 UTC - 02/12 23:30 UTC
22:40:05     INFO: Base cost:  564.7
22:40:05     INFO:
22:40:05     INFO: High Cost Usage Swaps
22:40:05     INFO: ---------------------
22:40:05     INFO:
22:40:06     INFO: 02/12 18:30 GMT:  1.33 kWh at  42.51p. <==> 02/12 02:30 GMT:  7.46p/kWh  9.88p  SOC:  20.0%-> 20.0% New SOC:  20.0%-> 32.2% Net:  560.0
22:40:06     INFO: 02/12 18:30 GMT:  1.33 kWh at  42.51p. <==> 02/12 02:30 GMT:  7.46p/kWh  9.88p  SOC:  20.0%-> 32.2% New SOC:  20.0%-> 44.4% Net:  544.0
22:40:06     INFO: 02/12 18:30 GMT:  1.33 kWh at  42.51p. <==> 02/12 02:30 GMT:  7.46p/kWh  9.88p  SOC:  20.0%-> 44.4% New SOC:  20.0%-> 47.6% Net:  538.4
22:40:06     INFO: 02/12 18:30 GMT:  1.33 kWh at  42.51p. <==> 02/12 03:30 GMT:  7.83p/kWh 10.38p  SOC:  45.5%-> 43.3% New SOC:  45.5%-> 57.7% Net:  520.8
22:40:06     INFO: 02/12 18:30 GMT:  1.33 kWh at  42.51p. <==> 02/12 03:30 GMT:  7.83p/kWh 10.38p  SOC:  45.5%-> 57.7% New SOC:  45.5%-> 69.9% Net:  503.3
22:40:07     INFO: 02/12 18:30 GMT:  1.33 kWh at  42.51p. <==> 02/12 03:30 GMT:  7.83p/kWh 10.38p  SOC:  45.5%-> 69.9% New SOC:  45.5%-> 73.1% Net:  499.1
22:40:07     INFO: 02/12 18:30 GMT:  1.33 kWh at  42.51p. <==> 02/12 01:30 GMT:  8.86p/kWh 11.75p  SOC:  20.0%-> 20.0% New SOC:  20.0%-> 32.2% Net:  486.9
22:40:07     INFO: 02/12 18:30 GMT:  1.33 kWh at  42.51p. <==> 02/12 01:30 GMT:  8.86p/kWh 11.75p  SOC:  20.0%-> 32.2% New SOC:  20.0%-> 44.4% Net:  472.0
22:40:07     INFO: 02/12 18:30 GMT:  1.33 kWh at  42.51p. <==> 02/12 01:30 GMT:  8.86p/kWh 11.75p  SOC:  20.0%-> 44.4% New SOC:  20.0%-> 47.6% Net:  466.7
22:40:08     INFO: 02/12 18:30 GMT:  1.33 kWh at  42.51p. <==> 02/12 04:30 GMT: 12.13p/kWh 16.07p  SOC:  95.5%-> 92.9% New SOC:  95.5%->100.0% Net:  452.7
22:40:08     INFO: 02/12 18:30 GMT:  1.33 kWh at  42.51p. <==> 02/12 06:30 GMT: 20.39p/kWh 27.03p  SOC:  91.9%-> 89.2% New SOC:  91.9%->100.0% Net:  431.7
22:40:08     INFO: 02/12 18:30 GMT:  1.33 kWh at  42.51p. <==> 02/12 11:00 GMT: 20.70p/kWh 27.43p  SOC:  84.8%-> 80.8% New SOC:  84.8%-> 90.9% Net:  402.7
22:40:08     INFO: 02/12 18:30 GMT:  1.33 kWh at  42.51p. <==> 02/12 11:00 GMT: 20.70p/kWh 27.43p  SOC:  84.8%-> 97.0% New SOC:  84.8%-> 95.0% Net:  397.6
22:40:09     INFO: 02/12 18:30 GMT:  1.31 kWh at  42.20p. <==> 02/12 15:00 GMT: 22.17p/kWh 29.13p  SOC:  65.1%-> 60.7% New SOC:  65.1%-> 77.2% Net:  383.3
22:40:10     INFO: 02/12 19:00 GMT:  0.37 kWh at   8.35p. <==> 02/12 15:00 GMT: 22.17p/kWh  8.29p  SOC:  65.1%-> 77.2% New SOC:  65.1%-> 80.6% Net:  383.3
22:40:10     INFO: 01/12 23:00 GMT:  0.37 kWh at   6.30p. <==> 01/12 22:30 GMT: 16.61p/kWh  6.21p  SOC:  20.0%-> 20.0% New SOC:  20.0%-> 23.4% Net:  383.2
22:40:10     INFO: 02/12 01:00 GMT:  0.31 kWh at   3.65p. <==> 02/12 00:00 GMT: 11.03p/kWh  3.37p  SOC:  20.0%-> 20.0% New SOC:  20.0%-> 22.8% Net:  382.5
22:40:11     INFO: 02/12 01:00 GMT:  0.24 kWh at   2.94p. <==> 02/12 00:00 GMT: 11.03p/kWh  2.62p  SOC:  20.0%-> 22.8% New SOC:  20.0%-> 25.0% Net:  382.2
22:40:18     INFO:
22:40:18     INFO: Low Cost Charging
22:40:18     INFO: ------------------
22:40:18     INFO:
22:40:18     INFO: Max export price when there is no forced charge: 15.00p/kWh.
22:40:18     INFO: 8 slots have an import price less than the max export price
22:40:19     INFO:
22:40:19     INFO: Charge net cost delta: -0.0p: < Pass Threshold 4.0p => Slots Excluded
22:40:19     INFO:
[... 366 lines cut ...]
22:41:45 WARNING pv_opt: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1045, in worker
    funcref(
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1597, in optimise_state_change
    self.optimise()
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1892, in optimise
    status = self.inverter.status
             ^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/solis.py", line 548, in status
    status = self._solis_state()
             ^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/solis.py", line 793, in _solis_state
    and status[direction]["current"] > 0
        ~~~~~~~~~~~~~~~~~^^^^^^^^^^^
KeyError: 'current'

22:41:45 WARNING pv_opt: ------------------------------------------------------------
```
This is after trying a fresh install.  Any idea what is happening?

philicibine, 2024-12-02:
Trying another fresh install.  Have deleted the mosquitto.db.  Looking at the log above it seems to get stuck and loop at comparing tarrifs.

philicibine, 2024-12-02:
Seeing alot of this now too

```
07:28:09 CRITICAL AppDaemon: Thread thread-0 has died
07:28:09 CRITICAL AppDaemon: Pinned apps were: ['pv_opt']
07:28:09 CRITICAL AppDaemon: Thread will be restarted
07:28:09 INFO AppDaemon: Adding thread 0
07:28:10 CRITICAL AppDaemon: Thread thread-0 has died
07:28:10 CRITICAL AppDaemon: Pinned apps were: ['pv_opt']
07:28:10 CRITICAL AppDaemon: Thread will be restarted
07:28:10 INFO AppDaemon: Adding thread 0
```

fboundy, 2024-12-02:
Which version number is this?

philicibine, 2024-12-02:
latest 3.18.2.  Just did a redownload

fboundy, 2024-12-02:
Hopefully fixed with 3.18.3 which I've just released

philicibine, 2024-12-03:
Good morning.  Just an update to say things seem to be running smoothly.  Thankyou for the fix.

closed 2024-12-03 by philicibine