stevebuk1/pv_opt #313: TypeError: can't multiply sequence by non-int of type 'float'

swests, 2024-12-03:
**Describe the bug**
Seeing these errors in the errorlog

cise description of what you expected to happen.

**Screenshots**
```

12:21:18 WARNING pv_opt: ------------------------------------------------------------
--
 
TypeError: can't multiply sequence by non-int of type 'float'
~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
return self.current_limit_amps * self.voltage
File "/homeassistant/appdaemon/apps/pv_opt/pvpy.py", line 455, in max_charge_power
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
min(self.battery.max_charge_power, self.inverter.charger_power)
File "/homeassistant/appdaemon/apps/pv_opt/pvpy.py", line 988, in optimised_force
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
self.flows[case] = self.pv_system.optimised_force(
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1822, in optimise
^^^^^^^^^^^^^^
return f(*args, **kw)
File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
self.optimise()
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1597, in optimise_state_change
^^^^^^^^^^^^^^
return f(*args, **kw)
File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
funcref(
File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1045, in worker
12:21:18 WARNING pv_opt: Traceback (most recent call last):
```

Latest HA and latest PVOPT

fboundy, 2024-12-03:
What have you got set for `battery_current_limit_amps` and `battery_voltage` in `config.yaml`?

swests, 2024-12-03:
`battery_voltage` comes from the SolaX Modbus integration - it's currently 50.1.
[image]

`battery_current_limit_amps` is 100A

Neither are explicitly set in config.yaml

fboundy, 2024-12-03:
OK - try 3.18.4 and let me know what it says. It's more of a workaround than a fix. It should put a warning in the log saying:
    
    Unable to calculate max_charge_power from current limit {self.current_limit_amps} x voltage {self.voltage}

Should give a clue as to where the error is coming from.

swests, 2024-12-03:
Thanks - installed and monitoring.

closed 2024-12-03 by swests