stevebuk1/pv_opt #319: Is there a walk through guide for configuration of AppDaemon ?

rosenightingale, 2024-12-05:
Hi, I'm trying to add this integration to HA in order to get my Solis inverter to charge batteries during Intelligent Octopus Go cheap slots which appear outside of the 11.30am to 5.30am fixed overnight period. 
Working through the installation, I have got stuck with configuring the AppDaemon. I'm not a developer or coder, so would appreciate a step by step guide to this please. I have opened File Editor on HA, but can't see the next step of where to do the configuration. Are there some instructions I can follow for this, please? 
Also, I downloaded and added an integration for Solcast today, it's showing a failed set up, so perhaps this is just waiting for the information to come online, but if I have done something wrong here, would appreciate knowing how to fix it. Many thanks. 
[image]

stevebuk1, 2024-12-05:
Re Solcast, have you visited www.solcast.com and setup a "Home User" account with details of your solar array (power, inclination and direction). The Solcast integration will use the solcast website to get a solar forecast for your setup at your exact location. 
The above is this bit in the Pv_opt setup guide:

[image]

From your Facebook post I remember you got stuck with locating /addon_configs so you could configure appdaemon?

[image]

This is most probably because theres a setting in file editor that you need to change so you can access the root directory. In HomeAssistant, go to Settings -> AddOns -> File Editor then click on the Configuration tab. The following image should display. 
Toggle "Force BasePath" to Off, like mine, then press Save at the bottom right:

[image]

That should allow you to find addon_configs.

rosenightingale, 2024-12-05:
Many thanks, yes I have set up the Solcast Hobby account and inputted my API, that was all fine.  I'm still getting the 'Failed Setup, Will retry' message, but will check again tomorrow in case that's cleared. 
Thank you for the information on changing the setup in the file editor. I did that, and can now see the following screen.  Sorry for the basic question, but do I now just copy and paste the configuration code into this white box? I can't see how to search for anything, does it look correct please? 
[image]

stevebuk1, 2024-12-05:
> Many thanks, yes I have set up the Solcast Hobby account and inputted my API, that was all fine. I'm still getting the 'Failed Setup, Will retry' message, but will check again tomorrow in case that's cleared. 

Ok, not sure of that, I still use the original Solcast integration (the one that is now deleted from HACS), lets see if it is an issue tomorrrow. 

Thank you for the information on changing the setup in the file editor. I did that, and can now see the following screen. Sorry for the basic question, but do I now just copy and paste the configuration code into this white box? I can't see how to search for anything, does it look correct please? 

So this is a file editor, like notepad in windows. The file we need to edit (`/addon_configs/a0d7b954_appdaemon/appdaemon.yaml`) already exists, so we need to navigate to its location and open it. 

Click the folder icon (leftmost icon in the blue bar at the top. Then click the left arrow until you get all the way to root, signified by a /, as below:

[image]

You'll see the folder addon_configs, click on it, then on a0d7b954_appdaemon, then on appdaemon.yaml (the file you want to edit), and it should then open:

[image]

Click anywhere on the right hand side, and you can now edit the file:

[image]

To save, its the floppy disc icon, top right (it will be red if there are edits to save)

rosenightingale, 2024-12-05:
That is incredibly helpful. I think I've now achieved this, copying across the code, my file editor now looks like this.  Hopefully ok. I think the next step on the list is secrets yaml (?) and onwards from there. Thank you for bearing with me!
[image]

stevebuk1, 2024-12-06:
If you don't have a secrets.yaml already, to save time you could insert your MQTT username and password in this file directly (as I did in mine, see my screenshot). 

Then carry on working down the instructions.

rosenightingale, 2024-12-06:
Many thanks. It looks like I don't have a username and password yet either, so although it's installed it's not configured. Will try to read up how to do that. In the meantime, still no luck connecting to the solcast unfortunately, despite installing and inserting API.
[image]

stevebuk1, 2024-12-23:
Hi Rose, where did you get to with this? Is it all working now or did you go another route?

rosenightingale, 2024-12-24:
Thanks for asking Steve, I really couldn't get this to work, unfortunately. The Solcast doesn't install properly although tried it several times. I've actually found another solution which is working well, using Solis Modbus and the times now change in line with the octopus dispatches, so very pleased that it's automated. Hopefully the solcast installation process can be sorted however, if others want to use it.

closed 2024-12-24 by rosenightingale

stevebuk1, 2024-12-24:
Thanks for letting me know and sorry you couldn't get Solcast working. Good you've got something up and running using the Octopus integration.