stevebuk1/pv_opt #321: Possible Incorrect Resulting SOC

Pyinthesky99, 2024-12-14:
**Describe the bug**
At 22:30 the plan was
22:30:13     INFO: Optimal forced charge/discharge slots:
22:30:13     INFO:   14-Dec 02:30 GMT - 14-Dec 03:30 GMT  Power:  1200W  SOC:   40% ->   52%  
22:30:13     INFO:   14-Dec 03:30 GMT - 14-Dec 06:00 GMT  Power:  4000W  SOC:   52% ->   95%  
Which seemed a good plan, although it was supposed to be sunny next day, starting the day in the 90s SOC will get me through to 8pm with batteries alone.
by just after midnight it had changed to 
00:20:39     INFO: 
00:20:39     INFO: Optimal forced charge/discharge slots:
00:20:39     INFO:   14-Dec 02:30 GMT - 14-Dec 03:30 GMT  Power:   100W  SOC:   44% ->   45%  <=
00:20:39     INFO:   14-Dec 03:30 GMT - 14-Dec 06:00 GMT  Power:  4000W  SOC:   45% ->   66%  

Which doesn't add up, 1 and a half hours at 4000W should give me at least 50% more charge.

At 2:20..
02:20:38     INFO:   14-Dec 02:30 GMT - 14-Dec 03:30 GMT  Power:   100W  SOC:   45% ->   46%  <=
02:20:38     INFO:   14-Dec 03:30 GMT - 14-Dec 06:00 GMT  Power:  4000W  SOC:   46% ->   68%  
02:20:38     INFO:   15-Dec 04:00 GMT - 15-Dec 05:30 GMT  Power:  4000W  SOC:   20% ->   77%  
02:20:38     INFO:   15-Dec 07:30 GMT - 15-Dec 08:30 GMT  Power:  4000W  SOC:   73% ->  100%  
02:20:38     INFO: 

At 3:30 was
03:30:38     INFO: Optimal forced charge/discharge slots:
03:30:38     INFO:   14-Dec 03:30 GMT - 14-Dec 06:00 GMT  Power:  4000W  SOC:   46% ->   69%  
03:30:38     INFO:   15-Dec 04:00 GMT - 15-Dec 05:30 GMT  Power:  4000W  SOC:   20% ->   77%  
03:30:38     INFO:   15-Dec 07:30 GMT - 15-Dec 08:30 GMT  Power:  4000W  SOC:   73% ->  100%  

We had very little usage yesterday which may have confused the issue. If it charging at 4000W from 46% for 1 and a half hours does it stop at the predicted outcome, which in this case appears incorrect.
So this morning I have 59% battery and it is dull outside, so could be in trouble later.

**To Reproduce**
Could be difficult as need same weather forecast and pricing?

**Expected behavior**
Stick to approximately same plan.

**Screenshots**
[pv_opt.log](https://github.com/user-attachments/files/18135511/pv_opt.log)

[error.log](https://github.com/user-attachments/files/18135514/error.log)

**Desktop (please complete the following information):**
 - OS: Widows 11
 - Browser Edge

**Smartphone (please complete the following information):**
 N/A

**Additional context**
Add any other context about the problem here.

Pyinthesky99, 2024-12-14:
Was OK at 
00:00:24     INFO: Optimal forced charge/discharge slots:
00:00:24     INFO:   14-Dec 02:30 GMT - 14-Dec 03:30 GMT  Power:  1200W  SOC:   44% ->   56%  
00:00:24     INFO:   14-Dec 03:30 GMT - 14-Dec 06:00 GMT  Power:  4000W  SOC:   56% ->   97% 
But at
00:10:39     INFO:   14-Dec 02:30 GMT - 14-Dec 03:30 GMT  Power:   100W  SOC:   44% ->   45%  <=
00:10:39     INFO:   14-Dec 03:30 GMT - 14-Dec 06:00 GMT  Power:  4000W  SOC:   45% ->   66%  
00:10:39     INFO:   15-Dec 04:00 GMT - 15-Dec 05:30 GMT  Power:  4000W  SOC:   20% ->   77%  
00:10:39     INFO:   15-Dec 07:30 GMT - 15-Dec 08:30 GMT  Power:  4000W  SOC:   73% ->  100%  
00:10:39     INFO: 
00:10:39     INFO: Plan time: 14-Dec 00:00 - 15-Dec 23:30 Initial SOC: 49.0 Base Cost: 430.9 Opt Cost: 295.5

fboundy, 2024-12-14:
Which version is this please?

Do you have the detailed logs for these periods?

Pyinthesky99, 2024-12-14:
Hi Francis,
Thanks for the response.
pv_opt.log and the error.log are in my opening post, would you like HA or something else?
[home-assistant_2024-12-14T15-38-57.233Z.log](https://github.com/user-attachments/files/18136442/home-assistant_2024-12-14T15-38-57.233Z.log)
I am running 3.19.0-Beta-18

fboundy, 2024-12-14:
That’s fine. On my phone so didn’t see them. I noticed that my optimiser was off with the very high prices. The fix I put out a couple of days ago may help with that. Basically it was still trying to use slots that were already maxed out and it ran out. Need to check wether you already had the patch when this happened

Pyinthesky99, 2024-12-14:
No worries, Francis
This is the current plan, which looks lightly odd especially the charge 04:30 to 21:00 at 4000w which takes us from 61% to 33%!!
[image]

So knocked off 'Include Export' and now have..
[image]

[pv_opt.log](https://github.com/user-attachments/files/18136901/pv_opt.log)

fboundy, 2024-12-14:
Looking at the recent cases: it seems to be a reporting issue as it's not charging 4:30 - 21:30, it's actually 4:30 to 5:30 same as the case with no export. I'll have a look at the code that does the rounding and see if there's anything obvious.

fboundy, 2024-12-14:
Can you remind me what integration you use to control your inverter?

fboundy, 2024-12-14:
Hmmm - @stevebuk1  has added a whole load of car-related stuff to that routine so best if he has a look

@stevebuk1 - I have merged my changes to `solis.py` into `dev` on my repo. It's working fine for me but I notice that you were using `hold_soc_old` which I have dropped. If you need this for the EV stuff then by all means reinstate it but let's call it `hold_soc_backup' or similar.

FYI - `dev` on my repo seems to be running pretty well on my system using the Solax integration. I also want to test it using Core modbus and SolisCloud but I'm hopeful it is nearly ready to release

stevebuk1, 2024-12-14:
> 00:20:39 INFO: 00:20:39 INFO: Optimal forced charge/discharge slots: 00:20:39 INFO: 14-Dec 02:30 GMT - 14-Dec 03:30 GMT Power: 100W SOC: 44% -> 45% <= 00:20:39 INFO: 14-Dec 03:30 GMT - 14-Dec 06:00 [...]

So this one is a reporting issue. At the 00:20 optimiser run this is the charge plan from 2:30 split into half hours:

```
                            import  export  Solcast  Solcast_p10  Solcast_p90  weighted  consumption     chg  chg_end      battery    grid  forced         soc     soc_end  carslot  period
2024-12-14 02:30:00+00:00  16.9785    15.0      0.0          0.0          0.0     0.000   161.976109  4211.3   4268.2  -125.054945   287.0     125   43.867708   44.460417        0       1
2024-12-14 03:00:00+00:00  16.9785    15.0      0.0          0.0          0.0     0.000   157.926706  4268.2   4325.1  -125.054945   283.0     125   44.460417   45.053125        0       1
2024-12-14 03:30:00+00:00  16.8420    15.0      0.0          0.0          0.0     0.000   153.877303  4325.1   6145.1 -4000.000000  4154.0    4000   45.053125   64.011458        0       2
2024-12-14 04:00:00+00:00  16.9785    15.0      0.0          0.0          0.0     0.000   149.827900  6145.1   6202.0  -125.054945   275.0     125   64.011458   64.604167        0       2
2024-12-14 04:30:00+00:00  16.9785    15.0      0.0          0.0          0.0     0.000   145.778498  6202.0   6258.9  -125.054945   271.0     125   64.604167   65.196875        0       2
2024-12-14 05:00:00+00:00  16.9785    15.0      0.0          0.0          0.0     0.000   141.729095  6258.9   6315.8  -125.054945   267.0     125   65.196875   65.789583        0       2
2024-12-14 05:30:00+00:00  16.9785    15.0      0.0          0.0          0.0     0.000   137.679692  6315.8   6372.7  -125.054945   263.0     125   65.789583   66.382292        0       2
```

Note the fact that six of the Agile slots have exactly the same price, to four decimal places. This means Pv_opt splits any high cost swaps equally between these six slots. What you actually have from 02:30 to 6:00 is 

- a 125W charge from 02:30 to 03:30 (eventually rounded to 100W)
- a 4kW charge from 03:30 to 04:00
- a 125W charge from 04:00 to 06:00 (also eventually rounded to 100W)

When the above is reported as charge windows, the first 125W period is shown separately but the 4kW slot and the 2nd 125W period are shown as one window. This is because each window is displayed based on whether "period" (final column above) increments. What I've always noticed is that this period is incremented when the charge is increased but NEVER when it is decreased, for reasons I don't understand. @fboundy I noticed this when i first coded the IOG stuff, I assume there must be a reason so I've always left it unchanged, but it would make a lot more sense if the 3 bulleted periods above were treated as 3 charge windows rather than two?  Issues #263 and #271 are queries on the same feature. Was it originally there to join together charge rates that effectively were "charge to 100% and stay there" to avoid unncessary writes to the inverter? Or is it just a bug?

fboundy, 2024-12-14:
Thanks Steve - I’ll have a look tomorrow
On 14 Dec 2024 at 20:36 +0000, fboundy/pv_opt ***@***.***>, wrote:
>
> rates

stevebuk1, 2024-12-14:
> At 22:30 the plan was
> 22:30:13 INFO: Optimal forced charge/discharge slots:
> 22:30:13 INFO: 14-Dec 02:30 GMT - 14-Dec 03:30 GMT Power: 1200W SOC: 40% -> 52%
> 22:30:13 INFO: 14-Dec 03:30 GMT - 14-Dec [...]

Setting aside the charge rates issue that I posted about above, there is indeed quite a big difference between the 22:30 plan (which is maintained in the the 00:00 plan), which charges you to 100% at 06:00am and the 00:10 plan which then only charges you to 66%.

The difference is that the 15th December Solar Forecast has now dropped in, which wasn't taken into account at 00:00 (nor 22:30)

Pyinthesky99, 2024-12-14:
Apologies @fboundy , been away :-)
SolarmanV2, but you probably know that already.

stevebuk1, 2024-12-14:
> @stevebuk1 - I have merged my changes to `solis.py` into `dev` on my repo. It's working fine for me but I notice that you were using `hold_soc_old` which I have dropped. If you need this for the EV stuff then by all means reinstate it but let's call it `hold_soc_backup` or similar.

@fboundy I kept it because in my brief time updating predbat I had a tete-a-tete with a user with a Solis inverter who commented that if hold slots were done with charge current = 0 rather than backup mode, any excess solar would go to grid rather than filling the battery. I felt he had a point, so maintained it in Dev. Not sure whats actually better for cost purposes. 

As all dedicated EV holding uses current = 0 and  99% of IOG cheap rates are at night I'm fine with using current = 0 for all holding - it really only gets any benefit in Agile. Do you want me to convert Dev to that strategy?

stevebuk1, 2024-12-14:
> FYI - `dev` on my repo seems to be running pretty well on my system using the Solax integration. I also want to test it using Core modbus and SolisCloud but I'm hopeful it is nearly ready to release

Agreed, it now feels stable and I'm not actively working on any code changes.  I think the only task left is the readme.md to add a section on EV charging, to explain how to use the EV charge plan generation when on Agile.

fboundy, 2024-12-14:
With the changes I’ve made it will be that way anyway. Would be good if you can test the dev from my repo

fboundy, 2024-12-14:
Ah, ok. In which case could you try the dev branch from the main repo rather than Steve’s fork? I’ve re-factored all the Solis control to a cleaner setup. I can test all the other modes but not solarman as my (3!) data sticks are all too new to support it.

stevebuk1, 2024-12-14:
> Ah, ok. In which case could you try the dev branch from the main repo rather than Steve’s fork? I’ve re-factored all the Solis control to a cleaner setup. I can test all the other modes but not solarman as my (3!) data sticks are all too new to support it.

No problem. I'm doing all development in the main repo now, but still then merging to my fork just to give me an easily accessible release archive.

stevebuk1, 2024-12-14:
Theres still a call here in Pv_opt.py:
`self.inverter.hold_soc_old(enable=True, soc=self.hold[0]["soc"])`

I'll correct and test tomorrow,

Pyinthesky99, 2024-12-15:
> Setting aside the charge rates issue that I posted about above, there is indeed quite a big difference between the 22:30 plan (which is maintained in the the 00:00 plan), which charges you to 100% [...]
Hi Steve
That's slightly curious, the forecast for the 15th was a lot worse than Saturday 14th, I think it was over 4 for Saturday and just over 1 for today. Perhaps last nights cheaper rates had something to do with it.
Regards
Alan

stevebuk1, 2024-12-15:
I mean that until 00:10, it hasn't loaded any solar data for the 15th, so
the battery charging plan was based on solar for the 14th only.

Pyinthesky99, 2024-12-15:
> I mean that until 00:10, it hasn't loaded any solar data for the 15th, so the battery charging plan was based on solar for the 14th only.
>
My thinking was if Sunday's Solar was going to worse than Saturday's, why reduce the charge? It worked out OK in the end as the sun finally peeped out just before lunchtime on Saturday and gassed up the batteries.

fboundy, 2024-12-15:
> No problem. I'm doing all development in the main repo now, but still then merging to my fork just to give me an easily accessible release archive.

You'll see that I've changed the structure of `solis.py` to get rid of all the `if`s and use class inheritance instead.

Pyinthesky99, 2024-12-15:
> Ah, ok. In which case could you try the dev branch from the main repo rather than Steve’s fork? I’ve re-factored all the Solis control to a cleaner setup. I can test all the other modes but not solarman as my (3!) data sticks are all too new to support it.
>
Interesting @fboundy what you say about Dataloggers, mine is a cheap looking silver effort with no visible LEDs, except you can see a green glow in the area of the antenna connector  in complete darkness. As far as I know it is a DLS-W which according to https://usservice.solisinverters.com/support/solutions/articles/73000593957-solis-data-loggers-explained
does not support settings changes and only one inverter.
I have found this, possibly a more likely candidate,, https://www.amazon.co.uk/dp/B0BTT1WCBV?tag=ceukreviews9056831-21&linkCode=ogi&th=1&psc=1     which says supports multiple inverters.
It looks much the same as mine except for the RJ45 looking connector at the antenna end, but it looks to me as if they have mixed pics for both WiFi and ethernet versions.
My solar install was by a group purchase scheme so it is possible the supplier bought a bunch of the cheapest available.
Regards
Alan

fboundy, 2024-12-15:
The Amazon add is confusing. The photos are of DLS-L and DLS-W’s or clones of them. These are both pretty old and no longer supported by Solis as far as I know.

This summarises the various Datalogger models: https://github.com/fboundy/ha_solis_overview

Pyinthesky99, 2024-12-15:
I guess you can add Solarman_ha (PV_OPT SolarmanV2) to your list!
I am still not convinced mine is an official DLS-W due to the lack of LEDs.

[image]

stevebuk1, 2024-12-15:
> You'll see that I've changed the structure of `solis.py` to get rid of all the `if`s and use class inheritance instead.

Much neater! I see you've got rid of the errant hold_soc_old call as well.
I've installed on my system without any issues (I did a push of some comment tidyups but no code changes), will run it over the next few days.

stevebuk1, 2024-12-15:
> My thinking was if Sunday's Solar was going to worse than Saturday's, why reduce the charge? It worked out OK in the end as the sun finally peeped out just before lunchtime on Saturday and gassed up the batteries.

There is certainly something strange going on. 

The 00:00 plan reports this:

```
00:00:23     INFO: Optimisation Summary
00:00:23     INFO: --------------------
00:00:23     INFO: 
00:00:23     INFO:   Base cost:                                444.7p
00:00:24     INFO:   Optimised cost (Optimised Charging):      292.1p
00:00:24     INFO:   Optimised cost (Optimised PV Export):     292.1p <=== Current Setup
00:00:24     INFO:   Optimised cost (Forced Discharge):        292.1p
```

```
The 00:10 plan reports this:

00:10:39     INFO: Optimisation Summary
00:10:39     INFO: --------------------
00:10:39     INFO: 
00:10:39     INFO:   Base cost:                                430.9p
00:10:39     INFO:   Optimised cost (Optimised Charging):      344.2p
00:10:39     INFO:   Optimised cost (Optimised PV Export):     295.5p <=== Current Setup
00:10:39     INFO:   Optimised cost (Forced Discharge):        295.5p
```

Whilst there is only 3pish difference between the two "Current Setup" plans (and I can believe this - the extra charging you had at 00:00 mostly covers off slots which will be at the same price once round trip losses are taken into account), I don't understand the large disparity between the two "(Optimised Pricing)" costs. 

The 00:00 plan makes sense. The Optimised Pv Export plan is the same price as the Optimised Charging plan, which you'd expect as the Optimised Pv Export plan  doenst actually involve any exporting. 

The 00:10 plan doesn't make sense. The Optimised Pv Export plan is cheaper than as the Optimised Charging plan, but it shouldn't be as the Optimised Pv Export plan again doesn't actually involve any exporting. 

Unfortunately whilst I have lots of detail on whatever is the "<===== Current Setup", I have none on the others as all that is logged is the cost value above. I'll see what I can elicit.

fboundy, 2025-01-03:
I've made some changes to the optimiser logic recently. Is this still a problem with 4.0.5?

Pyinthesky99, 2025-01-03:
Hi Francis
I am still running on 3.19.0-Beta-18 but am more than happy to move to 4.0.5 if advised.
The problem may be difficult to reproduce as happened under a certain circumstance, @stevebuk1 was doing some digging, as per his last post.

stevebuk1, 2025-01-03:
Hi Alan, you can now move to v4.0.5 - all the development work done in the Beta releases is now part of the main production release. 

You can use HACS to do the upgrade, noting you may need to do two upgrades to get to V4.0.5 (I did). 

I'm still looking at the logs you previously provided. I now understand why the Optimised Pv Export plan is cheaper than the Optimised Charging plan. I think its simply that the 00:10 plan used a very slightly different starting SOC, and whilst the charging plan result was quite different, the cost was not.

stevebuk1, 2025-01-03:
> You can use HACS to do the upgrade,

By this I mean you can use HA to to the upgrade - just click Settings in the left hand Menu bar and Pv_opt should appear in the list of available updates.

Pyinthesky99, 2025-01-03:
Thanks Steve.
Are we closing this issue?

Pyinthesky99, 2025-01-03:
Pv_opt not in list :-(

stevebuk1, 2025-01-03:
> Thanks Steve.
> Are we closing this issue?

No, not yet, was just giving a progress update on the logs you have provided. 

We should try to capture any future occurrences as well, but to now do this with the production release.

stevebuk1, 2025-01-03:
> Pv_opt not in list :-(

Strange - its called "HA Pv Optimiser", definitely not there? 

I don't have auto update switched on so my list is long - make sure you've pressed "show all updates"

Pyinthesky99, 2025-01-03:
Hi Steve
This is my list in 'Settings'
[image]
This is my list in 'HACS'
[image]

I don't have auto update on either, obviously :-)

Pyinthesky99, 2025-01-03:
Found it in HACS/Automation

stevebuk1, 2025-01-03:
hmmm, just in case this ends up doing a fresh install and creating havoc I'd just overwrite the files as we have been doing previously.

Links are now:

https://github.com/fboundy/pv_opt/blob/main/apps/pv_opt/pv_opt.py
https://github.com/fboundy/pv_opt/blob/main/apps/pv_opt/pvpy.py
https://github.com/fboundy/pv_opt/blob/main/apps/pv_opt/solis.py

Once up and running version should report v4.0.5.

Pyinthesky99, 2025-01-03:
v4.0.5 it is with no great havoc, except for restart of HA required.
It would have been a fresh install via HACS/Automation/ only option Download.
Regards
Alan

Pyinthesky99, 2025-01-03:
Only seems to be running a cycle if I change a slider
[error.log](https://github.com/user-attachments/files/18300217/error.log)
[pv_opt.log](https://github.com/user-attachments/files/18300218/pv_opt.log)

stevebuk1, 2025-01-03:
Thanks Alan. Looks like its falling over in compare_tariffs:

```
  File "/homeassistant/appdaemon/apps/pv_opt/pvpy.py", line 756, in calculate_flows
    solar = self.static_flows[solar_id]
            ~~~~~~~~~~~~~~~~~^^^^^^^^^^
TypeError: 'NoneType' object is not subscriptable
```

@fboundy. This looks like the same issue as being reported in #354.

stevebuk1, 2025-01-03:
Yes please, see if that fixes it. I won't get a chance to investigate till this evening.

Pyinthesky99, 2025-01-03:
Ran a cycle when I saved changes to config.yaml at 13.55 then again at 14:00 (but not at 13:58 obs!) so now looks ok
[error.log](https://github.com/user-attachments/files/18300432/error.log)
[pv_opt.log](https://github.com/user-attachments/files/18300433/pv_opt.log)

Pyinthesky99, 2025-01-04:
Didn't charge at all overnight :-(
[error.log](https://github.com/user-attachments/files/18306294/error.log)
[pv_opt.log](https://github.com/user-attachments/files/18306295/pv_opt.log)

stevebuk1, 2025-01-04:
Apologies - I think a few type conversions got deleted during the refactoring of the development code meaning the times are no longer being written. We weren't able to do any testing of the production release Solarman integration as I think you are the sole user. 

When you get a chance can you use these files and log during a charge slot.

(note the first is from a different path than the others). 
https://github.com/fboundy/pv_opt/blob/steveb-patch/apps/pv_opt/pv_opt.py
https://github.com/fboundy/pv_opt/blob/main/apps/pv_opt/pvpy.py
https://github.com/fboundy/pv_opt/blob/main/apps/pv_opt/solis.py

The first file has extra diagnostic logging but no fixes, hence the request just for a single slot. One attempted write to the inverter at the start of a slot should tell us enough to fix.

Pyinthesky99, 2025-01-04:
Thanks Steve, will probably give it a go tomorrow night as currently first charge slot is 03:30!

Pyinthesky99, 2025-01-05:
I have loaded the new scripts in anticipation of tonight's charge slot but, after the initial load at 12:52, it did not cycle at 13:00

[error.log](https://github.com/user-attachments/files/18310295/error.log)
[pv_opt.log](https://github.com/user-attachments/files/18310296/pv_opt.log)

stevebuk1, 2025-01-05:
I think you are still running solis.py at beta-18 which is the cause of the
error. Replacung it with the link above should solve the problem.

Pyinthesky99, 2025-01-05:
Didn't think so as I used the above links.

stevebuk1, 2025-01-05:
Apologies you are right, I read the previous error in error.log.

The issue is that compare tariffs isn't working in the production code,
this needs commenting out in config.yaml.

Pyinthesky99, 2025-01-05:
Apologies from me, I should have known that from a couple of days ago!!

Pyinthesky99, 2025-01-06:
I had hoped to use a 22:30 to 23:00 slot Sunday Evening but at 22:00 it disappeared. I tried playing around with solar confidence and daily consumption but couldn't get it back. I decided to revert to the beta-18 to get my overnight charge and go back to prod in the morning as there was a slot for Monday afternoon. This morning the afternoon slot has gone so will have to wait for another opportunity.  
Regards

stevebuk1, 2025-01-06:
No problem, whenever you can.  Appreciate you want to got to bed with
Beta-18 running.

stevebuk1, 2025-01-07:
WAside from the more recent issues of no Solarman functionality on v4.0.5, I've completed the review of the logs you first provided on the titled issue. 

> The 00:00 plan makes sense. The Optimised Pv Export plan is the same price as the Optimised Charging plan, which you'd expect as the Optimised Pv Export plan doenst actually involve any exporting. [...]

This does make sense now - the Optimised Pv Export plan generated didnt actually do any exporting:

```
00:10:16     INFO: Max export price when there is no forced charge: 15.00p/kWh.
00:10:16     INFO: 23 slots have an import price less than the max export price
00:10:16     INFO: 21 Min import price  7.85p/kWh at 15/12 05:00 UTC    0W   SOC:  20.0%-> 20.0% Net: 315.0 New SOC:  20.0%-> 39.0% Max export: -0W 
00:10:16     INFO: 20 Min import price  8.16p/kWh at 15/12 04:00 UTC    0W   SOC:  20.0%-> 20.0% Net: 297.0 New SOC:  20.0%-> 39.0% Max export: -0W 
00:10:16     INFO: 19 Min import price  8.84p/kWh at 15/12 04:30 UTC    0W   SOC:  39.0%-> 38.2% Net: 295.5 New SOC:  39.0%-> 57.9% Max export: -0W 
```

> Unfortunately whilst I have lots of detail on whatever is the "<===== Current Setup", I have none on the others as all that is logged is the cost value above. I'll see what I can elicit.

We previously said that the midnight plan (charges to 97%) and the ten past midnight plan (charges to 65%) differred in cost by around 3p. The actual reason for what appears to be a very different plan but for an inconsequential cost difference is simply the starting SOC - all other inputs (solar, consumption etc) were identical.  The midnight plan started with an SOC of 49.5%, whereas the ten past midnight plan started with 49%. This 0.5% difference is consistent for when actually within a 1/2 hour slot, its always 0.5% What to do in the first slot is always difficult when running an optimiser every 10 minutes, in fact we are just discussing this within #356. The current strategy is to try to fix the SOC so each optimiser run within the slot doesnt generate anything too differently, otherwise the calculations have to be become time aware which significantly complicates things but for effectively very little impact. I did manually check the cost of the grid draw throughout the planned period for the midnight plan versus the ten past midnight plan in a spreadsheet and found the same 3p difference. 

So, the conclusion is that for the case presented charging to 97% and charging to 65% actually makes very little cost difference because the slot prices available for charging do need to be quite a bit cheaper than the slot prices that the battery is then intended to cover to make it worthwhile (its a 10% loss each way).

Pyinthesky99, 2025-01-08:
Hi Steve,
Thanks for the exhaustive research. This is a bit of a minefield, I was wondering yesterday why we should import at 18p and charge up to 90 odd % when the solar forecast for today was over 7kWh but then you find production was 2.5! The retail cost today is through the roof so gassing up the batteries was probably a good thing. You could also try and count in 'depreciation' of the batteries is probably in the region of 6p per kWh cycle but can easily get paranoid about all this. Regarding this 'issue' the starting SOC is a moveable feast for the next slot as a 30-minute charge at 4000W can be 2 or 3% different from prediction to actual result 
 without a 'hold'? 
Was thinking about tinkering with the Thresholds but differences are probably minimal.
Best just consider how much the cost of electricity with solar and storage is **overall** over time compared with what it would have cost without.

Thanks again for your efforts, case closed?
Will keep an eye on #356 until my head explodes!
Regards
Alan

Pyinthesky99, 2025-01-09:
Thought, here's an opportunity to run a charge slot with v4.....
11:40:12     INFO: Optimal forced charge/discharge slots:
11:40:12     INFO:   09-Jan 20:30 GMT - 09-Jan 21:00 GMT  Power:  2600W  SOC:   63% ->   75%  
11:40:12     INFO:   09-Jan 21:30 GMT - 09-Jan 22:00 GMT  Power:  4000W  SOC:   67% ->   86%  
11:40:12     INFO:   09-Jan 22:30 GMT - 09-Jan 23:00 GMT  Power:  4000W  SOC:   82% ->  100%  
11:40:12     INFO:   10-Jan 20:30 GMT - 10-Jan 21:00 GMT  Power:   600W  SOC:   20% ->   23%  
11:40:12     INFO:   10-Jan 21:30 GMT - 10-Jan 22:00 GMT  Power:  4000W  SOC:   20% ->   39%  
11:40:12     INFO:   10-Jan 22:30 GMT - 10-Jan 23:00 GMT  Power:   400W  SOC:   34% ->   35%  <=
11:40:12     INFO: 
11:40:12     INFO: Plan time: 09-Jan 11:30 - 10-Jan 23:30 Initial SOC: 76.5 Base Cost: 251.5 Opt Cost: 246.7

Loaded v4 files and it came up with....
11:48:09     INFO: Optimal forced charge/discharge slots:
11:48:09     INFO:   10-Jan 03:30 GMT - 10-Jan 04:30 GMT  Power:  4000W  SOC:   31% ->   60%  
11:48:09     INFO: 
11:48:09     INFO: Plan time: 09-Jan 11:30 - 10-Jan 23:30 Initial SOC: 76.5 Base Cost: 143.1 Opt Cost:  86.8

[error.log](https://github.com/user-attachments/files/18361109/error.log)
[pv_opt.log](https://github.com/user-attachments/files/18361111/pv_opt.log)

If it is going to run just the one slot in the morning I might just let it ride but quite a disparity in the opt runs.

Regards
Alan

stevebuk1, 2025-01-09:
Just to explain, Agile predict is week ahead pricing for Agile, Pv_opt loads it from here: https://agilepredict.com/

Pv_opt makes use of the next 48 hours. At 4pm the next 24 hours(?) become fixed and won't change again, but further ahead than than its just a prediction based on market rates/weather and subject to change.

Pyinthesky99, 2025-01-09:
Hi Steve,
Thanks for the comprehensive explanation. Silly me, I thought it was because it had switched from 'Use Consumption History' from ON to Off, which are quite different usage forecasts.
Either way, hopefully it looks like I will get an opportunity to run a slot with v4 tonight.

stevebuk1, 2025-01-09:
> Silly me, I thought it was because it had switched from 'Use Consumption History' from ON to Off, which are quite different usage forecasts.

No not silly you - silly me! I didn't spot the different consumption values. I checked initial SOC (same), Solar (same) and then import price(different), the stopped checking there. That will certainly also be a reason and probably the bigger contributor.

Pyinthesky99, 2025-01-09:
> No not silly you - silly me! I didn't spot the different consumption values. I checked initial SOC (same), Solar (same) and then import price(different), then stopped checking there. That will certainly also be a reason and probably the bigger contributor.

All valid information though 😀

[84 more replies]

closed 2025-05-31 by github-actions[bot]