stevebuk1/pv_opt #322: Optimiser  re-booting..

Pyinthesky99, 2024-12-17:
**Describe the bug**
Apologies for raising another issue. Feel free to file this away as info only.
At around 21:42, the optimiser started rebooting. After around 10 minutes I decided to shut down AppDaemon and restart HA. It then came back with default values way off and said I had a Zappi, so it had not loaded my config.yaml. I decided to shutdown the entire VM and start it again and all came up as expected.

**To Reproduce**
Not Possible to reproduce at this time. 

[appdaemon.log](https://github.com/user-attachments/files/18165147/appdaemon.log)
[error.log](https://github.com/user-attachments/files/18165150/error.log)

[pv_opt.log](https://github.com/user-attachments/files/18165166/pv_opt.log)

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: Laptop
 - OS: Windows 11
 - Browser Edge

stevebuk1, 2024-12-18:
Up to 21:30, the optimiser was running every 10 minutes. Then, instead of waiting till 21:40, Pv_opt rebooted at 21:38:25....
```

21:30:16     INFO: 
21:30:16     INFO: Next charge window starts in 89.7 minutes . Nothing to do.
21:30:16     INFO: 
21:30:16     INFO: 
21:30:16     INFO: Updating EV charge control
21:30:16     INFO: --------------------------
21:30:16     INFO: 
21:30:16     INFO: Updating Car_Charging Slot Control to off
21:30:16     INFO: Output written to binary_sensor.pvopt_car_charging_slot
21:30:16     INFO: 
21:38:25     INFO: 
21:38:25     INFO: *************** PV Opt Version: v3.19.0-Beta-18 ***************
21:38:25     INFO: 
21:38:25     INFO: Pre-release version. Enabling debug logging
21:38:25     INFO: Debug Categories selected = OS
21:38:25     INFO: Debug Categories selected = OS
```

An inspection of Appdaemon.log shows appdaemon lost its connection with HA at 21:37:51 and then reconnected at 21:38:25. Appdeamon then restarted Pv_opt, which is the cause of the restart above. It then loses connection again, and restarts again at 21:39, this then repeats continuously every few minutes......

```
18:58:54 WARNING AppDaemon: Excessive time spent in callback '_compare_tariff_cb() in pv_opt', Thread 'thread.thread-0' - now complete after 54.211877 seconds (limit=45.0)
19:58:54 WARNING AppDaemon: Excessive time spent in callback '_compare_tariff_cb() in pv_opt', Thread 'thread.thread-0' - now complete after 54.231256 seconds (limit=45.0)
20:58:55 WARNING AppDaemon: Excessive time spent in callback '_compare_tariff_cb() in pv_opt', Thread 'thread.thread-0' - now complete after 55.178068 seconds (limit=45.0)
21:37:51 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
21:37:56 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
21:38:01 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
21:38:25 INFO HASS: Connected to Home Assistant 2024.9.1
21:38:25 INFO HASS: Evaluating startup conditions
21:38:25 INFO HASS: Startup condition met: hass state=RUNNING
21:38:25 INFO HASS: All startup conditions met
21:38:25 INFO AppDaemon: Processing restart for HASS
21:38:25 INFO AppDaemon: Terminating pv_opt
21:38:25 INFO AppDaemon: Reloading Module: /homeassistant/appdaemon/apps/pv_opt/pvpy.py
21:38:25 INFO AppDaemon: Reloading Module: /homeassistant/appdaemon/apps/pv_opt/solis.py
21:38:25 INFO AppDaemon: Reloading Module: /homeassistant/appdaemon/apps/pv_opt/pv_opt.py
21:38:25 INFO AppDaemon: Loading app pv_opt using class PVOpt from module pv_opt
21:38:25 INFO AppDaemon: Calling initialize() for pv_opt
21:38:51 INFO AppDaemon: pv_opt: Entity sensor.pvopt_opt_cost_Agile_Fix created in namespace: default
21:39:07 INFO AppDaemon: pv_opt: Entity sensor.pvopt_opt_cost_Eco7_Fix created in namespace: default
21:39:13 INFO AppDaemon: pv_opt: Entity sensor.pvopt_opt_cost_Flux created in namespace: default
21:39:21 INFO AppDaemon: pv_opt: Entity sensor.pvopt_opt_cost_Cosy created in namespace: default
21:39:22 INFO AppDaemon: pv_opt: Entity sensor.pvopt_tariff_import_OK created in namespace: default
21:39:22 INFO AppDaemon: pv_opt: Entity sensor.pvopt_tariff_export_OK created in namespace: default
21:39:38 WARNING AppDaemon: Excessive time spent in utility loop: 71495.0ms, 71495.0ms in check_app_updates(), 0.0ms in other
21:39:38 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
```

I initially thought this was down to the compare tariffs routine which as we know goes off at :58, but that only triggers once per hour, so wasn't due at 21:38. There aren't any other errors in appdaemon.log or error.log that appear to be the cause of the disconnection. 

There are other instances in the log where Appdaemon loses connection with HA, but most of these cases seem to trigger a full restart of appdeamon itself, which sucessfully clears the problem. 

As Pv_opt wasnt doing anything when this occured this isnt a Pv_opt problem as far as I can tell. If it happens again can you also post homeassistant.log, there might be something in there that might indicate whats going wrong.

Pyinthesky99, 2024-12-19:
Thanks for the response Steve.
Will await events.
Merry Christmas
Alan