stevebuk1/pv_opt #329: Question on best practice for overriding the generated charging scheme

swests, 2024-12-24:
Hi all - Happy Christmas!

Looking for guidance on best practice for overriding the generated charging scheme.

It's Christmas and I'm looking at the output thinking, well I'm going to be pulling power at peak cost, so I should top up the batteries while power is cheaper.  Wondering what's the 'correct' way to do this with PV_OPT doing its thing. 

I went into the Solisapp and manually set a charge window but it looks like PV_OPT resets what I manually set when it wakes up each 10 mins.  Got me thinking, is there a way I can tell PV_OPT that i'm overriding it - beyond switching it to read-only.  Perhaps something that I could automate with a simple UI for manual control.

Look forward ti thoughts/comments/pointers.

solarjones, 2024-12-27:
On the occasions I've wanted to steer PV_Opt differently (happened during negative pricing), I've manually switched to Readonly. As you identified, this stops sending the PV_Opt charging plan to the inverter. I then direct the invert accordingly but it doesn't feel the best as you have to then remember to reenable PV_Opt.

The other approach could be to set your daily consumption higher if you know it's going to be abnormally large for a day. I have mine set to not use consumption history, instead use a shaped profile for a fixed daily usage. Moving the daily consumption slider accordingly only amplifies the shaped profile evenly and won't necessarily give the spike of extra usage you wanted at a particular time. Guessing if you specify a really large consumption, it will demand more charging but this may well result in higher costs during periods you don't need the boost anymore. 

Maybe PV_Opt could benefit from a 'boost' button to fill the battery to max for a variable _n_ hours period, then revert to normal operation.  

Will let those that wrote PV_Opt suggest something.

Pyinthesky99, 2024-12-27:
Hi, and a Happy New Year to you.
I have the Solarman Integration and have entities that can set the time for all three pairs of charge/discharge time as well as the current.
I find if I set times on Timer 2 or 3 it doesn't upset pv_opt as the scheduled plans only use Timer 1.
Hope this helps.
[image]

swests, 2024-12-27:
Ah - thanks @Pyinthesky99.  Slot 2 and 3. Of course. Do you also override the charge amps? As I think that also gets managed by pv_opt.

Pyinthesky99, 2024-12-27:
Yes, @swests I set amps also.
ppv_opt sets them back to what it wants, I think not on the eval cycle but when it loads the plan to the inverter, I think 10 minutes before the slot is due to start. It resets amps to zero when the slot ends.

fboundy, 2024-12-27:
I think you've worked your way around this. I'm not minded to add too many "bespoke" features as it just complicates the app. I think using Read Only and possibly an Automation of your own is the way to go.

closed 2024-12-27 by fboundy