stevebuk1/pv_opt #330: Battery not charging / discharging

lazy-pete, 2024-12-30:
**Describe the bug**
A clear and concise description of what the bug is.

Hi - since 21/12 my battery doesn’t seem to have been charging / discharging. PV-Opt has been so solid lately I’d stopped my daily checks routine over the holidays. I’m not sure if it’s a PV-Opt thing or an inverter battery thing, so have opened a solid ticket as well (and reconnected to the Solis cloud for that purpose). 

It seems as if on the 22nd it just stopped working and is getting whatever PV charge happened that day but not discharging on demand to the house. Similarly, it doesn’t seem to have had received any programmed top ups from PV-Opt. 

I’m a bit confused. 

[image]
[error.log](https://github.com/user-attachments/files/18274707/error.log)
[main.log](https://github.com/user-attachments/files/18274708/main.log)
[pv_opt.log](https://github.com/user-attachments/files/18274709/pv_opt.log)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

lazy-pete, 2024-12-30:
Never mind - had a grid connection fault on the inverter, and it seems the main breaker to the house had tripped, presumably on the 21st. Shall have to set up an automation to alert me to that happening somehow.

stevebuk1, 2024-12-30:
Pleased you found the issue although hopefully the breaker trip was a one off! 

An HA automation is a good way. Or, to keep an eye on things I use an an old mobile phone pinned to the kitchen wall with the HomeAsssistant mobile app permanently displayed - with a few gauges on battery SOC, grid current (using a Shelly EM) and Solar power from the inverter, tends to let me know if everythings ok or somethings crashed/not working.

closed 2024-12-30 by stevebuk1

lazy-pete, 2024-12-30:
Good idea - thanks !