stevebuk1/pv_opt #352: Not using charging schedule

FarmingMartin, 2025-01-02:
I have found just recently thar after months of faultless operation my PV OPT is now not working correctly.
What i have noticed is that although a charging schedule is generated it appears that PV OPT is only using the first charge window and then ignoring all the others.
It is also not deleting the charge window ftom the charging plan once the charge time has passed.
You can see from the attached photos which I have attached shows the time as 18.15 on 2/1/25 but the next charge window showing is the one it used this morning.
This has resulted in my battery not being charged at cheap rated and power from the grid having to be used during peak times.
I have not altered anything in the settings other than installing the latest few updates, which coincidentally corresponds with hen the issues seem to have began.

**Screenshots**
Screenshots of my PV OPT charging plan and my inverter charging schedule showing last charge programed by PV OPT.

[image]
[image]
[image]
[image]
[image]

Restarting AppDaemon temporarily solves the issue by reloading and configuring charging plan but one the first charge window is used PV OPT get stuck as stated above and does not move onto the next one.

fboundy, 2025-01-02:
Can you post the log files please?

FarmingMartin, 2025-01-02:
Please find 2 log files attached.
Note appDaemon restarted around 18.30 to start Pv Opt

Log 1 is taken at 21.32 when charge window shows charge is scheduled but still at state of idle. (still also showing previous charge windows)

Log 2 is taken at 21.35 after restarting AppDaemon which optimised charge windows and instgated battery charge command to inverter.

FarmingMartin, 2025-01-02:
[Log  2.txt](https://github.com/user-attachments/files/18294768/Log.2.txt)
[Log 1.txt](https://github.com/user-attachments/files/18294769/Log.1.txt)

sorry forgot to attach logs. Here they are

fboundy, 2025-01-02:
Can you post the error.log file as well please

FarmingMartin, 2025-01-02:
where do i find that please?

fboundy, 2025-01-02:
It should be in the same folder is the pv_opt.log file

FarmingMartin, 2025-01-02:
sorry i cant find it or the PV opt log file

fboundy, 2025-01-02:
Where are you getting the log you’ve posted already?

FarmingMartin, 2025-01-02:
I'm have got the logs listed in Addon_configs/a0d7b954_appdaemon/log.logThey are not in a seperate folder.Unfortunatly I don't know how to set up the log folders using the /share/logs folder set up using samba share so that part i have not added to to AppDeamon configure yaml.Sent from my Galaxy

fboundy, 2025-01-02:
Can you get the error log through the appDaemon web UI?

FarmingMartin, 2025-01-02:
Just looking now. Just as an update battery charging was scheduled for 21.30 to 22.00 @ 3000wNext charge is 00.00 to 00.30 @ 3000w.It's now 23.00 and charging is still commencing Just restarting  appdeamon to reset everything.Sent from my Galaxy

FarmingMartin, 2025-01-02:
Had a look and nothing is showing.Just trying to see if i can alter the yaml to create error log file.It's now crested a pv opt log file but no error file.Sent from my Galaxy

FarmingMartin, 2025-01-02:
here is the latest log. still no error log showing . ill check tomorrow to see if it generates one over night
[log 3.txt](https://github.com/user-attachments/files/18295279/log.3.txt)

FarmingMartin, 2025-01-03:
No error log has been produced.
Update to issue.
I have found that if I turn on then off read only switch this forces PV OPT to rerun charging optimisation which then resets the charge windows and status, and updates the inverter.
This to me seems to suggest for some reason PV OPT is getting stuck in the last status.
Example today after the charge time had ended PV OPT status still showed "charging" eventhough charging had ended, and inverter charge time was not updated. The expired charge window was also still showing.
Once I had turned read only switch on and off then charging plan optimisation displayed, and the charging plan was updated, the inverter charge times were updated and pv_opt status changed to idle.
Other info that may help
Using bottlecap Dave octopus energy v13.5.1 and SolaX inverter modbus v2024.12.2 intergration.
Inverter Solis RAI 3K-48ES-5G

fboundy, 2025-01-03:
All of this suggests that there is an error. If you go to Addons/AppDaemon/Web UI/Logs you should be able to select “Error log”.

stevebuk1, 2025-01-03:
Whilst supporting issue #321 another user has reported the same symptoms - effectively the optimiser just runs once. 
His error.log shows the error is being generated in the compare_tarriffs. 

I don't normally run compare_tariffs, but have just enabled it and regenerated the same error.

@FarmingMartin In the interim, disable compare tariffs by commenting out the last 13 or so lines in config.yaml, starting with the line `  alt_tariffs:`. That should get you up and running again whilst we investigate a bugfix.

stevebuk1, 2025-01-03:
@fboundy my knowledge of classes is sketchy but this line within compare tariffs sets "static":

`static = pd.concat([solar, consumption], axis=1).set_axis(["solar", "consumption"], axis=1)`

In v3, `static` was passed over to the .flows call explicitly:
`base = self.pv_system.flows(initial_soc, static, solar="solar")`
and it picked it up with the name `static_flows`

but in the refactoring in v4, theres just a call to the new calculate_flows def with no parameters passed:

`self.pv_system.calculate_flows()`

Within calculate_flows I can see the use of self.static_flows, so there needs to be an assignment within compare tariff to that prior to calling `calculate_flows(),` something like `"self.static_flows = static"` so def calculate_flows can see `static`?

fboundy, 2025-01-03:
Thanks Steve - I've had it disabled myself so I will re-enable and look tomorrow

FarmingMartin, 2025-01-03:
stevebuk1 
thanks for the possible work around until the issue is fixed. i have amended my PV_OPT YAML as you have suggested.
Your findings do concur with an update of the bottletop dave Octopus energy intergration update to version 13.5.2 which could be what broke PV_OPT.

stevebuk1, 2025-01-06:
Did the workaround work?

FarmingMartin, 2025-01-06:
Sorry yes it's up and running now thanks.
I have also finally managed to get my PV OPT and Error logs folders created so in any future issues I should be able to supply them to assist with diagnosis.

stevebuk1, 2025-01-09:
If you are ok with replacing files directly then there is a Beta fix available for compare_tariffs that I've tried on my system and looks to be working, but it would be good if you could try it on yours and see if it does as you'd expect before formal release. I haven't used it before so unsure of expected output. 

There are 3 files that need updating, replace the ones on your system with these:

https://github.com/fboundy/pv_opt/blob/patch/apps/pv_opt/pv_opt.py
https://github.com/fboundy/pv_opt/blob/patch/apps/pv_opt/solis.py
https://github.com/fboundy/pv_opt/blob/patch/apps/pv_opt/pvpy.py

fboundy, 2025-01-11:
@stevebuk1 - this looks to be working OK on my system

fboundy, 2025-01-11:
Fixed in 4.0.7

closed 2025-01-11 by fboundy

stevebuk1, 2025-01-11:
I fixed it with commits in patch.