stevebuk1/pv_opt #353: Instability in High Cost Swaps routine

stevebuk1, 2025-01-02:
Running v4.0.4. 

There are large variations in charge plans between subsequent optimiser runs. 

One optimiser run will not schedule any charging at all in the IOG cheap rate period, whereas the next optimiser run will then generate a normal plan.

The issue seems to oscillate for 30minutes, i.e there are 3 optimiser runs with no plan, then 30 minutes with a normal plan, then a repeat. 

This is best illustrated by an HA plot of Pv_opt_charging_current:
[image]

19:00 to 19:30 is incorrect (0A, no plan)
19:30 to 20:00 is correct (26.58A)
20:00 to 20:30 is incorrect (0A, no plan)
20:30 to 21:00  is correct (26.42A) 

(The issue then seems to settle down again from 21:00)

As an example a correct High Cost Swaps log excerpt from 18:30 is fairly long and looks like this:

```
02/01, 18:00:02     INFO: High Cost Usage Swaps
02/01, 18:00:02     INFO: ---------------------
02/01, 18:00:02     INFO: 
02/01, 18:00:02     INFO:   1  60 02/01 20:30 GMT:  1.39 kWh at  31.64p. <==> 02/01 18:00 GMT: 25.63p/kWh 35.71p  SOC:  23.0%-> 17.7% No cheaper slots
02/01, 18:00:02     INFO:   2  60 02/01 22:00 GMT:  1.12 kWh at  25.31p. <==> 02/01 18:00 GMT: 25.63p/kWh 28.74p  SOC:  23.0%-> 16.0% No cheaper slots
02/01, 18:00:02     INFO: >>>   3 Slot: 02/01 23:30 UTC Factor: 0.083 Forced:      0W  End SOC: 16.0%  SPR:    125W  SCPA:   3100W  SAC:   1596W  Min Power:    125W RSC:   2975W
02/01, 18:00:02     INFO: >>>   3 Slot: 03/01 00:00 UTC Factor: 0.083 Forced:      0W  End SOC: 16.0%  SPR:    125W  SCPA:   3100W  SAC:   1596W  Min Power:    125W RSC:   2975W
02/01, 18:00:02     INFO: >>>   3 Slot: 03/01 00:30 UTC Factor: 0.083 Forced:      0W  End SOC: 16.0%  SPR:    125W  SCPA:   3100W  SAC:   1596W  Min Power:    125W RSC:   2975W
02/01, 18:00:02     INFO: >>>   3 Slot: 03/01 01:00 UTC Factor: 0.083 Forced:      0W  End SOC: 16.0%  SPR:    125W  SCPA:   3100W  SAC:   1596W  Min Power:    125W RSC:   2975W
02/01, 18:00:02     INFO: >>>   3 Slot: 03/01 01:30 UTC Factor: 0.083 Forced:      0W  End SOC: 16.0%  SPR:    125W  SCPA:   3100W  SAC:   1596W  Min Power:    125W RSC:   2975W
02/01, 18:00:02     INFO: >>>   3 Slot: 03/01 02:00 UTC Factor: 0.083 Forced:      0W  End SOC: 16.0%  SPR:    125W  SCPA:   3100W  SAC:   1596W  Min Power:    125W RSC:   2975W
02/01, 18:00:02     INFO: >>>   3 Slot: 03/01 02:30 UTC Factor: 0.083 Forced:      0W  End SOC: 16.0%  SPR:    125W  SCPA:   3100W  SAC:   1596W  Min Power:    125W RSC:   2975W
02/01, 18:00:02     INFO: >>>   3 Slot: 03/01 03:00 UTC Factor: 0.083 Forced:      0W  End SOC: 16.0%  SPR:    125W  SCPA:   3100W  SAC:   1596W  Min Power:    125W RSC:   2975W
02/01, 18:00:02     INFO: >>>   3 Slot: 03/01 03:30 UTC Factor: 0.083 Forced:      0W  End SOC: 16.0%  SPR:    125W  SCPA:   3100W  SAC:   1596W  Min Power:    125W RSC:   2975W
02/01, 18:00:02     INFO: >>>   3 Slot: 03/01 04:00 UTC Factor: 0.083 Forced:      0W  End SOC: 16.0%  SPR:    125W  SCPA:   3100W  SAC:   1596W  Min Power:    125W RSC:   2975W
02/01, 18:00:02     INFO: >>>   3 Slot: 03/01 04:30 UTC Factor: 0.083 Forced:      0W  End SOC: 16.0%  SPR:    125W  SCPA:   3100W  SAC:   1596W  Min Power:    125W RSC:   2975W
02/01, 18:00:02     INFO: >>>   3 Slot: 03/01 05:00 UTC Factor: 0.083 Forced:      0W  End SOC: 16.0%  SPR:    125W  SCPA:   3100W  SAC:   1596W  Min Power:    125W RSC:   2975W
02/01, 18:00:02     INFO:   3  60 03/01 06:00 GMT:  0.75 kWh at  16.99p. <==> 02/01 23:30 GMT:  7.00p/kWh  5.23p  SOC:  16.0%-> 16.0% New SOC:  16.0%-> 16.5% Net:  421.1
02/01, 18:00:02     INFO: >>>   4 Slot: 02/01 23:30 UTC Factor: 0.083 Forced:    125W  End SOC: 16.5%  SPR:    113W  SCPA:   2975W  SAC:   1587W  Min Power:    113W RSC:   2862W
02/01, 18:00:02     INFO: >>>   4 Slot: 03/01 00:00 UTC Factor: 0.083 Forced:    125W  End SOC: 17.0%  SPR:    113W  SCPA:   2975W  SAC:   1577W  Min Power:    113W RSC:   2862W
02/01, 18:00:02     INFO: >>>   4 Slot: 03/01 00:30 UTC Factor: 0.083 Forced:    125W  End SOC: 17.5%  SPR:    113W  SCPA:   2975W  SAC:   1568W  Min Power:    113W RSC:   2862W
02/01, 18:00:02     INFO: >>>   4 Slot: 03/01 01:00 UTC Factor: 0.083 Forced:    125W  End SOC: 18.0%  SPR:    113W  SCPA:   2975W  SAC:   1558W  Min Power:    113W RSC:   2862W
02/01, 18:00:02     INFO: >>>   4 Slot: 03/01 01:30 UTC Factor: 0.083 Forced:    125W  End SOC: 18.5%  SPR:    113W  SCPA:   2975W  SAC:   1549W  Min Power:    113W RSC:   2862W
02/01, 18:00:02     INFO: >>>   4 Slot: 03/01 02:00 UTC Factor: 0.083 Forced:    125W  End SOC: 19.0%  SPR:    113W  SCPA:   2975W  SAC:   1539W  Min Power:    113W RSC:   2862W
02/01, 18:00:02     INFO: >>>   4 Slot: 03/01 02:30 UTC Factor: 0.083 Forced:    125W  End SOC: 19.5%  SPR:    113W  SCPA:   2975W  SAC:   1530W  Min Power:    113W RSC:   2862W
02/01, 18:00:02     INFO: >>>   4 Slot: 03/01 03:00 UTC Factor: 0.083 Forced:    125W  End SOC: 20.0%  SPR:    113W  SCPA:   2975W  SAC:   1520W  Min Power:    113W RSC:   2862W
02/01, 18:00:02     INFO: >>>   4 Slot: 03/01 03:30 UTC Factor: 0.083 Forced:    125W  End SOC: 20.5%  SPR:    113W  SCPA:   2975W  SAC:   1511W  Min Power:    113W RSC:   2862W
02/01, 18:00:02     INFO: >>>   4 Slot: 03/01 04:00 UTC Factor: 0.083 Forced:    125W  End SOC: 21.0%  SPR:    113W  SCPA:   2975W  SAC:   1501W  Min Power:    113W RSC:   2862W
02/01, 18:00:02     INFO: >>>   4 Slot: 03/01 04:30 UTC Factor: 0.083 Forced:    125W  End SOC: 21.5%  SPR:    113W  SCPA:   2975W  SAC:   1492W  Min Power:    113W RSC:   2862W
02/01, 18:00:02     INFO: >>>   4 Slot: 03/01 05:00 UTC Factor: 0.083 Forced:    125W  End SOC: 22.0%  SPR:    113W  SCPA:   2975W  SAC:   1482W  Min Power:    113W RSC:   2862W
02/01, 18:00:03     INFO:   4  60 03/01 08:00 GMT:  0.68 kWh at  15.41p. <==> 02/01 23:30 GMT:  7.00p/kWh  4.76p  SOC:  16.0%-> 22.0% New SOC:  16.0%-> 16.9% Net:  410.6
02/01, 18:00:03     INFO: >>>   5 Slot: 02/01 23:30 UTC Factor: 0.083 Forced:    238W  End SOC: 16.9%  SPR:    113W  SCPA:   2862W  SAC:   1578W  Min Power:    113W RSC:   2749W
02/01, 18:00:03     INFO: >>>   5 Slot: 03/01 00:00 UTC Factor: 0.083 Forced:    238W  End SOC: 17.9%  SPR:    113W  SCPA:   2862W  SAC:   1560W  Min Power:    113W RSC:   2749W
02/01, 18:00:03     INFO: >>>   5 Slot: 03/01 00:30 UTC Factor: 0.083 Forced:    238W  End SOC: 18.9%  SPR:    113W  SCPA:   2862W  SAC:   1542W  Min Power:    113W RSC:   2749W
02/01, 18:00:03     INFO: >>>   5 Slot: 03/01 01:00 UTC Factor: 0.083 Forced:    238W  End SOC: 19.8%  SPR:    113W  SCPA:   2862W  SAC:   1524W  Min Power:    113W RSC:   2749W
02/01, 18:00:03     INFO: >>>   5 Slot: 03/01 01:30 UTC Factor: 0.083 Forced:    238W  End SOC: 20.8%  SPR:    113W  SCPA:   2862W  SAC:   1506W  Min Power:    113W RSC:   2749W
02/01, 18:00:03     INFO: >>>   5 Slot: 03/01 02:00 UTC Factor: 0.083 Forced:    238W  End SOC: 21.7%  SPR:    113W  SCPA:   2862W  SAC:   1488W  Min Power:    113W RSC:   2749W
02/01, 18:00:03     INFO: >>>   5 Slot: 03/01 02:30 UTC Factor: 0.083 Forced:    238W  End SOC: 22.6%  SPR:    113W  SCPA:   2862W  SAC:   1470W  Min Power:    113W RSC:   2749W
02/01, 18:00:03     INFO: >>>   5 Slot: 03/01 03:00 UTC Factor: 0.083 Forced:    238W  End SOC: 23.6%  SPR:    113W  SCPA:   2862W  SAC:   1452W  Min Power:    113W RSC:   2749W
02/01, 18:00:03     INFO: >>>   5 Slot: 03/01 03:30 UTC Factor: 0.083 Forced:    238W  End SOC: 24.6%  SPR:    113W  SCPA:   2862W  SAC:   1434W  Min Power:    113W RSC:   2749W
02/01, 18:00:03     INFO: >>>   5 Slot: 03/01 04:00 UTC Factor: 0.083 Forced:    238W  End SOC: 25.5%  SPR:    113W  SCPA:   2862W  SAC:   1416W  Min Power:    113W RSC:   2749W
02/01, 18:00:03     INFO: >>>   5 Slot: 03/01 04:30 UTC Factor: 0.083 Forced:    238W  End SOC: 26.5%  SPR:    113W  SCPA:   2862W  SAC:   1397W  Min Power:    113W RSC:   2749W
02/01, 18:00:03     INFO: >>>   5 Slot: 03/01 05:00 UTC Factor: 0.083 Forced:    238W  End SOC: 27.4%  SPR:    113W  SCPA:   2862W  SAC:   1379W  Min Power:    113W RSC:   2749W
02/01, 18:00:03     INFO:   5  60 03/01 08:00 GMT:  0.68 kWh at  15.41p. <==> 02/01 23:30 GMT:  7.00p/kWh  4.76p  SOC:  16.0%-> 27.4% New SOC:  16.0%-> 17.4% Net:  400.0
02/01, 18:00:03     INFO:   6  60 02/01 22:30 GMT:  0.66 kWh at  14.89p. <==> 02/01 18:00 GMT: 25.63p/kWh 16.84p  SOC:  23.0%-> 16.0% No cheaper slots
02/01, 18:00:03     INFO: >>>   7 Slot: 02/01 23:30 UTC Factor: 0.083 Forced:    351W  End SOC: 17.4%  SPR:    108W  SCPA:   2749W  SAC:   1569W  Min Power:    108W RSC:   2641W
[... cut ...]
02/01, 19:00:03     INFO: Optimisation Summary
02/01, 19:00:03     INFO: --------------------
```

Normally any charge planning from 4pm onwards is relatively stable with only minor current (A) changes as the cheap rate is approached. 

I've noted some refactoring in the High Cost Swaps routine alongside some code changes so I imagine the issue lies around there?

Log files attached:

[error.log](https://github.com/user-attachments/files/18294672/error.log)
[pv_opt.log](https://github.com/user-attachments/files/18294673/pv_opt.log)

fboundy, 2025-01-02:
Ok - I’ll have a look.

stevebuk1, 2025-01-02:
> Ok - I’ll have a look.

I can't immediately see it but gut reaction says a "No search Window" is being declared where it shouldn't. I see a few extra calls to self.flows and `if len(search_window) > 0` is now being done on more than what was just the index values (although I can't see how that would affect the result).

stevebuk1, 2025-01-02:
For completeness theres nothing in AppDeamon.log since the 1st of Jan at 21:23 where I installed 4.0.4 for the first time. 
(The 20:48 entry is me testing a bugfix to the EV dashboard display. )

```
01/01 21:23 installed version 4.0.4

02/01, 20:48:13 INFO AppDaemon: Terminating pv_opt
02/01, 20:48:13 INFO AppDaemon: Reloading Module: /homeassistant/appdaemon/apps/pv_opt/pv_opt.py
02/01, 20:48:13 INFO AppDaemon: Loading app pv_opt using class PVOpt from module pv_opt
02/01, 20:48:13 INFO AppDaemon: Calling initialize() for pv_opt
02/01, 20:48:26 WARNING AppDaemon: Excessive time spent in utility loop: 13934.0ms, 13933.0ms in check_app_updates(), 1.0ms in other
```

fboundy, 2025-01-02:
No that won’t change the logic. Mine seems to have been behaving but I haven’t looked that closely. I thought I’d fixed a bug in the optimiser logic but maybe I created another….

fboundy, 2025-01-03:
Hi Steve - yes I think you are right in that "No Search Window" shouldn't set `done` to `True`. I think it's happening because the most expensive slot is likely to be the current one during a charge window. Doesn't happen often on Agile so I missed it. See if 4.0.5 fixes it.

commit 3d04992d (2025-01-03): Attempt to fix #353
closed 2025-01-03 by fboundy