stevebuk1/pv_opt #355: Error in AppDaemon log

swests, 2025-01-05:
**Describe the bug**
Over the last few days I’ve woken to my battery at minimum SoC (16% in my case), even though pv_opt had a charge schedule set for overnight.  Been digging around and I noticed a lot of errors being logged by AppDaemon.  I get 100’s of the `Boolean Series Key` warning and a few for `DataFrame`

```
/homeassistant/appdaemon/apps/pv_opt/pvpy.py:950: UserWarning: Boolean Series key will be reindexed to match DataFrame index.
  x = x[available.loc[: max_slot - pd.Timedelta("30min")]]
/homeassistant/appdaemon/apps/pv_opt/pv_opt.py:3421: UserWarning: DataFrame columns are not unique, some columns will be omitted.
  | {col: df[["period_start", col]].to_dict("records") for col in cols if col in df.columns}

```
[image]

Unsure if these are the root cause, but a restart of AppDaemon seems to resolve things and a changing plan in created.

HA 2024.12.5
Pv_opt 4.0.4

Logs attached

[error.log](https://github.com/user-attachments/files/18310116/error.log)
[main.log](https://github.com/user-attachments/files/18310117/main.log)
[pv_opt.log](https://github.com/user-attachments/files/18310118/pv_opt.log)

punkymuzzle, 2025-01-06:
I have just updated to ver 4.0.4 and am getting the same errors in Appdaemon:

> /homeassistant/appdaemon/apps/pv_opt/pvpy.py:950: UserWarning: Boolean Series key will be reindexed to match DataFrame index.
  x = x[available.loc[: max_slot - pd.Timedelta("30min")]]
/homeassistant/appdaemon/apps/pv_opt/pvpy.py:950: UserWarning: Boolean Series key will be reindexed to match DataFrame index.
  x = x[available.loc[: max_slot - pd.Timedelta("30min")]]
/homeassistant/appdaemon/apps/pv_opt/pvpy.py:950: UserWarning: Boolean Series key will be reindexed to match DataFrame index.
  x = x[available.loc[: max_slot - pd.Timedelta("30min")]]
/homeassistant/appdaemon/apps/pv_opt/pvpy.py:950: UserWarning: Boolean Series key will be reindexed to match DataFrame index.
  x = x[available.loc[: max_slot - pd.Timedelta("30min")]]
/homeassistant/appdaemon/apps/pv_opt/pvpy.py:950: UserWarning: Boolean Series key will be reindexed to match DataFrame index.
  x = x[available.loc[: max_slot - pd.Timedelta("30min")]]
/homeassistant/appdaemon/apps/pv_opt/pv_opt.py:3421: UserWarning: DataFrame columns are not unique, some columns will be omitted.
  | {col: df[["period_start", col]].to_dict("records") for col in cols if col in df.columns}
/homeassistant/appdaemon/apps/pv_opt/pv_opt.py:3421: UserWarning: DataFrame columns are not unique, some columns will be omitted.
  | {col: df[["period_start", col]].to_dict("records") for col in cols if col in df.columns}

stevebuk1, 2025-01-06:
I am also getting both error message types. This is not affecting charging functionality, but I am on IOG.

swests, 2025-01-07:
Hummm.  I’m also seeing this 500 error in the main log for AppDaemon:

[image]

No overnight charging again last night.

swests, 2025-01-07:
Have been going some more digging, and I **think** it's when it calls `set_select` to set select.solmod_energy_storage_control_switch to Timed Charge/Discharge, as seen in the log.

When I look at the entity in HA, I see the following options:
[image]
[image]

(it's scrolls hence two images)

pv_opt is picking up the available options as per the log file:
[image]

Looking at solis.py, I see:
[image]

Seems default code 35 is defined twice...

Hope this helps?

swests, 2025-01-07:
Commenting out the `Timed Charge/Discharge` line in solis.py stops the 500 error showing in main.log.

So there are two issues in this.
1: the stream of UserWarnings in AppDaemon.log
2: the apparent mis-setting of `select.solmod_energy_storage_control_switch`

fboundy, 2025-01-07:
Thanks- will try to have a look this evening.

stevebuk1, 2025-01-07:
Some time last year the Wills106 Solax integration was updated and the names of some of the options in the dropdown changed. The code was updated to allow for both the old wills106 integration and the new one, hence the duplicate entries in the table. 

In v3.x.x there was quite a natty piece of code that 
1) Deduced the decimal value that needed to be written
2) Read the text options from the HomeAssistant energy control storage switch entity (pre Wills106 change there was 13, post change there are 11)
3) Parsed the list of (16) options looking for a match against both the desired string and the decimal value
4) Wrote the string value associated with the required decimal value back to the energy storage switch.

The code was this:

```
            entity_modes = self.host.get_state_retry(entity_id, attribute="options")
            modes = {INVERTER_DEFS[self.type]["codes"].get(mode): mode for mode in entity_modes}
            # mode = INVERTER_DEFS[self.type]["modes"].get(code)
            mode = modes.get(code)
            self.host.set_select("inverter_mode", mode)
```

The entry that Simon deleted as a tempory workaround/fix is the correct entry for the old version of wills106, which I'm still running:

[image]

I suspect therefore reason it doesnt flag an error on my system is either because of the above, or its because the inverter energy control switch was already in 35d so it now never needs to change it (as Pv_opt no longer uses backup mode). 

@fboundy the code exerrpt that does the lookups such that Pv_opt is agnostic to the wills106 versions seems to have disappeared from 4.x.x?

fboundy, 2025-01-07:
OK that makes sense. Let me have a look back at 3.x and I'll reinstate the workaround in 4.x.

fboundy, 2025-01-07:
As I recall it changed Selfuse to Self-Use or something similar

stevebuk1, 2025-01-07:
> As I recall it changed Selfuse to Self-Use or something similar

Two changes:

            "SelfUse" became "Self-Use - No Timed Charge/Discharge"

            "Timed Charge/Discharge" became "Self-Use"

Why he/she felt this change was really needed was a bit beyond me.......

The INVERTER_DEFs in v4.0.5 is correct, its just the code that does the search of INVERTER_DEFs is missing.

swests, 2025-01-07:
Thanks both.  Will run tonight with my edit and see if I get charged up, if all good will update to this release.

If I don’t change then something else is at play…

fboundy, 2025-01-07:
Hopefully v4.0.6 will fix this

stevebuk1, 2025-01-07:
Installed but have an error:

```
06/01, 00:10:00 WARNING pv_opt: ------------------------------------------------------------
07/01, 22:17:37 WARNING pv_opt: ------------------------------------------------------------
07/01, 22:17:37 WARNING pv_opt: Unexpected error running initialize() for pv_opt
07/01, 22:17:37 WARNING pv_opt: ------------------------------------------------------------
07/01, 22:17:37 WARNING pv_opt: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 162, in initialize_app
    await utils.run_in_executor(self, init)
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
    response = future.result()
               ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 589, in initialize
    self._load_inverter()
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1270, in _load_inverter
    self.inverter = create_inverter_controller(inverter_type=self.inverter_type, host=self)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/solis.py", line 313, in create_inverter_controller
    return SolisSolaxModbusInverter(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/solis.py", line 704, in __init__
    self._modes = {self._codes[code]: code for code in entity_modes}
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/solis.py", line 704, in <dictcomp>
    self._modes = {self._codes[code]: code for code in entity_modes}
                   ~~~~~~~~~~~^^^^^^
KeyError: 'Selfuse - No Grid Charging'

07/01, 22:17:37 WARNING pv_opt: ------------------------------------------------------------
```
Looking at my capture:

[image]

I think this line (line 32 of solis.py)
`            "SelfUse - No Grid Charging": 1,`
should read:
`            "Selfuse - No Grid Charging": 1,`

and line 36 of solis.py:
`        "SelfUse": 33,`
should read
`        "Selfuse": 33,`

i.e capital U changed to small in both cases. 

Not sure why this hasnt caused an error before now - did it used to parse numbers and then strings but now does it the other way around?

fboundy, 2025-01-07:
That’s odd. I copied the code pretty much verbatim. I’ll fix tomorrow. Best thing is actually to do the test all lower case.

stevebuk1, 2025-01-07:
Sorry, I forgot to add I modified my code as per the above and its all up and running again. 

v3:
            entity_modes = self.host.get_state_retry(entity_id, attribute="options")
            modes = {INVERTER_DEFS[self.type]["codes"].get(mode): mode for mode in entity_modes}
            mode = modes.get(code)
v4:
            entity_modes = self._host.get_state_retry(entity_id, attribute="options")
            self._codes = INVERTER_DEFS[inverter_type]["codes"][self._hmi_fb00]
            self._modes = {self._codes[code]: code for code in entity_modes}

Can't see the difference either.....

stevebuk1, 2025-01-08:
The change I made also had no adverse affects - charging happened as planned overnight. 

@fboundy suggest you make the change in a patch and provide a weblink where both myself and Simon can pickup the new Solis.py and test it before you release to main? (Or I can if you prefer)

fboundy, 2025-01-08:
Steve - happy for you do do that. Just update the patch branch and I can test too.

swests, 2025-01-08:
This might be slightly related, but I noticed another 500 error in main.log today for a call to `default/number/set_value` .  Digging in it I found this in pv_opt.log

```
2025-01-07 23:50:30.671978     INFO: Entity: number.solmod_timed_charge_current Value: 102.2  Old State: 100.0 New state: 100.0 Diff:  2.2 Tol:  0.1
2025-01-07 23:50:30.673016     INFO: Failed to write 102.2 to inverter
```

Which is called via solis.py -> _set_current

The max charge rate for my inverter is 100A (as picked up during initialisation in pv_opt.py -> _list_entities ).

Wouldn't it make sense to perform a range check e.g. if Calculated_Value > Max then Max.

OR - is this caused because of a configuration issue in my setup?  I've tried to unpick bur can't seem to resolve where is picking up the needed values.

stevebuk1, 2025-01-08:
> Steve - happy for you do do that. Just update the patch branch and I can test too.

@fboundy: patch branch updated to 4.0.7-Beta-1. 

@swests can you install these two files and test for 24 hours? 

https://github.com/fboundy/pv_opt/blob/patch/apps/pv_opt/pv_opt.py
https://github.com/fboundy/pv_opt/blob/patch/apps/pv_opt/solis.py

swests, 2025-01-08:
Installed and running

stevebuk1, 2025-01-08:
> Wouldn't it make sense to perform a range check e.g. if Calculated_Value > Max then Max.
> 
It would make sense to run a range check.  number.solmod_timed_charge_current is limited to 100A as you say, but that max limit is the same whatever inverter you have. My 3.6kW inverter has a 3kW DC charger, so in reality its never going to charge more than at 62.5A. However I've just tried setting it to 100A - the Solax integration accepts the number and the inverter front screen also accepts 100A. I know Francis also has a 3.6kW inverter so I suspect in reality we've never run into this issue and its only current that probably would benefit (i.e its not going to be an issue for writing times time or %)

There is a setting in config.yaml for charger power which will limit the maximum current so you could reduce that a bit, but as its power, the current is derived from voltage. Its at low SOC where the current will be set to highest and high SOC where the current will be lowest, so reducing the power to limit the current value written at low SOC will needlessly throttle your charge rate at high SOC and might even throw off the plan tracking a bit. 

However, more recently a separate setting for battery current was added, to accommodate users where their inverter/charger capacity exceeds the max charge and discharge rates of their batteries. If you set this to 100A then it will achieve the same thing as a range check, and will tell us 100% whether the error is due to this range problem or something else. 
Its called battery_current_limit_amps, you can add it to the plant parameters in config.yaml, for an example see the config.yaml here:

https://github.com/fboundy/pv_opt/blob/main/apps/pv_opt/config/config.yaml

> OR - is this caused because of a configuration issue in my setup? I've tried to unpick bur can't seem to resolve where is picking up the needed values.

As above I don't think its anything else other than the range, but the change to config.yaml above will tell us.

swests, 2025-01-08:
I’ve set the parameter in the config.  As it’s calculating higher than 100A (102.2A as above), there is clearly something changing. Given that A=W/V, it could be the voltage moving around for the charger (rated 48V but it only needs a small variance for the maths to calculate over 100A, especially as power watts is set at 5000 less any inefficiency).

stevebuk1, 2025-01-08:
> Given that A=W/V, it could be the voltage moving around for the charger (rated 48V but it only needs a small variance for the maths to calculate over 100A, especially as power watts is set at 5000 less any inefficiency).

Indeed, this is what I meant by low SOC (voltage less than 48V) and high SOC (voltage higher than 48V), hence using battery_current_limit_amps to avoid this problem. 

> I’ve set the parameter in the config. As it’s calculating higher than 100A (102.2A as above), there is clearly something changing.

Are you saying you've added  `battery_current_limit_amps: 100` to config.yaml and its still trying to load 102.2 to the inverter? 
Please post pv_opt.log and error.log.

swests, 2025-01-08:
Sorry - poorly constructed sentence.  I’ve set the limit, waiting for it do work its magic.

swests, 2025-01-08:
That said, I have just seen this…

[image]

swests, 2025-01-08:
But I assume that it’s battery current that I would be checking (0.8W)…. Please confirm

swests, 2025-01-09:
OK, results from overnight are not good.

[image]

There are no warnings in pv_opt.log and nothing in the appdaemon error.log, so digging further, I was matching up against the HA logs and saw this:

Logger: aiohttp.server
Source: /usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py:451
First occurred: 5 January 2025 at 17:40:34 (60 occurrences)
Last logged: 06:20:34

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py", line 480, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 86, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/api/__init__.py", line 430, in post
    response = await shield(
               ^^^^^^^^^^^^^
    ...<8 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2802, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2845, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1007, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1079, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 105, in async_set_value
    raise ServiceValidationError(
    ...<8 lines>...
    )
homeassistant.exceptions.ServiceValidationError: Value 101.2 for number.solmod_timed_discharge_current is outside valid range 0 - 100

There is a need for a range check as I set the charge limit in the config for pv_opt

stevebuk1, 2025-01-09:
Ok, there was no fix for the range checkin Beta-1, but can you attach
pv_opt.log so I can check the fix that was included is working?

stevebuk1, 2025-01-09:
> But I assume that it’s battery current that I would be checking (0.8A)…. Please confirm

The new limit in config.yaml will act on "Optimum Charge Current" on your dashboard. When a slot starts, then yes, battery current should match optimum charge current. 

I've had a longer look at how `battery_current_limit_amps` actually works and unfortunately its initially converted back to a power using the battery voltage so then a power for charging can be elicited from whichever is the smaller of battery power and charger power. This means it won't fix the problem. I'll have a look at a better way of fixing it.

stevebuk1, 2025-01-09:
> I've had a longer look at how `battery_current_limit_amps` actually works and unfortunately its initially converted back to a power using the battery voltage so then a power for charging can be [...]

Further inspection shows that whilst the above function was also present in v3.x.x,  there was an additonal line, just before the current was written:

`current = min(current, self.host.get_config("battery_current_limit_amps"))`

which is missing from v4.x.x.

@fboundy please let me know if this omission was deliberate but in my view has a setting of maximum current in A then that should be respected and this line needs to be re-established.

stevebuk1, 2025-01-09:
@swests:
v4.0.7 Beta-2 with a fix for battery_current_limit_amps so 100A should mean 100A. 

https://github.com/fboundy/pv_opt/blob/patch/apps/pv_opt/pv_opt.py
https://github.com/fboundy/pv_opt/blob/patch/apps/pv_opt/solis.py

swests, 2025-01-09:
@stevebuk1 - thanks.  Installed and running.  Will update tomorrow

swests, 2025-01-10:
Good morning folks.  Looks like the range fix has worked!  See attached log. A charge window starts at 00:00, charge current is calculated as 102.2A, but 100A is written to the one inverter.  I note the the logging is before the range fix.

[image]

[pv_opt.log](https://github.com/user-attachments/files/18372241/pv_opt.log)

Thank you.

stevebuk1, 2025-01-10:
Hi Simon,

Looks successful. Re the logging, the excerpt you included isn't the bit that writes the current, its a few seconds later here:

The third line shows the incoming value of 102.2 and the 13th line shows the current after the limit is invoked of 100. 

[image]

battery_current_limit_amps is defaulted to 100A if there is nothing in config.yaml, which explains why you havent seen the issue before with 3.x.x. However the BatteryModel stuff was added this year so I remain curious about why it wasnt an issue before then. It matters not. I need to check my logs also as I ran it overnight.

@fboundy @CraigCallender Prior to a PR I think I'll add some additional logging to indicate that current limiting has happenned, as I note that the use of battery_current_limit_amps in the power flows calculations is only part of low cost charging and not high cost swaps nor discharging. I would have expected all 3? This could cause issues for users where their charger power is significantly bigger than their max battery charge rate as the planning doesn't take it into account.

stevebuk1, 2025-01-10:
> Hi Simon,
> 
> Looks successful. Re the logging, the excerpt you included isn't the bit that writes the current, its a few seconds later here:
> 
> The third line shows the incoming value of 102.2 and the 13th line shows the current after the limit is invoked of 100.
> 
> ![image](https://private-user-images.githubusercontent.com/78320816/401872798-a1194256-6df8-413c-858a-e31572dcca4e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY0OTU3MjcsIm5iZiI6MTczNjQ5NTQyNywicGF0aCI6Ii83ODMyMDgxNi80MDE4NzI3OTgtYTExOTQyNTYtNmRmOC00MTNjLTg1OGEtZTMxNTcyZGNjYTRlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTEwVDA3NTAyN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ1NWQ4Mzg5NThlZTk5YzRkOTE0OWYzZDI1YjRkNWYzNDg4NWVlNjE5MTgyNjE4YzNjNjk0Mzg4NGU5Y2ZlMTYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.8Z4JLFgkaf6TDuXW_yRpyE3DKqaptkQk80WFzxu3wXk)
> 
> battery_current_limit_amps is defaulted to 100A if there is nothing in config.yaml, which explains why you havent seen the issue before with 3.x.x. However the BatteryModel stuff was added early 2024 so I remain curious about why it wasnt an issue for you before then. It matters not. I need to check my logs also as I ran it overnight.
> 
> @fboundy @CraigCallender Prior to a PR I think I'll add some additional logging to indicate that current limiting has happenned, as I note that the use of battery_current_limit_amps in the power flows calculations is only part of low cost charging and not high cost swaps nor discharging. I would have expected all 3? This could cause issues for users where their charger power is significantly bigger than their max battery charge rate as the planning doesn't take it into account.

CraigCallender, 2025-01-10:
> @fboundy @CraigCallender Prior to a PR I think I'll add some additional logging to indicate that current limiting has happenned, as I note that the use of battery_current_limit_amps in the power [...]

@stevebuk1 - I think this was one of my first contributions to this project so it's highly likely I didn't account for it in all three places.  The issue you're predicting is probably something I've been working around by lowering my charger power from 3,000W to 1800 and setting my battery_current_limit_amps to 37.5.  Technically I _should_ be able to set the first to the actual value of 3,000W.  So sorry for not bringing this up earlier!

stevebuk1, 2025-01-10:
> @stevebuk1 - I think this was one of my first contributions to this project so it's highly likely I didn't account for it in all three places. The issue you're predicting is probably something I've [...]

Hi Craig, thanks for the quick response, no apologies needed - its good to confirmation that it is indeed an issue that needs looking at. What you've explained you need to do as a workaround is exactly what I'd expect. Inserting the battery limits into High Cost Swaps is quite a complex fix (@fboundy looks like it needs to go into the new slot "Available" logic alongside SCPA) so one for another time. I'll raise it as a separate issue so we have a record but I suspect Francis will decide its one to solve for the Pv_opt integration hes working on (i.e Pv_opt that doesnt use AppDaemon).

stevebuk1, 2025-01-10:
> @stevebuk1 looks like you're on top of this so will leave you to finish off

the patch branch reinstantiates the current limit and sorts out the energy control switch names. I'll add the additional logging mentioned above, give that a test and raise a PR.

Regarding the two entries in the logging that this issue was raised against:

```
/homeassistant/appdaemon/apps/pv_opt/pvpy.py:950: UserWarning: Boolean Series key will be reindexed to match DataFrame index.
  x = x[available.loc[: max_slot - pd.Timedelta("30min")]]
```
and
```
/homeassistant/appdaemon/apps/pv_opt/pv_opt.py:3421: UserWarning: DataFrame columns are not unique, some columns will be omitted.
  | {col: df[["period_start", col]].to_dict("records") for col in cols if col in df.columns}
```
These are beyond my abilities to fix - but I don't think either of these are actually causing any problems.

stevebuk1, 2025-01-10:
> the patch branch reinstantiates the current limit and sorts out the energy control switch names. I'll add the additional logging mentioned above, give that a test and raise a PR.

The additional logging has been added to the `patch` branch and tested using an artificially low value of battery_current_limit_amps which is working. 
The logging reminds the user that the issue might be a cause of incorrect config.yaml setting. Theres a missing "A" which I've already fixed in `patch`. 

```
10/01, 22:30:16     INFO: battery_current_limit_amps of 10 is less than current of 14.1A required by charging plan.
10/01, 22:30:16     INFO: Reducing inverter charge current to 10A. 
10/01, 22:30:16     INFO: Check value of charger_power_watts in config.yaml if this is unexpected.
```
As said before I'll raise a seperate issue to add to the backlog for use of battery_current_limit_amps in charge planning.

fboundy, 2025-01-11:
Raised 3 new issues to cover the outstanding bits. So closing this one. 4.0.7 about to be release including patch code.

#364 
#365
#366

closed 2025-01-11 by fboundy