stevebuk1/pv_opt #356: First 1/2 hour of plan indicates phantom grid use

stevebuk1, 2025-01-06:
**Describe the bug**
in v4.0.5, the 1/2 hour optimisation summary shows grid use in the first 1/2 hour slot on every optimiser run (grid = 302 as shown below)

```
06/01, 21:10:09     INFO: 1/2 Hour Optimsation summary
06/01, 21:10:09     INFO: ----------------------------
06/01, 21:10:09     INFO: 
06/01, 21:10:09     INFO: 
                            import  export  consumption  Solcast  Solcast_p10  Solcast_p90    solar  consumption      chg  chg_end      battery    grid  forced        soc    soc_end  carslot  period
2025-01-06 21:00:00+00:00  25.6269       0   453.591383      0.0          0.0          0.0    0.000   453.591383   6042.0   5963.7   151.902000   302.0       0  53.000000  52.313158        0       0
2025-01-06 21:30:00+00:00  25.6269       0   443.371156      0.0          0.0          0.0    0.000   443.371156   5963.7   5735.2   443.290000     0.0       0  52.313158  50.308772        0       0
2025-01-06 22:00:00+00:00  25.6269       0   450.375953      0.0          0.0          0.0    0.000   450.375953   5735.2   5503.0   450.468000    -0.0       0  50.308772  48.271930        0       0
2025-01-06 22:30:00+00:00  25.6269       0   478.060626      0.0          0.0          0.0    0.000   478.060626   5503.0   5256.6   478.016000     0.0       0  48.271930  46.110526        0       0
2025-01-06 23:00:00+00:00  25.6269       0   411.076603      0.0          0.0          0.0    0.000   411.076603   5256.6   5044.7   411.086000    -0.0       0  46.110526  44.251754        0       0
2025-01-06 23:30:00+00:00   7.0003       0   422.438652      0.0          0.0          0.0    0.000   422.438652   5044.7   5569.3 -1152.967033  1575.0    1153  44.251754  48.853509        0       1
2025-01-07 00:00:00+00:00   7.0003       0   451.606537      0.0          0.0          0.0    0.000   451.606537   5569.3   6093.9 -1152.967033  1605.0    1153  48.853509  53.455263        0       1
2025-01-07 00:30:00+00:00   7.0003       0   770.784801      0.0          0.0          0.0    0.000   770.784801   6093.9   6618.5 -1152.967033  1924.0    1153  53.455263  58.057018        0       1
2025-01-07 01:00:00+00:00   7.0003       0   496.489884      0.0          0.0          0.0    0.000   496.489884   6618.5   7143.1 -1152.967033  1649.0    1153  58.057018  62.658772        0       1
2025-01-07 01:30:00+00:00   7.0003       0   504.163117      0.0          0.0          0.0    0.000   504.163117   7143.1   7667.7 -1152.967033  1657.0    1153  62.658772  67.260526        0       1
2025-01-07 02:00:00+00:00   7.0003       0   502.447571      0.0          0.0          0.0    0.000   502.447571   7667.7   8192.3 -1152.967033  1655.0    1153  67.260526  71.862281        0       1
2025-01-07 02:30:00+00:00   7.0003       0   663.150613      0.0          0.0          0.0    0.000   663.150613   8192.3   8716.9 -1152.967033  1816.0    1153  71.862281  76.464035        0       1
2025-01-07 03:00:00+00:00   7.0003       0  1029.966957      0.0          0.0          0.0    0.000  1029.966957   8716.9   9241.5 -1152.967033  2183.0    1153  76.464035  81.065789        1       1
2025-01-07 03:30:00+00:00   7.0003       0   427.050448      0.0          0.0          0.0    0.000   427.050448   9241.5   9766.1 -1152.967033  1580.0    1153  81.065789  85.667544        1       1
2025-01-07 04:00:00+00:00   7.0003       0   410.321547      0.0          0.0          0.0    0.000   410.321547   9766.1  10290.7 -1152.967033  1563.0    1153  85.667544  90.269298        1       1
2025-01-07 04:30:00+00:00   7.0003       0   438.243747      0.0          0.0          0.0    0.000   438.243747  10290.7  10815.3 -1152.967033  1591.0    1153  90.269298  94.871053        1       1
2025-01-07 05:00:00+00:00   7.0003       0  1000.891951      0.0          0.0          0.0    0.000  1000.891951  10815.3  11339.9 -1152.967033  2154.0    1153  94.871053  99.472807        1       1
2025-01-07 05:30:00+00:00  25.6269       0   268.167355      0.0          0.0          0.0    0.000   268.167355  11339.9  11201.7   268.108000     0.0       0  99.472807  98.260526        0       1
```

**To Reproduce**
I added logging to print out the base flows and found it was also present in Base, so should equally affect all tariffs and thus be immediately apparent for all users. (base log here is for same optimiser run as above, which also shows 302.0):

```
06/01, 21:10:09     INFO: 1/2 Hour Base
06/01, 21:10:09     INFO: -------------
06/01, 21:10:09     INFO: 
06/01, 21:10:09     INFO: 
                            import  export  consumption  Solcast  Solcast_p10  Solcast_p90    solar  consumption     chg  chg_end   battery    grid  forced        soc    soc_end
2025-01-06 21:00:00+00:00  25.6269     0.0   453.591383      0.0          0.0          0.0    0.000   453.591383  6042.0   5963.7   151.902   302.0       0  53.000000  52.313158
2025-01-06 21:30:00+00:00  25.6269     0.0   443.371156      0.0          0.0          0.0    0.000   443.371156  5963.7   5735.2   443.290     0.0       0  52.313158  50.308772
2025-01-06 22:00:00+00:00  25.6269     0.0   450.375953      0.0          0.0          0.0    0.000   450.375953  5735.2   5503.0   450.468    -0.0       0  50.308772  48.271930
2025-01-06 22:30:00+00:00  25.6269     0.0   478.060626      0.0          0.0          0.0    0.000   478.060626  5503.0   5256.6   478.016     0.0       0  48.271930  46.110526
2025-01-06 23:00:00+00:00  25.6269     0.0   411.076603      0.0          0.0          0.0    0.000   411.076603  5256.6   5044.7   411.086    -0.0       0  46.110526  44.251754
2025-01-06 23:30:00+00:00   7.0003     0.0   422.438652      0.0          0.0          0.0    0.000   422.438652  5044.7   4826.9   422.532    -0.0       0  44.251754  42.341228
2025-01-07 00:00:00+00:00   7.0003     0.0   451.606537      0.0          0.0          0.0    0.000   451.606537  4826.9   4594.1   451.632    -0.0       0  42.341228  40.299123
2025-01-07 00:30:00+00:00   7.0003     0.0   770.784801      0.0          0.0          0.0    0.000   770.784801  4594.1   4196.8   770.762     0.0       0  40.299123  36.814035
2025-01-07 01:00:00+00:00   7.0003     0.0   496.489884      0.0          0.0          0.0    0.000   496.489884  4196.8   3940.9   496.446     0.0       0  36.814035  34.569298
2025-01-07 01:30:00+00:00   7.0003     0.0   504.163117      0.0          0.0          0.0    0.000   504.163117  3940.9   3681.0   504.206    -0.0       0  34.569298  32.289474
2025-01-07 02:00:00+00:00   7.0003     0.0   502.447571      0.0          0.0          0.0    0.000   502.447571  3681.0   3422.0   502.460    -0.0       0  32.289474  30.017544
2025-01-07 02:30:00+00:00   7.0003     0.0   663.150613      0.0          0.0          0.0    0.000   663.150613  3422.0   3080.2   663.092     0.0       0  30.017544  27.019298
2025-01-07 03:00:00+00:00   7.0003     0.0  1029.966957      0.0          0.0          0.0    0.000  1029.966957  3080.2   2549.3  1029.946     0.0       0  27.019298  22.362281
2025-01-07 03:30:00+00:00   7.0003     0.0   427.050448      0.0          0.0          0.0    0.000   427.050448  2549.3   2329.2   426.994     0.0       0  22.362281  20.431579
2025-01-07 04:00:00+00:00   7.0003     0.0   410.321547      0.0          0.0          0.0    0.000   410.321547  2329.2   2117.7   410.310     0.0       0  20.431579  18.576316
2025-01-07 04:30:00+00:00   7.0003     0.0   438.243747      0.0          0.0          0.0    0.000   438.243747  2117.7   1891.8   438.246    -0.0       0  18.576316  16.594737
2025-01-07 05:00:00+00:00   7.0003     0.0  1000.891951      0.0          0.0          0.0    0.000  1000.891951  1891.8   1824.0   131.532   869.0       0  16.594737  16.000000
2025-01-07 05:30:00+00:00  25.6269     0.0   268.167355      0.0          0.0          0.0    0.000   268.167355  1824.0   1824.0     0.000   268.0       0  16.000000  16.000000
2025-01-07 06:00:00+00:00  25.6269     0.0   526.695465      0.0          0.0          0.0    0.000   526.695465  1824.0   1824.0     0.000   527.0       0  16.000000  16.000000
2025-01-07 06:30:00+00:00  25.6269     0.0   436.932775      0.0          0.0          0.0    0.000   436.932775  1824.0   1824.0     0.000   437.0       0  16.000000  16.000000
```

**Expected behavior**
If not  actively force charging or force discharging, all house use should be from battery. 

As its a single half hour slot I don't think its adversely impacting any charge plans, but its clearly incorrect and might have consequences I've yet to see. 

Logs:
[pv_opt.log](https://github.com/user-attachments/files/18324624/pv_opt.log)

fboundy, 2025-01-06:
OK this explains something that I'd seen which was a "sawtooth" in the 1 hour delayed SOC estimate and which was on my list to look at. It will be the scaling of the 1st slot between "soc_now" and "soc_initial" which I had a feeling had been corrupted in the re-factor.

I'll have a look now and see if it's obvious

stevebuk1, 2025-01-06:
Mmm I noticed self.soc_now and self.soc_initial were showing 'white' in MS Visual Studio code when editing pypy.py which implied there were not being correctly set, but as before classes are a bit lost on me. I can see you are attempting not to pass parameters when calling calculate_flows from pv_opt.py but not convinced its quite right.

stevebuk1, 2025-01-06:
To log base I inserted the following at line 2507 of pv_opt.py in main:

```
        self.base = self.flows["Base"]
        
        self.log("")
        self.ulog("1/2 Hour Base")
        self.log(f"\n{self.base.to_string()}")
        self.log("")
```

fboundy, 2025-01-06:
That is the source of the problem but I don't think I've actually changed any of this recently...

Going to have to have a bit more of a think about the whole "first slot" setup. The "right" way to do it it is to set the first index timestamp in the DataFrame to be the time that the optimisation runs and then to have a "slot duration" column. I will need to make sure that this is used throughout rather than a constant "freq".

fboundy, 2025-01-06:
This is all good stuff though because it will end up the the "integration" version which is slowly progressing

stevebuk1, 2025-01-07:
> Going to have to have a bit more of a think about the whole "first slot" setup. The "right" way to do it it is to set the first index timestamp in the DataFrame to be the time that the optimisation [...]

We've discussed this before - self.flows (now self.calculate_flows) is not time aware. At one point you were factoring the current slot if you were partway through, then I think you changed tack and just kept a consistent starting SOC for each of the 3 optimiser runs that occur in a half hour slot (start of slot, 10 mins in, 20 mins in). When I was having a bash at improving the former before I'd noticed the latter, I did start wondering if the whole thing would be better calculated in terms of energy (Wh) instead of power - this would at least better match with the preferred way of loading consumption and makes the introduction of time more simple, I think?

fboundy, 2025-01-07:
For everything other than the 1st slot energy vs power is a moot point because the period is fixed. I didn't get a chance to look at this tonight (work got in the way) but I will do over the next couple of days. It's a niggle but I'd like to get it fixed.

stevebuk1, 2025-01-08:
> For everything other than the 1st slot energy vs power is a moot point because the period is fixed.

True, but for anything other than the first slot, running the optimiser at a rate any faster than once every half hour is also moot. In essence the whole thing should probably be designed around what the first slot needs for this reason?

stevebuk1, 2025-01-08:
> I didn't get a chance to look at this tonight (work got in the way) but I will do over the next couple of days. It's a niggle but I'd like to get it fixed.

Understood - my work is quite quiet at the moment! To be honest its not really causing any effect for me as my charge rate is spread out over a 6 hour period and I run with a fairly high power margin % at this time of year. I've noticed no other adverse effects, the risk of which was the main reason for raising the issue.

fboundy, 2025-01-10:
Should be fixed in the `patch` now. Grateful if @stevebuk1  could test

stevebuk1, 2025-01-10:
> Should be fixed in the `patch` now. Grateful if @stevebuk1 could test. Did a bit of a tidy up of the calculate_flows method which was well overdue.

Ok will do.

stevebuk1, 2025-01-10:
Installed and a couple of optimiser runs show no zero errant grid use. 

I can see you've made a few changes to how the flows work in the current slot, looks like you taking a live SOC, moving the slot time forward and are factoring the battery flow by the slot time remaining? Looks good.  

I'll be interested to see when charging on IOG that the inverter charge current stays stable on each optimiser run. As my cars empty and Octopus have given me an 11pm slot I will probably  will be up for the 2nd optimiser run during a charge slot so will probably let you know later.

stevebuk1, 2025-01-10:
Re green line for Initial, is the starting value of zero at midnight intentional? It appears the same as the sawtooth from 0 you'd metioned, but now is at the graph X-axis origin. 

[image]

stevebuk1, 2025-01-10:
Made a few more pushes to `Patch` to clear 4 warnings in AppDeamon logs - looks mainly like futureproofing. 

```
/homeassistant/appdaemon/apps/pv_opt/pvpy.py:727: FutureWarning: ChainedAssignmentError: behaviour will change in pandas 3.0!
You are setting values through chained assignment. Currently this works in certain cases, but when using Copy-on-Write (which will become the default behaviour in pandas 3.0) this will never work to update the original DataFrame or Series, because the intermediate object on which we are setting values will behave as a copy.
A typical example is when you are setting values in a column of a DataFrame, like:
df["col"][row_indexer] = value
Use `df.loc[row_indexer, "col"] = values` instead, to perform the assignment in a single step and ensure this keeps updating the original `df`.
See the caveats in the documentation: https://pandas.pydata.org/pandas-docs/stable/user_guide/indexing.html#returning-a-view-versus-a-copy
  self.flows["forced"][chg_mask] = timed_slot_flows[chg_mask]

/homeassistant/appdaemon/apps/pv_opt/pvpy.py:727: SettingWithCopyWarning: 
A value is trying to be set on a copy of a slice from a DataFrame
See the caveats in the documentation: https://pandas.pydata.org/pandas-docs/stable/user_guide/indexing.html#returning-a-view-versus-a-copy
  self.flows["forced"][chg_mask] = timed_slot_flows[chg_mask]

You are setting values through chained assignment. Currently this works in certain cases, but when using Copy-on-Write (which will become the default behaviour in pandas 3.0) this will never work to update the original DataFrame or Series, because the intermediate object on which we are setting values will behave as a copy.
A typical example is when you are setting values in a column of a DataFrame, like:
df["col"][row_indexer] = value
Use `df.loc[row_indexer, "col"] = values` instead, to perform the assignment in a single step and ensure this keeps updating the original `df`.
See the caveats in the documentation: https://pandas.pydata.org/pandas-docs/stable/user_guide/indexing.html#returning-a-view-versus-a-copy
  self.flows["battery_temp"][chg_mask] = -timed_slot_flows[chg_mask]

/homeassistant/appdaemon/apps/pv_opt/pvpy.py:726: SettingWithCopyWarning: 
A value is trying to be set on a copy of a slice from a DataFrame
See the caveats in the documentation: https://pandas.pydata.org/pandas-docs/stable/user_guide/indexing.html#returning-a-view-versus-a-copy
  self.flows["battery_temp"][chg_mask] = -timed_slot_flows[chg_mask]
```

stevebuk1, 2025-01-10:
Inverter current and value of "forced" is stable, so closing issue. The graphing anomaly remains but is irrelevant to charge planning. Thanks very much.

closed 2025-01-10 by stevebuk1