stevebuk1/pv_opt #361: Unbelievable cost calcs !

lazy-pete, 2025-01-10:
Just looking at my PV-Opt today and I seem to have a baseline calculation of £1259.16 , which despite todays ridiculously high agile rates seems a bit implausible. 

I’m on V 3.16.1 as it has been pretty stable - been watching as the 4.x bugs are ironed out before updating. 

Can’t add log files right now - will do when I get home. 

Will also try and update as well, but hate the thought of breaking an otherwise functional setup - though it’s usually the Solax updates that break it tbf. 

[image]
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

fboundy, 2025-01-10:
The revelant code has been completely re-written in 4.0.x so I'm afraid I'm not going to revisit old code. If it's still an issue in 4.x then please re-open with logs

closed 2025-01-10 by fboundy

lazy-pete, 2025-01-10:
No problem - thanks.

P

Closed #361<https://github.com/fboundy/pv_opt/issues/361> as completed.