stevebuk1/pv_opt #362: Unbelievable Baseline Cost Calcs 4.0.6

lazy-pete, 2025-01-10:
**Describe the bug**
A clear and concise description of what the bug is.

Hi - I’ve got some truly unbelievable cost calcs today, even taking into account the wild Octopus Agile rates on a still Scottish winters day. 

Todays base £14331

I am also getting weird data in the HA native Energy dashboard, but all the linked sensors appear correct. 

I’ve updated PV-Opt and Octopus etc but no change. 

I did update the Solax Modbus integration but as usual that introduced a new Device and a bunch of unavailable sensors etc messing it all up, so I rolled back. 

[image]
[image]

[main.log](https://github.com/user-attachments/files/18382259/main.log)
[pv_opt.log](https://github.com/user-attachments/files/18382260/pv_opt.log)
[error.log](https://github.com/user-attachments/files/18382261/error.log)

Any bright ideas as to where things have gone wrong ? And any suggestions on how to update the Solis integrations without just adding another dysfunctional Device and confusing everything ?

cheers,

Pete

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

fboundy, 2025-01-11:
This appears to be a problem with either the the Lovelace card or the attribute that is being saved to Recorder rather than the actual calculations. The log is showing this:
    
    17:20:04     INFO: Optimisation Summary
    17:20:04     INFO: --------------------
    17:20:04     INFO: 
    17:20:04     INFO:   Base cost:                               1076.0p
    17:20:04     INFO:   Optimised cost (Optimised Charging):      964.6p
    17:20:04     INFO:   Optimised cost (Optimised PV Export):     964.6p
    17:20:04     INFO:   Optimised cost (Forced Discharge):        964.6p <=== Current Setup

Are you able to see what is saved to the attributes in Developer Tools\States are shown in the attached:

[image]

lazy-pete, 2025-01-11:
Is this what you were looking for ?

An issue with the cards I. General might make sense - the default energy
card is also showing me rubbish, yet the sensors it takes data from seem
correct.

I’ll update the Lovelace cards etc as well and see what happens.

stevebuk1, 2025-01-11:
> This appears to be a problem with either the the Lovelace card or the attribute that is being saved to Recorder rather than the actual calculations. The log is showing this:
>
> ```
> 17:20:04     INFO: [...]

There is an entry in Pv_opt.log showing the very high cost, 7th line below.....

```
17:10:05     INFO: Plan time: 10-Jan 17:00 - 11-Jan 23:30 Initial SOC: 18.5 Base Cost: 1073.9 Opt Cost: 964.1
17:10:05     INFO: 
17:10:05     INFO: Optimiser elapsed time 5.0 seconds
17:10:05     INFO: 
17:10:05     INFO: 
17:10:05     INFO: Output written to sensor.pvopt_optimiser_elapsed
17:10:05     INFO: Average unit cost today: 76613.10p/kWh
17:10:05     INFO: Output written to sensor.pvopt_unit_cost_today
17:10:05     INFO: Output written to sensor.pvopt_base_cost
17:10:05     INFO: Output written to sensor.pvopt_opt_cost
17:10:05     INFO: Output written to sensor.pvopt_charge_start
```

lazy-pete, 2025-01-11:
I really hope that wasn’t true !

Any idea where that comes from ?

I have also been seeing ludicrous values in the default HA Energy
dashboard.

fboundy, 2025-01-11:
If it’s across multiple integrations it sounds like a spike in the cumulative import data which they all rely on

lazy-pete, 2025-01-13:
Where might I find that data spike ? Not sure where to look to find the underlying problem. 

Some screenshots of the other place I am finding this weird - 

[image]
[image]
[image]
[image]

stevebuk1, 2025-02-01:
@lazy-pete did you manage to get this sorted out on your system?

closed 2025-04-10 by github-actions[bot]