stevebuk1/pv_opt #363: Solarman_V2 ERROR: Unable to write value to time registers

pyriel800, 2025-01-11:
Hi,
I've just started (yesterday) and managed to get the config working however I get this error for both charge and discharge time configuration - example is from this morning.

[image]

I'm using the current version of the solarmanv2 integration v24.12.22. All other things (octopus integration, HA etc.) are at current versions.

v.4.0.6 of pv_opt

Happy to supply anything else that might be useful

stevebuk1, 2025-01-11:
Hi, Solarman was working in the previous Pv_oot  release but after some major refactoring it's currently non functional, in the manner you describe. Issue #321 is tracking this.

pyriel800, 2025-01-11:
Ah okay. Thanks for that. I will continue to monitor that thread. And possibly use the solis cloud integration in the mean time

fboundy, 2025-01-11:
I'm not sure this is the same issues (though it may be related) so I'm re-opening this one.

I think the issue may be that we are trying to write a datetime to a time entity. I'll put a potential fix into. 4.0.7 which I'm about to release.

stevebuk1, 2025-01-11:
If I remember rightly I had a right faff with getting the time in the right
format to be accepted. It was working in dev but admit the code was
horrific.

fboundy, 2025-01-11:
The `write_and_poll_time` code looks OK and should handle it.  I think all I can really do now is add a bit more logging. 

@stevebuk1 - are you OK for the current `patch` to be pushed to production as 4.0.7? 

@pyriel800 - if you can try 4.0.7 and update with the logs hopefully we can track this down

stevebuk1, 2025-01-11:
Happy for 4.0.7 to be released. Inverter charging seemed fine last night.

Pyinthesky99, 2025-01-11:
And then there were two!!
@pyriel800 welcome to the Solarman club

pyriel800, 2025-01-11:
Thanks @Pyinthesky99 !

@fboundy Have updated and run. Log is attached. Still seems to have issue with the time though.

My logging options are WSOI - happy to change etc. Also apologies for my config errors

[pv_opt.log](https://github.com/user-attachments/files/18386455/pv_opt.log)

stevebuk1, 2025-01-11:
@fboundy this issue replicates part of the issues in #321, hence its title "Solarman integration non-functional in v4.x.x". 

The post with the logs showing the same failure to write to the time registers is here:
https://github.com/fboundy/pv_opt/issues/321#issuecomment-2570968668
log excerpt from that post in 321 shows:

```
04:11:22     INFO: Currently in charge/discharge/hold slot
04:11:22     INFO: Current charge/discharge window ends in 78.6 minutes.
04:11:22     INFO: Charge status is not active, setting start time value now
04:11:22     INFO: Setting start time to 04-Jan 04:11 GMT
04:11:22     INFO: Setting end time to 04-Jan 05:30 GMT
04:11:22     INFO: Setting power to 4000
04:11:22     INFO: Setting SOC to 62
04:11:22     INFO: >>> direction: charge
04:11:22     INFO: >>> times: {'start': Timestamp('2025-01-04 04:11:00+0000', tz='GB'), 'end': Timestamp('2025-01-04 05:30:00+0000', tz='GB')}
04:11:22     INFO: >>> current: 81.6
04:11:22     INFO: >>> 2025-01-04 04:11:00+00:00
04:11:22    ERROR: Unable to write value 2025-01-04 04:11:00+00:00 to entity time.solis_timed_charge_start
04:11:22     INFO: >>> 2025-01-04 05:30:00+00:00
04:11:22    ERROR: Unable to write value 2025-01-04 05:30:00+00:00 to entity time.solis_timed_charge_end
```
which matches the screengrab provided in the first post. 

@Pyinthesky99 @pyriel800 we will track the failure to write to the time registers here, @Pyinthesky99 we will keep #321 open solely for charge planning issues/queries.

stevebuk1, 2025-01-11:
Hi both, I've created a build v4.0.8-Beta-1 that no longer writes current to the inverter, and includes extra logging to help me debug whats going on. 

https://github.com/fboundy/pv_opt/blob/patch---solarman/apps/pv_opt/pv_opt.py
https://github.com/fboundy/pv_opt/blob/patch---solarman/apps/pv_opt/pvpy.py
https://github.com/fboundy/pv_opt/blob/patch---solarman/apps/pv_opt/solis.py

@Pyinthesky99 as mentioned in #321 this should enable you to use slot2 and 3 temporarily for any overnight charging, knowing that Pv_opt will no longer adjust the current. 

@pyriel800:

Solarman was in development over the last six months with @Pyinthesky99 and  we got a working build using a development release, but now its been merged with other fixes and enhancements its not working. I use Solax modbus for my Solis inverter and don't have a Solarman stick, so unfortunately can't test Solarman fixes myself, this makes the development debug process slow. 

Pv_opt is written in Python which is effectively a script rather than complied code - this means you can just overcopy source code files for special builds rather than needing to use HACS. You need to locate the directory where the pv_opt files are, on my RPi its /homeassistant/appdeamon/apps/pv_opt, where you'll find amoungs others these 3 files:

pv_opt.py
pvpy.py
solis.py

For test builds I'll ask you to overwrite one or more of these files with weblinks posted here, as I have done above. Appdaemon will automatically know something has changed and will reboot Pv_opt automatically, so normally its nothing more than just doing the file copy and then checking the dashboard.

fboundy, 2025-01-11:
hmm I thought I’d turned one”verbose” for that method so was expecting more detailed logging

The set time action/service wants a string of format “%H:%M” as input and the code should be generating one. Should just need a bit of logging to diagnose

stevebuk1, 2025-01-11:
Might be as simple as that there are two instances of `write_and_poll_time` in the pv_opt.py production code, one at 4338 and another one at line 4270. Not sure which one its picking up. The solarman patch I produced a week or two ago has the old one deleted - just need someone to test it.....

pyriel800, 2025-01-12:
@fboundy thanks for the files. I have now (pesky life getting in the way!) put them in and the log entries:

15:42:47 INFO: Waiting for inverter Read cycle: 72 seconds
15:42:47 ERROR: Unable to write value 2025-01-12 19:00:00+00:00 to entity time.solis_timed_discharge_end
15:42:47 ERROR: Unable to write value 2025-01-12 19:00:00+00:00 to entity time.solis_timed_discharge_start
15:42:47 INFO: Next charge window starts in 497.2 minutes but charge start and end times are different. but discharge start and end times are different. but inverter has a discharge slot before then. Disabling discharge.
15:42:47 ERROR: Unable to write value 2025-01-12 19:00:00+00:00 to entity time.solis_timed_discharge_end
15:42:47 ERROR: Unable to write value 2025-01-12 19:00:00+00:00 to entity time.solis_timed_discharge_start
15:42:47 INFO: Next charge window starts in 497.2 minutes but charge start and end times are different. but discharge start and end times are different.
15:42:47 ERROR: Unable to write value 2025-01-12 19:00:00+00:00 to entity time.solis_timed_charge_end
15:42:47 ERROR: Unable to write value 2025-01-12 19:00:00+00:00 to entity time.solis_timed_charge_start

are still there.

Happy to attach the full log if you want

stevebuk1, 2025-01-12:
Can you attach full logs please. There should be a pv_opt.log and error.log.

If you find multiple files for pv_opt.log called pv_opt.log.1, ov_opt.log.2 etc please attach those as well, but rename them by adding .log at the end otherwise GitHub won't allow them to be attached.

pyriel800, 2025-01-12:
@stevebuk1 no prob. Both logs attached.

[error.log](https://github.com/user-attachments/files/18390119/error.log)
[pv_opt-2.log](https://github.com/user-attachments/files/18390120/pv_opt-2.log)

stevebuk1, 2025-01-12:
Thanks for the logs. I think I've found the error. Please try new files here:

https://github.com/fboundy/pv_opt/blob/patch---solarman/apps/pv_opt/pv_opt.py
https://github.com/fboundy/pv_opt/blob/patch---solarman/apps/pv_opt/pvpy.py
https://github.com/fboundy/pv_opt/blob/patch---solarman/apps/pv_opt/solis.py

stevebuk1, 2025-01-12:
I mentioned in #321 but forgot to mention here that code above has inverter writes of charge current disabled. This is to allow start/end timer pairs 2 and 3 to be manually set to catch cheap overnight slots without Pv_opt changing the current, which is common to all 3. If the time settings now work then I'll renable and remove the logging.

pyriel800, 2025-01-13:
That seems to have done it - the times have been updated (my default was 7pm-4pm charge and 4-7 discharge):
[image]

Logs attached:
[pv_opt.log](https://github.com/user-attachments/files/18394587/pv_opt.log)
[error.log](https://github.com/user-attachments/files/18394589/error.log)

stevebuk1, 2025-01-13:
Many thanks, yes that looks all correct now. 

Tonight I'll renable current and tidyup the logging and then ask you to run for 24 hours prior to production release.

stevebuk1, 2025-01-13:
@fboundy if I do a pre-production release from a `patch` or `feature` branch, does it then appear in HACS? Was just wondering my method of weblinks to code for trials could instead utilise HACS. (I've worked out `main` is protected which is fair enough).

stevebuk1, 2025-01-13:
@pyriel800 @Pyinthesky99 that should be Solarman fully functional again.

Please can you both run for 24 hours as the final test, before the production version is released. 

Please overwrite your files with the following:
https://github.com/fboundy/pv_opt/blob/patch/apps/pv_opt/pv_opt.py
https://github.com/fboundy/pv_opt/blob/patch/apps/pv_opt/pvpy.py
https://github.com/fboundy/pv_opt/blob/patch/apps/pv_opt/solis.py

pyriel800, 2025-01-13:
@stevebuk1 All inserted now. Logs look okay. Fully handed over to pv_opt tonight - no other automations running. Fingers crossed - but should be okay. 

Thank you

stevebuk1, 2025-01-13:
Great, any issues post here. We'll keep this issue open until its released to production.

stevebuk1, 2025-01-13:
Theres a couple of problems noted on #321, one is Solarman related and another is a generic fix added thats turned out not to be quite complete. I'll have another Beta release tomorrow to test once I've tested the fixes for those two issues overnight.

stevebuk1, 2025-01-14:
v4.0.8-Beta-5 ready for test. 

https://github.com/fboundy/pv_opt/blob/patch/apps/pv_opt/pv_opt.py
https://github.com/fboundy/pv_opt/blob/patch/apps/pv_opt/pvpy.py
https://github.com/fboundy/pv_opt/blob/patch/apps/pv_opt/solis.py

pyriel800, 2025-01-15:
@stevebuk1 Have updated now. Thanks for the time on this!

I have a side question regarding the optimisation plan/choices/algorithm for calculating the best outcome - where's the best place to pick that up?

stevebuk1, 2025-01-15:
> I have a side question regarding the optimisation plan/choices/algorithm for calculating the best outcome - where's the best place to pick that up?

Best to raise a query on the optimiser function as a new issue.

fboundy, 2025-01-16:
> [@fboundy](https://github.com/fboundy) if I do a pre-production release from a `patch` or `feature` branch, does it then appear in HACS? Was just wondering my method of weblinks to code for trials could instead utilise HACS. (I've worked out `main` is protected which is fair enough).

@stevebuk1 - yes that should work and is a better way to go. I have set up a workflow that will automatically release pushes to `main` as a production release but pre-releases need to be done manually

stevebuk1, 2025-01-16:
> [@stevebuk1](https://github.com/stevebuk1) - yes that should work and is a better way to go. I have set up a workflow that will automatically release pushes to `main` as a production release but pre-releases need to be done manually

Thanks - it worked fine. Updated the yellow labels to state available on Beta.

fboundy, 2025-01-19:
How's the 4.08 beta looking? Let me know if and when it's ready to push to production.

pyriel800, 2025-01-19:
Hi @fboundy I ran beta 5 on Friday night (I updated via the HACS beta download route) and it went a bit sideways i.e. got a write failure on the time registers again.

Apologies for delay in coming back to you gents, but logs attached

[error.log](https://github.com/user-attachments/files/18470187/error.log)
[pv_opt.log](https://github.com/user-attachments/files/18470186/pv_opt.log)

stevebuk1, 2025-01-19:
Hi @pyriel800, looking your logs you installed Beta-5 around 08:55, where it scheduled a charge slot at 1am. Subsequent optimiser runs maintained that until around 20:50. 

At 20:56 Pv opt is restarted (no indication as to why) and runs ok until 21:10. Then it just stops. Nothing else is then put into the log until 19:16 (which must be a day later) where it records a Pv_opt restart at production release at v4.0.7. It is from that point on that time registers are showing problems, as the production release is the one thats not working. v4.0.7 is maintained until the end of the log. 

For the time Beta-5 was active there were no charge slots nearby, so no writes to the inverter were done, so theres no real evidence either way that Beta-5 maintains the time register fixes made in the previous Betas. (There is evidence within the log that v4.0.8-Beta-2 wrote to the time regsiters ok)

@pyriel800 if you can reestablish v4.0.8-Beta-5 (not sure what went wrong - try main.log (or appdaemon.log if you don't have main.log) and run through a charge slot. 

@fboundy - by the sounds of it, not quite yet. I was hoping to put a fix in for #378 before a PR from patch to main, but last nights testing didnt go well, not sure if its a consequence of the fix or more things that need aligning to a variable 1st slot start time.

fboundy, 2025-01-19:
> Just for info, my v4.0.8-Beta-5 is still running ok, with the exception of the time slipping 10 minutes during charge cycles. My install is a bit of a hybrid in that I haven't updated using HACS but just copied the three relevant py scripts each time. Hope this helps.
> error.log
> pv_opt.log
> pv_opt1.log

OK - I’ve added a small commit to fix a change in the Solax integration but it shouldn’t impact anything much.

pyriel800, 2025-01-21:
@stevebuk1 Yeah I think doing the redownload from inside HACS wasn't the best idea. Apologies for slowness on this but real life is an endless interrupt ;)

I just put the beta 5 files in manually and started it up. I've attached logs.

What I was immediately alarmed about was that it went full tilt hell for leather charging at 77pkWh! Seems a little agressive when there was enough juice for a while - more interestingly it then wanted to discharge hard 30 mins later...also I set the config to read: true (attached) which it didn't seem to care about.

[error.log](https://github.com/user-attachments/files/18494283/error.log)
[main.log](https://github.com/user-attachments/files/18494281/main.log)
[pv_opt-2.log](https://github.com/user-attachments/files/18494282/pv_opt-2.log)

[config.txt](https://github.com/user-attachments/files/18494287/config.txt)

stevebuk1, 2025-01-21:
> [@stevebuk1](https://github.com/stevebuk1) Yeah I think doing the redownload from inside HACS wasn't the best idea. Apologies for slowness on this but real life is an endless interrupt ;)

No problem!

> 
> I just put the beta 5 files in manually and started it up. I've attached logs.

Thanks!
> 
> What I was immediately alarmed about was that it went full tilt hell for leather charging at 77pkWh! Seems a little agressive when there was enough juice for a while - more interestingly it then wanted to discharge hard 30 mins later...

Somethings not right here. Pv_opt has picked up a price of 32.26p for the 5pm slot which explains the charging so it can dump it all again at 39p)

`45 Min import price 32.26p/kWh at 21/01 17:01`

but this is clearly incorrect as as you say, the Agile price was actually 77.6p for the 5pm slot. I can't tell why this is - perhaps because we were already inside 5pm on startup. 

In config.yaml please 
set `debug: false` to read `debug: true`
set `debug_categories: WSOIB` to read `debug_categories: O`

"O" will generate a summary of the plan by each half hour and is required as a minimum for any planning queries. (I leave my system in this logging mode permanently).  You can keep the other categories if you wish but it just means bigger logs so less historic time for investigations. 

> also I set the config to read: true (attached) which it didn't seem to care about.

When Pv_opt starts you can choose that it respects whats in HA already (i.e. on the dashboard), or reload whats in config.yaml. 

You have the former set via the line `overwrite_ha_on_restart: false`.

If you want each start of Pv_opt to load from config.yaml and update the dashboard, then change this to read  `overwrite_ha_on_restart: true`.  (I use this option)

stevebuk1, 2025-01-21:
I've released v4.0.9-Beta-1 to aid diagnosis of this, should it happen again. 

Either update through HACS, or replace files with the ones below:

https://github.com/fboundy/pv_opt/blob/patch2/apps/pv_opt/pv_opt.py
https://github.com/fboundy/pv_opt/blob/patch2/apps/pv_opt/pvpy.py
https://github.com/fboundy/pv_opt/blob/patch2/apps/pv_opt/solis.py

Please make the changes to config.yaml as outlined above.

pyriel800, 2025-01-22:
Okay, that's interesting and an issue my end - I actually have 2 accounts in my octopus integration, one is my father's account so I can keep an eye on it but I can remove that if it's causing an issue. Just for reference I also have 2 meters anyway, one is for import and a separate export one (although I think Octopus merges these 2 in the data).

I've added the T and grabbed logs from the updated run.

I will remove the second account if that helps.

[main.log](https://github.com/user-attachments/files/18503076/main.log)

pyriel800, 2025-01-22:
[pv_opt.log](https://github.com/user-attachments/files/18503123/pv_opt.log)

@stevebuk1 looks like pvopt log didn't upload

stevebuk1, 2025-01-22:
> Okay, that's interesting and an issue my end - I actually have 2 accounts in my octopus integration, one is my father's account so I can keep an eye on it but I can remove that if it's causing an issue. 

Yes it will cause an issue for Pv_opt. Best to remove it, but it might be possible to workaround it by turning off auto tariff location (`octopus_auto = False` in config.yaml) and instead inserting login details to Octopus, which will ensure only your account is visible. 

Just for reference I also have 2 meters anyway, one is for import and a separate export one (although I think Octopus merges these 2 in the data).

Ok useful to know, although I don't think this is causing an issue. The export tariff looks like it is being detected ok.

> 
> I've added the T and grabbed logs from the updated run.
> 
> I will remove the second account if that helps.
> 
> [main.log](https://github.com/user-attachments/files/18503076/main.log)

pyriel800, 2025-01-22:
@stevebuk1 i've deleted the account for now, easier and cleaner that way for this.

stevebuk1, 2025-01-22:
As v4.0.8 has been released to production that solves the title of this issue (Solarman time registers), I'm closing this. Feel free to raise new issues for anything else. Tariff inconsistencies please add to issue #380.

closed 2025-01-22 by stevebuk1