stevebuk1/pv_opt #366: Ensure battery current limit is used in charge planning

fboundy, 2025-01-11:
As per #355 - make sure that battery charge limits are honoured throughout

stevebuk1, 2025-01-11:
@CraigCallender just for my own curiosity, what is the use case for this? Pv_opt provides settings for setting the DC charger power, this being the charger that charges/discharges the batteries. On my 3.6kW inverter, the DC charger is 3kW, so its important to tell Pv_opt this otherwise it might assume we can charge batteries at 3.6kW. But if the batteries themselves can be charged/discharged at a rate thats less than the DC charger power, why not just set the DC charger power to that of the batteries?   In a schematic, the two are in series, the charge rate will always be the minimum of battery or DC charger, so they don't need to be set independently? Or did you just want a control in A rather than W? Reason for asking is that the answer will probably guide the fixes required.

CraigCallender, 2025-01-13:
@stevebuk1 - I think you're correct that I should have just used charger power instead of creating a new feature.  It was just me not being familiar with the system, so this could probably be mostly solved by removing the limit_amps feature and better documenting the charger_power setting that it should be configured as the lowest value between what your inverter is capable of vs your battery.

In _some_ ways it might make the system easier for users to retain the two settings.  If you think about it, having configuration that is specific to your inverter and another that is about your batteries prevents a complicated double question of, "Get the charging power of your inverter (normally specified in Watts) and the charging power of your battery (normally specified in Amps).  Now put the lower of the two on line 29...." ugh... it reminds me of manually filing my taxes in the USA...

Those are my thoughts!  Happy to help out more if I can.

Pyinthesky99, 2025-01-19:
Just my 2p
I have knocked Charger Power down to 4kW, although the Inverter is 5kW, just to give the batteries an easier life.
For the same reason, I have reset Inverter power back from 5k to 3.6kW as, although I don't use it routinely, on the one occasion I used pv_opt to force discharge (accidentally!) it started to dump the batteries at 5k.

closed 2025-06-18 by github-actions[bot]