stevebuk1/pv_opt #368: Charge settings not delivered

lazy-pete, 2025-01-11:
**Describe the bug**
A clear and concise description of what the bug is.

Now all is updated, including Solax integration, and producing charge calcs, however these don’t seem to be being delivered / acted upon. I should be charging now at 3200W but it’s just trickling in a little extra PV power. 

[image]
[image]
[image]
[main.log](https://github.com/user-attachments/files/18385896/main.log)
[pv_opt.log](https://github.com/user-attachments/files/18385898/pv_opt.log)
[error.log](https://github.com/user-attachments/files/18385899/error.log)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

lazy-pete, 2025-01-11:
Might have found the problem - 
sensor.solis_energy_storage_control_switch_2

HA seems to have (yet again) failed to remove all the old sensors when I deleted the Solax to add a new after updating. Most entities seemed to be fine, but this one is _2

I don’t seem to be able to relabel it in HA. Where would I change the target in PV_opt ? 

Cheers,

p

lazy-pete, 2025-01-11:
So, I’m wondering if the problem has arisen from at some point trying to install the Core PyModbus control method as well. I never got it to work, but it seems to be related to the same energy_charge_xxx entity thus may be why re-adding the Solax device ended up with _2. 

There are quite a few different versions of the energy control switch in the storage/core.entity_registry and I’m not quite sure which to remove. 

Would I be best to just remove all of both integrations in the hope that cleans up the registries and start again with just the Solax one ?

stevebuk1, 2025-01-11:
Hi, something else appears to be fighting Pv_opt in setting your inverter. Pv_opt writing start/end times and current, these are being read back correctly and even persist until the next optimiser slot, but then get reset to a start and end time of 00:00 by something else. I had this once when I was parallel running with predbat and it took itself out of readonly! 

As you suggest I think you need to remove all Solis integrations, clear out any remaining entities, then install one and ensure you can control the inverter manually with that, with Pv_opt in readonly mode.

fboundy, 2025-01-11:
Uninstall Solax and any other modbus integrations. Then restart HA and reinstall.

Integrations don’t get fully removed till you restart.

fboundy, 2025-01-12:
Looks like this is a problem with the integration rather than PV Opts so I am closing. Please re-open if it persists

closed 2025-01-12 by fboundy

lazy-pete, 2025-01-13:
I think it was a comms issue ultimately. I restarted the virtual machine in which HA runs, as well as restarting HA after updating the Solax-Modbus integration and then realised I’d left the Solis WiFi dongle also connected to my Waveshare 485 setup. After all that and disconnecting the dongle things seem to be working again. Apologies for any confusion caused.