stevebuk1/pv_opt #371: Solax Modbus integration: button.solis_update_charge_discharge_times button pressing

stevebuk1, 2025-01-11:
button.solis_update_charge_discharge_times needs pressing when different times have been written to the number entities. It does not need pressing for updates to inverter current. 

However, when the times are correct but the current is changed, PV_opt presses button.solis_update_charge_discharge_times:

```
11/01, 00:00:38     INFO: Currently in charge/discharge/hold slot
11/01, 00:00:38     INFO: Current charge/discharge window ends in 389.4 minutes.
11/01, 00:00:38     INFO: Charge status is active, not setting start time
11/01, 00:00:38     INFO: Setting end time to 11-Jan 06:30 GMT
11/01, 00:00:38     INFO: Setting power to 800
11/01, 00:00:38     INFO: Setting SOC to 82
11/01, 00:00:38     INFO: Entity: number.solis_timed_charge_end_hours Value:  6.0  Old State:  6.0 New state: None Diff:  0.0 Tol:  0.0
11/01, 00:00:38     INFO: Entity: number.solis_timed_charge_end_minutes Value: 30.0  Old State: 30.0 New state: None Diff:  0.0 Tol:  0.0
11/01, 00:00:39     INFO: Entity: number.solis_timed_charge_current Value: 15.7  Old State: 15.8 New state: 15.7 Diff:  0.1 Tol:  0.1
11/01, 00:00:39     INFO: Current 15.7A written to inverter
11/01, 00:00:39     INFO: Something changed - need to press the appropriate Button
11/01, 00:00:39     INFO: Successfully pressed button button.solis_update_charge_discharge_times
11/01, 00:00:39     INFO: Waiting for inverter Read cycle: 18 seconds
```
This also happens when the current actually need changing:

```
11/01, 02:00:14     INFO: Currently in charge/discharge/hold slot
11/01, 02:00:14     INFO: Current charge/discharge window ends in 269.8 minutes.
11/01, 02:00:14     INFO: Charge status is active, not setting start time
11/01, 02:00:14     INFO: Setting end time to 11-Jan 06:30 GMT
11/01, 02:00:14     INFO: Setting power to 900
11/01, 02:00:14     INFO: Setting SOC to 85
11/01, 02:00:14     INFO: Entity: number.solis_timed_charge_end_hours Value:  6.0  Old State:  6.0 New state: None Diff:  0.0 Tol:  0.0
11/01, 02:00:14     INFO: Entity: number.solis_timed_charge_end_minutes Value: 30.0  Old State: 30.0 New state: None Diff:  0.0 Tol:  0.0
11/01, 02:00:14     INFO: Entity: number.solis_timed_charge_current Value: 17.7  Old State: 17.6 New state: None Diff:  0.1 Tol:  0.1
11/01, 02:00:14     INFO: Inverter already at correct current
11/01, 02:00:14     INFO: Something changed - need to press the appropriate Button
11/01, 02:00:15     INFO: Successfully pressed button button.solis_update_charge_discharge_times
```

The issue is caused by `changed `being an or of the response from `_set_times` and `_set_current `where changed is consequently used to press the button. The button should only be pressed if `_set_times` reports a change.

fboundy, 2025-01-11:
Just need to check this with the newer solis firmware which is (more than a bit) flakey…

stevebuk1, 2025-01-11:
I imagined it might be needed for the new firmware, just concerned that for the old its more avoidable EEPROM writes.

fboundy, 2025-01-11:
Fair point.

lazy-pete, 2025-01-13:
Solis updated my firmware fairly recently while troubleshooting a different issue (which turned out to be an off-grid as the RCD had tripped). I wonder if that was when I started having issues with the Solax integration and pv_opt seemingly generating instructions that aren’t acted upon, plus the weird usage / pricing spikes I’ve been seeing.

stevebuk1, 2025-02-01:
Theres a trial fix for this in v4.0.9-Beta-5 in `patch2`. 

The logic for the newer solis firmware I've left unchanged via some additional if statements.

stevebuk1, 2025-03-28:
Released in v4.0.9, closing issue.

closed 2025-03-28 by stevebuk1