stevebuk1/pv_opt #372: Fixes for #356 mean carslot is incorrectly tagged

stevebuk1, 2025-01-11:
Fixes for #356 have involved modifying the index value in partial slots. This appears to have had a side effect on flagging an IOG carslot too early, putting the inverter in a current = 0 mode. 

```
10/01, 22:20:13     INFO:   Base cost:                                265.4p
10/01, 22:20:13     INFO:   Optimised cost (Optimised Charging):      137.9p <=== Current Setup
10/01, 22:20:13     INFO:   Optimised cost (Optimised PV Export):     137.9p
10/01, 22:20:13     INFO:   Optimised cost (Forced Discharge):        137.9p
10/01, 22:20:13     INFO: 
10/01, 22:20:13     INFO: Reloading IOG Car slots from Intelligent Dispatching Sensor
10/01, 22:20:13     INFO: -----------------------------------------------------------
10/01, 22:20:13     INFO: 
10/01, 22:20:13     INFO:   Current Dispatch Status (On/off) = : off
10/01, 22:20:13     INFO: 
10/01, 22:20:13     INFO:     Octopus Intelligent Go Smart Charging Schedule is.... 
10/01, 22:20:13     INFO:       22:30 to 23:00    Charge:   -1.72kWh
10/01, 22:20:13     INFO:       23:00 to 06:30    Charge:  -39.88kWh
10/01, 22:20:14     INFO: 
10/01, 22:20:14     INFO: 
10/01, 22:20:14     INFO: Checking EV Status (IOG tariff)
10/01, 22:20:14     INFO: -------------------------------
10/01, 22:20:14     INFO: 
10/01, 22:20:14     INFO: EV is connected but IOG tariff reload previously caried out. IOG tariff not reloaded
10/01, 22:20:14     INFO: 
10/01, 22:20:14     INFO: 
10/01, 22:20:14     INFO: 1/2 Hour Optimsation summary
10/01, 22:20:14     INFO: ----------------------------
10/01, 22:20:14     INFO: 
10/01, 22:20:14     INFO: 
                             solar  consumption  dt_hours  battery_grid_requirement  forced  battery_temp     chg  chg_end     battery    grid        soc    soc_end   import  export  carslot  period
2025-01-10 22:20:00+00:00    0.000   391.529042  0.166667                391.529042       0    391.529042  3762.0   3694.7   391.68600    -0.0  33.000000  32.409649  25.6269       0        1       0
2025-01-10 22:30:00+00:00    0.000   419.036068  0.500000                419.036068     750   -750.000000  3694.7   4036.0  -750.10989  1169.0  32.409649  35.403509   7.0003       0        1       1
2025-01-10 23:00:00+00:00    0.000   468.277757  0.500000                468.277757     750   -750.000000  4036.0   4377.2  -749.89011  1218.0  35.403509  38.396491   7.0003       0        1       1
```

The consequence of this is that an inverter hold slot is placed too early, and also for a fixed half hour:
```

10/01, 22:20:14     INFO: 
10/01, 22:20:14     INFO: Optimal forced charge/discharge slots:
10/01, 22:20:14     INFO:   10-Jan 22:20 GMT - 10-Jan 22:50 GMT  Power:     1W  SOC:   33% ->   33%  <=Car
10/01, 22:20:14     INFO:   10-Jan 22:30 GMT - 11-Jan 06:30 GMT  Power:   800W  SOC:   32% ->   80%  
```
[pv_opt.log.1.log](https://github.com/user-attachments/files/18387474/pv_opt.log.1.log)

stevebuk1, 2025-01-11:
@fboundy as this is EV related I'll take a look at this one.

stevebuk1, 2025-01-12:
Found the issue, will trial a fix overnight and raise a PR if successful.

swests, 2025-01-13:
@fboundy - A question or 2, if I may.  Just had a Zappi installed today (yay), is the code in a state to be used for EV charging (on Agile rather than IOG)? Would you like a motivated tester? Are there some docs, rough notes are fine, for using the EV side of the platform?  Thanks

stevebuk1, 2025-01-13:
> @fboundy - A question or 2, if I may. Just had a Zappi installed today (yay), is the code in a state to be used for EV charging (on Agile rather than IOG)? Would you like a motivated tester? Are there some docs, rough notes are fine, for using the EV side of the platform? Thanks

There is an Agile EV charger in the latest production release. The documentation fell behind but here are is the latest user guide.

https://github.com/fboundy/pv_opt/blob/steveb-patch/README.md#ev-charging-on-the-agile-tariff

I'm on IOG but I did some testing with my Zappi by forcing the Agile tariff. Other than that its untested. Any bugs, please raise as new issues.

stevebuk1, 2025-01-13:
Forgot to add (and the EV part documentation isn't clear), you'll need to get the myenergi HA integration installed if you haven't already. 

Also, all ideas / thoughts for improvements welcome (documentation as well as functionality). You might remember the feature was originally requested back in Feb 24 but the requester is still on IOG so it hasn't been used in anger. I can already see we should link Zappi control direct from Pv_opt to the myenergi integration rather than through an automation......

stevebuk1, 2025-01-13:
Last nights charging didnt trigger the right conditions for testing this (actual IOG car charging was well inside IOG cheap rate) and it might be some time before those conditions occur again, so closing this, assuming `patch `will get PR'd to `main` at some point. If it occurs again I'll reopen.

closed 2025-01-13 by stevebuk1