stevebuk1/pv_opt #376: ERROR: Tariff Inconsistency

punkymuzzle, 2025-01-13:
**Describe the bug**
Running pv-opt 4.0.7, and on the Intelligent Octopus Agile tariff, I get an error in the status ashboard saying "Error: Tariff Inconsistency"
Screenshots below show more detail.

Charging schedule appears to be configured but Inverter has not received the charging instruction before it reached 10% SoC and went idle.  SO had to manually add the schedule via the SolaX app.

AppDaemon errors also noted.

> /homeassistant/appdaemon/apps/pv_opt/pvpy.py:899: UserWarning: Boolean Series key will be reindexed to match DataFrame index.
  x = x[available.loc[: max_slot - pd.Timedelta("30min")]]
/homeassistant/appdaemon/apps/pv_opt/pvpy.py:899: UserWarning: Boolean Series key will be reindexed to match DataFrame index.
  x = x[available.loc[: max_slot - pd.Timedelta("30min")]]
/homeassistant/appdaemon/apps/pv_opt/pvpy.py:899: UserWarning: Boolean Series key will be reindexed to match DataFrame index.
  x = x[available.loc[: max_slot - pd.Timedelta("30min")]]
/homeassistant/appdaemon/apps/pv_opt/pvpy.py:899: UserWarning: Boolean Series key will be reindexed to match DataFrame index.
  x = x[available.loc[: max_slot - pd.Timedelta("30min")]]

**To Reproduce**
Steps to reproduce the behavior:
Running pv-opt 4.0.7 with IOG Tariff.

**Screenshots**
Status:
[image]

Charging schedule:
[image]

**Desktop (please complete the following information):**
RasPi 4B running HA

**Smartphone (please complete the following information):**
N/A

**Additional context**
Logs attached FYI
[error.log](https://github.com/user-attachments/files/18403353/error.log)
[main.log](https://github.com/user-attachments/files/18403354/main.log)
[pv_opt.log](https://github.com/user-attachments/files/18403355/pv_opt.log)

stevebuk1, 2025-01-14:
The tariff inconsistency checking is a compare from what is downloaded from the Octopus API via the Octopus website, versus what the BottlecapDave integration reports. 

On IOG, I've noted that if one plugs the car in and gets extra slots, the BottleCapDave integration picks up the extra slots but the Octopus API doesn't, as its a generic tariff download whereas the extra slots are unique to you at that moment in time. If that is the cause of the error you can ignore it, but given the screenshot charge plan above it doesnt look like you've got any extra slots awarded, so warrants further investigation. 

Unfortunately the logs you've provided contain optimiser runs that are exclusively Agile, so must come from before your switch over. The charge plans within are also Agile based. Can you attach a Pv_opt.log that contains a charge plan that matches your screenshot (almost always 23:30 to 05:30 unless extra slots are awarded). 

The Appdeamon errors you are report are already covered in #364. I get them too with v4.x.x, they are effectively warnings and are not the cause of this issue.

punkymuzzle, 2025-01-14:
OK great, will do thanks.
Regarding g the additional slots (other than 2330-0530) they did appear in
the charging schedule Sunday night, Monday morning and did set my inverter
to charge.
I'll look to start with an empty log file and capture what I can.
Thanks very much.

stevebuk1, 2025-01-18:
Note I've raised a seperate issue for the Tariff inconsistency logging (#380), but if you see it please post a log either here or against the new issue - I'll keep both open until we know what you are seeing is the same I've always seen.

punkymuzzle, 2025-01-19:
I'm still struggling with getting the logs to write (they were las written 2nd Jan 25....)
I note in #284 you have provided some ideas for me to try, so will take a look at this tomorrow.
At the moment pv-opt isn't a happy bunny.  it's not been working for a few days now.  I have restarted AppDaemon multiple times without success.
However, i restarted it earlier on and left it to do other things and it looks like it's up and running (sort of)
1. It's now creating a charging schedule.  However this keeps changing every cycle to slip each slot by 15 minutes - It was set to run at 00:01-05:30, then after the latest recalc it then said 00:15, so it's not updating the inverter.  So there is something very wrong with it at the moment.
2. Still no logs ( I know I have to take a look at the link you provided in #284).  I did ssh int the Raspi and took a look at the file permissions of the logfiles. I changed them to fully accessible (chmod 777) but pv-opt is still not writing to the logs.
3. AppDaemon is still showing an error in the log, however not sure this is affecting anything at the moment, but no 100% sure):

> /homeassistant/appdaemon/apps/pv_opt/pvpy.py:899: UserWarning: Boolean Series key will be reindexed to match DataFrame index.
>   x = x[available.loc[: max_slot - pd.Timedelta("30min")]]

stevebuk1, 2025-01-19:
> 2\. I did ssh int the Raspi and took a look at the file permissions of the logfiles. I changed them to fully accessible (chmod 777) but pv-opt is still not writing to the logs.

Advanced stuff! (takes me back to my uni days). I doubt this is file permissions, its more where Pv_opt is being told to write the files (via appdaemon.log). 

> 3\. AppDaemon is still showing an error in the log, however not sure this is affecting anything at the moment, but no 100% sure):

Definitely unrelated. My system generates this warning too, its new at v4.x.x. It has no effect on function or performance.

punkymuzzle, 2025-01-20:
> 2. I did ssh int the Raspi and took a look at the file permissions of the logfiles. I changed them to fully accessible (chmod 777) but pv-opt is still not writing to the logs.
>
>  Advanced stuff! [...]

:) I work in IT and used Unix years ago...

 

> 3. AppDaemon is still showing an error in the log, however not sure this is affecting anything at the moment, but no 100% sure):
>  /homeassistant/appdaemon/apps/pv_opt/pvpy.py:899: UserWarning: [...]

 

> Definitely unrelated. My system generates this warning too, its new at v4.x.x. It has no effect on function or performance.

OK thanks.

Incidentally I installed the latest update earlier and pv-opt seems to be working now, so thanks very much for this.
Logs still not updating though so will go through your earlier post some time tomorrow and report back.

Thanks very much for the support with this, really appreciate it.

punkymuzzle, 2025-01-20:
Can be closed now, original issue has been recorded as a bug in #380

closed 2025-01-20 by stevebuk1