stevebuk1/pv_opt #377: Stuck on Forced Discharge

pbcjh10, 2025-01-14:
Hello, 

The previous thread I was using for help was closed so making a new one. I have upgraded everything to the latest release. 
@stevebuk1 - can we carry on here please?
[pv_opt3.log](https://github.com/user-attachments/files/18408075/pv_opt3.log)

https://github.com/fboundy/pv_opt/issues/284

So running HA Core (on a HA Green) 2025.1.2, Supervisor 24.12.3 and OS 14.1
Home setup I have a solis hybrid connected a pylontec 4.8kw battery
I have 2 zappis 
We are on Intellegent Octopus Go

My essential problem is that it looks like despite optimising the charging plan, and coming up with a sensible looking plan (ie, over the cheap rate with battery reaching 100% as it ends) the Optimisation Breakdown has the <== current setup bit stuck on Forced Discharge. 
The battery never charges and sits at 19% slowly loosing charge. 

This is what my frontend shows me for the solis itself; 
[image]
[error(1).log](https://github.com/user-attachments/files/18408042/error.1.log)
[error2.log](https://github.com/user-attachments/files/18408043/error2.log)

With PV Opt
[image]

There are two possible things that I can see in logs. I attach both pvopt logs, but also HA logs
[error(1).log](https://github.com/user-attachments/files/18408042/error.1.log)
[error2.log](https://github.com/user-attachments/files/18408043/error2.log)
[pv_opt.log](https://github.com/user-attachments/files/18408056/pv_opt.log)
[main.log](https://github.com/user-attachments/files/18408057/main.log)
[pv_opt1.log](https://github.com/user-attachments/files/18408058/pv_opt1.log)
[pv_opt2.log](https://github.com/user-attachments/files/18408065/pv_opt2.log)
[pv_opt4.log](https://github.com/user-attachments/files/18408069/pv_opt4.log)
[pv_opt5.log](https://github.com/user-attachments/files/18408071/pv_opt5.log)

A few questions;

When I try and turn from shaped consumption to use consumption history, I get an immediate incorrect tarrif error. Switching back clears it. But there are over 10 days (I've tried 5-7) of data as I've been trying to get this running for months on and off. 

Are the new IOG charger settings active in yaml yet? Do things like copy across settings mirror IOG to your 2nd charger?

stevebuk1, 2025-01-14:
> @stevebuk1 - can we carry on here please?

No problem - the Github bot auto-shut the last issue due to inactivity. 

The issue is primarily caused by Pv_opt failing to read the status of the Ev charger selector, which means a check later on causes a fatal error just before the inverter is due to be updated with charge planning. I think we can work around this via a setting in config.yaml. 

Can you attach config.yaml so I can take a look?

pbcjh10, 2025-01-14:
@stevebuk1 Thanks!
Yep here's a config. [config.yaml (2).log](https://github.com/user-attachments/files/18409721/config.yaml.2.log)
Sinice posting above, I tried going through the github config compared to mine and have added in some sections, although they're largely commented out. 

Why is the pv opt failing to read the EV selector?

Another perhaps related issue is that despite me setting Zappi in charge in config.yaml, when i reboot it still is coming up empty on the dashboard until I select manually in dropdown box. Despite me having overwrite HA from yaml selected. Its like when I reboot its just ignoring bits/all? of the config.yaml?
[config.yaml (2).log](https://github.com/user-attachments/files/18409721/config.yaml.2.log)

This is my folder structure, which I think should be okay? 
[image]

stevebuk1, 2025-01-14:
> This is my folder structure, which I think should be okay?

All looks fine. Your config.yaml is being read, its just ev_charger needs to be read from HA correctly otherwise it won't look for it in config.yaml.

stevebuk1, 2025-01-14:
> Are the new IOG charger settings active in yaml yet? Do things like copy across settings mirror IOG to your 2nd charger?

When on IOG, Octopus is in control of your charger and Octopus can only control one charger. Pv_opt reads the Octopus planned dispatches in advance so it can plan prevent house battery discharge or charge at the cheap rates, and will read both Zappis to get house consumption data, but does not mirror the Zappi charging commands from Octopus as they occur in realtime from Octopus servers to MyEnergi servers to your Zappi. If a mirror functionality was in Pv_opt it would always be 10 minutes behind as this is the rate the optimiser runs at. 

If you want to mirror a 2nd Zappi to copy what the IOG controllled Zappi is doing you can probably do it by an HA automation that picks up the boost commands from one Zappi and mirrors them to the other.

pbcjh10, 2025-01-14:
Okay, so I added the new bits to config/yaml from the github version. 

I now am stuck with a load failure with this error in appdaemon;

<html><body>

AttributeError: 'PVOpt' object has no attribute 'zappi_plug_entity'
--
^^^^^^^^^^^^^^^^^^^^^^
if (len(self.zappi_plug_entity) > 0) and self.ev:
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 876, in _check_car_plugin_iog
self._check_car_plugin_iog()
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 2617, in optimise
^^^^^^^^^^^^^^
return f(*args, **kw)
File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
self.optimise()
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 2248, in optimise_time
^^^^^^^^^^^^^^
return f(*args, **kw)
File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
07:51:12 WARNING pv_opt: Traceback (most recent call last):
07:51:12 WARNING pv_opt: ------------------------------------------------------------
07:51:12  WARNING pv_opt: Worker Ags: {'id': 'c28c655b6a7741d29b3c9e7d8c1427fd',  'name': 'pv_opt', 'objectid': '9c97973b578e4d28b6e6291e8bdbc679',  'type': 'scheduler', 'function': <bound method PVOpt.optimise_time of  <pv_opt.PVOpt object at 0xffffb2b8f690>>, 'pin_app': True,  'pin_thread': 1, 'kwargs': {'interval': 600, '__thread_id': 'thread-1'}}
07:51:12 WARNING pv_opt: Unexpected error in worker for App pv_opt:
07:51:12 WARNING pv_opt: ------------------------------------------------------------
07:41:13 WARNING pv_opt: ------------------------------------------------------------
 

</body>
</html>

stevebuk1, 2025-01-14:
> 'scheduler', 'function': >, 'pin_app': True, 'pin_thread': 1, 'kwargs':

Yep, that's what the commit I made above  fixes. Ill post some new code
later tonight for you to try.

pbcjh10, 2025-01-14:
> Yep, that's what the commit I made above fixes. Ill post some new code later tonight for you to try.
> […](#)

So I should copy and paste the new pv_opt.opt?

stevebuk1, 2025-01-14:
> So I should copy and paste the new pv_opt.opt?

I did the fix at work and so wanted to test it on my system first. 

v4.0.8-Beta-5 contains a fix that should get the house battery charging up and running again. 
Overwrite the 3 files in `homeassistant/appdaemon/apps/pv_opt` with the following:

https://github.com/fboundy/pv_opt/blob/patch/apps/pv_opt/pv_opt.py
https://github.com/fboundy/pv_opt/blob/patch/apps/pv_opt/pvpy.py
https://github.com/fboundy/pv_opt/blob/patch/apps/pv_opt/solis.py

If on restart Ev Charger is still set to None, change it to Zappi.

pbcjh10, 2025-01-14:
Ok, have now updated a few bits (the solis_house_load) from november in the other issue. 
I've also adjusted the entity names so both chargers have zappi in rather than just one of them. 

This is yaml currently;
[config(3).yaml.log](https://github.com/user-attachments/files/18413563/config.3.yaml.log)

I've copied your 3 files above and have this;

`<html><body>

17:41:40 WARNING pv_opt: ------------------------------------------------------------
--
 
KeyError: 'Timed Charge/Discharge - No Grid Charging'
~~~~~~~~~~~^^^^^^
self._modes = {self._codes[code]: code for code in entity_modes}
File "/homeassistant/appdaemon/apps/pv_opt/solis.py", line 743, in <dictcomp>
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
self._modes = {self._codes[code]: code for code in entity_modes}
File "/homeassistant/appdaemon/apps/pv_opt/solis.py", line 743, in __init__
^^^^^^^^^^^^^^^^^^^^^^^^^
return SolisSolaxModbusInverter(
File "/homeassistant/appdaemon/apps/pv_opt/solis.py", line 314, in create_inverter_controller
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
self.inverter = create_inverter_controller(inverter_type=self.inverter_type, host=self)
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1274, in _load_inverter
self._load_inverter()
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 593, in initialize
^^^^^^^^^^^^^^
return f(*args, **kw)
File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
result = self.fn(*self.args, **self.kwargs)
File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
^^^^^^^^^^^^^^^
response = future.result()
File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
await utils.run_in_executor(self, init)
File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 162, in initialize_app
17:41:40 WARNING pv_opt: Traceback (most recent call last):
17:41:40 WARNING pv_opt: ------------------------------------------------------------
17:41:40 WARNING pv_opt: Unexpected error running initialize() for pv_opt
17:41:40 WARNING pv_opt: ------------------------------------------------------------
17:36:16 WARNING pv_opt: ------------------------------------------------------------
 
KeyError: 'Timed Charge/Discharge - No Grid Charging'
~~~~~~~~~~~^^^^^^
self._modes = {self._codes[code]: code for code in entity_modes}
File "/homeassistant/appdaemon/apps/pv_opt/solis.py", line 743, in <dictcomp>
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
self._modes = {self._codes[code]: code for code in entity_modes}
File "/homeassistant/appdaemon/apps/pv_opt/solis.py", line 743, in __init__
^^^^^^^^^^^^^^^^^^^^^^^^^
return SolisSolaxModbusInverter(
File "/homeassistant/appdaemon/apps/pv_opt/solis.py", line 314, in create_inverter_controller
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
self.inverter = create_inverter_controller(inverter_type=self.inverter_type, host=self)
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1274, in _load_inverter
self._load_inverter()
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 593, in initialize
^^^^^^^^^^^^^^
return f(*args, **kw)
File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
result = self.fn(*self.args, **self.kwargs)
File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
^^^^^^^^^^^^^^^
response = future.result()
File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
await utils.run_in_executor(self, init)
[... 183 lines cut ...]
return f(*args, **kw)
File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
result = self.fn(*self.args, **self.kwargs)
File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
^^^^^^^^^^^^^^^
response = future.result()
File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
await utils.run_in_executor(self, init)
File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 162, in initialize_app
16:43:39 WARNING pv_opt: Traceback (most recent call last):
16:43:39 WARNING pv_opt: ------------------------------------------------------------
16:43:39 WARNING pv_opt: Unexpected error running initialize() for pv_opt
16:43:39 WARNING pv_opt: ------------------------------------------------------------

<!--EndFragment-->
</body>
</html>`

This is in Appdaemon 
The PV Opt dashboard says "Initialising PV Model" while Appdaemon says initialisation error. 

[error(2).log](https://github.com/user-attachments/files/18414146/error.2.log)
[main(1).log](https://github.com/user-attachments/files/18414148/main.1.log)
[pv_opt(2).log](https://github.com/user-attachments/files/18414149/pv_opt.2.log)

pbcjh10, 2025-01-14:
This is from the Appdaemon error.log (in the appdaemon bit itself)

<html><body>

17:41:40 WARNING pv_opt: ------------------------------------------------------------
--
 
KeyError: 'Timed Charge/Discharge - No Grid Charging'
~~~~~~~~~~~^^^^^^
self._modes = {self._codes[code]: code for code in entity_modes}
File "/homeassistant/appdaemon/apps/pv_opt/solis.py", line 743, in <dictcomp>
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
self._modes = {self._codes[code]: code for code in entity_modes}
File "/homeassistant/appdaemon/apps/pv_opt/solis.py", line 743, in __init__
^^^^^^^^^^^^^^^^^^^^^^^^^
return SolisSolaxModbusInverter(
File "/homeassistant/appdaemon/apps/pv_opt/solis.py", line 314, in create_inverter_controller
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
self.inverter = create_inverter_controller(inverter_type=self.inverter_type, host=self)
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1274, in _load_inverter
self._load_inverter()
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 593, in initialize
^^^^^^^^^^^^^^
return f(*args, **kw)
File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
result = self.fn(*self.args, **self.kwargs)
File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
^^^^^^^^^^^^^^^
response = future.result()
File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
await utils.run_in_executor(self, init)
File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 162, in initialize_app
17:41:40 WARNING pv_opt: Traceback (most recent call last):
17:41:40 WARNING pv_opt: ------------------------------------------------------------
17:41:40 WARNING pv_opt: Unexpected error running initialize() for pv_opt
17:41:40 WARNING pv_opt: ------------------------------------------------------------
17:36:16 WARNING pv_opt: ------------------------------------------------------------

<!--EndFragment-->
</body>
</html>

pbcjh10, 2025-01-14:
[image]

pbcjh10, 2025-01-14:
Sorry silly question, which dropdown?

pbcjh10, 2025-01-14:
Ok, using agressive AI debugging I've removed the hmi_firmware_fb00_plus: True line that had come across from me going through the config yaml. 
I now have a system thats loaded and has planned a charge from 23:00 to 06:30 (cheap rates) and is in Forced Discharge mode currently.
The Forced Discharge bit was the problem before, shouldn't it be in self-use?

pbcjh10, 2025-01-14:
Yep self use here. 
[image]

With regards to IOG - pv_opt has been fairly consistent at detecting the additional charge blocks when IOG generate a plan if thats what you mean?
This from appdaemon live log looks promising?

<html><body>

20:00:43     INFO: ----------------------------
--
20:00:43     INFO: 1/2 Hour Optimsation summary
20:00:43     INFO:
20:00:43     INFO:
20:00:43     INFO: EV is connected but IOG tariff reload previously caried out. IOG tariff not reloaded

</body>
</html>

stevebuk1, 2025-01-14:
So each optimiser run is now completing successfully. 

Pv_opt will only write to the inverter 10 minutes before a charging or discharging is required. your log tells me you've got an extra slot at 11pm, so nothing else is going to happen inverter wise until 22:50. However Pv_opt defintely sorted out your inverter settings when it loaded so I'm confident it will work. 

Theres still something not quite right with the house consumption figures. Pv_opt is now downloading from both Zappis, but after subtracting this from the grid use we should be seeing what the house is using. Its coming up with a 10kW load on almost permanently, which I doubt is right. 

In config.yaml can you change this line:

`debug_categories: TBOC`

to 

`debug_categories: PO`
 which will log the detailed power consumption calculations Pv_opt is making with your Zappi consumption sensors. 

I only need one optimiser run for debugging as the data doesnt change much run to run, so give it 1/2 hour and post a pv_opt.log please.

stevebuk1, 2025-01-14:
> When I try and turn from shaped consumption to use consumption history, I get an immediate incorrect tarrif error. Switching back clears it. But there are over 10 days (I've tried 5-7) of data as I've been trying to get this running for months on and off.

Is this still an issue now or is it ok?

stevebuk1, 2025-01-15:
The data from the grid consumption sensor looks like this for 24 hours yesterday:

[image]

Y axis is in kWh. This means you are using 80kWh per day at a pretty constant rate. Is that what your Octopus App says? 
I'd appreciate a screengrab from the Octopus app usuage tab that shows a complete day. 
If it really is reality then fair enough, but your house battery is going to be flat in less than an hour and Pv_opt is predicting that correctly. 

It may be the EVs, but to get a straight line like that they'd both need to be charging all day. I assume that wasnt the case?

Unfortunately on startup misread ev_charger on startup and I don't think you changed it to "Zappi" on the dashboard, so the Zappi consumption data wasn't read nor calculated. 

It matters not as I made a mistake in the debug categories anyway....

Can you change

`debug_categories: PO`

to 

`debug_categories: POE`

Then:

1) Wait for PV_opt to read idle
2) Toggle Ev charger to Zappi
3) Wait for Pv_opt status to read idle again
4) post pv_opt.log after half hour of running. (More is fine, whenever is convenient)

pbcjh10, 2025-01-15:
Ok

It was already on Zappi, have toggled on and off back to zappi. 
[pv_opt(4).log](https://github.com/user-attachments/files/18423106/pv_opt.4.log)

Heres my electricity, its high, but none of the charts for entities seem to match.
We use a lot because its cold and ASHP + 2 cars. 

[image]
[image]
[image]

Here are two I think matching days from the inverter. 
[image]
[image]

Grid import yesterday is weird, perhaps because it wasn't setup up correctly. 
Today looks less weird?

[image]

[image]

[image]
[image]

I found this, are the units wrong?  It looks like it matches usage but should be in KW not W?
[image]

Although total house load is still unhelpful I think? 
[image]
[image]

stevebuk1, 2025-01-15:
Perfect, that gives me lots of data to work with. The logs now contain the Zappi logging so I can check out that part too.

stevebuk1, 2025-01-16:
To be honest it all looks fine.

Here is your smart meter consumption against what your inverter thinks it is importing from the grid:

[image]

The inverter is overeading the grid import as measured by your Smart meter by 6.5% on the 11th and 13% on the 12th. CT clamps are usually 10% accurate at best so this is what I'd expect (remembering the main reason that CT clamp exists is to keep the grid at zero when the battery is supplying the house).   The main thing is that sensor.solis_house_load_today is the right sensor so your config is correct. 

I've had a look at both Zappi imports and these are being summed together and subtracted from the grid correctly. 

Solis Inverter House load is a sensor returns instantaneous power. 1kW = 1000W so the scaling is correct, and it matches your Octopus app. We don't tend to encourage using it anymore as the file size it has to download is massive and it causes issues for Appdaemon which has a hard limit of 10seconds for any data read from HA. 

I think the only issue remaining is this:

> When I try and turn from shaped consumption to use consumption history, I get an immediate incorrect tarrif error. Switching back clears it. But there are over 10 days (I've tried 5-7) of data as I've been trying to get this running for months on and off.

I don't have a log that covers this.

All you need to do is 

- toggle consumption history off
- wait 2 minutes,
- toggle consumption history on
- wait 3 minutes
- send me pv_opt.log

stevebuk1, 2025-01-18:
> Consumption history is interesting. It was turned on and ok. I toggled off and it was fine. Toggled back on and tarrif inconsistency, but then 10 secs later its ok and is optimising charging plan. So i guess okay? To be continued?

Ok this is one of two things:

1) When IOG gives you an extra slot then a tariff inconsistency is raised. The tariff consistency checking was something introduced to check the Octopus Integration against the Octopus Website. For every other tariff, these will (should) always match. For IOG, they won't match as when you have an extra slot, the tariff is unique to you. I get the same on my system (I'm on IOG) when I get extra slots. It probably should be disabled for IOG. I just ignore it. 

2) Your log is producing this:

```
10:50:00     INFO: Checking tariff prices vs Octopus Energy Integration:
10:50:00     INFO: -----------------------------------------------------
10:50:00     INFO: 
10:50:00     INFO:   Import: Average Prices - PV_OPT: 22.08 p/kWh  OE Integration: 22.08 p/kWh  Mean difference:  0.00 p/kWh
10:50:00     INFO:   Export: Average Prices - PV_OPT: 13.00 p/kWh  OE Integration: 15.00 p/kWh  Mean difference:  2.00 p/kWh <<< ERROR
```
The Octopus Energy Integration is saying you have an export tariff of 15p, but Pv_opt is loading a 13p export tariff. The reason for this is your config.yaml there is a manual overide of your tariff set, probably inadvertently. Unless you are sure your export tariff code is E-1R-OUTGOING-FIX-12M-19-05-13-G, I'd just comment out the 3rd line below. 

```
  # octopus_import_tariff_code: E-1R-AGILE-23-12-06-G
  # # octopus_export_tariff_code: E-1R-OUTGOING-LITE-FIX-12M-23-09-12-G
  octopus_export_tariff_code: E-1R-OUTGOING-FIX-12M-19-05-13-G

  # octopus_import_tariff_code: E-1R-FLUX-IMPORT-23-02-14-G
  # octopus_export_tariff_code: E-1R-FLUX-EXPORT-23-02-14-G
```

stevebuk1, 2025-01-22:
Closing, as fixes released as v4.0.8. Any issues please raise a new bug.

closed 2025-01-22 by stevebuk1