stevebuk1/pv_opt #378: End time moves forward at end of charge window

stevebuk1, 2025-01-16:
- Version: v4.0.8-Beta-5

In the last 1/2 hour of charge period for the 2nd and 3rd optimiser runs in that period, the charge window does not match the plan, resulting in end time updates that are not necessary:

On the 5:10 optimiser run
Charge plan: (ends at 5:30)

[image]

Charge window: (ends at 5:50)

[image]

Writes to inverter:

[image]

The issue is one of curtailing unnecessary inverter writes only, as when the 05:30 optismer run happens it recognises the charge slot has finished and stops the charge anyway.

stevebuk1, 2025-01-18:
Fix implemented, on test tonight.

Pyinthesky99, 2025-01-23:
Another curiosity, had a charge slot scheduled for 14:30 t0 15:30 today, it rolled the finish time as  usual but then started to roll the hold time past the finish time. Corrected itself I think at 15:51

[image]

[error.log](https://github.com/user-attachments/files/18523123/error.log)
[pv_opt.log](https://github.com/user-attachments/files/18523122/pv_opt.log)

stevebuk1, 2025-01-26:
> Another curiosity, had a charge slot scheduled for 14:30 t0 15:30 today, it rolled the finish time as usual but then started to roll the hold time past the finish time. Corrected itself I think at 15:51

Not a different variant of the moving forward issue, its just that at 15:40 Pv_opt decided to schedule a very low charge (until 16:00), but it canned it at 15:50. It might have been caused by manual charge current edits as the start SOC would be higher.

stevebuk1, 2025-01-27:
A fix for this is ready for Beta test in `patch2`. I'll do this as 4.0.9-Beta-1, would appreciate it if someone on Agile could test it out.

Pyinthesky99, 2025-01-28:
Hi Steve
Will give it a go as soon as I can find it! :-)
Alan

Pyinthesky99, 2025-01-29:
Hi Steve.

Success, as far as I can tell, it appears to no longer roll forward and re-write the end times during a charge slot. It does seem to have a problem with the write current though and re-writes that every 10 minutes(because of the rounding?)

It does also appear to go 'off piste' a lot,
At 01:30 it was going to charge at 03:30 but at 01:40 it decided to do a charge from 01:40 to 02:00. At 02:30 it was happy with a charge at 03:30 but at 02:40 it decided to do a charge to 03:00. At 03:00 It abandoned the 03:30 charge, deciding on a 04:30 but changed it's mind at 03:10 and charged again until 03:30. At 03:31 it resumed it's plan on a 04:30 charge at 900W but at 3:40 it decided on a charge for 10 minutes!
At that point I gave up trying to follow it's course of action.

I did think the initial plan was OK, ending the session with around 80% SOC with a sunny day in prospect but it filled up the batteries by sun up.

[error.log](https://github.com/user-attachments/files/18586492/error.log)
[pv_opt.log](https://github.com/user-attachments/files/18586493/pv_opt.log)

Regards
Alan

stevebuk1, 2025-01-29:
Thanks for the test run on the issue in hand. 

Theres definitely something going on where the "low cost charging" routine is adding charging at 10 minutes into the slot because it shows a significant cost saving, but I'm not convinced that saving is true simply because if it was, it would have picked it up 10 mins earlier and it hasn't. To be honest I'd try setting "Include Export" to False for now until I get a chance to properly investigate.

Pyinthesky99, 2025-01-29:
Thanks for the info.
Tonight's charge slot was relatively simple. 29-Jan 22:30 GMT - 29-Jan 23:00 GMT  Power:  2200W  SOC:   36% ->   46% 

It wrote power..
22:30:03     INFO: Current 44.9A written to inverter
22:31:18     INFO: Current 43.3A written to inverter
22:40:03     INFO: Current 39.0A written to inverter
22:41:16     INFO: Current 39.1A written to inverter
22:50:05     INFO: Failed to write 0.0 to inverter to hold at 46% but actually wrote OK
23:00:02     INFO: Entity: number.solis_timed_charge_current Value:  0.0  Old State:  0.0 New state: None Diff:  0.0 Tol:  0.1
23:00:02     INFO: Inverter already at correct current

It 'held at 46%' but actually batteries were at 46%.

[image]

[error.log](https://github.com/user-attachments/files/18595281/error.log)
[pv_opt.log](https://github.com/user-attachments/files/18595282/pv_opt.log)

Pyinthesky99, 2025-01-30:
Hi Steve

Would you like me to raise another issue regarding the multiple writes of current?

stevebuk1, 2025-01-30:
> It 'held at 46%' but actually batteries were at EDIT 44%.

Hold mode is entered when target SOC is less than 3% away from actual SOC, so correct.

stevebuk1, 2025-01-30:
> Hi Steve
> 
> Would you like me to raise another issue regarding the multiple writes of current?

No its ok, I've noticed on my system there are far too many writes for fractions of an amp updates. Reason is that v4 has a tolerance of 0.1A whereas V3.x.x had a tolerance of 1A. I'm actually more tempted to use the power tolerance on the dashboard to make it user configurable, but need to think that through.

Pyinthesky99, 2025-01-30:
OK, thanks Steve.

Have set Include Export to False. Not too sure what it does if you can then select Optimise Discharging to On or Off anyway. A basic feature, I guess would be not to gas the batteries if the solar forecast for the next day was good and the Import price is greater than the Export price.

Good luck :-)

stevebuk1, 2025-01-30:
> Have set Include Export to False. Not too sure what it does if you can then select Optimise Discharging to On or Off anyway. 

You may be using an older dashboard, but on the current production dashvboard if you set "Include Export" to False, "Optimise Discharging" is removed from the dashboard. 
Behind the scenes, with Include Export set to off, the "low cost charging" part of the algorithm doesnt run, and it is this part that is generating big variations in charging. 

>  A basic feature, I guess would be not to gas the batteries if the solar forecast for the next day was good and the Import price is greater than the Export price.

As you know, my knowledge of low cost charging isnt great, but it will import additional charge if its cheaper. By definition you aren't going to self use this extra (because "high cost swaps" would have dealt with it), you are going to export it. The algorithm may well import charge if it means that it can then export more solar than it would have done if it didn't import - but only if it saves you money. Still need to get my head around it but right now given it varies wildly between one 10 minute slot to the next I'd say its been affected by the new partway through slot logic.

stevebuk1, 2025-01-30:
> Good luck :-)

Ha ha thanks! 

I've added some logging but it looks like power rounding to the power tolerance is applied to the resultant charge window before coverting to current, so using the same tolerance to then avoid needless inverter writes won't result in an inverter power thats much different from planned. Its complex because the plan is based on power but the inverter uses current, so as the battery charges more current is needed to achieve the same power. The impact of this is probably more apparent on overnight tariffs like IOG but the code has to work for everything. 

You are probably getting to the stage where I don't need to provide weblinks, just go into www.github.com/fboundy/pv_opt and select `patch2` from the dropdown which is invariably my latest code (I work in `patch2`, Francis works in `patch`). The logging I metioed before is there, just copy the usual 3 files from the `apps/pv_opt` area.

Pyinthesky99, 2025-01-31:
Had to tweak daily consumption to get it to generate a plan...

22:30:22     INFO: Current is 80.8080808080808
22:30:22     INFO: Current in solis.py = 80.8
22:30:24     INFO: Entity: number.solis_timed_charge_current Value: 80.8  Old State:  0.0 New state: 0.0 Diff: 80.8 Tol:  0.1
22:30:24     INFO: Failed to write 80.8 to inverter

22:31:37     INFO: Current is 80.8080808080808
22:31:37     INFO: Current in solis.py = 76.9
22:31:38     INFO: Entity: number.solis_timed_charge_current Value: 76.9  Old State: 80.7 New state: 76.9 Diff:  3.8 Tol:  0.1
22:31:38     INFO: Current 76.9A written to inverter

22:40:03     INFO: Current is 76.48183556405354
22:40:03     INFO: Current in solis.py = 76.5
22:40:04     INFO: Entity: number.solis_timed_charge_current Value: 76.5  Old State: 76.9 New state: 76.5 Diff:  0.4 Tol:  0.1
22:40:04     INFO: Current 76.5A written to inverter

22:41:17     INFO: Current is 76.48183556405354
22:41:17     INFO: Current in solis.py = 76.5
22:41:17     INFO: Entity: number.solis_timed_charge_current Value: 76.5  Old State: 76.5 New state: None Diff:  0.0 Tol:  0.1
22:41:17     INFO: Inverter already at correct current

22:50:03     INFO: Current is 76.04562737642586
22:50:03     INFO: Current in solis.py = 76.0
22:50:04     INFO: Entity: number.solis_timed_charge_current Value: 76.0  Old State: 76.5 New state: 76.0 Diff:  0.5 Tol:  0.1
22:50:04     INFO: Current 76.0A written to inverter

22:51:17     INFO: Current is 76.04562737642586
22:51:17     INFO: Current in solis.py = 76.0
22:51:17     INFO: Entity: number.solis_timed_charge_current Value: 76.0  Old State: 76.0 New state: None Diff:  0.0 Tol:  0.1
22:51:17     INFO: Inverter already at correct current

23:00:03     INFO: No charge/discharge windows planned  but charge start and end times are different.
23:00:05     INFO: Entity: number.solis_timed_charge_current Value:  0.0  Old State: 76.0 New state: 76.0 Diff: 76.0 Tol:  0.1
23:00:05     INFO: Failed to write 0 to inverter      
but obviously did.

To my simple mind, a tolerance of 1A looks as if it would be good and setting the plan at the start and letting it get on with it, perhaps checking every 30 minutes not 10.

Did it really also write end time, power and SOC twice every 10 minutes to the inverter?
22:30:22     INFO: Setting end time to 31-Jan 23:00 GMT
22:30:22     INFO: Setting power to 4000
22:30:22     INFO: Setting SOC to 80 etc.

Hope this helps
Alan

[error.log](https://github.com/user-attachments/files/18625791/error.log)
[pv_opt.log](https://github.com/user-attachments/files/18625790/pv_opt.log)

stevebuk1, 2025-01-31:
> To my simple mind, a tolerance of 1A looks as if it would be good and setting the plan at the start and letting it get on with it, perhaps checking every 30 minutes not 10.
> 

You could change the optimiser rate on the dashboard from 10 to 30 minutes to achieve this. 
As a trial I've gone for 100W (default power tolerance) which equates to roughly 2A, running 4.0.0-Beta-4 in `patch2` overnight on IOG to see what it does. But did think that power tolerance / 2 = 50W = 1A might be enough for Agile. 

> Did it really also write end time, power and SOC twice every 10 minutes to the inverter?

I think the double writes are a bug which I've raised as #371. I'm sure its writing times and current separately which it doesn't need to for the inverter firmware we are on I'm on but might need to be for the very latest (there is a six slot version available).

stevebuk1, 2025-01-31:
> 23:00:03 INFO: No charge/discharge windows planned but charge start and end times are different.
> 23:00:05 INFO: Entity: number.solis_timed_charge_current Value: 0.0 Old State: 76.0 New state: 76.0 Diff: 76.0 Tol: 0.1
> 23:00:05 INFO: Failed to write 0 to inverter
> but obviously did.

I see this on my system from time to time as well - it says its failed but the write goes through. I use modbus via Waveshares on a different timeout value to you so not really sure of the cause.

Pyinthesky99, 2025-02-01:
Interesting, set frequency to 30 mins around 11:24 and it still does 10 minute cycle.

[image]

[error.log](https://github.com/user-attachments/files/18628829/error.log)
[pv_opt.log](https://github.com/user-attachments/files/18628828/pv_opt.log)

Simple soul talking again, would it not be better to round the Amps rather than the Watts?  :-)
Guess it's not that simple!

stevebuk1, 2025-02-01:
> Interesting, set frequency to 30 mins around 11:24 and it still does 10 minute cycle.

Tested it on my system and found the same. 

Strange, as I'd previously tested mine  when issue #222 was raised on the same problem and it definitely was working then, but the problem went away for the originator so the issue it was closed.

The issue might be caused by a change I made in November to cure double optimiser runs. 

Have now fixed properly, try v4.0.9-Beta-5 on `patch2`.

stevebuk1, 2025-02-01:
> Simple soul talking again, would it not be better to round the Amps rather than the Watts? :-)
> Guess it's not that simple!

The final rounding is done on the amps, but the dashboard setting I want to use is "Power resolution" which is in Watts, hence needing the conversion. I could add an additional one for inverter amps but that would mean another configuration option and another slider so best use the one thats already there and just convert the units.

Pyinthesky99, 2025-02-01:
Hi Steve

Thanks for your efforts.

I installed v4.0.9-Beta-5 at 22:25 and it did an optimise cycle at 22:30 but not at 22:40 so all good so far.

With some sunshine in prospect tomorrow and prices rather high on Agile overnight there are no plans generated at present so will await events.

Thanks again
Alan

Pyinthesky99, 2025-02-02:
Hi Steve

It did come up with an 03:30 cycle plan.

I thought, in retrospect, I should have changed it back to 10 minute cycles to give it more bites at the currrent change cherry but...

03:00:05     INFO: Current tolerance is 2.0161290322580645A
03:00:07     INFO: Entity: number.solis_timed_charge_current Value: 14.1  Old State:  0.0 New state: 0.0 Diff: 14.1 Tol:  2.0
03:00:07     INFO: Failed to write 14.1 to inverter..........................but it obviously did

03:01:20     INFO: Current tolerance is 2.0161290322580645A
03:01:20     INFO: Entity: number.solis_timed_charge_current Value: 14.1  Old State: 14.1 New state: None Diff:  0.0 Tol:  2.0
03:01:20     INFO: Inverter already at correct current....................so that was good

Then at 03:30
03:30:05     INFO: Current is 14.08450704225352
03:30:05     INFO: Current tolerance is 2.0120724346076457A
03:30:05     INFO: Entity: number.solis_timed_charge_current Value: 14.1  Old State: 14.1 New state: None Diff:  0.0 Tol:  2.0
03:30:05     INFO: Inverter already at correct current

03:31:18     INFO: Current is 14.08450704225352
03:31:18     INFO: Current tolerance is 1.9880715705765408A
03:31:18     INFO: Entity: number.solis_timed_charge_current Value: 13.9  Old State: 14.1 New state: None Diff:  0.2 Tol:  2.0
03:31:18     INFO: Inverter already at correct current...................that was even better

04:00:05     INFO: Next charge window starts in 1949.9 minutes  but charge start and end times are different.
04:00:05     INFO: Current tolerance is 1.972386587771203A
04:00:08     INFO: Entity: number.solis_timed_charge_current Value:  0.0  Old State: 14.1 New state: 14.1 Diff: 14.1 Tol:  2.0
04:00:08     INFO: Failed to write 0 to inverter.........................but it obviously did
04:00:08     INFO: Waiting for inverter Read cycle: 72 seconds

So pretty successful?

[error.log](https://github.com/user-attachments/files/18632074/error.log)
[pv_opt.log](https://github.com/user-attachments/files/18632075/pv_opt.log)

stevebuk1, 2025-02-02:
> So pretty successful?

Yep thats doing what I expect. A tolerance of around 2A - i.e dont bother writing to the inverter for small changes - i.e. the 0.2A difference one in your report. It worked well for me last night too. 

The reporting of failed writes is a different issue. It actually may be accurate - Pv_opt makes a 2nd attempts to write if the first one fails, if the 2nd one goes through its not reported. Could be an actual failed write or just insufficient time waited between write and read.  I had to up the delay between write and read on the new time registers in Solarman but left the delay on current alone as its currently universal. If the delay is the issue then its a trade off - one might say if most writes go through then leave the delay as is an let the 2nd write catch it, as it means the whole process on average finishes quicker.

stevebuk1, 2025-03-02:
Re issue title (end time rolls forward on partial slots) a fix was implemented was for charging, but discharging and hold slots were missed. These have now also been fixed and under test., not on Beta release yet.

stevebuk1, 2025-03-28:
Released in v4.0.9, closing issue.

closed 2025-03-28 by stevebuk1

Pyinthesky99, 2025-03-29:
Thanks for the update, Steve.

If you would like me to test discharge/hold when it moves to Beta give me a shout.

Regards
Alan

stevebuk1, 2025-03-29:
Hi, sorry, it did get released to Beta at some point but is now part of the main production release at v4.0.9. If it doesnt work, reopen the issue.

Pyinthesky99, 2025-03-30:
Thanks for the info, Steve.

I am currently running v4.0.9 Beta 10. I had Optimise Discharge set and at around 16:30 yesterday, just by chance, noticed my battery SOC was 75%, I had expected 100% as it was a sunny. It was dumping my batteries down to 52%. I thought this was a bit premature as we us don't eat until fairly late, usually starting cooking, solely by electricity, around 19;30. As I am now a battery down, we would have been well into grid usage by bed-time. I killed the plan and turned the lights out at 22%!!

The plan was probably good in theory but I think I need a bit of tweaking on usage, I think need to shape my consumption profile but don't really understand the setting of hours and consumption in config.yaml

As my installer has waltzed off with one of my batteries, a failing cell, and gave no idea when it would be returned, I had knocked my battery capacity down to 7200 in pv_opt, I suppose I should knock Charger Power down, I originally choose 4000 as it was around 80% of 9600, I assume I am now limited to max 75Amps as down to 3 batteries, so probably need to go down to 3000, although I guess BMS sorts it out.
Install Release 4.0.9 now?
Regards
Alan

[pv_opt.log](https://github.com/user-attachments/files/19525141/pv_opt.log)