stevebuk1/pv_opt #379: Solax Modes Update

fboundy, 2025-01-17:
There is a small update to the Energy Storage Modes for Solis inverters using the Solax coming. It will need a small update to solis.py. Changes are unlikely to impact users as it only affects modes that aren't used by PV Opt

fboundy, 2025-01-19:
Fixed

closed 2025-01-19 by fboundy

stevebuk1, 2025-01-19:
Actually on 2nd thoughts as to what I said the Solarman time registers issue, I think patch is ready to push to main. 

I don't think there are any issues now with Solarman, it was all tested before in dev, and the only thing in the refactor that took it offline it was was an erroneoous `**kwargs` in the call `to write_and_poll_time`, which wasn't logged due to the try/except construct. 

The issue with the create_windows method adding fixed 30minutes between start and end is easily fixed, but I think other fixes are needed in high cost swaps, even if not its going to take a while to prove out, so I'll do this after v4.0.8 is released.

fboundy, 2025-01-19:
Ok - will push to main and release tomorrow

stevebuk1, 2025-01-21:
I did a PR, but not sure its helped given the error messages.

Next set of fixes I'll develop in `patch2`