stevebuk1/pv_opt #388: PV Opt tile flashes, requires restart of appdaemon to resolve

ThomasPrior, 2025-02-09:
**Describe the bug**

The PV Opt tile will flash in HomeAssistant. This happen about once every 24/48 hours and requires a restart of AppDaemon to resolve temporarily. 

Seems to be correlated with the appearance of the following error:

```
00:39:10 WARNING pv_opt: ------------------------------------------------------------
Exception: Unable to get expected response from Inverter Controller for {self.inverter_type}
raise Exception(e)
File "/conf/apps/pv_opt/pv_opt.py", line 605, in initialize
return f(*args, **kw)
File "/usr/local/lib/python3.10/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
result = self.fn(*self.args, **self.kwargs)
File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
response = future.result()
File "/usr/local/lib/python3.10/site-packages/appdaemon/utils.py", line 304, in run_in_executor
await utils.run_in_executor(self, init)
File "/usr/local/lib/python3.10/site-packages/appdaemon/app_management.py", line 162, in initialize_app
00:39:10 WARNING pv_opt: Traceback (most recent call last):
00:39:10 WARNING pv_opt: ------------------------------------------------------------
00:39:10 WARNING pv_opt: Unexpected error running initialize() for pv_opt
00:39:10 WARNING pv_opt: ------------------------------------------------------------
```

**To Reproduce**
Steps to reproduce the behaviour:
1. Wait 24/48 hours
2. Observe flashing tile

**Expected behavior**
No flashing tiles, PV Opt to not require a regular reboot stay running

**Screenshots**

https://github.com/user-attachments/assets/0c5c7bc8-39b0-4626-9c0a-6206e0fecc1d

**Setup**

Running on a Synology NAS in Docker, same Docker network:
- AppDaemon v4.4.2
- HomeAssistant 2025.1.0

**Additional context**
Tariff: Octopus Go (8.5p 00:30 to 05:30 UTC daily)
Inverter: Solis S5-EH1P5K-L
Connection: Modbus over Ethernet via https://github.com/alienatedsec/solis-ha-modbus-cloud

**Logs**
[pv_opt.log](https://github.com/user-attachments/files/18724696/pv_opt.log)

[error.log](https://github.com/user-attachments/files/18724700/error.log)

ThomasPrior, 2025-02-09:
This one didn't even wait 24 hours :)

https://github.com/user-attachments/assets/489c6717-4cdd-447d

[pv_opt.log](https://github.com/user-attachments/files/18726130/pv_opt.log)
[error.log](https://github.com/user-attachments/files/18726129/error.log)

stevebuk1, 2025-02-10:
Your first log shows Appdaemon jas rebooted Pv_opt at 21:38 and 00:38. 

When intitalising after the reboot, the inverter is not being found and PV_opt then stops with an error. 

The 2nd log shows exactly the same - reboots at 21:38 and 00:38. 

Theres no Pv_opt errors associated with the 21:38 and 00:38 reboots. 

You are running a 30 minute optimiser frequency which is running on the hour and 30 minutes past the hour, these are completing ok.

The tariff comparison function usually runs at 58 minutes past the hour, once per hour. 
I don't think its that, but I'm suspicious because of it being 8 minutes past something. 

Are you running tariff comparison? 

Can you post main.log (might be called appdaemon.log) to see what Appdaemon is doing?

stevebuk1, 2025-02-10:
Also, your log shows Pv_opt is in readonly mode. In readonly mode, Pv_opt won't look for the inverter. 

If you can include a log with Pv_opt with Readonly off, this will help diagnose the problem. 

(I'm also using the alienated_soc Waveshare based solution so should be able to replicate any issues)

ThomasPrior, 2025-02-10:
One main.log as requested. 
[main.log](https://github.com/user-attachments/files/18740851/main.log)

Am I running tariff comparison? Not sure what you mean here, can you clarify for a tired moron?

I've kept PV Opt in read only since switching from Agile to Go. I can turn it on for a bit, just turned it off as I had read the inverters have limited writes available and there have been occasions like this where I've woken up to a flat battery as the charge never got scheduled with PV Opt in a stuck state.

stevebuk1, 2025-02-10:
Thanks for the log, will take a look.

> Am I running tariff comparison? Not sure what you mean here, can you clarify for a tired moron?

Sorry. Its the last 13 lines of config.yaml. There is an option to log costs of different tariffs for comparison, these output to the dashboard and the log. I've got them switched off and there were problems with it in the initial v4.0.0 release. Disable it by commenting out the last 13 or so lines starting with "alt_tarriffs". Commenting out means the first character is #, as below

```
#  alt_tariffs:
#    - name: Agile_Fix
#      octopus_import_tariff_code: E-1R-AGILE-23-12-06-G
#      octopus_export_tariff_code: E-1R-OUTGOING-FIX-12M-19-05-13-G
#
#    - name: Eco7_Agile
#      octopus_import_tariff_code: E-2R-VAR-22-11-01-G
#      octopus_export_tariff_code: E-1R-AGILE-OUTGOING-19-05-13-G
#
#    - name: Eco7_Fix
#      octopus_import_tariff_code: E-2R-VAR-22-11-01-G
#      octopus_export_tariff_code: E-1R-OUTGOING-FIX-12M-19-05-13-G
#      
#    - name: Flux
#      octopus_import_tariff_code: E-1R-FLUX-IMPORT-23-02-14-G
#      octopus_export_tariff_code: E-1R-FLUX-EXPORT-23-02-14-G

```

ThomasPrior, 2025-02-10:
In that case no, I didn't know that was there but that is something I'd be interested in seeing the data for.

stevebuk1, 2025-02-10:
> One main.log as requested.
> [main.log](https://github.com/user-attachments/files/18740851/main.log)

Upon Appdaemon reboot, it seems to be having a problem loading one of the PV_opt files (solis.py).
Can you attach config.yaml please?

ThomasPrior, 2025-02-10:
Can't attach it as the filetype isn't allowed, but here's the contents:

```
# Internal configuration --- DO NOT EDIT ---
pvpy:
  module: pvpy
  global: true

inverters:
  module: inverters
  global: true

pv_opt:
  module: pv_opt
  class: PVOpt
  dependencies:
    - pvpy
    - solis

  log: pv_opt_log
  prefix: pvopt

# User configuration ---  EDIT AWAY! ---

  inverter_type: "SOLIS_SOLAX_MODBUS"
  # inverter_type: "SOLIS_CORE_MODBUS"#
  # inverter_type: SOLIS_SOLARMAN
  device_name: solis
  #consumption_history_days: 30

  # ========================================
  # Plant parameters
  # ========================================

  # All parameters can be a number or point to an entity UNLESS they start with 'id_'. All of
  # these are required but will be defaulted if not specified

  battery_capacity_wh: 16800
  # inverter_efficiency_percent: 97 # Default: 97
  # charger_efficiency_percent: 91 # Default: 91
  maximum_dod_percent: 10
  charger_power_watts: 5000
  inverter_power_watts: 5000
  # inverter_loss_watts: 100
  battery_current_limit_amps: 100
  
  # ========================================
  # Solcast configuration
  # ========================================
  #
  # id_solcast_today: sensor.solcast_pv_forecast_forecast_today
  # id_solcast_tomorrow: sensor.solcast_pv_forecast_forecast_tomorrow

  # ========================================
  # Solar and consumption forecast parameters
  # ========================================
  #
  # Valid options are:
  #
  # Solcast         - the Solcast mid-case forecast [Default]
  # Solcast_p90     - the Solcast high estimate
  # Solcast_p10     - the Solcast high estimate
  #
  # Set this using an "input_select" helper and it can be varied on the fly

  # solar_forecast:
  # - input_select.solar_forecast_source
  
  # # consumption estimation
  consumption_history_days:
[... 94 lines cut ...]
  
  # id_consumption:
  #   - sensor.{device_name}_house_load_power
  #   - sensor.{device_name}_backup_load_power

  # id_grid_power: sensor.{device_name}_meter_active_power
  # id_inverter_ac_power": sensor.{device_name}_inverter_ac_power
  # id_battery_soc: sensor.{device_name}_battery_soc
  # id_backup_mode_soc: 

  # id_timed_charge_start_hours: sensor.{device_name}_timed_charge_start_hour
  # id_timed_charge_start_minutes: sensor.{device_name}_timed_charge_start_minute
  # id_timed_charge_end_hours: sensor.{device_name}_timed_charge_end_hour
  # id_timed_charge_end_minutes: sensor.{device_name}_timed_charge_end_minute
  # id_timed_charge_current: sensor.{device_name}_timed_charge_current

  # id_timed_discharge_start_hours: sensor.{device_name}_timed_discharge_start_hour
  # id_timed_discharge_start_minutes: sensor.{device_name}_timed_discharge_start_minute
  # id_timed_discharge_end_hours: sensor.{device_name}_timed_discharge_end_hour
  # id_timed_discharge_end_minutes: sensor.{device_name}_timed_discharge_end_minute
  # id_timed_discharge_current: sensor.{device_name}_timed_discharge_current

  # id_inverter_mode: sensor.{device_name}_storage_control_mode

```

stevebuk1, 2025-02-10:
> In that case no, I didn't know that was there but that is something I'd be interested in seeing the data for.

Lets leave it disabled for now until you are up and running reliably again.
The tariff comparison functionality is explained here:

https://github.com/fboundy/pv_opt?tab=readme-ov-file#alternative-tariffs

ThomasPrior, 2025-02-10:
New main.log attached

[main.log](https://github.com/user-attachments/files/18741205/main.log)

stevebuk1, 2025-02-10:
> New main.log attached
> 
> [main.log](https://github.com/user-attachments/files/18741205/main.log)

Ok thats looking more like mine. Can you also post pv_opt.log so I can see what its done after reboot.

ThomasPrior, 2025-02-10:
Waiting until idle at 22:03, then restarted AppDaemon.

New logs attached.

[pv_opt.log](https://github.com/user-attachments/files/18741444/pv_opt.log)
[error.log](https://github.com/user-attachments/files/18741445/error.log)
[main.log](https://github.com/user-attachments/files/18741446/main.log)

stevebuk1, 2025-02-10:
Argghh, my bad, I copied and pasted one too many lines from my config.yaml and turned all debug text on which generates a very large file. 

Please comment out or remove  `debug: True`. 

After that, lets see what happens over the next 24 hours or so.

ThomasPrior, 2025-02-10:
Thank you, will do. Enjoy the rest of your evening!

ThomasPrior, 2025-02-16:
Had a couple of days of stability, unfortunately not resolved quite yet.

[pv_opt.log](https://github.com/user-attachments/files/18813212/pv_opt.log)
[error.log](https://github.com/user-attachments/files/18813213/error.log)
[main.log](https://github.com/user-attachments/files/18813214/main.log)

ThomasPrior, 2025-02-16:
This occurrence seems to start at 18:08

https://github.com/user-attachments/assets/8c5c3626-cd39-4f56-8b8c-e4679bc306e8

ThomasPrior, 2025-02-16:
Battery flat this morning as a result. Should've checked the charge schedule last night when I posted the logs. Whoops.

stevebuk1, 2025-02-16:
Ok so my fixes to config.yaml made no difference. 

AppDaemon is losing its connection to HA at 18:05:54

```
08:07:26 WARNING AppDaemon: Excessive time spent in utility loop: 12827.0ms, 12827.0ms in check_app_updates(), 0.0ms in other
18:05:54 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
18:05:59 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
18:06:04 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
18:06:09 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
18:06:15 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
18:06:20 INFO HASS: Connected to Home Assistant 2025.1.0
```

and reestablishes at 18:06:20. 

AppDaemon then does a stack of restarts (including Pv_opt). 

```
18:07:23 INFO HASS: All startup conditions met
18:07:23 INFO AppDaemon: Processing restart for HASS
18:07:23 INFO AppDaemon: Terminating pv_opt
18:07:23 INFO AppDaemon: Reloading Module: /conf/apps/pv_opt/pvpy.py
18:07:23 INFO AppDaemon: Reloading Module: /conf/apps/pv_opt/solis.py
18:07:23 INFO AppDaemon: Reloading Module: /conf/apps/pv_opt/pv_opt.py
18:07:23 INFO AppDaemon: Loading app pv_opt using class PVOpt from module pv_opt
18:07:23 INFO AppDaemon: Calling initialize() for pv_opt
18:08:24 INFO AppDaemon: pv_opt: Entity sensor.pvopt_status created in namespace: default
18:08:24 WARNING AppDaemon: Logged an error to /conf/logs/error.log
```

When Pv_opt reinitialises it can't find a numeric value for number.solis_battery_minimum_soc (this is the entity it uses to see if the Inverter is online), and generates the error you see at 18:08:24 because its returning unknown or unavailable. 

AppDaemon losing connection to HA at 18:05:54 is unrelated to PV_opt, I can't solve this, its not a Pv_opt problem. 

The fact that PV_opt doesnt restart properly when AppDaemon restarts is potentially something that can be fixed. 

number.solis_battery_minimum_soc should not be unknown/unavailable at this point, unless HA has restarted, and given your dashboard is live I can't see thats the issue. 

Can you have a look at the history for this entity and see what it was doing around 18:08? (You should be able to zoom in on this from HA).

ThomasPrior, 2025-02-16:
Good find, HA did restart at 18:06 according to the sensor.uptime entity. I didn't notice.

[image]

Status of number.solis_battery_minimum_soc at the requested timestamp below:

[image]

stevebuk1, 2025-02-16:
Ok, the above certainy explains it. AppDaemon and HA restarts don't synchronise well. 

Are you running the automation as recommended here?

https://github.com/fboundy/pv_opt?tab=readme-ov-file#13-add-an-automation-to-restart-appdaemon-when-ha-restarts-optional

If you are, changing `sensor.solis_battery_soc` to `number.solis_battery_minimum_soc` might then improve the situation.

stevebuk1, 2025-02-16:
Saying that, not sure if 

```
trigger:
  - event: start
    platform: homeassistant
```

would be triggered on an HA restart. The automation might need some mods.

ThomasPrior, 2025-02-16:
> Ok, the above certainy explains it. AppDaemon and HA restarts don't synchronise well.
>
> Are you running the automation as recommended here? [...]

Thank you, I'll give this a go. If memory serves I was doing this for a while, can't recall why I stopped.

stevebuk1, 2025-03-06:
Good news its solved. I don't see it as time wasted - it was a genuine problem that we were able to help with.